mstrens/oXs_on_RP2040 #3: PWM values output RP2040

nostromo123, 2022-11-21:
Prepared a demo board here with 6 servo channels with 'Test for 16 PWM'

Channels #1 to #4 and #6 on the transmitter output side 988µs - 1497µs - 2012µs
Output signal on RP2040 1.06ms - 1.59ms - 2.13ms.

Is that supposed to be so?

br KH

mstrens, 2022-11-21:
In fact oXs does not get PWM values.
It get values in a range 0/1024 when coding is on 10bits like ELRS). Those are converted (X 2) to a range 0/2048 to match the coding used by Sbus (11 bits)
Those Sbus values are then converted by oXs to generates PWM.
oXs performs the mapping with this instruction:
pwmValue = fmap( rcSbusOutChannels[i + config.gpio1 - 1] , 172, 1811, 988, 2012 );
So a value 172 is converted to 988 usec
And a value 1811 to 2012 usec.

On Tx side, there is a conversion in the opposite side. PWM values (usec) are converted to digital values (0/1024 or 0/2048).
If the 2 conversions (on TX and on RX) are not the related to each other, it is "normal" that the PWM on oXs are not equal to those on the TX.

The values used by oXs for  the mapping have been found in some document (I do not remember where) on internet.
You can change them easily if you compile the firmware by your self.

nostromo123, 2022-11-21:
Thank you for your explanation.

That line can be found in file '/oXs_on_RP2040-Test-for-16-PWM/src/sbus_out_pwm.cpp'

Tried to forward this and another question to you in a pm on RCN 2 weeks ago, but you did not answer, I thought this project to be dead. Good to see it still active!

nostromo123, 2022-11-21:
With those modified values I get 1ms .. 1.5ms .. 2 ms

pwmValue = fmap( rcSbusOutChannels[i] , 260, 2041, 988, 2012 ); //PWM mod.
//pwmValue = fmap( rcSbusOutChannels[i] , 172, 1811, 988, 2012 ); //original

but I would feel better, I this could be calculated and not just tried out.

br KH

closed 2022-11-21 by nostromo123