mstrens/oXs_on_RP2040 #6: Reset V1/4 to 255. May be an issue ....

nostromo123, 2022-11-24:
Playing with voltages on 'main' I seems that I cannot reset the values for V1/4 back to 255, once set to a pin.

```
Version = 0.2.9 
    Function              Code      Pin (255=disabled)
Primary channels input    PRI          9
Secondary channels input  SEC        255
Telemetry                 TLM         10
GPS Rx                    GPS_RX      11
GPS Tx                    GPS_TX      12
Sbus OUT                  SBUS_OUT   255
RPM                       RPM        255
SDA (baro sensor)         SDA         14
SCL (baro sensor)         SCL         15
PWM Channels 1, 2, 3 ,4   C1 / C4      0    1    2    3
PWM Channels 5, 6, 7 ,8   C5 / C8      4    5  255  255
PWM Channels 9,10,11,12   C9 / C12   255  255  255  255
PWM Channels 13,14,15,16  C13/ C16   255  255  255  255
Voltage 1, 2, 3, 4        V1 / V4     26   27   28  255

Protocol is CRSF (=ELRS)
CRSF baudrate = 420000
Voltage/RPM parameters:
    Scales : 1.000000 , 1.000000 , 1.000000 , 1.000000 
    Offsets: 0.000000 , 0.000000 , 0.000000 , 0.000000 
Baro sensor is detected using BMP280
    Sensitivity min = 100 (at 100)   , max = 300 (at 1000)
    Hysteresis = 5 
Foreseen GPS type is Ublox  :GPS is detected
Failsafe type is HOLD

Config parameters are OK

 >> 

```

With command C5=255 I can reset PWM #5 back to 255, but trying V1=255, V2=255 or V3=255 does not work here.

Might be a problem?!

br KH

nostromo123, 2022-11-24:
Don't want to harass you ... ;-)

Had a look into param.cpp, but I can not see a fault there, but this 

**Cmd to execute:   V1=255
Error : pin must be in range 26 / 29 or 255**

```
processing cmd

Cmd to execute:   V1=255
Error : pin must be in range 26 / 29 or 255

Version = 0.2.9 
    Function              Code      Pin (255=disabled)
Primary channels input    PRI          9
Secondary channels input  SEC        255
Telemetry                 TLM         10
GPS Rx                    GPS_RX      11
GPS Tx                    GPS_TX      12
Sbus OUT                  SBUS_OUT   255
RPM                       RPM        255
SDA (baro sensor)         SDA         14
SCL (baro sensor)         SCL         15
PWM Channels 1, 2, 3 ,4   C1 / C4      0    1    2    3
PWM Channels 5, 6, 7 ,8   C5 / C8    255    5  255  255
PWM Channels 9,10,11,12   C9 / C12   255  255  255  255
PWM Channels 13,14,15,16  C13/ C16   255  255  255  255
Voltage 1, 2, 3, 4        V1 / V4     26   27   28  255

Protocol is CRSF (=ELRS)
CRSF baudrate = 420000
Voltage/RPM parameters:
    Scales : 1.000000 , 1.000000 , 1.000000 , 1.000000 
    Offsets: 0.000000 , 0.000000 , 0.000000 , 0.000000 
Baro sensor is detected using BMP280
    Sensitivity min = 100 (at 100)   , max = 300 (at 1000)
    Hysteresis = 5 
Foreseen GPS type is Ublox  :GPS is detected
Failsafe type is HOLD

Config parameters are OK

 >> 
```

mstrens, 2022-11-24:
There is currently this line
} else if ( (!(ui>=26 and ui<=29)) or ui ==255) {

It should be
} else if ( ! ( ((ui>=26 and ui<=29)) or ui ==255) ) {

nostromo123, 2022-11-24:
OK, working now. ;-)

I have to apologize for not finding the issue myself. Last time I worked with software on a high level was ~1976, when I did my thesis as a student of mechanical engineering on a PDP 11/10 in FORTRAN IV.

br KH

closed 2022-11-24 by nostromo123