mstrens/oXs_on_RP2040 #7: Just a reminder

nostromo123, 2022-11-26:
Please do the correction in param.cpp 'main'  V1/4

I am uncertain about the shifted PWM values ... Have a look here  https://wiki.rc-network.de/wiki/PWM  
Leave it, like it is, or corect it?? I can do it here for me by modify the values ...

Good news, more interest in your project!!

https://www.rc-network.de/threads/expresslrs-crossfire-f%C3%BCr-fl%C3%A4chenmodelle.11918292/page-4

br KH

mstrens, 2022-11-26:
I just updated Main on github (for removing a Voltage pin)
I deleted the other branch.
I did not change scaling PWM because as you have seen there is no standard on the values to be used for min/max.
Scaling to apply could depend on the TX being used.

I expect that the fluctuations when connected to Gnd exist because the iRP2040 ADC is not perfect when voltage is very low.

FYI, there is also a ELRS receiver from BetaFPV with more redundancy  (true diversity because it uses 2 SX1280).

I have seen that the project oXs on RP2040 is mainly discussed with ELRS but you know that
- it can work with FRSKY/JETI telemetry
- it can also provide redundancy when connected to 2 receivers with sbus (could be Futaba,...)

Just decoding Sbus signal is also a good use case because this solution is cheap and does not require an additional tool to select the channels to decode (like the FRSKY SBUS/PWM decoder).
For decoding a few channels, it does not require an additional PCB because it is quite easy to solder servo extension wires directly to the RP2040.

nostromo123, 2022-11-26:
Thank you for answering fast again :-)

We keep the PWM thing!

Ordered a second ELRS receiver RX/TX for diversity. Only other ELRS receivers, I have here, are PWM type

I am also a bit busy with our old project here in Germany 'MPX telemetry with MPX rx and tx module on FrSky radios'. Now on ATTiny 841 and S-Bus. No code by me, I am too old for that, just testing and helping to get things going!

br KH

closed 2022-11-26 by nostromo123