mstrens/oXs_on_RP2040 #8: V3 deviation?

nostromo123, 2022-11-26:
Made the same basic settings in all 4 telemetry field on the radio.
(Hope, I did not make a mistake on my side...)

[image]

Could you please check, if there is a typo in the code with V3.

Thank you - KH

mstrens, 2022-11-26:
In oxs, the 4 voltages are measured exactly with the same code (it is a 4 X loop).
When data are transmitted to ELRS RX, they have to respect a CRSF format that is not foreseen for 4 voltages but for
- one volt (16 bits)
- one current (16 bits)
- one capacity (24 bits)
- one remain. (8 bits) (is supposed to be max 100%)

oXs fill the first 3 fields with the measured value from RP2040 (taking care of scaling and offset)
The 4 fields is first divided by 10 in order not to exceed the 8 bits length.

I just made some minor improvement in oXs code for better rounding. I put it on github.
So I hope e.g. that 6.6V would be rounded 7.0 instead of 6V.

The difference in the values being displayed (e.g. 70V for V3) have to be explained by the way openTX or Edge manage the CRSF values they receive.

nostromo123, 2022-11-26:
Solved, thank you!

br KH

closed 2022-11-26 by nostromo123