mstrens/oXs_on_RP2040 #9: Compilation issue

pierrotm777, 2022-11-26:
Hello mstrens,

When i compile this last version, i have a compilation error.

In file included from src\param.h:4:0,
                 from src\crsf.cpp:17:
src\crsf_frames.h:13:26: error: redeclaration of 'CRSF_FRAMETYPE_GPS'
     CRSF_FRAMETYPE_GPS = 0x02,
                          ^~~~
compilation terminated due to -Wfatal-errors.
*** [.pio\build\raspberry-mstrens\src\crsf.o] Error 1

nostromo123, 2022-11-26:
Sorry for stepping in here ...

Seeing your post here, I loaded the fresh 'main.zip', unpacked it, made no modifications and build and loaded it here to the RP2040. Working here!

Attch. protocol

[Neue Datei.txt](https://github.com/mstrens/oXs_on_RP2040/files/10096617/Neue.Datei.txt)

regards - KH

pierrotm777, 2022-11-26:
Sorry,

I had done a copy over my old files.
All is ok now.

Thanks

Pierre

mstrens, 2022-11-26:
This is very strange.
I can compile without error.

Did you completely replace all files in your project with those on github?
I checked and I do not see a redeclaration in the code.
You can try deleting all 4 files having Copie_cpp.txt or Copie_h.txt in their name.
You can also try deleting all files in your folder oxs_on_RP2040/.pio/.build
This would force a clear recompilation.

pierrotm777, 2022-11-26:
Did you completely replace all files in your project with those on github?

No, in fact, all is ok now.

Thanks

nostromo123, 2022-11-27:
@pierrotm777

If it is ok now, why not close this issue?

br KH

pierrotm777, 2022-11-27:
Thanks for your help

closed 2022-11-27 by pierrotm777