mstrens/oXs_on_RP2040 #10: Maybe I found something .... ?

nostromo123, 2022-11-29:
First of all, I tested V1/4.

V1/3 I could adjust up to 1/100V!! V4 is to say, usable for fuel tank level ...

But there are 3 very precise voltage inputs! For me, that is enough.

Was going to do a short test on RPM (What else could I test?)

RPM on PIN 8

See CONFIG 

[Neue Datei.odt](https://github.com/mstrens/oXs_on_RP2040/files/10112310/Neue.Datei.odt)

On the radio I see this here

[image]

br KH

mstrens, 2022-11-29:
ELRS uses CRSF protocol to communicates with handset.
CRSF protocol does not have a specific field for RPM.
So in order to transmit this data, I had to "misuse" another field supported by CRSF.
CRSF has a frame named "Attitude" that contains 3 fields (pitch, roll and yaw).
When you ask for RPM, oXs fill the value into Pitch field.
Note: to get some value you have to apply some pulses to the rpm pin. 
E.g. if you apply a signal at 100 hz (pulses between LOW and HIGH level), you should get this value on the handset.

nostromo123, 2022-11-29:
Thank you for this clarification, was ready to start a bl-motor with rpm sensor, but seeing that on the radio, I hesitated. I'll start the motor and report back.

I'll use my DIY 3.3V sensor https://www.rc-network.de/threads/openxsensor-f%C3%BCr-telemetriesensoren.686113/page-10#post-12402160 

br KH

nostromo123, 2022-11-29:
I see something on the radio, but to get those readings on the radio to make sense, I have to think a bit, how to adjust settings in CONFIG

> -To change RPM multiplicator, enter e.g. RPM_MULT=0.5 to divide RPM by 2

and in the radio telemetry.

[image]

[image]

I think, I'll use a cheap Chinese signal generator, I once bought. Can generate a stable signal.

[image]

Should make things much easier.

I close that issue here now. But which way is better to contact you, should further question arise? Here or on RCN or where?

br -KH

closed 2022-11-29 by nostromo123