mstrens/oXs_on_RP2040 #11: Still struggling with RPM/Hz

nostromo123, 2022-11-30:
I found this notes in the code about RPM:

> -To change RPM multiplicator, enter e.g. RPM_MULT=0.5 to divide RPM by 2
> 
> 
> // Note: ELRS has no field to transmit RPM; so RPM is sent in "attitude" frame as pitch, roll and yaw.  
> //       Those fiels can have a max value of about 16000 and some digits are considered as decimals and lost by openTx.
> //       Therefore RPM value (hetz) is transmitted 3 times (once in Hetz, once in 10 X Hetz and once in 100 X Hetz)
> //       So the user can decide which value best fit his need
> 
> 
> // One RPM (Hz) can be measured
> //       Take care to limit the voltage to the range 0-3V; so if you use capacitor coupling, add diodes and resistor to limit the voltage
> //       All pulsed are counted (no debouncing); so use a hardware low pass filter (resistor/capitor) to avoid dummy pulses reading
> //
> 
> //    - one RPM measurement; a scaling (SCALE4) can be used to take care e.g. of number of blades (optional)

I did set pitch, roll and yaw on same settings in the radio. I only see reaction in pitch. Nothing in roll or yaw.

I generate a square signal 3.3V pos. 100, 200, 300 .... 1000Hz

[image]

[image]

The radio displays with 100, 200,300 .... 1000Hz those values

1.00Hz, 2.10Hz, 3.21Hz,4.31Hz,5.42Hz,6.52Hz,7.63Hz.8.74Hz,9.84Hz,10.95Hz.

Here some of the screens.

[image]
[image]
[image]

With my calculation, 100Hz is a low rpm value, with a 14 magneto bl motor (ie. 7 pole) equals 857 rpm

Before looking into rpm instead of Hertz displayed, I would like your comments.

Was expecting to see 100Hz when applying 100Hz. 

br - KH

mstrens, 2022-11-30:
I think I changed the code but forgot to update the documentation.
Currently RPM is transmitted only in Pitch field (the 2 others just remains set to 0).
The max value that can be put by oXs in this field is about 32000 (it is a int16 and only positive values have sense for RPM).
I expect that some digits are lost or put as decimal in openTx/edge. 
Currently RPM was measured 10 times per second (interval is 100000usec = 0.1 sec)
At 100Hz it means that oXs can only count about 10 pulses (so value could easily become 9, 10 or 11).
For such a low frequency, it should make sense to increase the counting interval to e.g. 1 sec.
This was a parameter in rpm.cpp but I just moved it to _config.h (seems easier for a user)
When the number of pulses/second increases (e.g. 1000hz) an interval of 0.1sec makes sense.
Getting 10.95 for 1000hz is probably the result of your ratio 282. I presume a lower ratio could give you a value around 10.00
Getting 1000 on the display instead of 10.00 has to be solved on handset (perhaps calculated field)

nostromo123, 2022-12-02:
Did some further testing here ...

Setting in radio 

[image]

here a short table of settings. 

[image]

100Hz fq-generator > ratio 28 > display radio 0.10 Hz.

With higher values from the generator the difference increases and higher values than 3000Hz don't make sense.

Reduced ratio = 26 and now in range 3500 ... 100 Hz looks acceptable.

-------------------------------------------------------------------------------------------------------------------------------------------------------------

The I changed over to RPM instead of Hz on the radio,

I calculated (14 magnets / 7 poles)
these two settings for the fq-generator

1000 rpm | -----------------> | 117 Hz
5000 rpm | -----------------> | 583 Hz

117 Hz input

[image]

583 Hz input

[image]

My impression after several testings. Hz <> RPM on radio side.

Precision = Precision
Ratio = Blades/Poles
Offset = Multiplier 
in behavior!

Please comment. Can you do anything on your side or is the whole thing fixed by the CRSF protocol? What makes me suspicious is, that the Matek PWM converters offer most telemetry values, but no RPM!

br KH

mstrens, 2022-12-07:
As far I know, CRSF uses predefined telemetry formats to transmit the data.
There are a very limited number of formats, each containing a set of data related in some way to each other (e.g. there is a GPS format for all GPS, a "battery" format for voltage/current/capacity., ...)
There is no format foreseen for RPM and so I had to "misuse" an existing format named "Attitude" which is is supposed to contains degrees.
I do not see a better solution.
Please note that for RPM, in order to get as many digits as possible on tx side, you can use RPM_MULT= xxx.yyy in oXs in order to let oXs send a value that is as big as possible but still lower than 32000 (= max values in CRSF format).

nostromo123, 2022-12-07:
Just had another try fiddling with the values for RPM_MULT and Blades/Poles. Tested with 583Hz. Can get bigger values for RPM, but I see values jumping in between xxxx rpm and xx rpm.

What I see here has in my opinion nothing to do with measuring. Your "misuse" was in my eyes a bad idea. It would be better just to state, no RPM under CSRF and remove it from CSRF Config!

I'll do a post about rpm and CSRF in https://www.rc-network.de/threads/oxs-on-rp2040.11929246/ and https://www.rcgroups.com/forums/showthread.php?4130529-openXsensor-on-RP2040-board
hoping someone more clever then me finds a way to handle RPM_MULT and Blades/Poles.

br KH

mstrens, 2022-12-07:
I think it is still possible to "misuse" pitch in CRSF for RPM (with perhaps some limitations).
I expect that those that implemented CRSF on Tx side for pitch did it in order to display e.g. 12.34 when pitch is 12 degree, 345
I also expect that a a "common" RPM value could be e.g. 12340 tours/min. and that could be displayed 12.34 (meaning that the user has to read it as kilo tours/min (instead of degree).

We can split the problem in 2 parts:
- What is the value that oXs must sent in order to get the expected value on the Tx display
- How to let oXs generate this value.

I propose that you just make a test to help for the first part.
In file rpm.cpp, in line 52 there is:
fields[RPM].value = rpm; 

Could you change the line to
fields[RPM].value = 10000;

So oXs will for sure send a fix value equal to 10000  (if RPM pin is defined).
Let me know what is the value that is displayed on your TX when you do not apply any calculation/ratio/offset.
This would allow to see if it is possible to use the display in a meaningful way. 

For part 2, I expect that you want to display some values in tour/min (and not in Hz).
Let me also know what is the range of values that seems you realistic (e.g. max 25000, min= 2000) if if you would accept that it would be displayed respectively 25.00 and 2.00  

Base on this, I can look at the best values to use for RPM_MULT

nostromo123, 2022-12-08:
**Your 1. part.**

Changed line in rpm.cpp to fields[RPM].value = 10000;

Build and uploaded.

Had to reset RPM_MULT in CONFIG t0 1.0
```

 >> 
processing cmd

Cmd to execute:   RPM_MULT=1
New config has been saved
Device will reboot

Version = 0.3.0 
    Function              Code      Pin (255=disabled)
Primary channels input    PRI          9
Secondary channels input  SEC        255
Telemetry                 TLM         10
GPS Rx                    GPS_RX      11
GPS Tx                    GPS_TX      12
Sbus OUT                  SBUS_OUT   255
RPM                       RPM          8
SDA (baro sensor)         SDA         14
SCL (baro sensor)         SCL         15
PWM Channels 1, 2, 3 ,4   C1 / C4      0    1    2    3
PWM Channels 5, 6, 7 ,8   C5 / C8      4    5  255  255
PWM Channels 9,10,11,12   C9 / C12   255  255  255  255
PWM Channels 13,14,15,16  C13/ C16   255  255  255  255
Voltage 1, 2, 3, 4        V1 / V4     26   27   28   29

Protocol is CRSF (=ELRS)
CRSF baudrate = 420000
Voltage parameters:
    Scales : 0.972000 , 0.972000 , 0.972000 , 0.950000 
    Offsets: 6.000000 , 5.500000 , 6.000000 , 5.000000 
RPM multiplier = 1.000000
Baro sensor is not detected
Foreseen GPS type is Ublox  :GPS is not (yet) detected
Failsafe type is HOLD

Config parameters are OK

 >> 

```
Radio display with NO signal on pin 8

[image]

**About part 2**. I think, these could be realistic values for a simulation

5000 rpm | -----------------> | 583 Hz
10000 rpm | -----------------> | 1167 Hz
20000 rpm | -----------------> | 2333 Hz

The data sheet of the FrSky RPM sensor (for bl-motors) states that range 'Measures RPM from 1,000~30,000rpm'
(I could simulate that range too .....)

br - KH

nostromo123, 2022-12-08:
The display shows 0.40 NOT 0.04!

When I change the unit to V or ms, 0.40 stays, BUT now instead of 'Blades/Poles' 'Ratio' shows up, where  'Ratio' can be set to '-'. The min value, when RPM as unit  is selected for 'Blades/Poles' is 1, not '-'! 

In that case ('Ratio' = '-') display shows 1.00

[image]

You can calculate the value, add or multiply, BUT only with values available like V with A for calculating consumption or add times ... for total time.

My calculation for Hz <> rpm is based on that assumption bl motor 14 mag/7 pol

[image]

Crosschecked with MPX and Unisense-E.  Pic 2, 3 and 4. Unisense-E set to 14 mag, 583Hz >> 5000 rpm!

https://www.rc-network.de/threads/oxs-on-rp2040.11929246/page-2#post-12404723

I still think, that trying to have rpm measurement under ELRS/CSRF is a dead-end street.

RPM is NOT important in my eyes. Remove it from the ELRS/CSRF part of oXs_on_RP2040. All the other sensor values work!

br - KH

nostromo123, 2022-12-17:
Suggestion from me.

As CRSF at the moment seems not to be open for RPM measurement

in **README.md**

```
This project is foreseen to generate telemetry data (e.g. when a flight controller is not used) , PWM and Sbus signals.

For telemetry, it can provide

    up to 4 analog voltages measurement (with scaling and offset)
    the altitude and the vertical speed when connected to a pressure sensor (optional)
    GPS data (longitude, latitude, speed, altitude,...) (optional)
    RPM **- but at the moment NOT on CRSF/ELRS protocol**

```
and in **param.cpp**  line 104

printf("    RPM !not on CRSF/ELRS  RPM       0, 1, 2, ..., 29\n");

so it looks like that.

```
Commands can be entered to change the config parameters
- To activate a function, select the pin and enter function code = pin number (e.g. PRI=1)
    Function                  Code        Valid pins number
    Primary channels input    PRI       5, 9, 21, 25
    Secondary channels input  SEC       1, 13, 17, 29
    Telemetry                 TLM       0, 1, 2, ..., 29
    GPS Rx                    GPS_RX    0, 1, 2, ..., 29
    GPS Tx                    GPS_TX    0, 1, 2, ..., 29
    Sbus OUT                  SBUS_OUT  0, 1, 2, ..., 29
    RPM !not on CRSF/ELRS     RPM       0, 1, 2, ..., 29
    SDA (baro sensor)         SDA       2, 6, 10, 14, 18, 22, 26
    SCL (baro sensor)         SCL       3, 7, 11, 15, 19, 23, 27
    PWM Channels 1, ..., 16   C1 / C16  0, 1, 2, ..., 15
    Voltage 1, ..., 4         V1 / V4   26, 27, 28, 29
- To disable a function, use pin number = 255
```
If the situation with CRSL changes in future .....

But with these two remarks added, we could close this issue for now!

br - KH

mstrens, 2022-12-18:
Changed in version 0.3.5
Note: currently RPM is also not part of Jeti protocol.
So it is only for sport

nostromo123, 2022-12-18:
Thank you for that clarification.

br - KH

closed 2022-12-18 by nostromo123