mstrens/oXs_on_RP2040 #12: SBus PWM and SPort telemetry do not work simultaneously

chl-repo, 2022-12-04:
I have been trying to build a "full-featured" receiver with telemetry (Baro) and four PWM outputs based on a FrSky R-XSR.

Telemetry and PWM work each alone. But if I activate both at the same time, i.e. assign a pin to PRI and TLM, the PWM outputs work but no telemetry data is transmitted anymore. 

I have also tested this with a FrSyk X8R, with the same result. 

Please let me know if you need further information or if I should do some testing.
CHL

mstrens, 2022-12-04:
Thanks for the feed back.
I will check it (probably to morrow)

chl-repo, 2022-12-04:
Just for clarification:

Only the telemetry values generated by the RP2040 (Alt, VSpeed) are affected, but not those generated by the receiver (RSSI etc.).

mstrens, 2022-12-05:
can you provide me the configuration that you used?
Which pins are used and for which purpose.
Please note that PRI and TLM must be assigned to 2 different pins (PRI is connected to Sbus and TLM to Sport)

chl-repo, 2022-12-05:
The config is: 
PRI=5
TLM=1
SDA=10
SCL=11
C1=4
C2=6
C3=7
C4=8

mstrens, 2022-12-06:
I forgot to upload on github.
I did it now.

chl-repo, 2022-12-06:
It works.

Many thanks !

closed 2022-12-06 by chl-repo