mstrens/oXs_on_RP2040 #14: Diversity with ELRS

nostromo123, 2022-12-13:
(Put RPM aside, was hoping someone steps in with testing RPM, but ...)

I received my 2nd. BETAFPV nano RX. Both RX work as no. 1 RX with RP2040 on my test board.

For testing diversity, I disabled 'model match' in both RX and also disabled telemetry in no. 2 RX.

```
Version = 0.3.0 
Function Code Pin (255=disabled)
Primary channels input PRI 9
Secondary channels input SEC 13
Telemetry TLM 10
GPS Rx GPS_RX 11
GPS Tx GPS_TX 12
Sbus OUT SBUS_OUT 255
RPM RPM 255
SDA (baro sensor) SDA 14
SCL (baro sensor) SCL 15
PWM Channels 1, 2, 3 ,4 C1 / C4 0 1 2 3
PWM Channels 5, 6, 7 ,8 C5 / C8 4 5 255 255
PWM Channels 9,10,11,12 C9 / C12 255 255 255 255
PWM Channels 13,14,15,16 C13/ C16 255 255 255 255
Voltage 1, 2, 3, 4 V1 / V4 26 27 28 29

Protocol is CRSF (=ELRS)
CRSF baudrate = 420000
Voltage parameters:
Scales : 0.972000 , 0.972000 , 0.972000 , 0.950000 
Offsets: 6.000000 , 5.500000 , 6.000000 , 5.000000 
RPM multiplier = 1.000000
Baro sensor is detected using BMP280
Sensitivity min = 100 (at 100) , max = 300 (at 1000)
Hysteresis = 5 
Foreseen GPS type is Ublox :GPS is detected
Failsafe type is HOLD

Config parameters are OK
```

Tested like that:

- On testboard, no.1 RX TX nand RX connected to RP2040, no.2 RX TX **NOT** connected to pin 13.
- Radio ON!
- Testboard ON!
- All working normal, telemetry on radio active, PWM output on RP2040, both RX LED steady on, RP2040 green blinking

- Then I connect no.2 RX TX to pin 13.

- PWM freezes, RP2040 LED changing colors.

- Disconnect no.2 RX TX pin 13

- PWM working, LED2040 green.

Any idea?

br - KH

mstrens, 2022-12-13:
What happens when testboard 1 is OFF and only rx2 is connected to pin 13.
Note: in this case it is not required (but it can) to activate telemetry nor to connect the receiver to the TLM pin 10

nostromo123, 2022-12-13:
- Radio ON.

- RX1 disconnected 5V.  Only RX2 on 5V with TX -> pin 13 and GND.

- Power ON test board.

- RX2 starts blinking slow, trying to connect to radio, RP2040 does a short 1x time color cycle (green, red/blue, then blue) and the starts blinking green.

- RX2 connects, solid LED on RX2. RP2040 immediately changes over to multiple cycles  green, red/blue, then blue.

mstrens, 2022-12-13:
I found a bug. I just put a new version on github.
I did not test it but I hope it will fix the issue.

nostromo123, 2022-12-13:
Sorry the issue seems to be still there.TX RX2 --> pin 13. No boot, multiple cycles green, red/blue, then blue.

Then I pull TX RX2 and RP2040 stitches to green and test board working, reconnect TX RP2 and failure again.

Sorry -- good night ;-)

mstrens, 2022-12-13:
Thanks for having test.
I will look again at the issue.

mstrens, 2022-12-13:
I found another bug (due to cut and paste).
I put the new fix on github.

nostromo123, 2022-12-14:
Still not ok. Same as before.

mstrens, 2022-12-14:
Strange.
perhaps can you make one more tests to try to identify the reason of the bug:
What happens if you disable all sensors (so having GPS_TX, GPSRX, SCL, SDA, V1, V2, V3, V4 = 255) and set also PRI and TLM = 255.
So having only  SEC = 13 and some PWM pin defined.

nostromo123, 2022-12-14:
I de-cluttered the board a bit

before

[image]

now

[image]

```
Cmd to execute:   

Version = 0.3.1 
    Function              Code      Pin (255=disabled)
Primary channels input    PRI        255
Secondary channels input  SEC         13
Telemetry                 TLM        255
GPS Rx                    GPS_RX     255
GPS Tx                    GPS_TX     255
Sbus OUT                  SBUS_OUT   255
RPM                       RPM        255
SDA (baro sensor)         SDA        255
SCL (baro sensor)         SCL        255
PWM Channels 1, 2, 3 ,4   C1 / C4      0    1    2    3
PWM Channels 5, 6, 7 ,8   C5 / C8      4    5  255  255
PWM Channels 9,10,11,12   C9 / C12   255  255  255  255
PWM Channels 13,14,15,16  C13/ C16   255  255  255  255
Voltage 1, 2, 3, 4        V1 / V4    255  255  255  255

Protocol is CRSF (=ELRS)
CRSF baudrate = 420000
Voltage parameters:
    Scales : 0.972000 , 0.972000 , 0.972000 , 0.950000 
    Offsets: 6.000000 , 5.500000 , 6.000000 , 5.000000 
RPM multiplier = 1.000000
Baro sensor is not detected
Foreseen GPS type is Ublox  :GPS is not (yet) detected
Failsafe type is HOLD

Config parameters are OK
```
Same as before. Not working .....

nostromo123, 2022-12-14:
No luck. Only with SEC=13 and PWM channel 1 ... 4

Same as before. I tested with both RX and with both RP2040

When I power up the breadboard, the RP2040 does one cycle on color, then green blinking until the RX connects to the radio, then the color cycle starts ..... No PWM signal

nostromo123, 2022-12-14:
Fast cross check.

Changed to this setting. PRI instead of SEC

Have to push 1x RESET, but this is working!

```
Version = 0.3.1 
    Function              Code      Pin (255=disabled)
Primary channels input    PRI          9
Secondary channels input  SEC        255
Telemetry                 TLM        255
GPS Rx                    GPS_RX     255
GPS Tx                    GPS_TX     255
Sbus OUT                  SBUS_OUT   255
RPM                       RPM        255
SDA (baro sensor)         SDA        255
SCL (baro sensor)         SCL        255
PWM Channels 1, 2, 3 ,4   C1 / C4      0    1    2    3
PWM Channels 5, 6, 7 ,8   C5 / C8      4    5  255  255
PWM Channels 9,10,11,12   C9 / C12   255  255  255  255
PWM Channels 13,14,15,16  C13/ C16   255  255  255  255
Voltage 1, 2, 3, 4        V1 / V4    255  255  255  255

Protocol is CRSF (=ELRS)
CRSF baudrate = 420000
Voltage parameters:
    Scales : 0.972000 , 0.972000 , 0.972000 , 0.950000 
    Offsets: 6.000000 , 5.500000 , 6.000000 , 5.000000 
RPM multiplier = 1.000000
Baro sensor is not detected
Foreseen GPS type is Ublox  :GPS is not (yet) detected
Failsafe type is HOLD

Config parameters are OK
```

mstrens, 2022-12-14:
I do not understand anymore where is the bug.
Normally PRI and SEC uses now the same code except that PRI uses one of the hardware UART from the RP2040 and SEC the other one.
It seems that when SEC get ELRS frames, then program freezes and RP2040 restarts (as foreseen by a watchdog security process).
The question is : why does it freeze.
I will have to make a breadboard again to reproduce the issue and debug it.
Perhaps I can do it to morrow.

mstrens, 2022-12-14:
There is no trouble. I had never tested diversity but I though that it was OK because it was mainly a cut and paste of PRI code. I was wrong.
So, thanks again for testing.
From time to time I increase the version number but I am not 100% consistent because I do it not in all cases.
When the change is minor (one or 2 lines) or asks to activate some debug options, it is easy to explain which lines of code have to be changed but when there are many changes, it is better to just update all on github.
If I make so many changes that some main functions could be altered, best would be that I publish on github the changes in another branch (e.g. test instead of master).  
At this stage, I presume the project is used by many people and so the risk that someone download a new version having a new bug is quite limited.

mstrens, 2022-12-15:
I just made a test with this config
Version = 0.3.1
    Function              Code      Pin (255=disabled)
Primary channels input    PRI        255
Secondary channels input  SEC         13
Telemetry                 TLM        255
GPS Rx                    GPS_RX     255
GPS Tx                    GPS_TX     255
Sbus OUT                  SBUS_OUT   255
RPM                       RPM        255
SDA (baro sensor)         SDA        255
SCL (baro sensor)         SCL        255
PWM Channels 1, 2, 3 ,4   C1 / C4      0    1    2    3
PWM Channels 5, 6, 7 ,8   C5 / C8      4    5  255  255
PWM Channels 9,10,11,12   C9 / C12   255  255  255  255
PWM Channels 13,14,15,16  C13/ C16   255  255  255  255
Voltage 1, 2, 3, 4        V1 / V4    255  255  255  255

Protocol is CRSF (=ELRS)
CRSF baudrate = 420000
Voltage parameters:
    Scales : 1.000000 , 1.000000 , 1.000000 , 1.000000
    Offsets: 0.000000 , 0.000000 , 0.000000 , 0.000000
RPM multiplier = 1.000000
Baro sensor is not detected
Foreseen GPS type is Ublox  :GPS is not (yet) detected
Failsafe type is HOLD

Config parameters are OK

This config is in principe the same as your config (only SEC and some PWM)
On my side it works: the rp2040 led is blinking green and the servo I connect is just moving accordingly to the handset.
So, I do not understand what happens at your side.
Based on what you explained, it looks like your RP2040 is rebooting. Could it be that there is an issue with your power supply?
Still I do not understand why this would happen only with SEC and not with PRI.
Are you using 2 different models of elrs receiver?Did you already to try to exchange the receiver (connect to SEC the receiver being  currently connected to PRI)?
Can you also try to slow down the rate of your ELRS transmitter. I do not remember the rate I am using on my side but it should probably be less than 250hz

mstrens, 2022-12-15:
Sorry, I have perhaps an explanation.
I made my tests with the latest version I had on my laptop.
It seems that this version was not uploaded yesterday on github. I do not understand where I made the mistake with github.
I put this latest version 0.3.2 now on github.
Please test again with this version.

nostromo123, 2022-12-15:
Phew!

WORKING NOW with this setting

```
Version = 0.3.2 
    Function              Code      Pin (255=disabled)
Primary channels input    PRI          9
Secondary channels input  SEC         13
Telemetry                 TLM         10
GPS Rx                    GPS_RX     255
GPS Tx                    GPS_TX     255
Sbus OUT                  SBUS_OUT   255
RPM                       RPM        255
SDA (baro sensor)         SDA        255
SCL (baro sensor)         SCL        255
PWM Channels 1, 2, 3 ,4   C1 / C4      0    1    2    3
PWM Channels 5, 6, 7 ,8   C5 / C8    255  255  255  255
PWM Channels 9,10,11,12   C9 / C12   255  255  255  255
PWM Channels 13,14,15,16  C13/ C16   255  255  255  255
Voltage 1, 2, 3, 4        V1 / V4    255  255  255  255

Protocol is CRSF (=ELRS)
CRSF baudrate = 420000
Voltage parameters:
    Scales : 1.000000 , 1.000000 , 1.000000 , 1.000000 
    Offsets: 0.000000 , 0.000000 , 0.000000 , 0.000000 
RPM multiplier = 1.000000
Baro sensor is not detected
Foreseen GPS type is Ublox  :GPS is not (yet) detected
Failsafe type is HOLD

Config parameters are OK
```
Was a bit difficult, but good to know it works now ;-)

Regards and keep on the good work - KH

closed 2022-12-15 by nostromo123