mstrens/oXs_on_RP2040 #15: GPS does not seem to work with Jeti protocol

onki69, 2022-12-16:
I was triing to use a MS5611 together with a uBlox 6 GPS module and Jeti protocol.
According to the command prompt the GPS is detected correctly but I can't see any GPS related data on my transmitter.
Just vario and height and V1. Even with a valid 3D fix at the GPS no data
When I switch over with the same configuration to S.Port I can see all GPS data on my Transmitter (TX16S with R168).
So my hardware seems to work properly.

mstrens, 2022-12-16:
I do not have a Jeti TX/RX so I never could test the Jeti protocol.
I just copied the code for Jeti protocol from the AVR openXsensor project.

It is very strange that the jeti protocol works for other telemetry fields (vario and Voltage) and not for GPS because I do not remember having made changes for GPS on Jeti protocol.

If you have time for debugging, I could make a "test" branch on github and add some debug messages that should be displayed on the serial monitor in order to see if RP2040 get the data from the GPS and if it generates some messages on TLM port.

mstrens, 2022-12-16:
Perhaps I found why gps are not transmitted;
In file jeti.cpp, there is currently:
if ( gps.gpsInstalled ) {

can you replace it with
if ( config.pinGpsTx != 255 ) {

NB : I made the change also on github

onki69, 2022-12-16:
I tried to flash the UF" file but after that I end up in an endless boot cycle.
Will try to flash it through VSCode later.

mstrens, 2022-12-16:
I uploaded the version that is currently on github.
I did not connect the RP2040 to sensor nor receiver.
I used this setup and led is green blinking (so it seems OK)

Version = 0.3.3
    Function              Code      Pin (255=disabled)
Primary channels input    PRI     =  255
Secondary channels input  SEC     =   13
Telemetry                 TLM     =   10
GPS Rx                    GPS_RX  =    8
GPS Tx                    GPS_TX  =    7
Sbus OUT                  SBUS_OUT=  255
RPM                       RPM     =  255
SDA (baro sensor)         SDA     =  255
SCL (baro sensor)         SCL     =  255
PWM Channels 1, 2, 3 ,4   C1 / C4 =    1    2  255  255
PWM Channels 5, 6, 7 ,8   C5 / C8 =  255  255  255  255
PWM Channels 9,10,11,12   C9 / C12=  255  255  255  255
PWM Channels 13,14,15,16  C13/ C16=  255  255  255  255
Voltage 1, 2, 3, 4        V1 / V4 =  255  255  255  255

Protocol is CRSF (=ELRS)
CRSF baudrate = 420000
Voltage parameters:
    Scales : 1.000000 , 1.000000 , 1.000000 , 1.000000
    Offsets: 0.000000 , 0.000000 , 0.000000 , 0.000000
RPM multiplier = 1.000000
Baro sensor is not detected
Foreseen GPS type is Ublox  :GPS is not (yet) detected
Failsafe type is HOLD

Note : I just uploaded a new version on github but the only change is the version number (and adding some "=" on the display

onki69, 2022-12-17:
I had some time to do further tests. I currently use 0.2.9 in VS Code.
When I change the code in Jeti.cpp as suggested I do get the boot loop.
As soon as I use the original code (which might be wrong) the device runs normal, just without GPS Telemetry.
I just need to copy the latest revision for further testing.

mstrens, 2022-12-17:
Thanks. No need to do more tests at this stage. I can reproduce the issue.
I just realized that in my previous test, I did not activated the Jeti protocol.

mstrens, 2022-12-17:
I just put a new version on github.
It does not block anymore on my side.
Still I did not tested with a real GPS.
Please note that I made several other changes (that are not yet added to the doc):
Led is red fix when config is wrong
When config is OK, led is blinking and then color is
- red when oXs never got a RC frame
- green when oXs get a valid RC frame (
- blue when oXs generates failsafe (on Sbus or PWM)
- yellow or orange when oXs is configured to get both PRI and SEC and it get a valid frame only from one source and Failsafe or no frame from the other source. 
I also made changes in order to better manage priorities between PRI and SEC for Sbus (taking care e.g. on the missing and faisafe flags present in the 2 sbus frames.

mstrens, 2022-12-17:
Note: in current version, oXs does not provide the number of satellites in Jeti protocol.
When we are sure that this version works, I can try to add the number of sat for Jeti protocol.

onki69, 2022-12-18:
Was able to test this version.
Compiled it and flashed it to the RP2040.
After booting the red LED keeps on blinking which is normal as no RC signal is connected (Pri=255)
GPS is blinking (3D fix) but no telemetry on the transmitter.
Doiuble check with R168 / TX16s will follow to verify GPS setup.

mstrens, 2022-12-18:
Can you test this once more with version 0.3.5

onki69, 2022-12-18:
Bingo - with 0.3.5 it works. Just wait for the 3D fix to see if real data arrives.
Could take some time as the GPS reception is not that good at my workshop window.
Thanks anyway - I owe you a coffee or a beer.

Update: Data is OK with 3D fix

closed 2022-12-18 by onki69