mstrens/oXs_on_RP2040 #16: 0.3.4 not an advantage here. NO PWM signal!

nostromo123, 2022-12-17:
I am back to zero.

Put 0.3.4 on one of my RP2040. Now blinking red

I only have PRI on 9, TLM on 10 and SEC on 13.

No GPS, no baro, no voltages.

Start the radio and the power up the RP2040. PRI, TLM and SEC not! connected to RP2040.

The receivers connect to the radio. Then I connect receiver #1 TX to PIN 9 of RP2040.

Red blinking stops and blue LED.

NO PWM. 

I would like to have 0.3.3 back please!

i am on CRSF/ELRS ...

br - KH

mstrens, 2022-12-17:
I will make some more tests to morrow.
In the mean time, you can download the files from any version on github.
When you are on the main screen on github, there is a text (on the top of list, right) saying "80 commits" 
Click on it and you get a list with all update. 
On the item you want, click on "<>" and you get a list of all files at this timestamp.
You can then download  the files you want.

nostromo123, 2022-12-17:
The issue occurs between these two versions

[image]

Thank you for enlighten me how github works ;-)

br - KH  ......and a hopefully nice and warm weekend!

nostromo123, 2022-12-17:
OMG! Not only 1 or two line of code changed, but a lot!

[image]

mstrens, 2022-12-17:
comparing the 2 versions, I noticed a typo fout for Jeti protocol 
It should be (in jeti.cpp)
if ( config.pinGpsTx != 255 ) {

I updated on github.

You closed this isssue but I presume the bug is not solved (in latest version).
The update I just made concerns only Jeti and so should not explain the issue in CRSF.

nostromo123, 2022-12-17:
`You closed this isssue but I presume the bug is not solved (in latest version). The update I just made concerns only Jeti and so should not explain the issue in CRSF.`

You are right, so keep it open .....

mstrens, 2022-12-18:
I put a new version 0.3.5 on github.
I hope it works.
I tested it with ELSR protocol (LED seems OK and PWM is generated)

nostromo123, 2022-12-18:
WORKS!!

closed 2022-12-18 by nostromo123