mstrens/oXs_on_RP2040 #17: FeatureRequest: data logging on SDcard

NafetsRC, 2023-01-11:
Hi there,

this is my first posting here. please advise, if this is the wrong place…

just successfully installed the compiled firmware. a gps module is already ordered 😉

i was looking and would appreciate a possibility to log all the data directly in the RC Model on MicroSD. Similar to JLog or SM gps logger 

Sorry, i can‘t examine, whether this would be technically possible or not

regards 
stefan

pierrotm777, 2023-02-11:
May be with the add of an OpenLog module?
Good idea.

fourfingers4, 2023-03-24:
I believe onboard logging is the only way to go if you want to use data for tuning or analysis. 
RC telemetry systems I have access to, won´t log better than 2Hz on vital parameters for later analysis, making data too blocky and unreliable.

mstrens, 2023-11-21:
A logger functionality has been added to oXs (using a second RP2040).
Some users reported that it was OK.
If not, please open a new issue.

closed 2023-11-21 by mstrens