mstrens/oXs_on_RP2040 #19: MS5611 not working anymore

nostromo123, 2023-01-22:
I start an issue here on the trouble, we have lately with MS5611 sensors on oXs_on_RP2040.

See postings in RGB thread.

My equipment, I have here on the workbench ..

FrSky X10S Express running EdgeTX 2.7.1

board 1 ELRS

[image]

board 2 SPort

[image]

Vario sensors

[image]

mstrens, 2023-01-22:
I hope that version 0.4.3 that I uploaded now on github in hott branch solve the issue.

nostromo123, 2023-01-22:
Board 1 ELRS

```
Version = 0.4.3 
    Function              Code      Pin (255=disabled)
Primary channels input    PRI     =    9
Secondary channels input  SEC     =  255
Telemetry                 TLM     =   10
GPS Rx                    GPS_RX  =   11
GPS Tx                    GPS_TX  =   12
Sbus OUT                  SBUS_OUT=  255
RPM                       RPM     =  255
SDA (baro sensor)         SDA     =   14
SCL (baro sensor)         SCL     =   15
PWM Channels 1, 2, 3 ,4   C1 / C4 =    0    1    2    3
PWM Channels 5, 6, 7 ,8   C5 / C8 =    4    5  255  255
PWM Channels 9,10,11,12   C9 / C12=  255  255  255  255
PWM Channels 13,14,15,16  C13/ C16=  255  255  255  255
Voltage 1, 2, 3, 4        V1 / V4 =   26   27   28   29

Protocol is CRSF (=ELRS)
CRSF baudrate = 420000
Voltage parameters:
    Scales : 0.972000 , 0.972000 , 0.972000 , 0.950000 
    Offsets: 6.000000 , 5.500000 , 6.000000 , 5.000000 
RPM multiplier = 1.000000
Baro sensor is detected using MS5611
    Sensitivity min = 100 (at 100)   , max = 300 (at 1000)
    Hysteresis = 5 
First analog to digital sensor is not detected
Second analog to digital sensor is not detected
Foreseen GPS type is Ublox  :GPS is detected
Failsafe type is HOLD

Config parameters are OK

 >> 

vspeed=3
vspeed=3
vspeed=3
vspeed=3
vspeed=3
vspeed=3
vspeed=3
vspeed=9
vspeed=16
vspeed=23
vspeed=29
vspeed=35
vspeed=42
vspeed=49
vspeed=56
vspeed=62
vspeed=69
vspeed=69
vspeed=69
vspeed=69
vspeed=61
vspeed=61
vspeed=54
vspeed=47
vspeed=40
vspeed=34
vspeed=34
vspeed=27
vspeed=19
vspeed=2
vspeed=-7
vspeed=-17
vspeed=-33
vspeed=-33
vspeed=-33
vspeed=-39
vspeed=-39
vspeed=-39
vspeed=-45
vspeed=-45
vspeed=-45
vspeed=-45
vspeed=-45
vspeed=-37
vspeed=-30
vspeed=-21
vspeed=-3
vspeed=6
vspeed=16
vspeed=25
vspeed=43
vspeed=50
vspeed=56
vspeed=56
vspeed=62
vspeed=62
vspeed=62
vspeed=62
vspeed=62
vspeed=49
vspeed=42
vspeed=33
vspeed=12
vspeed=3
vspeed=-5
vspeed=-13
vspeed=-29
vspeed=-36
vspeed=-42
vspeed=-48
vspeed=-48
vspeed=-54
vspeed=-54
vspeed=-54
vspeed=-54
vspeed=-47
vspeed=-39
vspeed=-32
vspeed=-13
vspeed=-3
vspeed=14
vspeed=27
vspeed=44
vspeed=56
vspeed=64
vspeed=70
vspeed=76
vspeed=82
vspeed=82
vspeed=82
vspeed=82
vspeed=74
vspeed=61
vspeed=55
vspeed=46
vspeed=27
vspeed=21
vspeed=4
vspeed=-5
vspeed=-23
vspeed=-32
vspeed=-39
vspeed=-47
vspeed=-54
vspeed=-61
vspeed=-61
vspeed=-61
vspeed=-67
vspeed=-67
vspeed=-67
vspeed=-67
vspeed=-60
vspeed=-52
vspeed=-45
vspeed=-39
vspeed=-39
vspeed=-33
vspeed=-33
vspeed=-27
vspeed=-27
vspeed=-21
vspeed=-21
vspeed=-21
vspeed=-15
vspeed=-15
vspeed=-15
vspeed=-15
vspeed=-15
vspeed=-9
vspeed=-9
vspeed=-9
vspeed=-9
vspeed=-9
vspeed=-9
vspeed=-9
vspeed=-9
vspeed=-9
vspeed=-3
vspeed=-3
vspeed=-3
vspeed=-3
vspeed=3
vspeed=3
vspeed=3
vspeed=3
vspeed=3
```

Board 2 SPort

```
Version = 0.4.3 
    Function              Code      Pin (255=disabled)
Primary channels input    PRI     =    9
Secondary channels input  SEC     =  255
Telemetry                 TLM     =   10
GPS Rx                    GPS_RX  =  255
GPS Tx                    GPS_TX  =  255
Sbus OUT                  SBUS_OUT=  255
RPM                       RPM     =  255
SDA (baro sensor)         SDA     =   14
SCL (baro sensor)         SCL     =   15
PWM Channels 1, 2, 3 ,4   C1 / C4 =    0    1    2    3
PWM Channels 5, 6, 7 ,8   C5 / C8 =  255  255  255  255
PWM Channels 9,10,11,12   C9 / C12=  255  255  255  255
PWM Channels 13,14,15,16  C13/ C16=  255  255  255  255
Voltage 1, 2, 3, 4        V1 / V4 =  255  255  255  255

Protocol is Sport (Frsky)
CRSF baudrate = 420000
Voltage parameters:
    Scales : 1.000000 , 1.000000 , 1.000000 , 1.000000 
    Offsets: 0.000000 , 0.000000 , 0.000000 , 0.000000 
RPM multiplier = 1.000000
Baro sensor is detected using MS5611
    Sensitivity min = 100 (at 100)   , max = 300 (at 1000)
    Hysteresis = 5 
First analog to digital sensor is not detected
Second analog to digital sensor is not detected
Foreseen GPS type is Ublox  :GPS is not (yet) detected
Failsafe type is HOLD

Config parameters are OK

 >> 
```
**No additional output here on board 2 SPort in Terminal.

AT THE MOMENT on both boards, MS5611 ALT and VSPD WORKING!**

nostromo123, 2023-01-22:
> I hope that version 0.4.3 that I uploaded now on github in hott branch solve the issue.

Could you please give me the lines to modify in VERSION "0.3.6". MS5611.h/MS5611.cpp

I can modify 'main' here, build it and test it locally before you modify 'main' on github -----

mstrens, 2023-01-22:
I made quite many changes/unchanges to debug so I do not remember what has exactly been changes but I think that this is the critical part to fix the bug (in MS5611.cpp):
int64_t dT = ((int64_t)((_D2+ _D2Prev) >> 1 )) - (( (uint64_t)_calibrationData[5]) << 8) ;
  int32_t TEMP = 2000 + ((dT * ((int64_t)_calibrationData[6])) >> 23)  ;
  temperature = TEMP;
  _D2Prev = _D2 ;
  int64_t OFF  = (((int64_t)_calibrationData[2]) << 16) + ((_calibrationData[4] * dT) >> 7);
  int64_t SENS = (((int64_t)_calibrationData[1]) << 15) + ((_calibrationData[3] * dT) >> 8);
  int64_t rawPressure= (((((((int64_t) _D1) * SENS) >> 21) - OFF) * 10000 ) >> 15) ; // 1013.25 mb gives 1013250000 is a factor to keep higher precision (=1/100 cm).
  
Note: I expect to put hott branch into main branch quite soon.

nostromo123, 2023-01-22:
Modified that part in 0.3.6 MS5611.cpp according to your proposal

```
  int64_t dT = ((int64_t)((_D2+ _D2Prev) >> 1 )) - (( (uint64_t)_calibrationData[5]) << 8);
  int32_t TEMP = 2000 + ((dT * ((int64_t)_calibrationData[6])) >> 23);
  temperature = TEMP;
  _D2Prev = _D2 ;
  int64_t OFF = (((int64_t)_calibrationData[2]) << 16) + ((_calibrationData[4] * dT) >> 7);
  int64_t SENS = (((int64_t)_calibrationData[1]) << 15) + ((_calibrationData[3] * dT) >> 8);
  int64_t rawPressure= (((((((int64_t) _D1) * SENS) >> 21) - OFF) * 10000 ) >> 15) ; // 1013.25 mb gives 1013250000 is a factor to keep higher precision (=1/100 cm).
   
//  int64_t dT = ((_D2+ _D2Prev) >> 1 ) - ((long)_calibrationData[5] << 8);
//  int32_t TEMP = (2000 + (((int64_t)dT * (int64_t)_calibrationData[6]) >> 23)) / (float) 1.0 ;
//  temperature = TEMP;
//  _D2Prev = _D2 ;
//  int64_t OFF  = (((int64_t)_calibrationData[2]) << 16) + ((_calibrationData[4] * dT) >> 7);
//  int64_t SENS = (((int64_t)_calibrationData[1]) << 15) + ((_calibrationData[3] * dT) >> 8);
//  int64_t rawPressure= (((((((int64_t) _D1) * (int64_t) SENS) >> 21) - OFF) * 10000 ) >> 15) ; // 1013.25 mb gives 1013250000 is a factor to keep higher precision (=1/100 cm).
```

Building and upload ok.

Working now on both breadboards. 

As temp here is below 20°C, this would maybe also explain, why my old MS5611.ino sketch refuses to work properly.

br KH

nostromo123, 2023-01-22:
> As temp here is below 20°C, this would maybe also explain, why my old MS5611.ino sketch refuses to work properly.

Put the Arduino and the MS5611 on a warm place, now that 'old' MS5611 test sketch is working again ...

closed 2023-01-22 by nostromo123