mstrens/oXs_on_RP2040 #20: Problems on Ver0.4.3_hott

Satcomix, 2023-01-24:
Hello Michel Strens,
I've spent the last few weeks getting a little more involved with openXsensor_on_328P and oXs_on_RP2040.
I tested with several RC systems.
On the one hand the S.Port protocol with FrSky X20S Ethos 1.4.6 and TD-R10 receiver and on the other hand the HOTT protocol with a Graupner MC28 4D with GR-24 receiver.
I tested with firmware up to 0.3.6_main and up to 0.4.3_hott.
As a test board I have several
Pi Pico,
Waveshare RP2040Plus,
Cytron Maker RP2040,
Cytron Maker Pico Base and my favorite board,
Pimoroni Servo2040.
All functions were tested several times on all boards.
With version 0.3.6 I could not find any noticeable error. All sensors were displayed in the FrSky Ethos telemetry with plausible values and the SBUS in or out showed me 16 channels. Can i change the predefiened PWM Output from 992 to 1500???
Ver.0.4.0 can be loaded without any problems, from Ver.0.4.1 the problems with the hardware start on some boards, with the Pimoroni Servo2040 there are massive problems. USB logs in and out, the board can no longer be addressed. These problems do not exist with Ver 0.3.6 all ports work fine, even when I reprogrammed these IO ports with Platformio, e.g. RGB LED IO 16 to 18 or 28 and I2C1 to I2C0 for the sensors, because of the onboard QT connector. Everything worked until version 0.4.1.
There are also small blemishes in the code, so offset and scale are not displayed correctly by the ADS1115.
Offset is contained twice in the Platformio script, I exchanged it with Scale.
I am available for more tests with S.Port and HOTT protocol.
I hope i can help you a little bit.
Thank you very much for your effort,
Best regards,
Torsten

In Version = 0.3.6 everything is ok!!!!
Function Code Pin (255=disabled)
Primary channel input PRI = 5
Secondary channels input SEC = 1
Telemetry TLM = 0
GPS Rx GPS_RX = 16
GPS Tx GPS_TX = 17
Sbus OUT SBUS_OUT= 11
RPM RPM = 255
SDA (baro sensor) SDA = 20
SCL (baro sensor) SCL = 21
PWM channels 1, 2, 3, 4 C1 / C4 = 2 3 4 6
PWM channels 5, 6, 7, 8 C5 / C8 = 7 8 9 10
PWM Channels 9,10,11,12 C9 / C12= 255 255 255 255
PWM Channels 13,14,15,16 C13/ C16= 255 255 255 255
Voltage 1, 2, 3, 4 V1 / V4 = 26 27 28 29

Protocol is Sport (Frsky)
CRSF baud rate = 420000
Voltage parameter:
    Scales : 10.000000 , 0.100000 , 0.200000 , 3.000000
    Offsets: -25.000000 , -0.250000 , 0.000000 , 0.000000
RPM multiplier = 1.000000
Baro sensor is detected using MS5611
    Sensitivity min = 100 (at 100) , max = 300 (at 1000)
    Hysteresis = 5
Foreseen GPS type is Ublox :GPS is detected
Failsafe uses predefined values
     Chan 1...4 = 992 , 992 , 992 , 992
     Chan 5...8 = 992 , 992 , 992 , 992
     Chan 9...12 = 992 , 992 , 992 , 992
     Chan 13...16= 992 , 992 , 992 , 992

Config parameters are OK

Version = 0.4.3 
    Function              Code      Pin (255=disabled)
Primary channels input    PRI     =    5
Secondary channels input  SEC     =  255
Telemetry                 TLM     =   12
GPS Rx                    GPS_RX  =    0
GPS Tx                    GPS_TX  =    1
Sbus OUT                  SBUS_OUT=   13
RPM                       RPM     =  255
SDA (baro sensor)         SDA     =    2
SCL (baro sensor)         SCL     =    3
PWM Channels 1, 2, 3 ,4   C1 / C4 =  255  255   14   15
PWM Channels 5, 6, 7 ,8   C5 / C8 =  255  255  255  255
PWM Channels 9,10,11,12   C9 / C12=  255  255  255  255
PWM Channels 13,14,15,16  C13/ C16=  255  255  255  255
Voltage 1, 2, 3, 4        V1 / V4 =   26   27   28   29

Protocol is Sport (Frsky)
CRSF baudrate = 420000
Voltage parameters:
    Scales : 10.000000 , 0.100000 , 0.200000 , 0.435500 
    Offsets: -15.000000 , -0.200000 , -0.400000 , -0.026500 
RPM multiplier = 1.000000
Baro sensor is detected using MS5611
    Sensitivity min = 100 (at 100)   , max = 300 (at 1000)
    Hysteresis = 5 
First analog to digital sensor is detected using ads1115
    Measurement setup: 4 , 5 , 8 ,8
    Gains: 2 , 2 , 2 ,2
    Rates: 3 , 3 , 4 ,4
    Offsets: 1.000000 , 1.000000 , 0.000000 ,0.000000
    Scales: 10.000000 , 0.100000 , 1.000000 ,1.000000
    Averaged on: 1 , 1 , 1 ,1
Second analog to digital sensor is detected using ads1115
    Measurement setup: 4 , 5 , 8 , 8
    Gains: 2 , 2 , 2 , 2
    Rates: 3 , 3 , 4 , 4
    Offsets: 1.000000 , 1.000000 , 0.000000 , 0.000000
    Scales: 10.000000 , 0.100000 , 1.000000 , 1.000000
    Averaged on: 1 , 1 , 1 , 1
Foreseen GPS type is Ublox  :GPS is detected
Failsafe uses predefined values
     Chan 1...4  = 992 , 992 , 992 , 992
     Chan 5...8  = 992 , 992 , 992 , 992
     Chan 9...12 = 992 , 992 , 992 , 992
     Chan 13...16= 992 , 992 , 992 , 992

Config parameters are OK

 >>

mstrens, 2023-01-24:
Thanks for the feedback.
I will try to find the bugs you noticed.
I am currently trying to add MPU6050 to calculate Vspeed as requested by a user.

Hott protocol has not yet been tested (by me) and so it is not "abnormal" that it does not yet work.

About the issue with 0.4.1, I have no idea yet what is causing the issue.
Could it be that it you first load 0.3.6 (and set up a first config - pin, protocol,...) with this version), then you can perhaps load the latest version (0.4.x). If so, it mean that the error should be in the way the config parameters are saved/retrieved.
I saw on rcgroups such a report.

Satcomix, 2023-01-25:
Hello mstrens,
Thank you for the fast feedback.
I did some tests again with the S.Port protocol and versions 0.4.0-hott and 0.4.3-hott.
Before each test setup, the board was set to the delivery state with flash_nuke.uf2
Conclusion:
the UART TXD0 GpIO 16 and RXD0 GpIO 17 do not send any data to the GPS. No GPS Sensor Telemetrie on these IO to X20S with version 0.4.0! I2C1 connection to GY86 works. 
GpIO 12 and GpIO13 worked fine and sent data to GPS for set baud rate to 38.400 and FrSky X20S Ethos 1.4.6 received telemetry data.
With an old config (e.g. 0.3.6) on the board, Ver.0.4.3 can be uploaded and used without any problems, also when you make a reset.
After a factory reset with flash_nuke.uf2 and installing Ver.0.4.3 there are problems with the USB Uart. on/off every second.
Something is wrong with the uarts since Ver.0.4.0., on Ver.0.3.6 i can use every IO pin without Problems.
Where can I download the complete version 0.4.0 as a zip?
Best regards,
Torsten

mstrens, 2023-01-25:
To get 0.4.0, you can:
- first select hott branch on github
- click on "88 commits"
- on 20/1 there have been 2 PR. On the one having an id begining with 72...., click on <>
- You have now access to all files from this version.

Satcomix, 2023-01-25:
Now,I get Ver.0.4.0 working in Platformio.
Thank you very much.

mstrens, 2023-01-25:
I think I fixed some bugs that caused the RP2040 to reboot.
New version (0.4.5) is on github in hott branch.

This version contains some code to support mpu6050 but this part is not yet completed.

Satcomix, 2023-01-26:
Hello mstrens,
Thanks for the Ver.0.4.5-hott.
I have tested the new software with Graupner MC28-GR24 HOTT protocol and FrSky X20S and S.Port protocol and have not found any noticeable errors in S.Port protocol so far, HOTT doesnt work.
The new version can now be uploaded to a fresh board without any problems, without a pre-config of a previous version.
The only thing I noticed is the error message about the SBUS "sbusQueue try add error sbusQueue try add error......." when you want to configure V1 to V4. But it is definitely related to the output of the ADS1 and ADS2 voltages.
Too much traffic on the UART?
Due to the innovations with ADS1115 (and MPU6050), the restart of the board after a reset has become a bit longer, recognizable by the RGB Led.
In param.cpp "change TX Pin" for "change SBUS_Out Pin" is present twice (GPS-SBUS).
In the ADS1115 settings line 645 and 656 change ads_Offset to ads_Scale.
Thank you for your great work.
Regards,
Torsten

ADC 1 : 771 mVoltAdc 0 : 773 mVoltAdc 1 : 771 mVoltAdc 0 : 773 mVolt
Cmd to execute:   

Version = 0.4.5 
    Function              Code      Pin (255=disabled)
Primary channels input    PRI     =    5 -> ok
Secondary channels input  SEC     =    1 -> ok
Telemetry                 TLM     =    0 -> ok
GPS Rx                    GPS_RX  =   16 -> ok
GPS Tx                    GPS_TX  =   17 -> ok
Sbus OUT                  SBUS_OUT=    2 -> all 16ch ok
RPM                       RPM     =  255
SDA (baro sensor)         SDA     =   14 -> GY86- 2xADS1115 ok
SCL (baro sensor)         SCL     =   15 -> GY86- 2xADS1115 ok
PWM Channels 1, 2, 3 ,4   C1 / C4 =  255  255    6    7 -> all 16ch ok
PWM Channels 5, 6, 7 ,8   C5 / C8 =  255  255  255  255
PWM Channels 9,10,11,12   C9 / C12=  255  255  255  255
PWM Channels 13,14,15,16  C13/ C16=  255  255  255  255
Voltage 1, 2, 3, 4        V1 / V4 =   26   27   28   29 -> ok

PinIO 16 RGB LED all 6 colours ok

Protocol is Sport (Frsky) -> ok
CRSF baudrate = 420000
Voltage parameters:
    Scales : 10.000000 , 0.100000 , 0.200000 , 3.000000 
    Offsets: -25.000000 , -0.250000 , 0.000000 , 0.000000 
RPM multiplier = 1.000000
Baro sensor is detected using MS5611 -> ok
    Sensitivity min = 100 (at 100)   , max = 300 (at 1000)
    Hysteresis = 5 
First analog to digital sensor is detected using ads1115 -> ok
    Measurement setup: 4 , 8 , 8 ,8
    Gains: 2 , 2 , 2 ,2
    Rates: 3 , 3 , 3 ,3
    Offsets: 0.000000 , 0.000000 , 0.000000 ,0.000000
    Scales: 10.000000 , 1.000000 , 0.100000 ,4.000000
    Averaged on: 10 , 10 , 10 ,10
Second analog to digital sensor is detected using ads1115 -> ok
    Measurement setup: 4 , 8 , 8 ,8
    Gains: 2 , 2 , 2 ,2
    Rates: 3 , 3 , 3 ,3
    Offsets: 0.000000 , 0.000000 , 0.000000 ,0.000000
    Scales: 10.000000 , 1.000000 , 0.200000 ,3.900000
    Averaged on: 10 , 10 , 10 ,10
Foreseen GPS type is Ublox  :GPS is detected -> ok
Failsafe uses predefined values
     Chan 1...4  = 992 , 992 , 992 , 992
     Chan 5...8  = 992 , 992 , 992 , 992
     Chan 9...12 = 992 , 992 , 992 , 992
     Chan 13...16= 992 , 992 , 992 , 992

Config parameters are OK

 >> 
Adc 0 : 773 mVoltAdc 1 : 771 mVoltAdc 0 : 773 mVoltAdc 1 : 771 mVolt

mstrens, 2023-01-26:
Thanks for the feedback.
Please note that I do not yet looked why hott protocol is not working.
When done, I will put a message on github issues section.
I do not have the hardware to test this part. So I can just look at the code and add some debug messages.
So probably I will require your help.

I fixed the typo fout in param.cpp. Will be in next version.
Increase of the starting time is mainly due to the fact that I compiled this version with the the DEBUG option (in config).
This added a delay of 1.5 sec to be sure that platformio monitor is up and running and so I can get all messages.
In production version, this should be avoided.

Still when MPU6050 is present, it can take some time to upload the DMP firmware into the MPU6050.
I do not yet know if I will use the DMP option for MPU6050 or if I will use my own algorithm to calculate the 3D positions.

Currently I am trying to let all the code about sensors (voltage, baro, imu, gps) run a the second core of the RP2040.
So the first core will have more "free" time to respect all timings of communications with receiver, Sbus and PWM.
I think this could be useful because handling MPU6050 requires a lot of calculation.

closed 2023-01-28 by Satcomix