mstrens/oXs_on_RP2040 #21: ADS1115 with 0.4.5 on ELRS here

nostromo123, 2023-01-27:
Did setup ADS1 in config.h these  2 lines as follows

```
#define ADS1_OFFSET 0.0, 0.0 , 0.0 , 0.0 // can be a float (positive or negative)
#define ADS1_SCALE 1.0, 1.0, 1.0, 1.0 // can be a float 
```
In the terminal I see

```
First analog to digital sensor is detected using ads1115
Measurement setup: 4 , 8 , 8 ,8
Gains: 0 , 0 , 2 ,2
Rates: 3 , 3 , 7 ,7
Offsets: 0.000000 , 0.000000 , 0.000000 ,0.000000
Scales: 0.000000 , 0.000000 , 0.000000 ,0.000000
Averaged on: 20 , 20 , 50 ,50 
```

Changed in config.h to

```
#define ADS1_OFFSET 0.5, 0.0 , 0.0 , 0.0 // can be a float (positive or negative)
#define ADS1_SCALE 1.0, 1.0, 1.0, 1.0 // can be a float 
```

and in terminal I get

```
First analog to digital sensor is detected using ads1115
Measurement setup: 4 , 8 , 8 ,8
Gains: 0 , 0 , 2 ,2
Rates: 3 , 3 , 7 ,7
Offsets: 0.500000 , 0.000000 , 0.000000 ,0.000000
Scales: 0.500000 , 0.000000 , 0.000000 ,0.000000
Averaged on: 20 , 20 , 50 ,50 
```

Satcomix just told me in German forum, that this has been reported before and it is just a simple thing to solve this issue. But as it is out-of-my-scope to search for causes and it does not make sense to do further testing with a modify your code here, it would be nice if you could make the necessary changes.

******************************************

But there seems to be a 2. issue here, which Satcomix does not see on his testboard.

Adding in config.h a second input

`#define ADS1_MEASURE A0_TO_GND , ADS1_MEASURE A1_TO_GND , ADS_OFF , ADS_OFF`

the here compiler drops out

[image]

br KH

Satcomix, 2023-01-27:
By your first issue i can help.
When you change it, everything is ok in Terminal.
    In "param.cpp" Line 645 and 656 there are some mistakes in  printf(" Scales: it must be "ads_Scale" and not "ads_Offset"
  
    if (adc1.adsInstalled) {
        printf("First analog to digital sensor is detected using ads1115\n")  ;
        printf("    Measurement setup: %i , %i , %i ,%i\n", ads_Measure[0][0], ads_Measure[0][1], ads_Measure[0][2], ads_Measure[0][3]) ;
        printf("    Gains: %i , %i , %i ,%i\n", ads_Gain[0][0], ads_Gain[0][1], ads_Gain[0][2], ads_Gain[0][3]) ;
        printf("    Rates: %i , %i , %i ,%i\n", ads_Rate[0][0], ads_Rate[0][1], ads_Rate[0][2], ads_Rate[0][3]) ;
        printf("    Offsets: %f , %f , %f ,%f\n", ads_Offset[0][0], ads_Offset[0][1], ads_Offset[0][2], ads_Offset[0][3]) ;
        printf("    Scales: %f , %f , %f ,%f\n", ads_Scale[0][0], ads_Scale[0][1], ads_Scale[0][2], ads_Scale[0][3]) ;<-----------------here
        printf("    Averaged on: %i , %i , %i ,%i\n", ads_MaxCount[0][0], ads_MaxCount[0][1], ads_MaxCount[0][2], ads_MaxCount[0][3]) ;
    } else {
        printf("First analog to digital sensor is not detected\n")  ;
    }
    if (adc2.adsInstalled) {
        printf("Second analog to digital sensor is detected using ads1115\n")  ;
        printf("    Measurement setup: %i , %i , %i ,%i\n", ads_Measure[1][0], ads_Measure[1][1], ads_Measure[1][2], ads_Measure[1][3]) ;
        printf("    Gains: %i , %i , %i ,%i\n", ads_Gain[1][0], ads_Gain[1][1], ads_Gain[1][2], ads_Gain[1][3]) ;
        printf("    Rates: %i , %i , %i ,%i\n", ads_Rate[1][0], ads_Rate[1][1], ads_Rate[1][2], ads_Rate[1][3]) ;
        printf("    Offsets: %f , %f , %f ,%f\n", ads_Offset[1][0], ads_Offset[1][1], ads_Offset[1][2], ads_Offset[1][3]) ;
        printf("    Scales: %f , %f , %f ,%f\n", ads_Scale[1][0], ads_Scale[1][1], ads_Scale[1][2], ads_Scale[1][3]) ;<-----------------here
        printf("    Averaged on: %i , %i , %i ,%i\n", ads_MaxCount[1][0], ads_MaxCount[1][1], ads_MaxCount[1][2], ads_MaxCount[1][3]) ;
    }  else {
        printf("Second analog to digital sensor is not detected\n")  ;

br
Torsten

nostromo123, 2023-01-27:
Thank you for posting here, so this info is available not only for users of the German RCN forum.

I'll concentrate now on my 2. point, activating a second input, where the compiler here drops out.

By the way, I am running 

```
$ pio system info
--------------------------  -------------------------------------------------
PlatformIO Core             6.1.5
Python                      3.9.2-final.0
System Type                 linux_x86_64
Platform                    Linux-6.0.0-0.deb11.6-amd64-x86_64-with-glibc2.31
File System Encoding        utf-8
Locale Encoding             UTF-8
PlatformIO Core Directory   /home/kalle/.platformio
PlatformIO Core Executable  /home/kalle/.platformio/penv/bin/platformio
Python Executable           /home/kalle/.platformio/penv/bin/python
Global Libraries            0
Development Platforms       6
Tools & Toolchains          22
```

cu KH

nostromo123, 2023-01-27:
Made the correction in param.cpp, made two typos, but the compiler directly pointed out to the mistakes, I made.
But this is ok now.

I activated now additionally to A0   A1, A2 and A3 and get an error compiling 

The small sign "**^**" in the 3rd line points to what is causing the error.

```
src/ads1115.h: At global scope:
src/config.h:145:34: error: 'ADS1_MEASURE' was not declared in this scope
 #define ADS1_MEASURE A0_TO_GND , ADS1_MEASURE A1_TO_GND , ADS1_MEASURE A2_TO_GND , ADS1_MEASURE A3_TO_GND // select 4 values between A0_TO_A1, A0_TO_A3, A1_TO_A3, A2_TO_A3, A0_TO_GND, A1_TO_GND, A2_TO_GND, A3_TO_GND, ADS_OFF
                                  ^
src/ads1115.h:89:39: note: in expansion of macro 'ADS1_MEASURE'
 const uint8_t ads_Measure[2][4] = { { ADS1_MEASURE} , {ADS2_MEASURE } }; //  how to configure the multiplexer
                                       ^~~~~~~~~~~~
compilation terminated due to -Wfatal-errors.
*** [.pio/build/raspberry-mstrens/src/main.o] Error 1
In file included from src/vario.h:5:0,
                 from src/crsf_frames.h:3,
                 from src/param.h:4,
                 from src/param.cpp:2:
src/config.h:145:34: error: 'ADS1_MEASURE' was not declared in this scope
 #define ADS1_MEASURE A0_TO_GND , ADS1_MEASURE A1_TO_GND , ADS1_MEASURE A2_TO_GND , ADS1_MEASURE A3_TO_GND // select 4 values between A0_TO_A1, A0_TO_A3, A1_TO_A3, A2_TO_A3, A0_TO_GND, A1_TO_GND, A2_TO_GND, A3_TO_GND, ADS_OFF
                                  ^
src/ads1115.h:89:39: note: in expansion of macro 'ADS1_MEASURE'
 const uint8_t ads_Measure[2][4] = { { ADS1_MEASURE} , {ADS2_MEASURE } }; //  how to configure the multiplexer
                                       ^~~~~~~~~~~~
compilation terminated due to -Wfatal-errors.
*** [.pio/build/raspberry-mstrens/src/param.o] Error 1
```

Next test: Changed entry 'ADS1_MEASURE A1_TO_GND' to 'ADS_OFF', now the error marking points to 'ADS1_MEASURE A2_TO_GND'

```
src/ads1115.h: At global scope:
src/config.h:145:44: error: 'ADS1_MEASURE' was not declared in this scope
 #define ADS1_MEASURE A0_TO_GND , ADS_OFF , ADS1_MEASURE A2_TO_GND , ADS1_MEASURE A3_TO_GND // select 4 values between A0_TO_A1, A0_TO_A3, A1_TO_A3, A2_TO_A3, A0_TO_GND, A1_TO_GND, A2_TO_GND, A3_TO_GND, ADS_OFF
                                            ^
src/ads1115.h:89:39: note: in expansion of macro 'ADS1_MEASURE'
 const uint8_t ads_Measure[2][4] = { { ADS1_MEASURE} , {ADS2_MEASURE } }; //  how to configure the multiplexer
                                       ^~~~~~~~~~~~
compilation terminated due to -Wfatal-errors.
In file included from src/vario.h:5:0,
                 from src/crsf_frames.h:3,
                 from src/param.h:4,
                 from src/param.cpp:2:
src/config.h:145:44: error: 'ADS1_MEASURE' was not declared in this scope
 #define ADS1_MEASURE A0_TO_GND , ADS_OFF , ADS1_MEASURE A2_TO_GND , ADS1_MEASURE A3_TO_GND // select 4 values between A0_TO_A1, A0_TO_A3, A1_TO_A3, A2_TO_A3, A0_TO_GND, A1_TO_GND, A2_TO_GND, A3_TO_GND, ADS_OFF
                                            ^
src/ads1115.h:89:39: note: in expansion of macro 'ADS1_MEASURE'
 const uint8_t ads_Measure[2][4] = { { ADS1_MEASURE} , {ADS2_MEASURE } }; //  how to configure the multiplexer
                                       ^~~~~~~~~~~~
compilation terminated due to -Wfatal-errors.
*** [.pio/build/raspberry-mstrens/src/main.o] Error 1
```
My general settings here:

```
Version = 0.4.5 
    Function              Code      Pin (255=disabled)
Primary channels input    PRI     =    9
Secondary channels input  SEC     =  255
Telemetry                 TLM     =   10
GPS Rx                    GPS_RX  =   11
GPS Tx                    GPS_TX  =   12
Sbus OUT                  SBUS_OUT=  255
RPM                       RPM     =  255
SDA (baro sensor)         SDA     =   14
SCL (baro sensor)         SCL     =   15
PWM Channels 1, 2, 3 ,4   C1 / C4 =    0    1    2    3
PWM Channels 5, 6, 7 ,8   C5 / C8 =    4    5  255  255
PWM Channels 9,10,11,12   C9 / C12=  255  255  255  255
PWM Channels 13,14,15,16  C13/ C16=  255  255  255  255
Voltage 1, 2, 3, 4        V1 / V4 =   26   27   28   29

Protocol is CRSF (=ELRS)
CRSF baudrate = 420000
Voltage parameters:
    Scales : 0.972000 , 0.972000 , 0.972000 , 0.950000 
    Offsets: 6.000000 , 5.500000 , 6.000000 , 5.000000 
RPM multiplier = 1.000000
Baro sensor is detected using BMP280
    Sensitivity min = 100 (at 100)   , max = 300 (at 1000)
    Hysteresis = 5 
First analog to digital sensor is detected using ads1115
    Measurement setup: 4 , 8 , 8 ,8
    Gains: 0 , 0 , 2 ,2
    Rates: 3 , 3 , 7 ,7
    Offsets: 0.000000 , 0.000000 , 0.000000 ,0.000000
    Scales: 1.000000 , 1.000000 , 1.000000 ,1.000000
    Averaged on: 20 , 20 , 50 ,50
Second analog to digital sensor is not detected
Foreseen GPS type is Ublox  :GPS is detected
Failsafe type is HOLD

Config parameters are OK

```

mstrens, 2023-01-27:
ADS1_MEASURE may not be repeated.
So you should have something like 
#define ADS1_MEASURE A0_TO_GND , A1_TO_GND ,  A2_TO_GND ,  A3_TO_GND

nostromo123, 2023-01-27:
Thank you, works now. Sorry for bothering ...

Close this now.

br KH

nostromo123, 2023-01-27:
Sorry, but the issues seem to go on.

Same as reported by Satcomix, no new values/sensors for ADS1115 on the radio under ELRS.

Satcomix, 2023-01-27:
There is no telemetry output for your Transmitter yet, therefore you have the Debugger output on terminal for testing the different features on ADC1115
Best regards, 
Torsten

mstrens, 2023-01-27:
ELRS has a very limited set of telemetry fields.
Apart GPS and Vario, there is only one field for battery.
There is also just one field for current, one for "capacity" and one for "remain" (between 0 and 100%).
So I do not see how I could send all fields from er has toads1115.

Frsky protocol offers more possibilities but first it must be decided which data should be sent.
The easiest would be to sent each of the 2 X 4 values but then the user has to make some setup (complex?) on Tx side.

nostromo123, 2023-01-27:
So, simply saying, ads1115 in combination with ELRS does not make sense, right?
RPM was another thing, which does not give any reasonable values in combination with ELRS. We learned that.

I think, those 'limitations' should be communicated. As I expressed earlier, I don't see RPM or an up to 8 cell lipo monitoring in any way as essential. What oXs_on_RP2040 is offering  in combination with ELRS is, in my eyes, totally enough!

Thank you for the comments, any reason to keep this issue open? From my side at the moment - no!

br - KH

nostromo123, 2023-01-28:
Then I close this ....

closed 2023-01-28 by nostromo123