mstrens/oXs_on_RP2040 #22: Problems with Ver.0.4.6-hott

Satcomix, 2023-01-28:
Good morning Mstrens,
I decided to test the new version with my FrSky X20S and S.Port protocol.
1. Boot process takes up to 10sec.
2. GPS is displayed in the telemetry for 2-5 seconds, then the sensor get lost. No improvement even after changing the IO port.
3. Voltages V1 to V4 are displayed correctly.
4. Vario GY86-MS5611 first changes the values ​​properly, but then remains at these values. Rate of climb Vario remains at 10m/s, flight altitude changes slightly.
5. 2xADS1115 is initialized and shows plausible values in debug mode.
6. 2 RGB LEDs are suddenly displayed on my test board. Function OK, all colors are displayed.
7. SBUS in and out every 16CH OK
8. After some time, maybe 5min. all Sensors get lost, and came back (without GPS) after a few seconds.

Thank you for your work.
Many greetings,
Torsten

Satcomix, 2023-01-28:
ADC-Pitch-Roll-ACC??????

Adc 0 : 92 mVoltAdc 0 : 84 mVoltpitch, roll, acc:     15   -173      2      5
Adc 1 : 82 mVoltAdc 1 : 17 mVoltAdc 0 : 100 mVoltAdc 0 : 76 mVoltAdc 1 : 764 mVoltpitch, roll, acc:     15   -173    -11     -9
Adc 0 : 92 mVoltAdc 0 : 84 mVoltAdc 0 : 76 mVoltAdc 1 : 764 mVoltpitch, roll, acc:     15   -173     -1      1
Adc 0 : 100 mVoltpitch, roll, acc:     15   -173      0     -3
Adc 0 : 84 mVoltAdc 1 : 82 mVoltAdc 0 : 92 mVoltAdc 1 : 1535 mVoltAdc 0 : 76 mVoltAdc 1 : 764 mVoltpitch, roll, acc:     15   -173      3      2
Adc 0 : 76 mVoltAdc 1 : 764 mVoltpitch, roll, acc:     15   -173      3     -0
Adc 0 : 84 mVoltAdc 0 : 100 mVoltAdc 0 : 92 mVoltpitch, roll, acc:     15   -173     -2     -2
Adc 1 : 82 mVoltAdc 1 : 17 mVoltAdc 0 : 76 mVoltAdc 1 : 764 mVoltpitch, roll, acc:     15   -173      3      6
Adc 0 : 84 mVoltAdc 0 : 92 mVoltAdc 0 : 100 mVoltpitch, roll, acc:     15   -173      3     -0
Adc 0 : 76 mVoltAdc 1 : 764 mVoltAdc 0 : 84 mVoltpitch, roll, acc:     15   -173     -6     -3
Adc 1 : 82 mVoltAdc 0 : 76 mVoltAdc 1 : 764 mVoltAdc 0 : 92 mVoltpitch, roll, acc:     15   -173     -1      1
Adc 0 : 100 mVoltAdc 0 : 84 mVoltpitch, roll, acc:     15   -173      2      1
Adc 0 : 76 mVoltAdc 1 : 764 mVoltAdc 0 : 92 mVoltpitch, roll, acc:     15   -173     -4     -2
Adc 0 : 84 mVoltAdc 1 : 82 mVoltAdc 1 : 17 mVoltAdc 0 : 100 mVoltAdc 1 : 1535 mVoltAdc 0 : 76 mVoltAdc 1 : 764 mVoltpitch, roll, acc:     15   -173     -0     -1
pitch, roll, acc:     15   -173     -2      1
Adc 0 : 92 mVoltAdc 0 : 76 mVoltAdc 1 : 764 mVoltAdc 0 : 84 mVoltpitch, roll, acc:     15   -173     -3     -2
Adc 0 : 100 mVoltAdc 1 : 82 mVoltAdc 0 : 76 mVoltAdc 1 : 764 mVoltpitch, roll, acc:     15   -173      8      5
Adc 0 : 84 mVoltAdc 0 : 92 mVoltpitch, roll, acc:     15   -173     -1     -5
Adc 0 : 76 mVoltAdc 1 : 764 mVoltprocessing cmd

Commands can be entered to change the config parameters
- To activate a function, select the pin and enter function code = pin number (e.g. PRI=1)
    Function                  Code        Valid pins number
    Primary channels input    PRI     = 5, 9, 21, 25
    Secondary channels input  SEC     = 1, 13, 17, 29
    Telemetry                 TLM     = 0, 1, 2, ..., 29
    GPS Rx                    GPS_RX  = 0, 1, 2, ..., 29
    GPS Tx                    GPS_TX  = 0, 1, 2, ..., 29
    Sbus OUT                  SBUS_OUT= 0, 1, 2, ..., 29
    RPM (only for Sport)      RPM     = 0, 1, 2, ..., 29
    SDA (baro sensor)         SDA     = 2, 6, 10, 14, 18, 22, 26
    SCL (baro sensor)         SCL     = 3, 7, 11, 15, 19, 23, 27
    PWM Channels 1, ..., 16   C1 / C16= 0, 1, 2, ..., 15
    Voltage 1, ..., 4         V1 / V4 = 26, 27, 28, 29
- To disable a function, set pin number to 255

-To debug on USB/serial the telemetry frames, enter DEBUGTLM=Y or DEBUGTLM=N (default)
-To change the protocol, enter PROTOCOL=x where x=S for Sport, C for CRSF/ELRS, J for Jeti or H for Hott
-To change the CRSF baudrate, enter e.g. BAUD=420000
-To change voltage scales, enter SCALEx=nnn.ddd e.g. SCALE1=2.3 or SCALE3=0.123
     Enter SCALEx=0 to avoid sending voltage x to the Transmitter (for Frsky or Jeti)
-To change voltage offset, enter OFFSETx=nnn.ddd e.g. OFFSET1=0.6789
-To change GPS type: for an Ublox, enter GPS=U and for a CADIS, enter GPS=C
-To change RPM multiplicator, enter e.g. RPM_MULT=0.5 to divide RPM by 2
-To select the failsafe mode to HOLD, enter FAILSAFE=H
-To set the failsafe values on the current position, enter SETFAILSAFE
   Note: some changes require a reset to be applied

Cmd to execute:   

Version = 0.4.6 
    Function              Code      Pin (255=disabled)
Primary channels input    PRI     =    5
Secondary channels input  SEC     =    1
Telemetry                 TLM     =    0
GPS Rx                    GPS_RX  =   12
GPS Tx                    GPS_TX  =   13
Sbus OUT                  SBUS_OUT=    2
RPM                       RPM     =  255
SDA (baro sensor)         SDA     =   14
SCL (baro sensor)         SCL     =   15
PWM Channels 1, 2, 3 ,4   C1 / C4 =  255  255    6    7
PWM Channels 5, 6, 7 ,8   C5 / C8 =  255  255  255  255
PWM Channels 9,10,11,12   C9 / C12=  255  255  255  255
PWM Channels 13,14,15,16  C13/ C16=  255  255  255  255
Voltage 1, 2, 3, 4        V1 / V4 =   26   27   28   29

Protocol is Sport (Frsky)
[... 28 lines cut ...]

Config parameters are OK

 >> 
Adc 0 : 100 mVoltpitch, roll, acc:     15   -173     -6     -4
Adc 0 : 84 mVoltAdc 0 : 92 mVoltAdc 1 : 82 mVoltAdc 1 : 17 mVoltAdc 0 : 76 mVoltAdc 1 : 764 mVoltpitch, roll, acc:     15   -173      3      4
Adc 0 : 84 mVoltpitch, roll, acc:     15   -173      5      1
Adc 0 : 100 mVoltAdc 0 : 76 mVoltAdc 1 : 764 mVoltAdc 0 : 92 mVoltpitch, roll, acc:     15   -173      8      5
Adc 0 : 84 mVoltAdc 1 : 82 mVoltAdc 1 : 1535 mVoltAdc 0 : 76 mVoltAdc 1 : 764 mVoltpitch, roll, acc:     15   -173      6      6
Adc 0 : 100 mVoltAdc 0 : 92 mVoltpitch, roll, acc:     15   -173     -4     -3
Adc 0 : 76 mVoltAdc 1 : 764 mVoltAdc 0 : 84 mVoltpitch, roll, acc:     15   -173      4      6
Adc 1 : 82 mVoltAdc 0 : 92 mVoltAdc 1 : 17 mVoltAdc 0 : 100 mVoltAdc 0 : 76 mVoltAdc 1 : 764 mVoltpitch, roll, acc:     15   -173     -2     -1
Adc 0 : 84 mVoltpitch, roll, acc:     15   -173     -4     -2
Adc 0 : 76 mVoltAdc 1 : 764 mVoltpitch, roll, acc:     15   -173     -0      2
Adc 0 : 92 mVoltAdc 0 : 84 mVoltAdc 0 : 100 mVoltAdc 1 : 82 mVoltpitch, roll, acc:     15   -173      7      5
[... cut ...]
Adc 0 : 100 mVoltAdc 0 : 76 mVoltAdc 1 : 764 mVoltAdc 0 : 84 mVoltpitch, roll, acc:     15   -173      4      4
Adc 0 : 92 mVoltpitch, roll, acc:     15   -173     -2     -3
Adc 1 : 82 mVoltAdc 0 : 76 mVoltAdc 1 : 764 mVoltAdc 0 : 84 mVoltAdc 0 : 100 mVoltpitch, roll, acc:     15   -173      0     -0
Adc 0 : 92 mVoltAdc 0 : 76 mVoltAdc 1 : 764 mVoltpitch, roll, acc:     15   -173      2     -0
Adc 0 : 84 mVoltpitch, roll, acc:     15   -173     -2     -3
Adc 1 : 82 mVoltAdc 1 : 17 mVoltAdc 0 : 100 mVoltAdc 0 : 76 mVoltAdc 1 : 764 mVoltAdc 0 : 92 mVoltpitch, roll, acc:     15   -173

mstrens, 2023-01-28:
It was a debug version. It was printing a lot of info on the USB port.
When oXs is not connected to the USB, I suspect that some buffer becomes full and slow down the process.
In your first test, was oXs connected to the USB port of the PC?
 
I made some minor changes and disable the DEBUG option in a version 0.4.7. 
Perhaps it is better now.

Satcomix, 2023-01-28:
Hello Mstrens,
The board is connected all the time to usb port of the PC.
For testing outside is has a lipo onboard ;-)
Greatings,
Torsten

Satcomix, 2023-01-28:
Hello Mstrens,
Next Test! Ver.0.4.7 FrSky X20S S.Port
GPS only works **without** activating the I2C1 port and **without** activating V1 to V4, 
by setting up **SETFAILSAFE** i become **sbusQueue try add error**, 
after reboot everything seems OK, without I2C1 and V1-4.

Version = 0.4.7 
    Function              Code      Pin (255=disabled)
Primary channels input    PRI     =    5
Secondary channels input  SEC     =    1
Telemetry                 TLM     =    0
GPS Rx                    GPS_RX  =   12
GPS Tx                    GPS_TX  =   13
Sbus OUT                  SBUS_OUT=    2
RPM                       RPM     =  255
SDA (baro sensor)         SDA     =  255
SCL (baro sensor)         SCL     =  255
PWM Channels 1, 2, 3 ,4   C1 / C4 =  255  255    6    7
PWM Channels 5, 6, 7 ,8   C5 / C8 =  255  255  255  255
PWM Channels 9,10,11,12   C9 / C12=  255  255  255  255
PWM Channels 13,14,15,16  C13/ C16=  255  255  255  255
Voltage 1, 2, 3, 4        V1 / V4 =  255  255  255  255

Protocol is Sport (Frsky)
CRSF baudrate = 420000
Voltage parameters:
    Scales : 1.000000 , 1.000000 , 1.000000 , 1.000000 
    Offsets: 0.000000 , 0.000000 , 0.000000 , 0.000000 
RPM multiplier = 1.000000
Baro sensor is not detected
First analog to digital sensor is not detected
Second analog to digital sensor is not detected
Foreseen GPS type is Ublox  :GPS is detected
Failsafe uses predefined values
     Chan 1...4  = 992 , 992 , 992 , 992
     Chan 5...8  = 992 , 992 , 992 , 992
     Chan 9...12 = 992 , 992 , 992 , 992
     Chan 13...16= 992 , 992 , 992 , 992

Config parameters are OK

mstrens, 2023-01-28:
it is strange!
I do not understand the issue.
Did you try to de-activate GPS (GPS pin = 255) and to activate only voltage?
If this work, could you also activate only I2C?
And finally I2C+voltage.

I was testing on a breadboard with only I2C and it seameds OK.
I will start activating more options.

Version = 0.4.7
    Function              Code      Pin (255=disabled)
Primary channels input    PRI     =  255
Secondary channels input  SEC     =  255
Telemetry                 TLM     =  255
GPS Rx                    GPS_RX  =  255
GPS Tx                    GPS_TX  =  255
Sbus OUT                  SBUS_OUT=  255
RPM                       RPM     =  255
SDA (baro sensor)         SDA     =    2
SCL (baro sensor)         SCL     =    3
PWM Channels 1, 2, 3 ,4   C1 / C4 =  255  255  255  255
PWM Channels 5, 6, 7 ,8   C5 / C8 =  255  255  255  255
PWM Channels 9,10,11,12   C9 / C12=  255  255  255  255
PWM Channels 13,14,15,16  C13/ C16=  255  255  255  255
Voltage 1, 2, 3, 4        V1 / V4 =  255  255  255  255

Protocol is Sport (Frsky)
CRSF baudrate = 420000
Voltage parameters:
    Scales : 1.000000 , 1.000000 , 1.000000 , 1.000000
    Offsets: 0.000000 , 0.000000 , 0.000000 , 0.000000
RPM multiplier = 1.000000
Baro sensor is detected using MS5611
    Sensitivity min = 100 (at 100)   , max = 300 (at 1000)
    Hysteresis = 5
First analog to digital sensor is not detected
Second analog to digital sensor is not detected
Foreseen GPS type is Ublox  :GPS is not (yet) detected
Failsafe type is HOLD

Config parameters are OK

Satcomix, 2023-01-28:
Hello Mstrens,
I think its a "buffer overflow" of the UARTS.
I can connect the board with all sensors, but after 3 seconds I hear sensor lost from the X20S, V1 to V4 everything ok, vario everything ok, but GPS has completely logged off, no more live data. After some time, the FIX SAT reappears and logs in and out regularly, while the other GPS data no longer comes back.
V1 to V4 looks fine without GPS, but when I power up the GPS beforehand and then configure V1, sbusQueue try add error pops up.
I am now trying to change everything from I2C1 to I2C0, but it takes a while.

Satcomix, 2023-01-28:
Only GPS without I2C1 and without V1- V4 everything is OK
Only I2C1 (GY-86 + 2xADS1115) with V1- V4 but without GPS UART everything is OK.
GPS with V1-4 error
GPS with I2C1 error

Version = 0.4.7 
    Function              Code      Pin (255=disabled)
Primary channels input    PRI     =    5
Secondary channels input  SEC     =    1
Telemetry                 TLM     =    0
GPS Rx                    GPS_RX  =   12
GPS Tx                    GPS_TX  =   13
Sbus OUT                  SBUS_OUT=    2
RPM                       RPM     =  255
SDA (baro sensor)         SDA     =  255
SCL (baro sensor)         SCL     =  255
PWM Channels 1, 2, 3 ,4   C1 / C4 =  255  255    6    7
PWM Channels 5, 6, 7 ,8   C5 / C8 =  255  255  255  255
PWM Channels 9,10,11,12   C9 / C12=  255  255  255  255
PWM Channels 13,14,15,16  C13/ C16=  255  255  255  255
Voltage 1, 2, 3, 4        V1 / V4 =  255  255  255  255

Protocol is Sport (Frsky)
CRSF baudrate = 420000
Voltage parameters:
    Scales : 1.000000 , 1.000000 , 1.000000 , 1.000000 
    Offsets: 0.000000 , 0.000000 , 0.000000 , 0.000000 
RPM multiplier = 1.000000
Baro sensor is not detected
First analog to digital sensor is not detected
Second analog to digital sensor is not detected
Foreseen GPS type is Ublox  :GPS is detected
Failsafe uses predefined values
     Chan 1...4  = 992 , 992 , 992 , 992
     Chan 5...8  = 992 , 992 , 992 , 992
     Chan 9...12 = 992 , 992 , 992 , 992
     Chan 13...16= 992 , 992 , 992 , 992

Config parameters are OK

mstrens, 2023-01-28:
In a version, I changed the setup of the gps in order to get another type of frame (for ublox type M10). This frame is very long.
Perhaps, the queue becomes full.
I will have to check.
One test could be to disable this new frame.

Satcomix, 2023-01-28:
Maybe i put gps.cpp and gps.h vom Ver.0.3.6 into 0.4.7

mstrens, 2023-01-28:
No, there are some changes and this would not help.
To disable the new type of frame, you can put the following lines as comment:
0xB5,0x62,0x06,0x01,0x08,0x00,0x01,0x07,0x00,0x01,0x00,0x00,0x00,0x00,0x18,0xE1, //        NAV_PVT 
So having 
//0xB5,0x62,0x06,0x01,0x08,0x00,0x01,0x07,0x00,0x01,0x00,0x00,0x00,0x00,0x18,0xE1, //        NAV_PVT
Take care to do this twice (there are 2 identical lines)

mstrens, 2023-01-28:
I think I understand:
In the main loop, I read and process only one char (per loop) from the gps queue.
So when main loop take some more time, queue becomes full.
I should process all bytes in the gps queue at each loop.
I will try to change this.

mstrens, 2023-01-28:
I put on github a version 0.4.8 in order to read all GPS data in the GPS queue at each loop.
I hope it should solve the issue.
I did not tested it.
I did not put the 2 lines in comment to avoid the new frame. So this version should work with a m10 ublox.

Satcomix, 2023-01-28:
I disabled the Line
//0xB5,0x62,0x06,0x01,0x08,0x00,0x01,0x07,0x00,0x01,0x00,0x00,0x00,0x00,0x18,0xE1, // NAV_PVT
make new, get it on the board,
and than i try to set the uart for GPS (the IO for RX and TX is 255) to IO0 for RX , since that command in terminal, I no longer have access over terminal to the board.
busQueue try add error
sbusQueue try add error
sbusQueue try add error
sbusQueue try add error
sbusQueue try add error
sbusQueue try add error
sbusQueue try add error
sbusQueue try add error
sbusQueue try add error.................

Satcomix, 2023-01-28:
Ver.0.4.8.
This version behaves even worse than previous versions.
I also disabled the two //NAV_PVT lines.
sbusQueue try add error
sbusQueue try add error
sbusQueue..........................

mstrens, 2023-01-28:
I expect that the message is generated because core0 does not run the main loop fast enough.
Then the character received from the Sbus can't be added to a queue anymore because queue is full.
So, primary, it is not related to the uart.
We must identify the process that blocks/slows down the core0 that manage all communication.

It is strange that process on core0 blocks depending on GPS.
GPS is handled on core1 and even if core1 is blocked, it should not block core0 that runs the communication.
 
Can you run 0.4.8 starting with no parameters (all pins = 255).
Then add options but only one by one.
- first select protocol
- then add one voltage pin.
- then add tlm pin
- I expect that at this stage you should get one voltage on the handset and that RP2040 would be blinking (red)
- then add Tx and RX GPS pins. Do you get GPS data (+ previous voltage) and is led still blinking (red)
- then add 3 more voltage pins. Normally it should still work
- then add I2C

Satcomix, 2023-01-28:
Hello Mstrens,
Can you run 0.4.8 starting with no parameters (all pins = 255). **-->I did a factory reset on the board.**
Then add options but only one by one.

    first select protocol -->It was on S.Port OK
    then add one voltage pin. -->activate V1 GPio26 OK
    then add tlm pin --> activate TLM GPio12 OK
    I expect that at this stage you should get one voltage on the handset and that RP2040 would be blinking (red) -->yes,OK
    then add Tx and RX GPS pins. Do you get GPS data (+ previous voltage) and is led still blinking (red) -->activate GPio0+1 OK
    then add 3 more voltage pins. Normally it should still work -->activateV2 till V4 GPio27-29 OK, no problems so far.
    then add I2C -->activate SDA GPio2, than all Sensors get lost  :-(   **NOT OK**  :-(

mstrens, 2023-01-28:
I am not sure that your last test is really an issue.
When SDA=255 and SCL is defined, then oXs considers that the config is not valid and stop every process except looking at USB commands in order to correct the wrong config.
So normally you can press ENTER and you should get the current config.
Then sending SCL=3 (or similar) should correct the config and oXs should process the sensors.

mstrens, 2023-01-28:
I put a wrong explanation in previous post.
Please read again.
I just edited it.

Satcomix, 2023-01-28:
Entered SDA again, the sensors get all lost, than entered SCL, sensors all again at display, but without I2C, put the board off/on, everything seems fine.

Satcomix, 2023-01-28:
Now everything should be OK.
After every command i get SBusQueue................,but after one second it was ok, when i make a Board-Reset.

Cmd to execute:   

Version = 0.4.9 
    Function              Code      Pin (255=disabled)
Primary channels input    PRI     =    5
Secondary channels input  SEC     =   13
Telemetry                 TLM     =   12
GPS Rx                    GPS_RX  =    0
GPS Tx                    GPS_TX  =    1
Sbus OUT                  SBUS_OUT=   14
RPM                       RPM     =  255
SDA (baro sensor)         SDA     =    2
SCL (baro sensor)         SCL     =    3
PWM Channels 1, 2, 3 ,4   C1 / C4 =  255  255   15  255
PWM Channels 5, 6, 7 ,8   C5 / C8 =  255  255  255  255
PWM Channels 9,10,11,12   C9 / C12=  255  255  255  255
PWM Channels 13,14,15,16  C13/ C16=  255  255  255  255
Voltage 1, 2, 3, 4        V1 / V4 =   26   27   28   29

Protocol is Sport (Frsky)
CRSF baudrate = 420000
Voltage parameters:
    Scales : 1.000000 , 1.000000 , 1.000000 , 0.100000 
    Offsets: 0.000000 , 0.000000 , 0.000000 , 0.000000 
RPM multiplier = 1.000000
Baro sensor is detected using MS5611
    Sensitivity min = 100 (at 100)   , max = 300 (at 1000)
    Hysteresis = 5 
Acc/Gyro is detected using MP6050
First analog to digital sensor is detected using ads1115
    Measurement setup: 0 , 8 , 8 ,8
    Gains: 2 , 2 , 2 ,2
    Rates: 7 , 4 , 4 ,7
    Offsets: 0.000000 , 0.000000 , 0.000000 ,0.000000
    Scales: 1.000000 , 1.000000 , 1.000000 ,1.000000
    Averaged on: 10 , 20 , 30 ,50
Second analog to digital sensor is detected using ads1115
    Measurement setup: 0 , 8 , 8 ,8
    Gains: 2 , 2 , 2 ,2
    Rates: 7 , 4 , 4 ,7
    Offsets: 0.000000 , 0.000000 , 0.000000 ,0.000000
    Scales: 1.000000 , 1.000000 , 1.000000 ,1.000000
    Averaged on: 10 , 20 , 30 ,50
Foreseen GPS type is Ublox  :GPS is detected
Failsafe uses predefined values
     Chan 1...4  = 992 , 992 , 992 , 992
     Chan 5...8  = 992 , 992 , 992 , 992
     Chan 9...12 = 992 , 992 , 992 , 992
     Chan 13...16= 992 , 992 , 992 , 992

Config parameters are OK

 >>

mstrens, 2023-01-28:
It can happen that I2C (on sensor device side ) remains blocked when RP2040 is reset or reflashed and that there is no power down.
It is safe to always make a power down to be sure to reset all the external components (MS5611/ADS1115/MPU6050/GPS).

Does it mean that currently you have all I2C + GPS + voltage working?

If so, you can try to add (step by step)
- Primary input pin (from Rx - for a Sbus or a ELRS signal )
- one PWM pin
- Sbus out pin
- Second input pin (from Rx - for a Sbus or a ELRS signal)

Satcomix, 2023-01-28:
LooK at my previous post, every Port is connected, without RPM. Imake this before you write to me. :-)

mstrens, 2023-01-28:
Having Sbusqueue error after entering a command that force saving the config is quite "normal".
Saving the config takes a relatively long time and this could explain why the SbusQueue becomes full.
When config is saved, the process should continue. 

I put a version 0.4.9 in order to let oXs says if it detect or not a MP6050

Satcomix, 2023-01-28:
The previous post of mine show the Ver.0.4.9 with S.Port, SBus in and out, GPS, I2C1 with Gy86 incl. MS5611 and MP6050, and PWM, V1 to V4, everything is fine. In FrSky X20s all Sensors show plausible Values.
I learned not to make a Button Reset, a Power off and on is the better Way.
Thank you.

Satcomix, 2023-01-28:
Hello Mstrens,
I think the Issue "Problems with Ver.04.6" can be closed, because now we have Ver.04.9-hott
I hope, that i can help you something with my Tests.
Thank you for your work and time.
br,
Torsten

closed 2023-01-28 by Satcomix