mstrens/oXs_on_RP2040 #23: GPS Alt value Ver.0.5.2-HOTT

Satcomix, 2023-01-29:
Hello Mstrens,
I made some Test with the new Release 0.5.2-hott
With my FrSky X20S S.Port and SBUS ,i get very high GPS Alt value. Normaly ca.40m but with Ver.0.5.2-hott ca. 96-147m
With my MC28 GR24 HOTT, it shows 40m GPS Alt.
Have you any idea?
br,
Torsten

mstrens, 2023-01-29:
I can't explain the difference.
The GPS frame is filled in the same way on oXs on RP2040 as the pro mini version.
The GPS is normally captured in cm and is converted to m for sending in Hott.
Furthermore, in Hott I add an offset of 500m to keep data positive (that is supposed to be deducted on handset side).
So a Gps value 12345 (cm) is transmitted as 623(500+123).
In order to better analyse the issue, I put now on github a version 0.5.4 that print (to pc) the GPS altitude in meter (so for 12345cm, it is supposed to print 123). So you can compare the value on pc with the value on handset.

Satcomix, 2023-01-29:
Hello Mstrens,
Thanks for the Ver.0.5.4-hott. The problem does not exist in the HOTT protocol, everything is displayed properly there with plausible values.
The problem is in the S.PORT protocol.
The error is a bit confusing. Once the GPS starts at -100 meters, once at +97 meters, once the value levels off at +10 meters, once the value wanders between +97 meters and 147 meters.
When I switch to the HOTT protocol, the MC28 shows me values ​​between 25 and 48 meters, which fits. 
After every restart, i.e. power off/on, not just reset button, it always shows me something else. I don't know what to do anymore, very strange error. Will try out several boards tomorrow, up to Ver.0.3.6 the GPS always showed me the correct height.
Thank you very much and greetings,
Torsten

mstrens, 2023-01-29:
So, I presume the issue is in the way the altitude is provided by the GPS.
Initially oXs got only one altitude from the GPS. It was based on frame ubx_nav_posllh.
typedef struct {
    uint32_t time;              // GPS msToW
    int32_t longitude;                 // in degree with 7 decimals
    int32_t latitude;                  // in degree with 7 decimals 
    int32_t altitude_ellipsoid;        // in mm
    int32_t altitude_msl;              // in mm
    uint32_t horizontal_accuracy;
    uint32_t vertical_accuracy;        // in mm
} __attribute__((__packed__)) ubx_nav_posllh;

In newer version (for compatibility reason), I also decode another frame: ubx_nav_pvt
This frame contains 2 altitudes:
int32_t height;
int32_t height_above_sea_level;
Currently oXs uses also height from this frame.
Perhaps this is not the right value and I should use the other one.
I can make a version to print the 3 values and so we can compare.
I can also just discard the "height" values from the  ubx_nav_pvt because the other frame exist in all type of ublox GPS (M6...M10)

Satcomix, 2023-01-29:
Thanks for the reply,
tomorow, i will make a small change, like yesterday in the script with the two lines of // nav_pvt.
Maybe it will help.
br,
Torsten

mstrens, 2023-01-29:
Sorry, I was partly wrong.
It is right that I also use a new frame (PVT) but I don't use height from PVT. I use it only for number of sat.
I only use height from the original frame.
So, this can't explain the issue.
I put a version 0.5.5 on github that should print on PC the height as received from GPS (just dividing the value by 10 in order to get cm)

Satcomix, 2023-01-29:
Hello Mstrens,
Thank you for Ver.0.5.5

Begin:
height POSLLH=2507
height POSLLH=2506
height POSLLH=2509

height POSLLH=3828
height POSLLH=3833
height POSLLH=3838

height POSLLH=5826
height POSLLH=5828
height POSLLH=5821

height POSLLH=-167
height POSLLH=-114
height POSLLH=-65
height POSLLH=-40
height POSLLH=7
height POSLLH=70
height POSLLH=131
height POSLLH=180
height POSLLH=210
height POSLLH=262

Very strange, the value wander from minus 20meters to plus .....
height POSLLH=6005
height POSLLH=6012
height POSLLH=6016
height POSLLH=6006  shows 35meters on Display

mstrens, 2023-01-29:
Strange!
I presume that the huge variation at the beginning results from a poor reception of satellites.

At this stage, I can't explain why 6000 (= 60m) is displayed as 35m.

There is probably an issue because another user says me that the Sport value is 10X the Hott value.
To be further analysed.

Satcomix, 2023-01-30:
Good morning Mstrens,
The board has been factory restored multiple times using flash_nuke.uf2.
After that it was de-energized, not just button reset.
Then oXs Ver.0.5.5 was installed.
First the TLM port and then GPS_RX/TX was set up.
No other ports were used for the measurements.
After the measurements, all other ports were occupied and measured again. No change in the measurement result was found.
**The problem does not exist in the HOTT protocol with MC28, everything is displayed properly there with plausible values.**
Very Strange. 
**Equipment: FrSky X20S S.Port, oXs Ver.0.5.5, GPS Beitian BN220**

height POSLLH=-1700 -Cold Start GPS BN220
height POSLLH=-1700
height POSLLH=-1700
height POSLLH=1657399   **3 Sats** 
height POSLLH=1657391
height POSLLH=1657380
height POSLLH=1657370   **4 Sats**
height POSLLH=1657360
height POSLLH=-52
height POSLLH=-118
height POSLLH=-76
height POSLLH=-81
height POSLLH=-9
height POSLLH=-3
height POSLLH=-25
height POSLLH=-35
height POSLLH=-85
height POSLLH=-12516
height POSLLH=-12511
height POSLLH=-12518
height POSLLH=-12517
height POSLLH=-12506
height POSLLH=-7860
height POSLLH=-7871
height POSLLH=-7883
height POSLLH=-1813
height POSLLH=132
height POSLLH=1086   **6 Sats**
height POSLLH=1660
height POSLLH=2039
height POSLLH=3629
height POSLLH=3634
height POSLLH=3634
height POSLLH=3638
height POSLLH=3641
height POSLLH=3641
height POSLLH=3629
height POSLLH=3731
height POSLLH=3729   **7 Sats**
height POSLLH=3724
height POSLLH=3725  **174m 7 Sats**
height POSLLH=4071
height POSLLH=4072
height POSLLH=4077
height POSLLH=4077  **175m 7 Sats**
height POSLLH=4206
height POSLLH=4205
height POSLLH=4209
height POSLLH=4209
height POSLLH=4213
height POSLLH=4210  **176m 7 Sats**
height POSLLH=4216
height POSLLH=4218
height POSLLH=4217
height POSLLH=4637
height POSLLH=4640
height POSLLH=4640
height POSLLH=4642  **180m 7 Sats**
height POSLLH=4645
height POSLLH=4647
height POSLLH=4649
height POSLLH=4649
height POSLLH=5056
height POSLLH=5057
height POSLLH=5058
height POSLLH=5059
height POSLLH=5062
height POSLLH=5061  **185m 7 Sats**
height POSLLH=5063
height POSLLH=5063
height POSLLH=5065
height POSLLH=5473
height POSLLH=5470
height POSLLH=5469  **189m 7 Sats**
height POSLLH=5471
height POSLLH=5471
height POSLLH=4787
height POSLLH=4784  **182m 8 Sats**
height POSLLH=4782
height POSLLH=4782
height POSLLH=4779
height POSLLH=4776
height POSLLH=4588
height POSLLH=4589
height POSLLH=4590  **180m 8 Sats 15min after Coldstart Range between 172m to 208m**
height POSLLH=4593
height POSLLH=4596 
height POSLLH=8964
height POSLLH=8968
height POSLLH=8970  **220m 6 Sats**
height POSLLH=8985
height POSLLH=9004
height POSLLH=8964
height POSLLH=8968  **225m 8 Sats 25min after Coldstart** 
height POSLLH=8970
height POSLLH=8985
height POSLLH=9004
height POSLLH=4885
height POSLLH=4902
height POSLLH=4913  **182m 10 Sats 45min after Coldstart**
height POSLLH=4917
height POSLLH=4924
height POSLLH=4941
height POSLLH=4958

**Hottstart 60min after  Coldstart**
height POSLLH=3023
height POSLLH=3038
height POSLLH=3047
height POSLLH=3058  **164m 10 Sats 5min after Hottstart**
height POSLLH=3072
height POSLLH=3077

br,
Torsten

mstrens, 2023-01-30:
Huge height variations at the beginning seem me normal because the number of sat is to low to calculate it.
This should be the case for all protocol because data received from the GPS are the same for all protocols.

As long there is no 3D fix, oXs should not send GPS data except number of sat at least in Sport and Jeti protocols (in ELRS and Hott, this is not possible because all GPS data are in only one frame and so I can only keep the uncertain values = 0).
This was normally the case in 0.3.6 version but not anymore in 0.4.x and above.
I can now make a change (in 0.5.6) in order to restore the same process as 0.3.6 (that seems me better) for Sport.
In this version, for analyse purpose, I print now the 2 heights that the gps provides ( above sea level and ellipsoid) in cm.
I print them now only when there is a 3d fix (so when height should also be sent in Sport).

I still can't explain the quite huge variations when there are 6 sat or more. Anyway, this is related to the GPS itself and not to oXs.
I can't explain why a value like 3058 becomes 164m on the handset display in Sport protocol.
In the hott protocol, do you get on handset a height value that match the value provided by GPS (e.g. 1234cm on PC should become 12m on handset)

Satcomix, 2023-01-30:
Hello Mstrens,
I did some tests with the U_Center to rule out that the GPS device has a hardware error.
The U_Center shows GPS Altitude 87,600m, GPS Altitude (MSL) 41,100m, these values ​​match the location. Sats received = 10.
Terminal output after 15minutes:
POSLLH alt_msl  alt_ellipsoid = 4067  8719
POSLLH alt_msl  alt_ellipsoid = 4071  8724
POSLLH alt_msl  alt_ellipsoid = 4073  8725
POSLLH alt_msl  alt_ellipsoid = 4079  8731
POSLLH alt_msl  alt_ellipsoid = 4086  8738
POSLLH alt_msl  alt_ellipsoid = 4091  8743
FrSky X20S Display = -15meters
Fix Sats = 10
br,
Torsten

mstrens, 2023-01-30:
Alt_msl is the value used by oXs and seems OK.
About the Frsky value, the issue is perhaps on the handset side.
Perhaps the handset takes the first value it get as a reference and then calculates the values to display subtracting it from the new incoming telemetry values.
If so, it should be solved with a "reset" of telemetry on the handset.

Satcomix, 2023-01-30:
Hello Mstrens,
I reset the X20S telemetry, no success, I downgraded a previous firmware version from Ethos 1.4.6 to 1.4.5, no success.
Board RP2040 changed, no success.
Always the same values.
11 satellite fix.
Display shows minus 0.50 meters
POSLLH alt_msl alt_ellipsoid = 4557 9209
POSLLH alt_msl alt_ellipsoid = 4553 9206
POSLLH alt_msl alt_ellipsoid = 4548 9200
POSLLH alt_msl alt_ellipsoid = 4544 9196
POSLLH alt_msl alt_ellipsoid = 4543 9196
POSLLH alt_msl alt_ellipsoid = 4541 9193
POSLLH alt_msl alt_ellipsoid = 4542 9194
br,
Torsten

Satcomix, 2023-01-30:
Hello Mstrens,
I made some more Tests with Ver.0.5.6-hott.
1. Test with MC28 HOTT all values are displayed properly there with plausible values, the GPS Alt is MSL "original hight".
2. Test with X20S S.Port all values are displayed properly there with plausible values, the GPS Alt is "relative" after Reset
3. I dont know why the Alt value goes from -100 to +200, i think it is a Software Problem with Ethos, because my RC crashed after upgrade from 1.4.5 to 1.4.6 and everything in the Handheld must be installed again. I will ask Bsongis (FrSky) for switching absolute (MSL) to relative GPS in Ethos.
I think this version 0.5.6-hott is ok, so i can close the error.
The good thing is, i learned something about your software.
Thank you very much for your help and time.
br,
Torsten

closed 2023-01-30 by Satcomix