mstrens/oXs_on_RP2040 #25: oXs Ver.0.5.8 -hott

Satcomix, 2023-01-31:
Hello Mstrens,
I read your post in RC Groups Forum.
I have a Account, but "YOUR ACCOUNT IS CURRENTLY FLAGGED AS UNCONFIRMED!" , i wait in vain three weeks for  an Answer or Email from any Admin.
Something happened in Ver.0.5.8 with my FrSky X20S Handheld.
Normaly,
oXs V1 is VFAS in Volt 
oXs V2 is FASS in Ampere 
oXs V3 is Temp2 in Degree °
oXs V4 is  Fuel in %

Now Temp2/V3 doesnt show me anything ---
and Temp1 was FIX SATELLITES now ---
I made a new search for Sensors, and Reset the board several times, but nothing happened.
I read the telemetry_doc.txt but i dont understand, why the values disappear.

br,
Torsten

Satcomix, 2023-01-31:
Next Test with Graupner MC28-4D GR24 HOTT,
GPS show plausible values like, number of Satellites, GPS speed, GPS Altitude MSL above NN, GPS Dir in degree, Home Altitude, Home Dir in degree.
Vario GY86 incl. MS5611 and MP6050 show plausible values, like Altitude in meters and Vspeed in m/s
oXs V1 is AKK1 Sensor1 Voltage1 Scale 1/10
oXs V2 is Current Sensor in Ampere Scale 1/10
oXs V3 is AKK2 Sensor2 Voltage2 Scale 1/10
oXs V4 is Voltage3 over Current to get Capacity in mAh, but there is something wrong:
V4 2,2V       V4 2,2V
V2 1,66A     V2 0,1A
-----------   ---------
1660mAh       160mAh 
I think the problem is V4 instead of V1 or V3
br,
Torsten

mstrens, 2023-01-31:
In 0.5.8 I tried to avoid misuse of standard Frsky fields.
E.g. I tried to avoid sending number of sat in Temp1.
Idem for V3 and V4.
In some documentation, I saw that there was a range of code (fields ID) foreseen for "DIY" sensors.
So I changed the setup in order to use those DIY ID's for some fields (number of sat, V3 and V4) that are not predefined by Frsky.

I documented this in the new file "telemetry_doc.txt" (available on github).
Still I am not sure that openTx is able to discover those "DIY" ID's.
To check if openTx can discover them, best would be to first ask it to delete all sensors and then ask for a new discovery.
If they are not discovered, I will have to change again and use some ID's that are supported by openTx even it is a misuse.

Satcomix, 2023-01-31:
Hallo Mstrens,
Bei FrSky S.PORT sehe ich in 0.5.8 Temp1/Anzahl  Satelliten, V3 und V4 nicht mehr, habe Telemetrie ersetzt, kein Erfolg bei neuer Sensor Suche. Habe Ver.0.5.7 wieder in Gebrauch, diese Version läuft flüssiger. 
BR,
Torsten

Satcomix, 2023-01-31:
Hello Mstrens,
I read the new Document, see my first post.
I have had no Temp1 / Fix Satellites, V3 Temp2 and V4 capacity in % on my FrSky X20S S.Port since version 0.5.8,
I reverted to Ver.0.5.7, since a telemetry reset with a new search was unsuccessful.
thanks for the work
br,
Torsten

mstrens, 2023-01-31:
A member just post a message on Rc group to say that sensor ID's 5100, 5101, 5102 are discovered.
I do not know wich version of OTX or Edge he is using

Satcomix, 2023-01-31:
I checked all my ID Lists for FrSky Ethos but i cant find those ID`s, maybe it is easer to integrate it in OpenTX or Edge, sorry i use Ethos. 
I like to help with testing for FrSky S.PORT and Graupner HOTT, but when you change the ID´s and i dont see the sensors, i cant help anymore.
I think most openXsensor users were happy about a new faster, better RP2040 board, hoping to keep the old familiar configuration steps and versatile settings.
Thank you for the work,
br,
Torsten

mstrens, 2023-01-31:
Here is where I found the ID's
https://github.com/opentx/opentx/blob/d67d1aa0c09d2f485c3ef7ca8c4fb1e9214a2ee7/radio/src/telemetry/frsky.h

If they are not supported by Ethos, I can change in order to use other ID's
Just need to see which one. I presume there is no documentation about the Id's supported by Ethos.

Satcomix, 2023-01-31:
I know that ID List and Pawelsky shows the same, but the sensors, 5100, 5101 and 5102 are new, i think. 
ok, thank you My FrSky Handheld with Ethos dont find them.
br,
Torsten

mstrens, 2023-01-31:
Perhaps you can ask to Ethos developper team if they plan to support them in the future.

Satcomix, 2023-01-31:
I will ask the FrSky Team, and i will look tomorow in Ethos software, i think it is for a future Release of Ethos or for the diy Sensors in Lua, but normaly Ethos find an ID, that you can make your own sensor.
Thank you very much,
br,
Torsten

Satcomix, 2023-02-01:
Good morning Mstrens,
I did a full factory reset of the RP2040 and then installed Ver.0.5.9, after that my FrSky X20S Ethos was able to find the sensors in the DIY Sensor section. I can see 3 new sensors, 5100, 5101 and 5102.
5100 is shown as Fix Satellites. So everything is fine again with the latest Ver.0.5.9.
No idea why it didn't work yesterday.
Thank you for this work.
br,
Torsten

mstrens, 2023-02-01:
Fine. Thanks for the feedback.
I do not think that a full factory reset of the RP2040 has changed something because the definition of the telemetry fields to be sent is hard coded (and not part of the setup parameters to be updated with the serial terminal and saved in the RP2040 outside of the firmware)

closed 2023-02-01 by Satcomix