mstrens/oXs_on_RP2040 #26: Featurerequest Support I-Bus Protocol

doloebig, 2023-01-31:
it would be nice if oXs would support I-Bus for the Flysky RX. They running often on Opentx with MPM

doloebig, 2023-02-02:
found some code and projekts on github, maybe it helps
https://github.com/bmellink/IBusBM
https://github.com/Yenya/ibus-altitude-sensor
https://github.com/aanon4/FlySkyIBus

pierrotm777, 2023-02-11:
Yes, I I should like a Flysky telemetry too!

pierrotm777, 2023-02-11:
If it's can help, i have this lib.
[iBUSTelemetry.zip](https://github.com/mstrens/oXs_on_RP2040/files/10714276/iBUSTelemetry.zip)

mstrens, 2023-02-20:
I just try to add it in version 1.2.0.
I can't test it.
If it do not work, please make a capture with a logic analyser of the signal on the "tlm" pin.
It is supposed to be a uart signal at 115200 baud 8N1.
Then send me the captured file.

pierrotm777, 2023-02-20:
Thanks mstrens for this new add.
In fact, i receive no telemetry to my fs-ia6b.
In automatic, i receive a signal in 11384 baud.
[24 MHz, Ibus.zip](https://github.com/mstrens/oXs_on_RP2040/files/10782886/24.MHz.Ibus.zip)
[image]

doloebig, 2023-02-20:
i can confirm, also noch data received by handset
Cmd to execute:   

Version = 1.2.1 
    Function                Pin   Change entering XXX=yyy (yyy=255 to disable)
Primary channels input    =  255  (PRI     = 5, 9, 21, 25)
Secondary channels input  =  255  (SEC     = 1, 13, 17, 29)
Telemetry . . . . . . . . =    7  (TLM     = 0, 1, 2, ..., 29)
GPS Rx  . . . . . . . . . =    3  (GPS_RX  = 0, 1, 2, ..., 29)
GPS Tx  . . . . . . . . . =    2  (GPS_TX  = 0, 1, 2, ..., 29)
Sbus OUT  . . . . . . . . =  255  (SBUS_OUT= 0, 1, 2, ..., 29)
RPM   . . . . . . . . . . =    6  (RPM     = 0, 1, 2, ..., 29)
SDA (I2C sensors) . . . . =  255  (SDA     = 2, 6, 10, 14, 18, 22, 26)
SCL (I2C sensors) . . . . =  255  (SCL     = 3, 7, 11, 15, 19, 23, 27)
PWM Channels 1, 2, 3 ,4   =  255  255  255  255 (C1 / C16= 0, 1, 2, ..., 15)
PWM Channels 5, 6, 7 ,8   =  255  255  255  255
PWM Channels 9,10,11,12   =  255  255  255  255
PWM Channels 13,14,15,16  =  255  255  255  255
Voltage 1, 2, 3, 4        =   26   27   28   29 (V1 / V4 = 26, 27, 28, 29)

Protocol is ibus(Flysky)
CRSF baudrate = 420000
Voltage parameters:
    Scales : 5.700000 , 10.000000 , 0.100000 , 0.100000 
    Offsets: 0.000000 , 5000.000000 , 0.000000 , 0.000000 
    Temperature sensors are connected on V3 and V4
RPM multiplier = 1.000000
Baro sensor is not detected
Acc/Gyro is not detected
First analog to digital sensor is not detected
Second analog to digital sensor is not detected
Foreseen GPS type is Ublox  :GPS is detected but has not (yet) a fix
Failsafe type is HOLD

Config parameters are OK
Press ? + Enter to get help about the commands

 >> 
FV
processing cmd

Cmd to execute:   FV

GPS Num sat. = 0 cm
GPS Date J M A = 18 10 15 
GPS Time H M S = 0 6 42 
Volt 1 = 158 mVolt
Current (Volt 2) = -27 mA
Capacity (using current) = 0 mAh
Temp 1 (Volt 3) = 26 degree
Temp 2 (Volt 4) = 26 degree
RPM = 0 Hertz

DEBUGTLM=Y not output

reg
Doro

mstrens, 2023-02-20:
I fixed 2 bugs.
Perhaps there are some other.
Now version 1.2.2.
The capture helped to find those 2 bugs.

doloebig, 2023-02-20:
DEBUGTLM=Y bring up the configscreen
no data on the handset

Cmd to execute:   

Version = 1.2.2 
    Function                Pin   Change entering XXX=yyy (yyy=255 to disable)
Primary channels input    =  255  (PRI     = 5, 9, 21, 25)
Secondary channels input  =  255  (SEC     = 1, 13, 17, 29)
Telemetry . . . . . . . . =    7  (TLM     = 0, 1, 2, ..., 29)
GPS Rx  . . . . . . . . . =    3  (GPS_RX  = 0, 1, 2, ..., 29)
GPS Tx  . . . . . . . . . =    2  (GPS_TX  = 0, 1, 2, ..., 29)
Sbus OUT  . . . . . . . . =  255  (SBUS_OUT= 0, 1, 2, ..., 29)
RPM   . . . . . . . . . . =    6  (RPM     = 0, 1, 2, ..., 29)
SDA (I2C sensors) . . . . =  255  (SDA     = 2, 6, 10, 14, 18, 22, 26)
SCL (I2C sensors) . . . . =  255  (SCL     = 3, 7, 11, 15, 19, 23, 27)
PWM Channels 1, 2, 3 ,4   =  255  255  255  255 (C1 / C16= 0, 1, 2, ..., 15)
PWM Channels 5, 6, 7 ,8   =  255  255  255  255
PWM Channels 9,10,11,12   =  255  255  255  255
PWM Channels 13,14,15,16  =  255  255  255  255
Voltage 1, 2, 3, 4        =   26   27   28   29 (V1 / V4 = 26, 27, 28, 29)

Protocol is ibus(Flysky)
CRSF baudrate = 420000
Voltage parameters:
    Scales : 5.700000 , 10.000000 , 0.100000 , 0.100000 
    Offsets: 0.000000 , 5000.000000 , 0.000000 , 0.000000 
    Temperature sensors are connected on V3 and V4
RPM multiplier = 1.000000
Baro sensor is not detected
Acc/Gyro is not detected
First analog to digital sensor is not detected
Second analog to digital sensor is not detected
Foreseen GPS type is Ublox  :GPS is detected but has not (yet) a fix
Failsafe type is HOLD

Config parameters are OK
Press ? + Enter to get help about the commands

 >> 
FV
processing cmd

Cmd to execute:   FV

GPS Num sat. = 0 cm
GPS Date J M A = 18 10 15 
GPS Time H M S = 0 1 34 
Volt 1 = 154 mVolt
Current (Volt 2) = -229 mA
Capacity (using current) = 0 mAh
Temp 1 (Volt 3) = 28 degree
Temp 2 (Volt 4) = 27 degree
RPM = 0 Hertz
DEBUGTLM=Y
processing cmd

Cmd to execute:   DEBUGTLM=Y

Version = 1.2.2 
    Function                Pin   Change entering XXX=yyy (yyy=255 to disable)
Primary channels input    =  255  (PRI     = 5, 9, 21, 25)
Secondary channels input  =  255  (SEC     = 1, 13, 17, 29)
Telemetry . . . . . . . . =    7  (TLM     = 0, 1, 2, ..., 29)
GPS Rx  . . . . . . . . . =    3  (GPS_RX  = 0, 1, 2, ..., 29)
GPS Tx  . . . . . . . . . =    2  (GPS_TX  = 0, 1, 2, ..., 29)
Sbus OUT  . . . . . . . . =  255  (SBUS_OUT= 0, 1, 2, ..., 29)
RPM   . . . . . . . . . . =    6  (RPM     = 0, 1, 2, ..., 29)
SDA (I2C sensors) . . . . =  255  (SDA     = 2, 6, 10, 14, 18, 22, 26)
SCL (I2C sensors) . . . . =  255  (SCL     = 3, 7, 11, 15, 19, 23, 27)
PWM Channels 1, 2, 3 ,4   =  255  255  255  255 (C1 / C16= 0, 1, 2, ..., 15)
PWM Channels 5, 6, 7 ,8   =  255  255  255  255
PWM Channels 9,10,11,12   =  255  255  255  255
PWM Channels 13,14,15,16  =  255  255  255  255
Voltage 1, 2, 3, 4        =   26   27   28   29 (V1 / V4 = 26, 27, 28, 29)

pierrotm777, 2023-02-20:
Always no data for me .

mstrens, 2023-02-20:
can you provide a new capture?

pierrotm777, 2023-02-20:
[24 MHz_IBUS_2.zip](https://github.com/mstrens/oXs_on_RP2040/files/10784054/24.MHz_IBUS_2.zip)

mstrens, 2023-02-20:
perhaps better with next version 1.2.3
Please provide a capture.
Have also a look at the messages displayed on the pc.
I added some debug messages for ibus when some error are detected.
If some messages exist, please make a copy/paste of some of them.

pierrotm777, 2023-02-20:
I receive that when i send command 'temp=0'.
[code]
Cmd to execute:   TEMP=0
Number of temperature sensors = 0
config has been saved
Device will reboot

Version = 1.2.3 
    Function                Pin   Change entering XXX=yyy (yyy=255 to disable)
Primary channels input    =    5  (PRI     = 5, 9, 21, 25)
Secondary channels input  =   13  (SEC     = 1, 13, 17, 29)
Telemetry . . . . . . . . =    0  (TLM     = 0, 1, 2, ..., 29)
GPS Rx  . . . . . . . . . =    8  (GPS_RX  = 0, 1, 2, ..., 29)
GPS Tx  . . . . . . . . . =    7  (GPS_TX  = 0, 1, 2, ..., 29)
Sbus OUT  . . . . . . . . =  255  (SBUS_OUT= 0, 1, 2, ..., 29)
RPM   . . . . . . . . . . =   14  (RPM     = 0, 1, 2, ..., 29)
SDA (I2C sensors) . . . . =    2  (SDA     = 2, 6, 10, 14, 18, 22, 26)
SCL (I2C sensors) . . . . =    3  (SCL     = 3, 7, 11, 15, 19, 23, 27)
PWM Channels 1, 2, 3 ,4   =  255  255  255  255 (C1 / C16= 0, 1, 2, ..., 15)
PWM Channels 5, 6, 7 ,8   =  255  255  255  255
PWM Channels 9,10,11,12   =  255  255  255  255
PWM Channels 13,14,15,16  =  255  255  255  255
Voltage 1, 2, 3, 4        =  255  255   28   29 (V1 / V4 = 26, 27, 28, 29)

Protocol is ibus(Flysky)
CRSF baudrate = 420000
Voltage parameters:
    Scales : 1.000000 , 1.000000 , 1.000000 , 1.000000 
    Offsets: 0.000000 , 0.000000 , 0.000000 , 0.000000 
    No temperature sensors are connected on V3 and V4
RPM multiplier = 1.000000
Baro sensor is detected using MS5611
    Sensitivity min = 100 (at 100)   , max = 300 (at 1000)
    Hysteresis = 5 
Acc/Gyro is detected using MP6050
     Acceleration offsets X, Y, Z = -6502 , 1062 , 2346
     Gyro offsets         X, Y, Z = 185 , -36 , -10
First analog to digital sensor is detected using ads1115
    Measurement setup: 4 , 5 , 6 ,7
    Gains: 1 , 1 , 1 ,1
    Rates: 5 , 5 , 5 ,5
    Offsets: 0.000000 , 0.000000 , 0.000000 ,0.000000
    Scales: 1.000000 , 1.000000 , 1.000000 ,1.000000
    Averaged on: 10 , 10 , 10 ,10
Second analog to digital sensor is not detected
Foreseen GPS type is Ublox  :GPS is not (yet) detected
Failsafe type is HOLD

Config parameters are OK
Press ? + Enter to get help about the commands

 >> 
ibus pooling with checksum error
ibus pooling with checksum error
[/code]
[24 MHz,IBUS_3.zip](https://github.com/mstrens/oXs_on_RP2040/files/10784483/24.MHz.IBUS_3.zip)

doloebig, 2023-02-20:
same here
Cmd to execute:   OFFSET2=0
config has been saved
Device will reboot

Version = 1.2.3 
    Function                Pin   Change entering XXX=yyy (yyy=255 to disable)
Primary channels input    =  255  (PRI     = 5, 9, 21, 25)
Secondary channels input  =  255  (SEC     = 1, 13, 17, 29)
Telemetry . . . . . . . . =    7  (TLM     = 0, 1, 2, ..., 29)
GPS Rx  . . . . . . . . . =    3  (GPS_RX  = 0, 1, 2, ..., 29)
GPS Tx  . . . . . . . . . =    2  (GPS_TX  = 0, 1, 2, ..., 29)
Sbus OUT  . . . . . . . . =  255  (SBUS_OUT= 0, 1, 2, ..., 29)
RPM   . . . . . . . . . . =    6  (RPM     = 0, 1, 2, ..., 29)
SDA (I2C sensors) . . . . =  255  (SDA     = 2, 6, 10, 14, 18, 22, 26)
SCL (I2C sensors) . . . . =  255  (SCL     = 3, 7, 11, 15, 19, 23, 27)
PWM Channels 1, 2, 3 ,4   =  255  255  255  255 (C1 / C16= 0, 1, 2, ..., 15)
PWM Channels 5, 6, 7 ,8   =  255  255  255  255
PWM Channels 9,10,11,12   =  255  255  255  255
PWM Channels 13,14,15,16  =  255  255  255  255
Voltage 1, 2, 3, 4        =   26   27   28   29 (V1 / V4 = 26, 27, 28, 29)

Protocol is ibus(Flysky)
CRSF baudrate = 420000
Voltage parameters:
    Scales : 5.700000 , 10.000000 , 0.100000 , 0.100000 
    Offsets: 0.000000 , 0.000000 , 0.000000 , 0.000000 
    Temperature sensors are connected on V3 and V4
RPM multiplier = 1.000000
Baro sensor is not detected
Acc/Gyro is not detected
First analog to digital sensor is not detected
Second analog to digital sensor is not detected
Foreseen GPS type is Ublox  :GPS is detected but has not (yet) a fix
Failsafe type is HOLD

Config parameters are OK
Press ? + Enter to get help about the commands

 >> 
ibus pooling with checksum error
ibus pooling with checksum error

mstrens, 2023-02-20:
Do you get a lot of "ibus pooling with checksum error" messages or only 2?

mstrens, 2023-02-20:
If you can recompile by yourself, can you try changing this line (in ibus.cpp)
There is currently:
case 0X80: // request discovering next sensor ; we reply the same value to confirm it exists

It should be 08  instead of 80
case 0X08: // request discovering next sensor ; we reply the same value to confirm it exists

Please send again a capture.

If you can't compile yourself, I can compile for you. Just ask for

doloebig, 2023-02-20:
only times, and only if change the config by command (in my case Offset2=0)

i have no environment for compile
if i need on what is running on windows?

Doro

pierrotm777, 2023-02-20:
I receive only 2 lignes and only when i send temp=x

Cmd to execute:   TEMP=0
Number of temperature sensors = 0
config has been saved
Device will reboot

Version = 1.2.3 
    Function                Pin   Change entering XXX=yyy (yyy=255 to disable)
Primary channels input    =    5  (PRI     = 5, 9, 21, 25)
Secondary channels input  =   13  (SEC     = 1, 13, 17, 29)
Telemetry . . . . . . . . =    0  (TLM     = 0, 1, 2, ..., 29)
GPS Rx  . . . . . . . . . =    8  (GPS_RX  = 0, 1, 2, ..., 29)
GPS Tx  . . . . . . . . . =    7  (GPS_TX  = 0, 1, 2, ..., 29)
Sbus OUT  . . . . . . . . =  255  (SBUS_OUT= 0, 1, 2, ..., 29)
RPM   . . . . . . . . . . =   14  (RPM     = 0, 1, 2, ..., 29)
SDA (I2C sensors) . . . . =    2  (SDA     = 2, 6, 10, 14, 18, 22, 26)
SCL (I2C sensors) . . . . =    3  (SCL     = 3, 7, 11, 15, 19, 23, 27)
PWM Channels 1, 2, 3 ,4   =  255  255  255  255 (C1 / C16= 0, 1, 2, ..., 15)
PWM Channels 5, 6, 7 ,8   =  255  255  255  255
PWM Channels 9,10,11,12   =  255  255  255  255
PWM Channels 13,14,15,16  =  255  255  255  255
Voltage 1, 2, 3, 4        =  255  255  255   29 (V1 / V4 = 26, 27, 28, 29)

Protocol is ibus(Flysky)
CRSF baudrate = 420000
Voltage parameters:
    Scales : 1.000000 , 1.000000 , 1.000000 , 1.000000 
    Offsets: 0.000000 , 0.000000 , 0.000000 , 0.000000 
    No temperature sensors are connected on V3 and V4
RPM multiplier = 1.000000
Baro sensor is detected using MS5611
    Sensitivity min = 100 (at 100)   , max = 300 (at 1000)
    Hysteresis = 5 
Acc/Gyro is detected using MP6050
     Acceleration offsets X, Y, Z = -6502 , 1062 , 2346
     Gyro offsets         X, Y, Z = 185 , -36 , -10
First analog to digital sensor is detected using ads1115
    Measurement setup: 4 , 5 , 6 ,7
    Gains: 1 , 1 , 1 ,1
    Rates: 5 , 5 , 5 ,5
    Offsets: 0.000000 , 0.000000 , 0.000000 ,0.000000
    Scales: 1.000000 , 1.000000 , 1.000000 ,1.000000
    Averaged on: 10 , 10 , 10 ,10
Second analog to digital sensor is not detected
Foreseen GPS type is Ublox  :GPS is detected and has a fix
Failsafe type is HOLD

Config parameters are OK
Press ? + Enter to get help about the commands

 >> 
ibus pooling with checksum error
ibus pooling with checksum error

Cmd to execute:   FV

GPS Latitude = xx.xxxxxx degree
GPS Longitude = xx.xxxxxx degree
GPS Groundspeed = 52 cm/s
GPS Heading = 288.520000 degree
GPS Altitude = 1024 cm
GPS Num sat. = 106 cm
GPS Date J M A = 20 2 23 
GPS Time H M S = 13 29 38 
GPS Pdop = 444 
GPS Home bearing = 164 degree
GPS Home distance = 25 m
Volt 3 = 578 mVolt
Volt 4 = 557 mVolt
Temp 1 (Volt 3) = 592 degree
Vspeed = 3 cm/s
Baro Rel altitude = 21 cm
Pitch = 0 degree
Roll = 4 degree
RPM = 0 Hertz
Ads 1 1 = 0 mVolt
Ads 1 2 = 0 mVolt
Ads 1 3 = 0 mVolt
Ads 1 4 = 0 mVolt

mstrens, 2023-02-20:
This is strange. I do not understand why there are error and why only twice.
I made a change and added more info in the debug message in version 1.2.4
Can you try it.

pierrotm777, 2023-02-20:
Last firmware return 👍 
ibus request for sensor adr 1
ibus request for sensor adr 1
ibus request for sensor adr 1
ibus request for sensor adr 1
ibus request for sensor adr 1
ibus request for sensor adr 1
ibus request for sensor adr 1
ibus request for sensor adr 1

doloebig, 2023-02-20:
continuous "ibus request for sensor adr 1"
no input possible

pierrotm777, 2023-02-20:
[24 MHz,IBUS_4.zip](https://github.com/mstrens/oXs_on_RP2040/files/10784996/24.MHz.IBUS_4.zip)

mstrens, 2023-02-20:
There is a small progress.
At least a first ibus command is recognized.
Can you make a new capture taking care that the capture 
- begins while oXs is not yet powered on and
- ended while the message ibus request for sensor adr is already displayed.

mstrens, 2023-02-20:
Sorry I did not saw the zip file
I will first look at it.

pierrotm777, 2023-02-20:
Just one thing, this my discovered values without RP2040. 
[image]

pierrotm777, 2023-02-20:
Another thing, my sensor  FS-CVT01 (voltage sensor) is discovered as A3 value !

pierrotm777, 2023-02-20:
Always prints:
ibus request for sensor adr 1
ibus request for sensor adr 1
ibus request for sensor adr 1
ibus request for sensor adr 1
ibus request for sensor adr 1

pierrotm777, 2023-02-20:
If i comment the lignes 313 to 316 i have a new print:
error ibus request 0X09 for sensor adr 10
error ibus request 0X09 for sensor adr 10
error ibus request 0X09 for sensor adr 10
error ibus request 0X09 for sensor adr 10
error ibus request 0X09 for sensor adr 10
error ibus request 0X09 for sensor adr 10
error ibus request 0X09 for sensor adr 10
error ibus request 0X09 for sensor adr 10
error ibus request 0X09 for sensor adr 10
error ibus request 0X09 for sensor adr 10
error ibus request 0X09 for sensor adr 10
error ibus request 0X09 for sensor adr 10
error ibus request 0X09 for sensor adr 10

mstrens, 2023-02-20:
Can you try version 1.2.6
I added some more debug messages.
Note: there should be a line at the begin with "List of ibus fields xxxxxx"
It would be good to know the rest of the line too (+ the repeating messages)

pierrotm777, 2023-02-20:
Always this print:
error ibus request 0X09 for sensor adr 10
error ibus request 0X09 for sensor adr 10
error ibus request 0X09 for sensor adr 10
error ibus request 0X09 for sensor adr 10
error ibus request 0X09 for sensor adr 10

pierrotm777, 2023-02-20:
I have new things sorry:
ibus frame sent: 4  8a  71  ff 
end sending
error ibus request 0X09 for sensor adr 10
error ibus request 0X09 for sensor adr 10
error ibus request 0X09 for sensor adr 10
error ibus request 0X09 for sensor adr 10
error ibus request 0X09 for sensor adr 10
error ibus request 0X09 for sensor adr 10
error ibus request 0X09 for sensor adr 10
error ibus request 0X09 for sensor adr 10
error ibus request 0X09 for sensor adr 10
error ibus request 0X09 for sensor adr 10
ibus frame sent: 4  8a  71  ff 
end sending
error ibus request 0X09 for sensor adr 10
error ibus request 0X09 for sensor adr 10
error ibus request 0X09 for sensor adr 10
error ibus request 0X09 for sensor adr 10
error ibus request 0X09 for sensor adr 10
error ibus request 0X09 for sensor adr 10
error ibus request 0X09 for sensor adr 10
error ibus request 0X09 for sensor adr 10
error ibus request 0X09 for sensor adr 10
error ibus request 0X09 for sensor adr 10
ibus frame sent: 4  8a  71  ff 
end sending
error ibus request 0X09 for sensor adr 10
error ibus request 0X09 for sensor adr 10
error ibus request 0X09 for sensor adr 10
error ibus request 0X09 for sensor adr 10
error ibus request 0X09 for sensor adr 10
error ibus request 0X09 for sensor adr 10
error ibus request 0X09 for sensor adr 10
error ibus request 0X09 for sensor adr 10
error ibus request 0X09 for sensor adr 10
error ibus request 0X09 for sensor adr 10
ibus frame sent: 4  8a  71  ff 
end sending
error ibus request 0X09 for sensor adr 10
error ibus request 0X09 for sensor adr 10
error ibus request 0X09 for sensor adr 10
error ibus request 0X09 for sensor adr 10
error ibus request 0X09 for sensor adr 10
error ibus request 0X09 for sensor adr 10
error ibus request 0X09 for sensor adr 10
error ibus request 0X09 for sensor adr 10
error ibus request 0X09 for sensor adr 10
error ibus request 0X09 for sensor adr 10
ibus frame sent: 4  8a  71  ff 
end sending
error ibus request 0X09 for sensor adr 10
error ibus request 0X09 for sensor adr 10
error ibus request 0X09 for sensor adr 10
error ibus request 0X09 for sensor adr 10

pierrotm777, 2023-02-20:
I don't know if it's can help you, see here, line [477](https://github.com/adis1313/iBUSTelemetry-Arduino/blob/master/iBUSTelemetry.cpp) and after.

mstrens, 2023-02-20:
So it looks like the receiver detected that oXs sent a reply for 9 types of fields. There are some progress.
Can you also provide the beginning of the debug messages.
Can you also provide a capture taking care that the capture:
- begins before power on oXs
- end when error isbus request... are displayed.

mstrens, 2023-02-20:
The link you provided is the same I used.

pierrotm777, 2023-02-20:
Version = 1.2.6 
    Function                Pin   Change entering XXX=yyy (yyy=255 to disable)
Primary channels input    =    5  (PRI     = 5, 9, 21, 25)
Secondary channels input  =   13  (SEC     = 1, 13, 17, 29)
Telemetry . . . . . . . . =    0  (TLM     = 0, 1, 2, ..., 29)
GPS Rx  . . . . . . . . . =    8  (GPS_RX  = 0, 1, 2, ..., 29)
GPS Tx  . . . . . . . . . =    7  (GPS_TX  = 0, 1, 2, ..., 29)
Sbus OUT  . . . . . . . . =  255  (SBUS_OUT= 0, 1, 2, ..., 29)
RPM   . . . . . . . . . . =   14  (RPM     = 0, 1, 2, ..., 29)
SDA (I2C sensors) . . . . =    2  (SDA     = 2, 6, 10, 14, 18, 22, 26)
SCL (I2C sensors) . . . . =    3  (SCL     = 3, 7, 11, 15, 19, 23, 27)
PWM Channels 1, 2, 3 ,4   =  255  255  255  255 (C1 / C16= 0, 1, 2, ..., 15)
PWM Channels 5, 6, 7 ,8   =  255  255  255  255
PWM Channels 9,10,11,12   =  255  255  255  255
PWM Channels 13,14,15,16  =  255  255  255  255
Voltage 1, 2, 3, 4        =  255  255  255   29 (V1 / V4 = 26, 27, 28, 29)

Protocol is ibus(Flysky)
CRSF baudrate = 420000
Voltage parameters:
    Scales : 1.000000 , 1.000000 , 1.000000 , 1.000000 
    Offsets: 0.000000 , 0.000000 , 0.000000 , 0.000000 
    No temperature sensors are connected on V3 and V4
RPM multiplier = 1.000000
Baro sensor is detected using MS5611
    Sensitivity min = 100 (at 100)   , max = 300 (at 1000)
    Hysteresis = 5 
Acc/Gyro is detected using MP6050
     Acceleration offsets X, Y, Z = -6502 , 1062 , 2346
     Gyro offsets         X, Y, Z = 185 , -36 , -10
First analog to digital sensor is detected using ads1115
    Measurement setup: 4 , 5 , 6 ,7
    Gains: 1 , 1 , 1 ,1
    Rates: 5 , 5 , 5 ,5
    Offsets: 0.000000 , 0.000000 , 0.000000 ,0.000000
    Scales: 1.000000 , 1.000000 , 1.000000 ,1.000000
    Averaged on: 10 , 10 , 10 ,10
Second analog to digital sensor is not detected
Foreseen GPS type is Ublox  :GPS is detected but has not (yet) a fix
Failsafe type is HOLD

Config parameters are OK
Press ? + Enter to get help about the commands
ibus frame sent: 4  81  7a  ff 
end sending
ibus frame sent: 4  81  7a  ff 
end sending
ibus frame sent: 4  81  7a  ff 
end sending
ibus frame sent: 4  81  7a  ff 
end sending
ibus frame sent: 4  81  7a  ff 
end sending
ibus frame sent: 4  81  7a  ff 
end sending
ibus frame sent: 4  81  7a  ff 
end sending
ibus frame sent: 4  81  7a  ff 
end sending
ibus frame sent: 4  81  7a  ff 
end sending
ibus frame sent: 4  81  7a  ff 
end sending
ibus frame sent: 4  81  7a  ff 
end sending
ibus frame sent: 4  81  7a  ff 
end sending
ibus pooling with checksum error 4 81 4 91 
ibus frame sent: 6  91  83  4  e1  fe 
end sending
ibus frame sent: 4  82  79  ff 
end sending
ibus frame sent: 6  92  9  2  5c  ff 
end sending
ibus frame sent: 4  83  78  ff 
end sending
ibus frame sent: 6  93  7  2  5d  ff 
end sending
ibus frame sent: 4  84  77  ff 
end sending
ibus frame sent: 6  94  81  4  e0  fe 
end sending
ibus frame sent: 4  85  76  ff 
end sending
ibus frame sent: 6  95  80  4  e0  fe 
end sending
ibus frame sent: 4  86  75  ff 
end sending
ibus frame sent: 6  96  b  2  56  ff 
end sending
ibus frame sent: 4  87  74  ff 
end sending
ibus frame sent: 6  97  13  2  4d  ff 
end sending
ibus frame sent: 4  88  73  ff 
end sending
ibus frame sent: 6  98  a  2  55  ff 
end sending
ibus frame sent: 4  89  72  ff 
end sending
ibus frame sent: 6  99  82  4  da  fe 
end sending
ibus frame sent: 4  8a  71  ff 
end sending
error ibus request 0X09 for sensor adr 10
error ibus request 0X09 for sensor adr 10
error ibus request 0X09 for sensor adr 10

pierrotm777, 2023-02-20:
[24 MHz,IBUS_5.zip](https://github.com/mstrens/oXs_on_RP2040/files/10785405/24.MHz.IBUS_5.zip)

mstrens, 2023-02-20:
Thanks. This helped.
Now a new version 1.2.8 to test.
Same report as before could help if this does not yet work.

pierrotm777, 2023-02-20:
I think it's correct because i receive altitude , but the value is 0.0  :-)

YES !!!!

pierrotm777, 2023-02-20:
And the print is:
Setup1 takes 1200030 usec
List of ibus fields :  0  19  18  23  1  0  5  2  3  4 

Version = 1.2.5 
    Function                Pin   Change entering XXX=yyy (yyy=255 to disable)
Primary channels input    =    5  (PRI     = 5, 9, 21, 25)
Secondary channels input  =   13  (SEC     = 1, 13, 17, 29)
Telemetry . . . . . . . . =    0  (TLM     = 0, 1, 2, ..., 29)
GPS Rx  . . . . . . . . . =    8  (GPS_RX  = 0, 1, 2, ..., 29)
GPS Tx  . . . . . . . . . =    7  (GPS_TX  = 0, 1, 2, ..., 29)
Sbus OUT  . . . . . . . . =  255  (SBUS_OUT= 0, 1, 2, ..., 29)
RPM   . . . . . . . . . . =   14  (RPM     = 0, 1, 2, ..., 29)
SDA (I2C sensors) . . . . =    2  (SDA     = 2, 6, 10, 14, 18, 22, 26)
SCL (I2C sensors) . . . . =    3  (SCL     = 3, 7, 11, 15, 19, 23, 27)
PWM Channels 1, 2, 3 ,4   =  255  255  255  255 (C1 / C16= 0, 1, 2, ..., 15)
PWM Channels 5, 6, 7 ,8   =  255  255  255  255
PWM Channels 9,10,11,12   =  255  255  255  255
PWM Channels 13,14,15,16  =  255  255  255  255
Voltage 1, 2, 3, 4        =  255  255  255   29 (V1 / V4 = 26, 27, 28, 29)

Protocol is ibus(Flysky)
CRSF baudrate = 420000
Voltage parameters:
    Scales : 1.000000 , 1.000000 , 1.000000 , 1.000000 
    Offsets: 0.000000 , 0.000000 , 0.000000 , 0.000000 
    No temperature sensors are connected on V3 and V4
RPM multiplier = 1.000000
Baro sensor is detected using MS5611
    Sensitivity min = 100 (at 100)   , max = 300 (at 1000)
    Hysteresis = 5 
Acc/Gyro is detected using MP6050
     Acceleration offsets X, Y, Z = -6502 , 1062 , 2346
     Gyro offsets         X, Y, Z = 185 , -36 , -10
First analog to digital sensor is detected using ads1115
    Measurement setup: 4 , 5 , 6 ,7
    Gains: 1 , 1 , 1 ,1
    Rates: 5 , 5 , 5 ,5
    Offsets: 0.000000 , 0.000000 , 0.000000 ,0.000000
    Scales: 1.000000 , 1.000000 , 1.000000 ,1.000000
    Averaged on: 10 , 10 , 10 ,10
Second analog to digital sensor is not detected
Foreseen GPS type is Ublox  :GPS is not (yet) detected
Failsafe type is HOLD

Config parameters are OK
Press ? + Enter to get help about the commands
error ibus request 0X09 for sensor adr 10
error ibus request 0X09 for sensor adr 10
error ibus request 0X09 for sensor adr 10
error ibus request 0X09 for sensor adr 10
error ibus request 0X09 for sensor adr 10
error ibus request 0X09 for sensor adr 10
error ibus request 0X09 for sensor adr 10
error ibus request 0X09 for sensor adr 10
error ibus request 0X09 for sensor adr 10
error ibus request 0X09 for sensor adr 10

mstrens, 2023-02-20:
I forgot to put the last version on github.
I did it now (version 1.2.8)
Can you make the test again and provide 
- the log (messages on pc)
- the capture (as before - begin before oXs power on and end after messages are displayed.

pierrotm777, 2023-02-20:
Test with 12.8:

ibus frame sent: 6  98  a  2  55  ff 
end sending
ibus frame sent: 4  89  72  ff 
end sending
ibus frame sent: 6  99  82  4  da  fe 
end sending
ibus request for sensor adr 10 max=9
ibus frame sent: 8  a1  82  0  0  0  56  ff 
end sending
ibus frame sent: 8  a1  82  ff  ff  ff  5a  fb 
end sending
ibus frame sent: 4  81  7a  ff 
end sending
ibus frame sent: 6  91  83  4  e1  fe 
end sending
ibus frame sent: 4  82  79  ff 
end sending
ibus frame sent: 6  92  9  2  5c  ff 
end sending
ibus frame sent: 4  83  78  ff 
end sending
ibus frame sent: 6  93  7  2  5d  ff 
end sending
ibus frame sent: 4  84  77  ff 
end sending
ibus frame sent: 6  94  81  4  e0  fe 
end sending
ibus frame sent: 4  85  76  ff 
end sending
ibus frame sent: 6  95  80  4  e0  fe 
end sending
ibus frame sent: 4  86  75  ff 
end sending
ibus frame sent: 6  96  b  2  56  ff 
end sending
ibus frame sent: 4  87  74  ff 
end sending
ibus frame sent: 6  97  13  2  4d  ff 
end sending
ibus frame sent: 4  88  73  ff 
end sending
ibus frame sent: 6  98  a  2  55  ff 
end sending
ibus frame sent: 4  89  72  ff 
end sending
ibus frame sent: 6  99  82  4  da  fe 
end sending
ibus request for sensor adr 10 max=9
ibus frame sent: 8  a1  82  fd  ff  ff  5c  fb 
end sending
ibus frame sent: 8  a1  82  0  0  0  56  ff 
[24 MHz_IBUS6.zip](https://github.com/mstrens/oXs_on_RP2040/files/10786227/24.MHz_IBUS6.zip)

mstrens, 2023-02-20:
It seems that:
- oXs sent some data back to the Rx. Do you get values on the handset?
- your capture was not long enough to cover the debug messages sent to the USB
I made a version 1.2.9 that generates more messages to better understand the logic of ibus.
Can you repeat the test?

doloebig, 2023-02-20:
i get sensors on the Handset but no values
[image]

[image]

doloebig, 2023-02-20:
ibus request for sensor adr 12 max=11
ibus frame sent: 6  a1  82  1  57  ff 
end sending
ibus frame sent: 6  a1  82  1  57  ff 
end sending
ibus frame sent: 4  81  7a  ff 
end sending
ibus frame sent: 6  91  3  2  63  ff 
end sending
ibus frame sent: 4  82  79  ff 
end sending
ibus frame sent: 6  92  5  2  60  ff 
end sending
ibus frame sent: 4  83  78  ff 
end sending
ibus frame sent: 6  93  1  2  63  ff 
end sending
ibus frame sent: 4  84  77  ff 
end sending
ibus frame sent: 6  94  1  2  62  ff 
end sending
ibus frame sent: 4  85  76  ff 
end sending
ibus frame sent: 6  95  7  2  5b  ff 
end sending
ibus frame sent: 4  86  75  ff 
end sending
ibus frame sent: 6  96  81  4  de  fe 
end sending
ibus frame sent: 4  87  74  ff 
end sending
ibus frame sent: 6  97  80  4  de  fe 
end sending
ibus frame sent: 4  88  73  ff 
end sending
ibus frame sent: 6  98  b  2  54  ff 
end sending
ibus frame sent: 4  89  72  ff 
end sending
ibus frame sent: 6  99  13  2  4b  ff 
end sending
ibus frame sent: 4  8a  71  ff 
end sending
ibus frame sent: 6  9a  a  2  53  ff 
end sending
ibus frame sent: 4  8b  70  ff 
end sending
ibus frame sent: 6  9b  82  4  d8  fe 
end sending
ibus request for sensor adr 12 max=11
ibus frame sent: 6  a1  82  1  57  ff 
end sending
ibus frame sent: 4  81  7a  ff 
end sending
ibus frame sent: 6  91  3  2  63  ff 
end sending

pierrotm777, 2023-02-20:
FV return me without my GPS:
Cmd to execute:   FV

GPS Num sat. = 0 cm
GPS Date J M A = 18 10 15 
GPS Time H M S = 0 0 19 
Volt 4 = 572 mVolt
Vspeed = -2 cm/s
Baro Rel altitude = -8 cm
Pitch = -86 degree
Roll = -126 degree
RPM = 0 Hertz
Ads 1 1 = 0 mVolt
Ads 1 2 = 0 mVolt
Ads 1 3 = 0 mVolt
Ads 1 4 = 0 mVolt
[24 MHz_IBUS7.zip](https://github.com/mstrens/oXs_on_RP2040/files/10786571/24.MHz_IBUS7.zip)
[debug_ibus7.txt](https://github.com/mstrens/oXs_on_RP2040/files/10786575/debug_ibus7.txt)

doloebig, 2023-02-20:
[image]
[image]

doloebig, 2023-02-20:
on github is still 1.2.8

pierrotm777, 2023-02-20:
Idem, i discover now to TPM1 if TEMP=2.

pierrotm777, 2023-02-20:
doloebig,
You discover a RPM sensor or it's a Flysky sensor ?

doloebig, 2023-02-20:
it is discvered.
have only the oXs sensor on RX.

mstrens, 2023-02-20:
can you try version 1.2.10 on github and make the same report as before.

pierrotm777, 2023-02-20:
List of ibus fields :  0  11  12  16  19  18  23  1  0  5  2  3  4 

Version = 1.2.8 
    Function                Pin   Change entering XXX=yyy (yyy=255 to disable)
Primary channels input    =    5  (PRI     = 5, 9, 21, 25)
Secondary channels input  =   13  (SEC     = 1, 13, 17, 29)
Telemetry . . . . . . . . =    0  (TLM     = 0, 1, 2, ..., 29)
GPS Rx  . . . . . . . . . =    8  (GPS_RX  = 0, 1, 2, ..., 29)
GPS Tx  . . . . . . . . . =    7  (GPS_TX  = 0, 1, 2, ..., 29)
Sbus OUT  . . . . . . . . =  255  (SBUS_OUT= 0, 1, 2, ..., 29)
RPM   . . . . . . . . . . =   14  (RPM     = 0, 1, 2, ..., 29)
SDA (I2C sensors) . . . . =    2  (SDA     = 2, 6, 10, 14, 18, 22, 26)
SCL (I2C sensors) . . . . =    3  (SCL     = 3, 7, 11, 15, 19, 23, 27)
PWM Channels 1, 2, 3 ,4   =  255  255  255  255 (C1 / C16= 0, 1, 2, ..., 15)
PWM Channels 5, 6, 7 ,8   =  255  255  255  255
PWM Channels 9,10,11,12   =  255  255  255  255
PWM Channels 13,14,15,16  =  255  255  255  255
Voltage 1, 2, 3, 4        =   26   27   28  255 (V1 / V4 = 26, 27, 28, 29)

Protocol is ibus(Flysky)
CRSF baudrate = 420000
Voltage parameters:
    Scales : 1.000000 , 1.000000 , 1.000000 , 1.000000 
    Offsets: 0.000000 , 0.000000 , 0.000000 , 0.000000 
    One temperature sensor is connected on V3
RPM multiplier = 1.000000
Baro sensor is detected using MS5611
    Sensitivity min = 100 (at 100)   , max = 300 (at 1000)
    Hysteresis = 5 
Acc/Gyro is detected using MP6050
     Acceleration offsets X, Y, Z = -6502 , 1062 , 2346
     Gyro offsets         X, Y, Z = 185 , -36 , -10
First analog to digital sensor is detected using ads1115
    Measurement setup: 4 , 5 , 6 ,7
    Gains: 1 , 1 , 1 ,1
    Rates: 5 , 5 , 5 ,5
    Offsets: 0.000000 , 0.000000 , 0.000000 ,0.000000
    Scales: 1.000000 , 1.000000 , 1.000000 ,1.000000
    Averaged on: 10 , 10 , 10 ,10
Second analog to digital sensor is not detected
Foreseen GPS type is Ublox  :GPS is detected but has not (yet) a fix
Failsafe type is HOLD

Config parameters are OK
Press ? + Enter to get help about the commands

With my config, i discover:
I discover: A3, TMP1, Alt, Curr but not RPM !

mstrens, 2023-02-20:
I do not yet know why RPM is not discovered.
I would like first to get the results of version 1.2.10

doloebig, 2023-02-20:
ibus frame sent: 6  a3  1a  0  3c  ff 
ibus get 4 a4 57 ff
ibus frame sent: 6  a4  1b  0  3a  ff 
ibus get 4 a5 56 ff
ibus frame sent: 6  a5  0  0  54  ff 
ibus get 4 a6 55 ff
ibus get 4 a6 55 ff
ibus get 4 a6 55 ff
ibus get 4 a6 55 ff
ibus get 4 a6 55 ff
ibus get 4 a6 55 ff
ibus get 4 a6 55 ff
ibus get 4 a6 55 ff
ibus get 4 a6 55 ff
ibus get 4 a6 55 ff
ibus get 4 86 75 ff
ibus frame sent: 4  86  75  ff 
ibus get 4 96 65 ff
ibus frame sent: 6  96  81  4  de  fe 
ibus get 4 87 74 ff
ibus frame sent: 4  87  74  ff 
ibus get 4 97 64 ff
ibus frame sent: 6  97  80  4  de  fe 
ibus get 4 88 73 ff
ibus frame sent: 4  88  73  ff 
ibus get 4 98 63 ff
ibus frame sent: 6  98  b  2  54  ff 
ibus get 4 89 72 ff
ibus frame sent: 4  89  72  ff 
ibus get 4 99 62 ff
ibus frame sent: 6  99  13  2  4b  ff 
ibus get 4 8a 71 ff
ibus frame sent: 4  8a  71  ff 
ibus get 4 9a 61 ff
ibus frame sent: 6  9a  a  2  53  ff 
ibus get 4 8b 70 ff
ibus frame sent: 4  8b  70  ff 
ibus get 4 9b 60 ff
ibus frame sent: 6  9b  82  4  d8  fe 
ibus get 4 8c 6f ff
ibus get 4 a1 5a ff
ibus frame sent: 6  a1  1  0  57  ff 
ibus get 4 a2 59 ff
ibus frame sent: 6  a2  f3  1  63  fe 
ibus get 4 a3 58 ff
ibus frame sent: 6  a3  1a  0  3c  ff 
ibus get 4 a4 57 ff
ibus frame sent: 6  a4  1b  0  3a  ff 
ibus get 4 a5 56 ff
ibus frame sent: 6  a5  0  0  54  ff 
ibus get 4 a6 55 ff
ibus get 4 a6 55 ff
ibus get 4 a6 55 ff
ibus get 4 a6 55 ff
ibus get 4 a6 55 ff
ibus get 4 a6 55 ff
ibus get 4 a6 55 ff
ibus get 4 a6 55 ff
ibus get 4 a6 55 ff
ibus get 4 a6 55 ff
ibus get 4 86 75 ff
ibus frame sent: 4  86  75  ff 
ibus get 4 96 65 ff
ibus frame sent: 6  96  81  4  de  fe 
ibus get 4 87 74 ff
ibus frame sent: 4  87  74  ff 
ibus get 4 97 64 ff
ibus frame sent: 6  97  80  4  de  fe 
ibus get 4 88 73 ff
ibus frame sent: 4  88  73  ff 
ibus get 4 98 63 ff
ibus frame sent: 6  98  b  2  54  ff 
ibus get 4 89 72 ff
ibus frame sent: 4  89  72  ff 
ibus get 4 99 62 ff
ibus frame sent: 6  99  13  2  4b  ff 
ibus get 4 8a 71 ff
ibus frame sent: 4  8a  71  ff 
ibus get 4 9a 61 ff
ibus frame sent: 6  9a  a  2  53  ff 
ibus get 4 8b 70 ff
ibus frame sent: 4  8b  70  ff 
ibus get 4 9b 60 ff
ibus frame sent: 6  9b  82  4  d8  fe

how can i interupt the messages?

pierrotm777, 2023-02-20:
[24 MHz_IBUS8.zip](https://github.com/mstrens/oXs_on_RP2040/files/10786814/24.MHz_IBUS8.zip)
[debug_ibus8.txt](https://github.com/mstrens/oXs_on_RP2040/files/10786829/debug_ibus8.txt)

doloebig, 2023-02-20:
Setup1 takes 1100032 usec
List of ibus fields :  0  11  12  16  17  23  1  0  5  2  3  4 

Version = 1.2.10 
    Function                Pin   Change entering XXX=yyy (yyy=255 to disable)
Primary channels input    =  255  (PRI     = 5, 9, 21, 25)
Secondary channels input  =  255  (SEC     = 1, 13, 17, 29)
Telemetry . . . . . . . . =    7  (TLM     = 0, 1, 2, ..., 29)
GPS Rx  . . . . . . . . . =    3  (GPS_RX  = 0, 1, 2, ..., 29)
GPS Tx  . . . . . . . . . =    2  (GPS_TX  = 0, 1, 2, ..., 29)
Sbus OUT  . . . . . . . . =  255  (SBUS_OUT= 0, 1, 2, ..., 29)
RPM   . . . . . . . . . . =    6  (RPM     = 0, 1, 2, ..., 29)
SDA (I2C sensors) . . . . =  255  (SDA     = 2, 6, 10, 14, 18, 22, 26)
SCL (I2C sensors) . . . . =  255  (SCL     = 3, 7, 11, 15, 19, 23, 27)
PWM Channels 1, 2, 3 ,4   =  255  255  255  255 (C1 / C16= 0, 1, 2, ..., 15)
PWM Channels 5, 6, 7 ,8   =  255  255  255  255
PWM Channels 9,10,11,12   =  255  255  255  255
PWM Channels 13,14,15,16  =  255  255  255  255
Voltage 1, 2, 3, 4        =   26   27   28   29 (V1 / V4 = 26, 27, 28, 29)

Protocol is ibus(Flysky)
CRSF baudrate = 420000
Voltage parameters:
    Scales : 5.700000 , 10.000000 , 0.100000 , 0.100000 
    Offsets: 0.000000 , 0.000000 , 0.000000 , 0.000000 
    Temperature sensors are connected on V3 and V4
RPM multiplier = 1.000000
Baro sensor is not detected
Acc/Gyro is not detected
First analog to digital sensor is not detected
Second analog to digital sensor is not detected
Foreseen GPS type is Ublox  :GPS is detected but has not (yet) a fix
Failsafe type is HOLD

Config parameters are OK
Press ? + Enter to get help about the commands
ibus pooling with checksum error 4 81 4 81 
ibus get 4 81 7a ff
ibus frame sent: 4  81  7a  ff 
ibus get 4 91 6a ff
ibus frame sent: 6  91  3  2  63  ff 
ibus get 4 82 79 ff
ibus frame sent: 4  82  79  ff 
ibus get 4 92 69 ff
ibus frame sent: 6  92  5  2  60  ff 
ibus get 4 83 78 ff
ibus frame sent: 4  83  78  ff 
ibus get 4 93 68 ff
ibus frame sent: 6  93  1  2  63  ff 
ibus get 4 84 77 ff
ibus frame sent: 4  84  77  ff 
ibus get 4 94 67 ff
ibus frame sent: 6  94  1  2  62  ff 
ibus get 4 85 76 ff
ibus frame sent: 4  85  76  ff 
ibus get 4 95 66 ff
ibus frame sent: 6  95  7  2  5b  ff 
ibus get 4 86 75 ff
ibus frame sent: 4  86  75  ff 
ibus get 4 96 65 ff
ibus frame sent: 6  96  81  4  de  fe 
ibus get 4 87 74 ff
ibus frame sent: 4  87  74  ff 
ibus get 4 97 64 ff
ibus frame sent: 6  97  80  4  de  fe 
ibus get 4 88 73 ff
ibus frame sent: 4  88  73  ff 
ibus get 4 98 63 ff
ibus frame sent: 6  98  b  2  54  ff 
ibus get 4 89 72 ff
ibus frame sent: 4  89  72  ff 
ibus get 4 99 62 ff
ibus frame sent: 6  99  13  2  4b  ff 
ibus get 4 8a 71 ff
ibus frame sent: 4  8a  71  ff 
ibus get 4 9a 61 ff
ibus frame sent: 6  9a  a  2  53  ff 
ibus get 4 8b 70 ff
ibus frame sent: 4  8b  70  ff 
ibus get 4 9b 60 ff
ibus frame sent: 6  9b  82  4  d8  fe 
ibus get 4 8c 6f ff
ibus get 4 a1 5a ff
ibus frame sent: 6  a1  1  0  57  ff 
ibus get 4 a2 59 ff
ibus frame sent: 6  a2  ea  1  6c  fe 
ibus get 4 a3 58 ff
ibus frame sent: 6  a3  1a  0  3c  ff 
ibus get 4 a4 57 ff
ibus frame sent: 6  a4  1b  0  3a  ff 
ibus get 4 a5 56 ff
ibus frame sent: 6  a5  0  0  54  ff 
ibus get 4 a6 55 ff
ibus get 4 a6 55 ff
ibus get 4 a6 55 ff
ibus get 4 a6 55 ff
ibus get 4 a6 55 ff
ibus get 4 a6 55 ff
ibus get 4 a6 55 ff
ibus get 4 a6 55 ff
ibus get 4 a6 55 ff
ibus get 4 a6 55 ff
ibus get 4 86 75 ff
ibus frame sent: 4  86  75  ff 
ibus get 4 96 65 ff
ibus frame sent: 6  96  81  4  de  fe 
ibus get 4 87 74 ff
ibus frame sent: 4  87  74  ff 
ibus get 4 97 64 ff
ibus frame sent: 6  97  80  4  de  fe 
ibus get 4 88 73 ff
ibus frame sent: 4  88  73  ff 
ibus get 4 98 63 ff
ibus frame sent: 6  98  b  2  54  ff 
ibus get 4 89 72 ff
ibus frame sent: 4  89  72  ff 
ibus get 4 99 62 ff
ibus frame sent: 6  99  13  2  4b  ff 
ibus get 4 8a 71 ff
ibus frame sent: 4  8a  71  ff 
ibus get 4 9a 61 ff
ibus frame sent: 6  9a  a  2  53  ff 
ibus get 4 8b 70 ff
ibus frame sent: 4  8b  70  ff 
ibus get 4 9b 60 ff
ibus frame sent: 6  9b  82  4  d8  fe 
ibus get 4 8c 6f ff
ibus get 4 a1 5a ff
ibus frame sent: 6  a1  1  0  57  ff 
ibus get 4 a2 59 ff
ibus frame sent: 6  a2  ec  1  6a  fe 
ibus get 4 a3 58 ff
ibus frame sent: 6  a3  1a  0  3c  ff 
ibus get 4 a4 57 ff
ibus frame sent: 6  a4  1b  0  3a  ff 
ibus get 4 a5 56 ff
ibus frame sent: 6  a5  0  0  54  ff 
ibus get 4 a6 55 ff
ibus get 4 a6 55 ff
ibus get 4 a6 55 ff
ibus get 4 a6 55 ff
ibus get 4 a6 55 ff
ibus get 4 a6 55 ff
ibus get 4 a6 55 ff
ibus get 4 a6 55 ff
ibus get 4 a6 55 ff
ibus get 4 a6 55 ff
ibus get 4 86 75 ff
ibus frame sent: 4  86  75  ff 
ibus get 4 96 65 ff
ibus frame sent: 6  96  81  4  de  fe 
ibus get 4 87 74 ff
ibus frame sent: 4  87  74  ff 
ibus get 4 97 64 ff
ibus frame sent: 6  97  80  4  de  fe 
ibus get 4 88 73 ff
ibus frame sent: 4  88  73  ff 
ibus get 4 98 63 ff
ibus frame sent: 6  98  b  2  54  ff 
ibus get 4 89 72 ff
ibus frame sent: 4  89  72  ff 
ibus get 4 99 62 ff
ibus frame sent: 6  99  13  2  4b  ff 
ibus get 4 8a 71 ff
ibus frame sent: 4  8a  71  ff 
ibus get 4 9a 61 ff
ibus frame sent: 6  9a  a  2  53  ff 
ibus get 4 8b 70 ff
ibus frame sent: 4  8b  70  ff 
ibus get 4 9b 60 ff
ibus frame sent: 6  9b  82  4  d8  fe 
ibus get 4 8c 6f ff
ibus get 4 a1 5a ff
ibus frame sent: 6  a1  1  0  57  ff 
ibus get 4 a2 59 ff
ibus frame sent: 6  a2  eb  1  6b  fe 
ibus get 4 a3 58 ff
ibus frame sent: 6  a3  1a  0  3c  ff 
ibus get 4 a4 57 ff

pierrotm777, 2023-02-20:
Just a remark, altitude returns a good value :-)

mstrens, 2023-02-20:
can you try version 1.2.11.
I had foreseen to send a value only when a new value was calculated.
This does not seems to work for ibus. It seems that the Rx restart discovering all sensors each time a value is not provided for a sensor that has already been discovered.
I change this:
- I send now 0 when a value is not yet available
- I repeat the previous value when it has been at least one available.
Test version 1.2.11.
If it works, I can then remove the debug messages in next version.

pierrotm777, 2023-02-20:
YES, i discover now, RPM and GPSAlt, GPSSpd Sats Hdg and more, Curr , Capa etc...

pierrotm777, 2023-02-20:
All GPS values are 0 but i suppose it's normal ?

[32 more replies]

closed 2023-02-22 by doloebig