mstrens/oXs_on_RP2040 #28: USB connection lost inbetween commands

nostromo123, 2023-02-09:
I'll put MPX for the moment aside and will concentrate on ELRS.

I nuked the RP2040 and rebuild the breadboard here with RP2040, ELRS RX, MS5611 and MPU6050.

Ver.0.6.12 and only RP2040 connected during configuration. I see a strange effect during configuration.

[image]

After each entered command the USB connection drops out. In picture /dev/ttyACM0 disconnects and I have to reconnect to /dev/ttyACM1.

That behavior is totally new here. I think, Torsten mention a similar performance.

Regards - KH

mstrens, 2023-02-09:
I added 2 delays when the config is saved.
Perhaps those delays create this issue.
If you can compile by your self, yo can try to remove those 2 delays.
It is in param.cpp at lines 767 and 779.
sleep_ms(1000); 

Just add "//" in front of each line so having
//sleep_ms(1000);  

If you can't compile yourself, I will change this in next version.

nostromo123, 2023-02-09:
OK, I'll do so and come back. No problem doing that short modification here ...

nostromo123, 2023-02-10:
Good morning.

I think, there is a typo in the code ver.0.6.12. Can not build!

Cross checked with 0.6.10. No issue building that!

As I don't want to mess around in your code, please have a look

[image]

[Neue Datei.txt](https://github.com/mstrens/oXs_on_RP2040/files/10705286/Neue.Datei.txt)

Tell me please, what to correct in 0.6.12 and I'll give it a try.

Regards - KH

mstrens, 2023-02-10:
I just uploaded version 0.7.0.
I explained on rcgroups what I changed (including the proposal for USB connection).
Probably best to test with this version.

nostromo123, 2023-02-10:
OK, with 0.7.0 don't see this issue no more.

Was a short test, but anyway I'll close this issue now.

closed 2023-02-10 by nostromo123