mstrens/oXs_on_RP2040 #29: ELRS 0.6.12 - YAW, ROLL and PITCH showing on transmitter, but values static

nostromo123, 2023-02-09:
With a reset breadboard with RP2040, ELRS RX, MS5611 and MPU6050 I now think I managed to get a valid MPU6050 sensor calibration and in terminal I see values for x and y axis, which make sense. But on the radio i see not moving values like this

[image]

I think, this was also remaked by GRacing in RCG.

> 
> Version = 0.6.12 
>     Function                Pin   Change entering XXX=yyy (yyy=255 to disable)
> Primary channels input    =    9  (PRI     = 5, 9, 21, 25)
> Secondary channels input  =  255  (SEC     = 1, 13, 17, 29)
> Telemetry . . . . . . . . =   10  (TLM     = 0, 1, 2, ..., 29)
> GPS Rx  . . . . . . . . . =  255  (GPS_RX  = 0, 1, 2, ..., 29)
> GPS Tx  . . . . . . . . . =  255  (GPS_TX  = 0, 1, 2, ..., 29)
> Sbus OUT  . . . . . . . . =  255  (SBUS_OUT= 0, 1, 2, ..., 29)
> RPM   . . . . . . . . . . =  255  (RPM     = 0, 1, 2, ..., 29)
> SDA (I2C sensors) . . . . =   14  (SDA     = 2, 6, 10, 14, 18, 22, 26)
> SCL (I2C sensors) . . . . =   15 (SCL     = 3, 7, 11, 15, 19, 23, 27)
> PWM Channels 1, 2, 3 ,4   =    0    1    2    3 (C1 / C16= 0, 1, 2, ..., 15)
> PWM Channels 5, 6, 7 ,8   =  255  255  255  255
> PWM Channels 9,10,11,12   =  255  255  255  255
> PWM Channels 13,14,15,16  =  255  255  255  255
> Voltage 1, 2, 3, 4        =  255  255  255  255 (V1 / V4 = 26, 27, 28, 29)
> 
> Protocol is CRSF (=ELRS)
> CRSF baudrate = 420000
> Voltage parameters:
>     Scales : 1.000000 , 1.000000 , 1.000000 , *
> 1.000000 
>     Offsets: 0.000000 , 0.000000 , 0.000000 , 0.000000 
> RPM multiplier = 1.000000
> Baro sensor is detected using MS5611
>     Sensitivity min = 100 (at 100)   , max = 300 (at 1000)
>     Hysteresis = 5 
> Acc/Gyro is detected using MP6050
>      Acceleration offsets X, Y, Z = -21754 , 238 , -5938
>      Gyro offsets         X, Y, Z = 90 , 19 , -9
> First analog to digital sensor is not detected
> Second analog to digital sensor is not detected
> Foreseen GPS type is Ublox  :GPS is not (yet) detected
> Failsafe type is HOLD
> 
> Config parameters are OK
> Press ? + Enter to get help about the commands
> 
>  >> 
> *
> *
> *
> *
> *
> *
> *
> *
> p r= -88 175
> p r= -88 174
> p r= -88 173
> p r= -88 172
> p r= -88 175
> p r= -88 174
> p r= -56 179
> p r= -10 -179
> p r= -3 -176
> p r= -5 -161
> p r= -5 -161
> p r= -8 -176
> p r= -18 -166
> p r= -22 -169
> p r= -17 -176
> p r= -14 -160
> p r= -11 -152
> p r= -67 -149
> p r= -52 -93
> p r= -81 -96
> p r= -66 -93
> p r= -86 -122
> p r= -37 -157
> p r= -66 -157
> p r= -83 -96
> p r= -84 -151
> p r= -88 177
> p r= -88 174
> p r= -88 175
> p r= -88 174
> p r= -88 174
> p r= -88 175
> p r= -88 174
> p r= -88 174
> p r= -88 175
> p r= -88 175
> p r= -88 173
> p r= -88 175
> p r= -88 175
> p r= -88 173
> p r= -88 173
> p r= -88 174
> p r= -88 175
> p r= -88 176
> p r= -88 175
> p r= -88 174
> p r= -88 174
> p r= -88 172
> p r= -88 174
> p r= -88 174
> p r= -88 174
> p r= -88 172
> p r= -88 175
> p r= -88 175
> p r= -88 174
> p r= -88 173
> p r= -88 174
> p r= -88 173
> p r= -88 173
> p r= -88 174
> p r= -88 174
> p r= -88 175
> p r= -88 175
> p r= -88 174
> p r= -88 175
> p r= -88 173
> p r= -88 174
> p r= -88 175
> p r= -88 174
> p r= -88 174
> p r= -88 174
> p r= -88 173
> p r= -88 175
> p r= -88 173
> p r= -88 173
> p r= -88 174
> p r= -88 173
> p r= -88 173
> p r= -88 173
> p r= -88 174
> p r= -88 173
> p r= -86 -140
> p r= -89 -160
> p r= -86 106
> p r= -81 99

br - KH

mstrens, 2023-02-09:
The lines p r= .... are generated at the same location as the code that fill the frame to be sent to ELRS receiver for pitch and roll.
As those lines are generated with valid values, it means that oXs reach this part of the program.

It is even possible to get the content of the frames being sent. For this you could remove the // before lines 248/250 from crsf_out.cpp.

So having 
printf("Attitude: ");
for (uint8_t i = 0; i< CRSFBufferLength ; i++) printf( " %02X ", CRSFBuffer[i]);
printf("\n");

You can try this and post the messages displayed on the pc.

At this stage, I do not understand where I can make a mistake.
It is very strange that the handset discover those sensors (pitch/roll) but does not update the values.

nostromo123, 2023-02-09:
No problem, I'll make the modification and come back.

Good night and best regards - KH

mstrens, 2023-02-09:
I test my self and the data are valid.
I checked some other internet code and it seems that the values pitch and roll in the frame may not be in degree (as I expected) but in radian (and with a factor 1000 or 10000).
Therefore, the values to be transmitted have to be multiplied by 174,533.
So you can try to change those lines from crsf_out.cpp
if ( fields[PITCH].available ) {
        fillBufferI16( (int16_t) (fields[PITCH].value)) ; //pitch  
    } else {
        fillBufferI16( (int16_t) 0);
    }
    if ( fields[ROLL].available ) {
        fillBufferI16( (int16_t) (fields[ROLL].value)) ; //roll  
    } else {
        fillBufferI16( (int16_t) 0);
    }
    
to

if ( fields[PITCH].available ) {
        fillBufferI16( (int16_t) (fields[PITCH].value * 175)) ; //pitch  
    } else {
        fillBufferI16( (int16_t) 0);
    }
    if ( fields[ROLL].available ) {
        fillBufferI16( (int16_t) (fields[ROLL].value * 175)) ; //roll  
    } else {
        fillBufferI16( (int16_t) 0);
    }

mstrens, 2023-02-10:
I just included those changes in version 0.7.0 (on git hub)

nostromo123, 2023-02-10:
OK, I see moving values now with 0.7.0

But I'll leave this issue open.

Have to nuke the board, re calibrate and the as the output is in rad, not in degrees, have to do some calculations ....

mstrens, 2023-02-10:
I had expected (not tested) that the handset would automatically convert the radian to degree (knowing the protocol being used).
I am surprised it is not the case.

nostromo123, 2023-02-10:
Important, that now changing values show up ....

nostromo123, 2023-02-10:
Nuked the RP2040, build and uploaded 0.7.3 with platformio. Configured my settings for ELRS, RX, MS5611 and MPU6050. Did a MPUCAL, in the radio (edgeTX) set the ROLL and PITCH sensor from rad to °, RATIO is off and now things here make sense, finally! 

You wrote on RCG "- I removed RPM from ELRS protocol because I noticed that it was not a valid solution to "misuse" the YAW field for RPM. The yaw field is supposed to be in 1/10000 of radian and so I do not think that it is possible get a meaningful RPM value on handset display. (perhaps still with a lua script)"

but a YAW value under ELRS is still transmitted?? Could you explain please. Later I will close this issue.

Regards - KH

mstrens, 2023-02-10:
In ELRS protocol you can't send each field individually.
Several fields are grouped in some frames.
So, Pitch, Roll and Yaw are part of the same frame.
If I have to send Pitch and Roll I must also put Yaw in the same frame.
Currently, I fill Yaw with 0.
I could fill it with RPM but then I do not know if the value can be converted into a meaningful value as the handset makes afterwards some conversion on the received value.

nostromo123, 2023-02-11:
Thank you for clarification. I'll close this issue now.

br KH

closed 2023-02-11 by nostromo123