mstrens/oXs_on_RP2040 #30: oXs_RP2040 Ver.0.7.1-hott

Satcomix, 2023-02-10:
Hello Mstrens,
I want to test Ver.0.7.1-hott, but i do not get access to terminal port, and the handheld show nothing, and the GPS is running at 9600kbs.
br,
Torsten

mstrens, 2023-02-10:
I just tried it.
Indeed there is a major bug.
I will check it

nostromo123, 2023-02-10:
I just tried to load uf2 0.7.1 Does not work, then I tried building in platformio

6.10 OK, but 6.12 ERROR, 7.0 ERROR and finally 7.1 ERROR

See attached file

[Neue Datei.txt](https://github.com/mstrens/oXs_on_RP2040/files/10710094/Neue.Datei.txt)

mstrens, 2023-02-10:
It means that some files are missing.
I expect that you did not make a full download from github.
 Make a full download of hott branch (latest version 0.7.3) and it should work.
I see that the files exist on github.

Satcomix, 2023-02-10:
Hello Mstrens,
I have tested the new Ver.0.7.3 with FrSky X20S Ethos S.PORT Protocol and Graupner MC28 GR24 HOTT Protocol, and every value is displayed in the handheld, also all DIY ID from the telemetry.txt from the DOC folder.
Thank you very much for the work, and after a few tests with other Protocols, we have a new MAIN.
Best regards,
Torsten

nostromo123, 2023-02-10:
I see 0.7.2.

uf2 loads

[image]

but building here >> ERROR!

[Neue Datei.txt](https://github.com/mstrens/oXs_on_RP2040/files/10710237/Neue.Datei.txt)

Satcomix, 2023-02-10:
That is what i see, 0.7.2 not 0.7.3
processing cmd

Cmd to execute:   

Version = 0.7.2 
    Function                Pin   Change entering XXX=yyy (yyy=255 to disable)
Primary channels input    =    5  (PRI     = 5, 9, 21, 25)
Secondary channels input  =    1  (SEC     = 1, 13, 17, 29)
Telemetry . . . . . . . . =    0  (TLM     = 0, 1, 2, ..., 29)
GPS Rx  . . . . . . . . . =   12  (GPS_RX  = 0, 1, 2, ..., 29)
GPS Tx  . . . . . . . . . =   13  (GPS_TX  = 0, 1, 2, ..., 29)
Sbus OUT  . . . . . . . . =    2  (SBUS_OUT= 0, 1, 2, ..., 29)
RPM   . . . . . . . . . . =  255  (RPM     = 0, 1, 2, ..., 29)
SDA (I2C sensors) . . . . =   14  (SDA     = 2, 6, 10, 14, 18, 22, 26)
SCL (I2C sensors) . . . . =   15 (SCL     = 3, 7, 11, 15, 19, 23, 27)
PWM Channels 1, 2, 3 ,4   =    7  255    6  255 (C1 / C16= 0, 1, 2, ..., 15)
PWM Channels 5, 6, 7 ,8   =  255  255  255  255
PWM Channels 9,10,11,12   =  255  255  255  255
PWM Channels 13,14,15,16  =  255  255  255  255
Voltage 1, 2, 3, 4        =   26   27   28   29 (V1 / V4 = 26, 27, 28, 29)

Protocol is Sport (Frsky)
CRSF baudrate = 420000
Voltage parameters:
    Scales : 1.000000 , 1.000000 , 1.000000 , 1.000000 
    Offsets: 0.000000 , 0.000000 , 0.000000 , 0.000000 
RPM multiplier = 1.000000
Baro sensor is not detected
Acc/Gyro is not detected
First analog to digital sensor is not detected
Second analog to digital sensor is not detected
Foreseen GPS type is Ublox  :GPS is not (yet) detected
Failsafe uses predefined values
     Chan 1...4  = 992 , 992 , 992 , 992
     Chan 5...8  = 992 , 992 , 992 , 992
     Chan 9...12 = 992 , 992 , 992 , 992
     Chan 13...16= 992 , 992 , 992 , 992

Config parameters are OK
Press ? + Enter to get help about the commands

 >>

nostromo123, 2023-02-10:
@Satcomix.

Can you build from source?

mstrens, 2023-02-10:
Sorry I made a mistake in the version nummer in my last post. I should wrote 0.7.2 instead of 0.7.3.
Anyway, I just reloaded the same firmware but I changed the version to 0.7.3.
So both are just identical (except the version number).

Satcomix, 2023-02-10:
I build my own ver.0.7.3

processing cmd

Cmd to execute:   

Version = 0.7.3 
    Function                Pin   Change entering XXX=yyy (yyy=255 to disable)
Primary channels input    =    5  (PRI     = 5, 9, 21, 25)
Secondary channels input  =    1  (SEC     = 1, 13, 17, 29)
Telemetry . . . . . . . . =    0  (TLM     = 0, 1, 2, ..., 29)
GPS Rx  . . . . . . . . . =   12  (GPS_RX  = 0, 1, 2, ..., 29)
GPS Tx  . . . . . . . . . =   13  (GPS_TX  = 0, 1, 2, ..., 29)
Sbus OUT  . . . . . . . . =    2  (SBUS_OUT= 0, 1, 2, ..., 29)
RPM   . . . . . . . . . . =  255  (RPM     = 0, 1, 2, ..., 29)
SDA (I2C sensors) . . . . =   14  (SDA     = 2, 6, 10, 14, 18, 22, 26)
SCL (I2C sensors) . . . . =   15 (SCL     = 3, 7, 11, 15, 19, 23, 27)
PWM Channels 1, 2, 3 ,4   =    7  255    6  255 (C1 / C16= 0, 1, 2, ..., 15)
PWM Channels 5, 6, 7 ,8   =  255  255  255  255
PWM Channels 9,10,11,12   =  255  255  255  255
PWM Channels 13,14,15,16  =  255  255  255  255
Voltage 1, 2, 3, 4        =   26   27   28   29 (V1 / V4 = 26, 27, 28, 29)

Protocol is Sport (Frsky)
CRSF baudrate = 420000
Voltage parameters:
    Scales : 1.000000 , 1.000000 , 1.000000 , 1.000000 
    Offsets: 0.000000 , 0.000000 , 0.000000 , 0.000000 
RPM multiplier = 1.000000
Baro sensor is detected using BMP280
    Sensitivity min = 100 (at 100)   , max = 300 (at 1000)
    Hysteresis = 5 
Acc/Gyro is detected using MP6050
     Acceleration offsets X, Y, Z = 3134 , 340 , 1000
     Gyro offsets         X, Y, Z = -7 , 41 , 51
First analog to digital sensor is not detected
Second analog to digital sensor is not detected
Foreseen GPS type is Ublox  :GPS is detected
Failsafe uses predefined values
     Chan 1...4  = 992 , 992 , 992 , 992
     Chan 5...8  = 992 , 992 , 992 , 992
     Chan 9...12 = 992 , 992 , 992 , 992
     Chan 13...16= 992 , 992 , 992 , 992

Config parameters are OK
Press ? + Enter to get help about the commands

 >>

mstrens, 2023-02-10:
If you make a download of all files from github, I do not see why you could not build from source.

nostromo123, 2023-02-10:
Loading the 0.7.2 ZIP file from github, I assume I have all the files ....

Satcomix, 2023-02-10:
@Kalle
Thats the way i do it every time when a new version is out, i have no Problems with the .zip and can loading everything in PlatformIO, o i can change 0.7.2 to 0.7.3 by myself.
br,
Torsten

nostromo123, 2023-02-10:
0.6.10 in left terminal, 0.7.3 in right terminal

[image]

Have a nice evening. Bye bye ;-)

Satcomix, 2023-02-10:
@Kalle
The warnings with the EMF Button is since a long time, since the BOOT button becomes a user-button for failsafe.
I can make a build by my own , and its working like a charm.
Nice evening at all,
Torsten

mstrens, 2023-02-10:
The error is about KalmanFilterD4.
Don't you have another error message (in read) before the one at the end?
Did you remove all files in your directory before copying the zip file from github? 
If not please install all in a clean directory.

nostromo123, 2023-02-10:
The error occurs in 0.6.12.

I don't know, what has been changed in between 0.6.10 and 0.6.12?

I see this message here when trying to build 0.6.12

```
Compiling .pio/build/raspberry-mstrens/src/KalmanFilter4D.o
src/KalmanFilter4D.cpp:6:10: fatal error: Kalmanfilter4d.h: No such file or directory

************************************************************************
* Looking for Kalmanfilter4d.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:Kalmanfilter4d.h"
* Web  > https://registry.platformio.org/search?q=header:Kalmanfilter4d.h
*
************************************************************************

 #include "Kalmanfilter4d.h"
          ^~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio/build/raspberry-mstrens/src/KalmanFilter4D.o] Error 1
Archiving .pio/build/raspberry-mstrens/lib422/libwiring.a
Indexing .pio/build/raspberry-mstrens/lib422/libwiring.a
```
[image]

mstrens, 2023-02-10:
There is an error in the file name KalmanFilter4D.h about capital letter.
Once it is with filter and once with Filter. So the compiler does not find what is requested.
I think I made the mistake in some version but this error does not exist anymore in 0.7.3

nostromo123, 2023-02-10:
> Did you remove all files in your directory before copying the zip file from github?
> If not please install all in a clean directory.

I download the *.zip file, extract it to the desktop, rename it from oXs_on_RP2040-hott to oXs_on_RP2040-hott_0.7.3 and then move it my platformio folder ...
[image]

For building I move into the selected folder and open terminal there and try to build. Until 6.10 ok, since 6.12 ERROR!

Satcomix, 2023-02-10:
I have a look in Platformio Build Ver 0.7.3 no Warnings or Errors while Platformio is compiling/build and upload.

Satcomix, 2023-02-10:
> I download the *.zip file, extract it to the desktop, rename it from oXs_on_RP2040-hott to oXs_on_RP2040-hott_0.7.3 and then move it my platformio folder ... [image]
>
> For building I move into the [...]

Did you open a new oXs_0.7.3 Project under Projects and Folders in Platformio, because you rename the folder by yourself.

nostromo123, 2023-02-10:
I copied the building of 0.7.3 here in a docx file and marked the error messages in red.

Nice Torsten, that you don't have any errors. Lucky you ;-)

[Neue Datei.docx](https://github.com/mstrens/oXs_on_RP2040/files/10710757/Neue.Datei.docx)

mstrens, 2023-02-10:
I think I found the reason of your compilation error.
In file KalmanFilter4D.cpp, you have a line with
#include "kalmanfilter4d.h"

It should be (3 X caps).
#include "KalmanFilter4D.h"

It probably the difference between windows and linux

Satcomix, 2023-02-10:
I have only two folders in my Platformio Projects.
First Folder is .main
Second Folder is .hott
I replace the prebuild .hott every time when a new prebuild is born, but i do not rename it.
Just drag and trop with replace.
br,
Torsten

mstrens, 2023-02-10:
I will fix this bug in next version but if you change it manually in your version I expect it should be ok.

nostromo123, 2023-02-10:
That was it Michel!  KalmanFilter4D you introduced in 6.12. Building now is ok.

nostromo123, 2023-02-10:
> I have only two folders in my Platformio Projects. First Folder is .main Second Folder is .hott I replace the prebuild .hott every time when a new prebuild is born, but i do not rename it. Just drag and drop with replace.

Yes, you are the best!

Satcomix, 2023-02-11:
After all issues have been clarified and fixed in the new version 0.7.3-hott, I can close the post.
Thank you Mstrens for the work.
Best regards,
Torsten

closed 2023-02-11 by Satcomix