mstrens/oXs_on_RP2040 #31: Ver.0.7.3 TEMP=1-3 missing values

Satcomix, 2023-02-11:
Hello Mstrens,
I think that on the terminal interface there is **an addition** for the 
GPIO pins 26-29 
TEMP=0
TEMP=1
TEMP=2
**is missing.**
The definition can be found in param.ccp.
// change number of temperature sensors
    if ( strcmp("TEMP", pkey) == 0 ) {
        ui = strtoul(pvalue, &ptr, 10);
        if ( *ptr != 0x0){
            printf("Error : value must be an unsigned integer\n");
        } else if ( !(ui==0 or ui==1 or ui==2 or ui ==255)) {
            printf("Error : value must be 0, 1, 2 or 255\n");
        } else {
            config.temperature;
            printf("Number of temperature sensors = %u\n" , config.temperature );
            updateConfig = true;
        }
br,
Torsten

mstrens, 2023-02-11:
I think I had a wrong setup in a table for T2 (cut and past error)
Still I expected that when a pin is defined for V3 AND if TEMP= 1 (or), then the field T1 should be displayed in sport instead of 5113.
If this is not the case, then it is an unknown error.
Anyway, it is perhaps best to wait a little because I am currently rewriting part of Sport telemetry in order to :
- reduce error set up
- allow to transmit more fields per sec using 3 device ID instead of 1.
It will perhaps be programmed in 1 or 2 hours.

Satcomix, 2023-02-11:
I will wait until a new S.PORT version with new features appears and then test again, test..... :-)
Thank you for the work and time.
br,
Torsten

mstrens, 2023-02-11:
I just put 0.7.4 on github.
I had no time to test it.
I hope it works.
Note : in this version oXs simulates up to 3 Sport devices. It is supposed to response to 3 polling codes from RX.
This should allow to transmit more data and better manage priorities of some fields (e.g. Vspeed is sent more often than GPS date).

mstrens, 2023-02-11:
I saw a bug: the value entered with TEMP=x was not saved.
So oXs never sent temperature instead of volt3/4
Fixed in 0.7.5 version (loaded)
Sorry for this.

Satcomix, 2023-02-11:
Hello Mstrens,
In the latest version 0.7.5-hott, all telemetry values ​​are available or listed twice in the handheld display.
Sensor lost appears constantly.
br,
Torsten

Satcomix, 2023-02-11:
Hello Mstrens,
In the latest version 0.7.5-hott I have now deactivated V3 and V4 =255. 
VSpeed, FASS,Altitude,GPS,Fuel, every value is displayed four times
Sensor lost appears constantly.
br,
Torsten

mstrens, 2023-02-11:
did you deleted all sensors before asking to discover them again?
It is strange, because:
- I did not change Hott protocol
- in Hott I just sent 2 types frames (GAM and GPS) that contains all data without an Id before each field.
I have to leave home for some hours. I can only look afterwards.

Satcomix, 2023-02-11:
Hello Mstrens,
It is the -hott Version, not the HOTT Protocol, 
its FrSky S.PORT Protocol.
I deleted all sensors before, than make a new search.
Than every sensor is displayed twice or more.
br,Torsten

mstrens, 2023-02-11:
Ok, I will have a look.

Satcomix, 2023-02-11:
Ver.07.5
TLM=0
V1=26
V2=27
all other options are not in use!!!
Telemetry sensors deleted first, than new search.
Telemetry display X20S S.PORT:
VFAS  00  210
FASS  06  200
FUEL  00  600
FASS  01  200
FUEL  01  600
VFAS  01  210
FASS  00  200

br,
Torsten

mstrens, 2023-02-11:
Thanks for this info.
It helped me to find at least one bug.
Still not 100% sure it is totally fixed (I do not understand origin of FASS 06 200.
Best to perform first the previous test again with version 0.7.6 (now on github

Satcomix, 2023-02-11:
Ver.0.7.6
X20S S.PORT
Clean Telemetry Sensors, than new search.
TLM=10
V1=26
V2=27
all other optios deativated =255

FUEL  03  0600
VFAS  01  0210 
FASS  03  0200

than activate
V3=28
V4=29
no extra Sensor appear,
but sensor conflict, every 5 sec

mstrens, 2023-02-11:
Thanks.
So, with only V1 and V2 it is Ok

About V3/V4, what was the value of parameter TEMP
Depending the value of TMP, I expect different sensors ID

Satcomix, 2023-02-11:
I dont see the TEMP values after a new search when i activate V3 and/or V4

mstrens, 2023-02-11:
I display the value only when it is 1 or 2.
So, it means that it had to display V3 and V4.
I will probably have to connect my board to a receiver to debug this because I do not see the error in the code.

Satcomix, 2023-02-11:
I dont see values by activating V3 alone, V4 alone, or both V3 and V4
Only V1 and V2 values are shown.

Satcomix, 2023-02-11:
I see Temp1 and Temp2 by activating V3 and V4, forget to activate TEMP=2, sorry
Temp1  03  0400
Temp2  03  0410

Satcomix, 2023-02-11:
Version = 0.7.6 
    Function                Pin   Change entering XXX=yyy (yyy=255 to disable)
Primary channels input    =  255  (PRI     = 5, 9, 21, 25)
Secondary channels input  =  255  (SEC     = 1, 13, 17, 29)
Telemetry . . . . . . . . =    0  (TLM     = 0, 1, 2, ..., 29)
GPS Rx  . . . . . . . . . =  255  (GPS_RX  = 0, 1, 2, ..., 29)
GPS Tx  . . . . . . . . . =  255  (GPS_TX  = 0, 1, 2, ..., 29)
Sbus OUT  . . . . . . . . =  255  (SBUS_OUT= 0, 1, 2, ..., 29)
RPM   . . . . . . . . . . =  255  (RPM     = 0, 1, 2, ..., 29)
SDA (I2C sensors) . . . . =  255  (SDA     = 2, 6, 10, 14, 18, 22, 26)
SCL (I2C sensors) . . . . =  255 (SCL     = 3, 7, 11, 15, 19, 23, 27)
PWM Channels 1, 2, 3 ,4   =  255  255  255  255 (C1 / C16= 0, 1, 2, ..., 15)
PWM Channels 5, 6, 7 ,8   =  255  255  255  255
PWM Channels 9,10,11,12   =  255  255  255  255
PWM Channels 13,14,15,16  =  255  255  255  255
Voltage 1, 2, 3, 4        =   26   27   28   29 (V1 / V4 = 26, 27, 28, 29)

Protocol is Sport (Frsky)
CRSF baudrate = 420000
Voltage parameters:
    Scales : 1.000000 , 1.000000 , 1.000000 , 1.000000 
    Offsets: 0.000000 , 0.000000 , 0.000000 , 0.000000 
    **Temperature sensors are connected on V3 and V4**
RPM multiplier = 1.000000
Baro sensor is not detected
Acc/Gyro is not detected
First analog to digital sensor is not detected
Second analog to digital sensor is not detected
Foreseen GPS type is Ublox  :GPS is not (yet) detected
Failsafe type is HOLD

Config parameters are OK
Press ? + Enter to get help about the commands

 >>

Satcomix, 2023-02-11:
Everything is activated, no Problems
processing cmd

Cmd to execute:   

Version = 0.7.6 
    Function                Pin   Change entering XXX=yyy (yyy=255 to disable)
Primary channels input    =    5  (PRI     = 5, 9, 21, 25)
Secondary channels input  =    1  (SEC     = 1, 13, 17, 29)
Telemetry . . . . . . . . =    0  (TLM     = 0, 1, 2, ..., 29)
GPS Rx  . . . . . . . . . =   12  (GPS_RX  = 0, 1, 2, ..., 29)
GPS Tx  . . . . . . . . . =   13  (GPS_TX  = 0, 1, 2, ..., 29)
Sbus OUT  . . . . . . . . =    2  (SBUS_OUT= 0, 1, 2, ..., 29)
RPM   . . . . . . . . . . =  255  (RPM     = 0, 1, 2, ..., 29)
SDA (I2C sensors) . . . . =   14  (SDA     = 2, 6, 10, 14, 18, 22, 26)
SCL (I2C sensors) . . . . =   15 (SCL     = 3, 7, 11, 15, 19, 23, 27)
PWM Channels 1, 2, 3 ,4   =  255  255    6  255 (C1 / C16= 0, 1, 2, ..., 15)
PWM Channels 5, 6, 7 ,8   =  255  255  255  255
PWM Channels 9,10,11,12   =  255  255  255  255
PWM Channels 13,14,15,16  =  255  255  255  255
Voltage 1, 2, 3, 4        =   26   27   28   29 (V1 / V4 = 26, 27, 28, 29)

Protocol is Sport (Frsky)
CRSF baudrate = 420000
Voltage parameters:
    Scales : 1.000000 , 1.000000 , 1.000000 , 1.000000 
    Offsets: 0.000000 , 0.000000 , 0.000000 , 0.000000 
    Temperature sensors are connected on V3 and V4
RPM multiplier = 1.000000
Baro sensor is detected using MS5611
    Sensitivity min = 100 (at 100)   , max = 300 (at 1000)
    Hysteresis = 5 
Acc/Gyro is detected using MP6050
     Acceleration offsets X, Y, Z = 0 , 0 , 0
     Gyro offsets         X, Y, Z = 0 , 0 , 0
First analog to digital sensor is not detected
Second analog to digital sensor is not detected
Foreseen GPS type is Ublox  :GPS is detected
Failsafe type is HOLD

Config parameters are OK
Press ? + Enter to get help about the commands

 >>

Satcomix, 2023-02-11:
Ver.0.7.6
New Sensor Search
TEMP=0  no V3 and V4 Sensor in Display
TEMP=1  V3=Temp1 and V4 show nothing
TEMP=2  V3=Temp1 and V4=Temp2

mstrens, 2023-02-11:
I tested on my side (only with V1, V2, V3, V4, with TEMP=0, 1, 2) and it seems OK.
Fo safety I put 0.7.7 on github. (Still have to delete a list that is displayed one at power for debuging)  
For clarity, I just display some more text when there TEMP=0 or 255 (no temp sensor).

I did not yet tested with vario and GPS.
Best to delete all sensors on handset each time you change to the list of sensors to be sent.

Satcomix, 2023-02-11:
FrSky X20S S.PORT Protocol
Everything works fine!
After searching DIY Sensors you will find,
ID 5100,
ID 5113,
ID 5114,
ID 5120,
ID 5121, 

processing cmd

Cmd to execute:   

Version = 0.7.7 
    Function                Pin   Change entering XXX=yyy (yyy=255 to disable)
Primary channels input    =    5  (PRI     = 5, 9, 21, 25)
Secondary channels input  =    1  (SEC     = 1, 13, 17, 29)
Telemetry . . . . . . . . =    0  (TLM     = 0, 1, 2, ..., 29)
GPS Rx  . . . . . . . . . =   12  (GPS_RX  = 0, 1, 2, ..., 29)
GPS Tx  . . . . . . . . . =   13  (GPS_TX  = 0, 1, 2, ..., 29)
Sbus OUT  . . . . . . . . =    2  (SBUS_OUT= 0, 1, 2, ..., 29)
RPM   . . . . . . . . . . =  255  (RPM     = 0, 1, 2, ..., 29)
SDA (I2C sensors) . . . . =   14  (SDA     = 2, 6, 10, 14, 18, 22, 26)
SCL (I2C sensors) . . . . =   15  (SCL     = 3, 7, 11, 15, 19, 23, 27)
PWM Channels 1, 2, 3 ,4   =  255  255    6  255 (C1 / C16= 0, 1, 2, ..., 15)
PWM Channels 5, 6, 7 ,8   =  255  255  255  255
PWM Channels 9,10,11,12   =  255  255  255  255
PWM Channels 13,14,15,16  =  255  255  255  255
Voltage 1, 2, 3, 4        =   26   27   28   29 (V1 / V4 = 26, 27, 28, 29)

Protocol is Sport (Frsky)
CRSF baudrate = 420000
Voltage parameters:
    Scales : 1.000000 , 1.000000 , 1.000000 , 1.000000 
    Offsets: 0.000000 , 0.000000 , 0.000000 , 0.000000 
    Temperature sensors are connected on V3 and V4
RPM multiplier = 1.000000
Baro sensor is detected using MS5611
    Sensitivity min = 100 (at 100)   , max = 300 (at 1000)
    Hysteresis = 5 
Acc/Gyro is detected using MP6050
     Acceleration offsets X, Y, Z = 3274 , 328 , 1006
     Gyro offsets         X, Y, Z = -2 , 41 , 49
First analog to digital sensor is not detected
Second analog to digital sensor is not detected
Foreseen GPS type is Ublox  :GPS is detected
Failsafe uses predefined values
     Chan 1...4  = 992 , 992 , 992 , 992
     Chan 5...8  = 992 , 992 , 992 , 992
     Chan 9...12 = 992 , 992 , 992 , 992
     Chan 13...16= 992 , 992 , 992 , 992

Config parameters are OK
Press ? + Enter to get help about the commands

 >>

mstrens, 2023-02-11:
On my side (with openTX), I had V1 up to V4 defined and TMP=2.
I had on screen VFAS, current, fuel, T1 and T2 (that is ok)
When I enter TEMP=0,  a "*" before T1 and T2 does not not blink (normal because those fields are not sent anymore)
If I delete all sensors and ask for discovery of sensor, I got VFAS, current, V3 and V4 in the list (but with their DIY ID for the 2 last).
So on my side all seems normal.
To get V3 (or T1), a pin must be defined for V3.
Same principle for V4 (or T2).
For T2, V3 and V4 pins must both be defined.
In fact, with TEMP=x, X = the number of temperature sensors connected.

Satcomix, 2023-02-11:
Next Test with **Graupner MC28 GR24 HOTT**,
Receiver,
General Modul,
GPS 
are found in Telemetry Sensors,
all values are seen on display, everything works fine.

Cmd to execute:   PROTOCOL=H
config has been saved
Device will reboot

Version = 0.7.7 
    Function                Pin   Change entering XXX=yyy (yyy=255 to disable)
Primary channels input    =    5  (PRI     = 5, 9, 21, 25)
Secondary channels input  =    1  (SEC     = 1, 13, 17, 29)
Telemetry . . . . . . . . =    0  (TLM     = 0, 1, 2, ..., 29)
GPS Rx  . . . . . . . . . =   12  (GPS_RX  = 0, 1, 2, ..., 29)
GPS Tx  . . . . . . . . . =   13  (GPS_TX  = 0, 1, 2, ..., 29)
Sbus OUT  . . . . . . . . =    2  (SBUS_OUT= 0, 1, 2, ..., 29)
RPM   . . . . . . . . . . =  255  (RPM     = 0, 1, 2, ..., 29)
SDA (I2C sensors) . . . . =   14  (SDA     = 2, 6, 10, 14, 18, 22, 26)
SCL (I2C sensors) . . . . =   15  (SCL     = 3, 7, 11, 15, 19, 23, 27)
PWM Channels 1, 2, 3 ,4   =  255  255    6  255 (C1 / C16= 0, 1, 2, ..., 15)
PWM Channels 5, 6, 7 ,8   =  255  255  255  255
PWM Channels 9,10,11,12   =  255  255  255  255
PWM Channels 13,14,15,16  =  255  255  255  255
Voltage 1, 2, 3, 4        =   26   27   28   29 (V1 / V4 = 26, 27, 28, 29)

Protocol is Hott
CRSF baudrate = 420000
Voltage parameters:
    Scales : 1.000000 , 1.000000 , 1.000000 , 1.000000 
    Offsets: 0.000000 , 0.000000 , 0.000000 , 0.000000 
    No temperature sensors are connected on V3 and V4
RPM multiplier = 1.000000
Baro sensor is detected using MS5611
    Sensitivity min = 100 (at 100)   , max = 300 (at 1000)
    Hysteresis = 5 
Acc/Gyro is detected using MP6050
     Acceleration offsets X, Y, Z = 3274 , 328 , 1006
     Gyro offsets         X, Y, Z = -2 , 41 , 49
First analog to digital sensor is not detected
Second analog to digital sensor is not detected
Foreseen GPS type is Ublox  :GPS is detected
Failsafe uses predefined values
     Chan 1...4  = 992 , 992 , 992 , 992
     Chan 5...8  = 992 , 992 , 992 , 992
     Chan 9...12 = 992 , 992 , 992 , 992
     Chan 13...16= 992 , 992 , 992 , 992

Config parameters are OK
Press ? + Enter to get help about the commands

 >>

Satcomix, 2023-02-11:
Now everything works fine in S.PORT and HOTT protocol.
@Mstrens, thank you very much for the work.
I will close this post.

closed 2023-02-11 by Satcomix