mstrens/oXs_on_RP2040 #32: OXS 2040 with Jeti

Klippeneck, 2023-02-11:
I have also built the OXS sensor with SW 0.61 and higher.
Configured with MS 5611 and GPS, telemetry is Jeti.
The sensor logs on with the values for the GPS, Vario I see no values although in the configuration of the MS5611 is detected and displayed. With SW 0.36 on the other hand, the vario is there, but is extremely slow in response.
Has anyone else already tested with Jeti?
 
Quote
Report

mstrens, 2023-02-11:
I think there were some tests done with Jeti but it was with quite old version.
Can you try latest version (currently 0.7.7 in branch "hott").

When you press "enter" on a serial terminal, you get the config being used.
Are both the GPS and the baro declared as discovered in the list being displayed on the PC?

What is exactly discovered on the Jeti display?

pierrotm777, 2023-02-11:
Hello Klippeneck,
I have just tested my module in Jeti mode.
It  wasn't my transmitter so my tests was very quick and short.
By I have seen all gps values.
And sorry, I don't remember if baro values are discovered but I have seen Vspeed and Altitude.
Pierre

mstrens, 2023-02-11:
I presume that you know that there are 2 jeti protocols for telemetry.
oXs is foreseen to simulate the "old" protocol so not the EX bus.
I have no experience with jeti but I expect that some receivers support both protocol (probably based on some config to be defined by the user)

pierrotm777, 2023-02-11:
The transmitter used for this test was a Jeti dc16 and my receiver is a REX7.

pierrotm777, 2023-02-11:
Sorry, just som adds.
I used 0.7.5 version.
My devices are:
- GPS
- GY-86 (baro, acc etc...)
- ADS1115 (anaglog/digital converter)

About data refreshed , ii suppose that was fast enough (not enough tests done)

Klippeneck, 2023-02-11:
Performed tests with SW 0.77.
Transmitter is Jeti DS 16II, receiver REX 10, both with latest SW.
Configuration Telemetry input receiver is Jeti EX, not EX bus.
In sensor login I see oXs, then 1 Speed, 2 Course, 3 Gps Alt,
4 Gps Long and Gps Alt. No registration for altitude and vario.
Attached is the config. 
processing cmd

Cmd to execute:   

Version = 0.7.7 
    Function                Pin   Change entering XXX=yyy (yyy=255 to disable)
Primary channels input    =  255  (PRI     = 5, 9, 21, 25)
Secondary channels input  =  255  (SEC     = 1, 13, 17, 29)
Telemetry . . . . . . . . =    2  (TLM     = 0, 1, 2, ..., 29)
GPS Rx  . . . . . . . . . =   28  (GPS_RX  = 0, 1, 2, ..., 29)
GPS Tx  . . . . . . . . . =   29  (GPS_TX  = 0, 1, 2, ..., 29)
Sbus OUT  . . . . . . . . =  255  (SBUS_OUT= 0, 1, 2, ..., 29)
RPM   . . . . . . . . . . =  255  (RPM     = 0, 1, 2, ..., 29)
SDA (I2C sensors) . . . . =    6  (SDA     = 2, 6, 10, 14, 18, 22, 26)
SCL (I2C sensors) . . . . =    7  (SCL     = 3, 7, 11, 15, 19, 23, 27)
PWM Channels 1, 2, 3 ,4   =  255  255  255  255 (C1 / C16= 0, 1, 2, ..., 15)
PWM Channels 5, 6, 7 ,8   =  255  255  255  255
PWM Channels 9,10,11,12   =  255  255  255  255
PWM Channels 13,14,15,16  =  255  255  255  255
Voltage 1, 2, 3, 4        =  255  255  255  255 (V1 / V4 = 26, 27, 28, 29)

Protocol is Jeti (Ex)
CRSF baudrate = 420000
Voltage parameters:
    Scales : 1.000000 , 1.000000 , 1.000000 , 1.000000 
    Offsets: 0.000000 , 0.000000 , 0.000000 , 0.000000 
    No temperature sensors are connected on V3 and V4
RPM multiplier = 1.000000
Baro sensor is detected using MS5611
    Sensitivity min = 100 (at 100)   , max = 300 (at 1000)
    Hysteresis = 5 
Acc/Gyro is not detected
First analog to digital sensor is not detected
Second analog to digital sensor is not detected
Foreseen GPS type is Ublox  :GPS is not (yet) detected
Failsafe type is HOLD

Config parameters are OK
Press ? + Enter to get help about the commands

 >>

Klippeneck, 2023-02-11:
Have performed tests with SW 0.36.
Transmitter is Jeti DS 16II, receiver REX 10, both with latest SW.
Configuration Telemetry input receiver is Jeti EX, not EX bus.
Hardware is the same as in test with SW 0.77.
In sensor login I see oXs, then 1 altitude, 2 vario, 3 speed, 4 heading, 5 gps alt,
6 Gps Long, 7 Gps Lat. 
I think this rules out a HW problem.

mstrens, 2023-02-12:
I think I understand why Alt and Vspeed does not appear in latest versions.
I am looking for a solution.
I will come back.

Klippeneck, 2023-02-12:
O.K., thank you very much

mstrens, 2023-02-12:
I put on github (in hott branch) a version 0.7.8 that should fix the issue about missing Alt and Vspeed.
Please test it and let me know.

Klippeneck, 2023-02-12:
I have tested with SW 1.0.0, altitude and vario are now logging in.
Telemetry values for altitude are ok, also the update speed.
Vario values are noh not plausible, direction is correct but update is 
extremely slow, see attached telemetry log. 
[image]

mstrens, 2023-02-12:
So I presume it is the same issue as with 03.6.
This would be logic because my last change concerns only the wrong detection of the baro sensor but not sending the values.
 
Is vspeed also very slow when you do not have GPS connected?

Klippeneck, 2023-02-12:
Just tested without GPS enabled, same result.
It looks like the vario is always at Max. Values and witer this max. values outputs until a reset or even higher values come.

mstrens, 2023-02-12:
Indeed.
In fact the bug is not in the jeti protocol but in the way vspeed is calculated when there is no MP6050.
I expect that other users tested with MP6050 and so did not saw the bug.
I will try to fix it this PM

mstrens, 2023-02-12:
I hope it is fixed.
It is on github in main (not yet in test branch!!)

mstrens, 2023-02-12:
version is 1.0.1

Satcomix, 2023-02-12:
Version 1.0.1 works in S.PORT and HOTT with only BMP280, and without MP6050

mstrens, 2023-02-12:
@Satcomix
So it is ok with BMP280 without MP6050.
I presume it is also ok with baro + MP6050

Satcomix, 2023-02-12:
@Mstrens,
Yes it works well in S.PORT and HOTT **with** or **without** MP6050
br,
Torsten

mstrens, 2023-02-12:
Ok thanks

Klippeneck, 2023-02-13:
Yes, problem with Jeti in 1.03 solved. GPS and vario with MS 5611 work as desired.
Thanks a lot.

nostromo123, 2023-02-13:
@Klippeneck 

> Yes, problem with Jeti in 1.03 solved. GPS and vario with MS 5611 work as desired. Thanks a lot.

Then please close this issue ;-)

closed 2023-02-13 by Klippeneck