mstrens/oXs_on_RP2040 #36: V1.0.1 MPX V1 <> V2 mixed up?

nostromo123, 2023-02-12:
Testing with that CONFIG.

> Version = 1.0.1 
>     Function                Pin   Change entering XXX=yyy (yyy=255 to disable)
> Primary channels input    =  255  (PRI     = 5, 9, 21, 25)
> Secondary channels input  =  255  (SEC     = 1, 13, 17, 29)
> Telemetry . . . . . . . . =   10  (TLM     = 0, 1, 2, ..., 29)
> GPS Rx  . . . . . . . . . =  255  (GPS_RX  = 0, 1, 2, ..., 29)
> GPS Tx  . . . . . . . . . =  255  (GPS_TX  = 0, 1, 2, ..., 29)
> Sbus OUT  . . . . . . . . =  255  (SBUS_OUT= 0, 1, 2, ..., 29)
> RPM   . . . . . . . . . . =  255  (RPM     = 0, 1, 2, ..., 29)
> SDA (I2C sensors) . . . . =   14  (SDA     = 2, 6, 10, 14, 18, 22, 26)
> SCL (I2C sensors) . . . . =   15  (SCL     = 3, 7, 11, 15, 19, 23, 27)
> PWM Channels 1, 2, 3 ,4   =  255  255  255  255 (C1 / C16= 0, 1, 2, ..., 15)
> PWM Channels 5, 6, 7 ,8   =  255  255  255  255
> PWM Channels 9,10,11,12   =  255  255  255  255
> PWM Channels 13,14,15,16  =  255  255  255  255
> Voltage 1, 2, 3, 4        =   26   27  255  255 (V1 / V4 = 26, 27, 28, 29)
> 
> Protocol is Mpx
> CRSF baudrate = 420000
> Voltage parameters:
>     Scales : 1.000000 , 1.000000 , 1.000000 , 1.000000 
>     Offsets: 0.000000 , 0.000000 , 0.000000 , 0.000000 
>     No temperature sensors are connected on V3 and V4
> RPM multiplier = 1.000000
> Baro sensor is detected using MS5611
>     Sensitivity min = 100 (at 100)   , max = 300 (at 1000)
>     Hysteresis = 5 
> Acc/Gyro is not detected
> First analog to digital sensor is not detected
> Second analog to digital sensor is not detected
> Foreseen GPS type is Ublox  :GPS is not (yet) detected
> Failsafe type is HOLD
> 
> Config parameters are OK

First of all, MS5611 and BMP280 working!

By accident I had V1 AND V2 active and I see this on the radio

[image]

When I de-active V2, Curr and Capa disappear. When I reactivate V2 and put a low voltage on pin 27, Curr changes, but Capa stays fixed. So I think, you mixed up V1 with V2 in source.

br - KH

mstrens, 2023-02-12:
Indeed there was a bug.
I calculated the capacity but I put V2 in the transmitted field.
Should be solved in version 1.0.2

mstrens, 2023-02-12:
I put version 1.0.2 on github (in test branch)

nostromo123, 2023-02-12:
With 1.0.2

> Version = 1.0.2 
>     Function                Pin   Change entering XXX=yyy (yyy=255 to disable)
> Primary channels input    =  255  (PRI     = 5, 9, 21, 25)
> Secondary channels input  =  255  (SEC     = 1, 13, 17, 29)
> Telemetry . . . . . . . . =   10  (TLM     = 0, 1, 2, ..., 29)
> GPS Rx  . . . . . . . . . =   12  (GPS_RX  = 0, 1, 2, ..., 29)
> GPS Tx  . . . . . . . . . =   13  (GPS_TX  = 0, 1, 2, ..., 29)
> Sbus OUT  . . . . . . . . =  255  (SBUS_OUT= 0, 1, 2, ..., 29)
> RPM   . . . . . . . . . . =  255  (RPM     = 0, 1, 2, ..., 29)
> SDA (I2C sensors) . . . . =   14  (SDA     = 2, 6, 10, 14, 18, 22, 26)
> SCL (I2C sensors) . . . . =   15  (SCL     = 3, 7, 11, 15, 19, 23, 27)
> PWM Channels 1, 2, 3 ,4   =  255  255  255  255 (C1 / C16= 0, 1, 2, ..., 15)
> PWM Channels 5, 6, 7 ,8   =  255  255  255  255
> PWM Channels 9,10,11,12   =  255  255  255  255
> PWM Channels 13,14,15,16  =  255  255  255  255
> **Voltage 1**, 2, 3, 4        =   **26**  255  255  255 (V1 / V4 = 26, 27, 28, 29)
> 
> Protocol is Mpx
> CRSF baudrate = 420000
> Voltage parameters:
>     Scales : 1.000000 , 1.000000 , 1.000000 , 1.000000 
>     Offsets: 0.000000 , 0.000000 , 0.000000 , 0.000000 
>     No temperature sensors are connected on V3 and V4
> RPM multiplier = 1.000000
> Baro sensor is detected using MS5611
>     Sensitivity min = 100 (at 100)   , max = 300 (at 1000)
>     Hysteresis = 5 
> Acc/Gyro is not detected
> First analog to digital sensor is not detected
> Second analog to digital sensor is not detected
> Foreseen GPS type is Ublox  :GPS is detected and has a fix
> Failsafe type is HOLD
> 
> Config parameters are OK
> Press ? + Enter to get help about the commands
> 
>  >> 
> da id= 3 18
> da id= 4 2
> da id= 7 9
> da id= 8 19
> da id= 13 10
> da id= 1 11
> da id= 3 18
> da id= 4 2
> .
> .

This is, what I see here with oXs disconnected

[image]

and with oXs connected

[image]

Nothing visible like V1 here.

By the way, those debug messages are a bit annoying and could  not be stopped by 

_-To debug on USB/serial the telemetry frames, enter DEBUGTLM=Y or DEBUGTLM=N (default)_

mstrens, 2023-02-12:
oXs currently transmits V1 as line nr 1.
It could be that this is not allowed because the first number (0, 1, and perhaps 2) are reserved by MPX for the internal measurements done by the Rx itself.
I can try to move V1 to another line to see if this solve the issue.
I will also remove the debug info I had added.

You can now try version 1.0.3 (on main or test it is currently the same)

mstrens, 2023-02-12:
You can also try adding V2.
V2 is supposed to be the current on line 2.
If my change solved the issue for V1 and if V2 does not appear (when configured), then I probably will have to moved it also to another line number.

mstrens, 2023-02-12:
I presume that V2 is OK because I see it on another screen shot you put at the beginning of this issue

nostromo123, 2023-02-12:
Yes had it on V2, put pin 27 on GND and then tested it with a 1.5V battery. This seemed working. OK, did not do any settings with SCALE or OFFSET on oXs side and with the settings on the TX side. You don't have to put new versions out, just tell me, what modifications to make for a check here .... I'll check and report back.

Just advice and I'll check tomorrow ... ;-) 

PS. Still curious, why I had the the issue here building with the 'KalmanFilter4D.cpp' thing with the missing capital letters in #include "KalmanFilter4D.h"??

1. It is your code and I saw here that something with KalmanFilter4D.h was not OK.

2. And obviously you and Torsten did not have any issues building. You don't want to say, that Windows is less critical than Linux with these typos?

Good night - KH

nostromo123, 2023-02-13:
I think, we should keep to your documentation with V1 as the momentary voltage output. Hopefully at least a V2 will be available in future  for current transmission ...

> VOLT1       0.001V           V1 pin is defined                            Index = 1                   0.1V
> VOLT2       0.001V=0.001A    V2 pin is defined                            not transmitted currently
> Capacity    mAh              V2 pin is defined (reset after power down)   not transmitted currently
> VOLT3       0.001V           V3 pin is defined & TEMP=0 or =255           not transmitted currently
>             represent temp   V3 pin is defined & TEMP=1 or =2             Index = 6                  0.1 degree
> VOLT4       0.001V           V4 pin is defined                            not transmitted currently

mstrens, 2023-02-13:
About KalmanFilter4D issue, I think that:
- #include is an instruction for the compiler make a kind of (temporary) copy paste of the content of the file specified by the include into the .cpp file.
- so the compiler ask to the OS to retrieve this .h file and to copy it.
- this task is executed by the OS. 
- in windows, OS does not take care of the caps in the file name and so it find the file KalmanFilter4D even if the include says "kalmanfilter4d.h"
- in linux, I expect that the OS takes care of the caps when looking for a file. And so if caps does not match, the #include fails

mstrens, 2023-02-13:
> I think, we should keep to your documentation with V1 as the momentary voltage output. Hopefully at least a V2 will be available in future for current transmission ...

Sorry, I do not understand.
In MPX protocol I have to specify a line number (from 0 to 15) for each field.
In latest version, I still continue to measure V1 and to transmit it. Still, instead of sending it as line 1, I moved it to line 9 (because I expected that line 1 is reserved for Rx internal usage)

I kept V2 for measuring a current (mA) and it remains on line 2 
When V2 is defined I also calculate a current consumption (= capacity) that is transmitted on line 11.

It is strange that currently you do not get anymore current and capacity because I did not changed this part and they where present in the first screen shot from this post.
Note: I just noticed that the fiel fields per protocol.txt was not updated. I will fix this in next version.

mstrens, 2023-02-13:
Sorry, I wrote last post in Fr. Here the same in english
I do not understand.
In MPX protocol, each field get a line (=index) number (in range 0/15).
In last version, I still measure and transmit V1 but I put it in line 9 instead of 1 (because I expected that line 1 is for internal use)

I still use V2 to measure a current and it stays on line 2
When V2 is defined I also calculate the consumption (=capacity) on line 11.

It is strange that you do not get V1, current and capacity because I did not change this code.
When V3 is defined and TEMP=1, TEMP1 should also be transmitted in line 6
When RPM is defined, it should be transmitted in line 5 

Note: I noticed some errors in the file fields per protocole.txt. I will fix it in next version.

nostromo123, 2023-02-13:
Your latest version I see and have here is 1.0.3.

So I'll wait for the next version to test again V1 / V2. Breadboard is ready and can wait ...

PS. Thank you for explaining this building error in platformio under Linux. Experienced the same failure with the Arduino IDE under Linux and normally I fix this myself. 

br KH

mstrens, 2023-02-13:
I put version 1.0.4 in test branch.
Changes are only about MPX protocol.
I updated the doc about fields per protocol.
I also tried to send some MPX fields (with a special code  NO_VALUE) when some fields are not immediately available (like for GPS). It could be that MPX discards fields that are not present within some delay after power on. I am not sure but this could perhaps explains why some user does not get GPS data.

nostromo123, 2023-02-13:
Sorry, but 1.0.4 seems to be a setback.

> Version = 1.0.4 
>     Function                Pin   Change entering XXX=yyy (yyy=255 to disable)
> Primary channels input    =  255  (PRI     = 5, 9, 21, 25)
> Secondary channels input  =  255  (SEC     = 1, 13, 17, 29)
> Telemetry . . . . . . . . =   10  (TLM     = 0, 1, 2, ..., 29)
> GPS Rx  . . . . . . . . . =   12  (GPS_RX  = 0, 1, 2, ..., 29)
> GPS Tx  . . . . . . . . . =   13  (GPS_TX  = 0, 1, 2, ..., 29)
> Sbus OUT  . . . . . . . . =  255  (SBUS_OUT= 0, 1, 2, ..., 29)
> RPM   . . . . . . . . . . =  255  (RPM     = 0, 1, 2, ..., 29)
> SDA (I2C sensors) . . . . =   14  (SDA     = 2, 6, 10, 14, 18, 22, 26)
> SCL (I2C sensors) . . . . =   15  (SCL     = 3, 7, 11, 15, 19, 23, 27)
> PWM Channels 1, 2, 3 ,4   =  255  255  255  255 (C1 / C16= 0, 1, 2, ..., 15)
> PWM Channels 5, 6, 7 ,8   =  255  255  255  255
> PWM Channels 9,10,11,12   =  255  255  255  255
> PWM Channels 13,14,15,16  =  255  255  255  255
> Voltage 1, 2, 3, 4        =   26   27  255  255 (V1 / V4 = 26, 27, 28, 29)
> 
> Protocol is Mpx
> CRSF baudrate = 420000
> Voltage parameters:
>     Scales : 1.000000 , 1.000000 , 1.000000 , 1.000000 
>     Offsets: 0.000000 , 0.000000 , 0.000000 , 0.000000 
>     No temperature sensors are connected on V3 and V4
> RPM multiplier = 1.000000
> Baro sensor is detected using MS5611
>     Sensitivity min = 100 (at 100)   , max = 300 (at 1000)
>     Hysteresis = 5 
> Acc/Gyro is not detected
> First analog to digital sensor is not detected
> Second analog to digital sensor is not detected
> Fore> Version = 1.0.4 
>     Function                Pin   Change entering XXX=yyy (yyy=255 to disable)
> Primary channels input    =  255  (PRI     = 5, 9, 21, 25)
> Secondary channels input  =  255  (SEC     = 1, 13, 17, 29)
> Telemetry . . . . . . . . =   10  (TLM     = 0, 1, 2, ..., 29)
> GPS Rx  . . . . . . . . . =   12  (GPS_RX  = 0, 1, 2, ..., 29)
> GPS Tx  . . . . . . . . . =   13  (GPS_TX  = 0, 1, 2, ..., 29)
> Sbus OUT  . . . . . . . . =  255  (SBUS_OUT= 0, 1, 2, ..., 29)
> RPM   . . . . . . . . . . =  255  (RPM     = 0, 1, 2, ..., 29)
> SDA (I2C sensors) . . . . =   14  (SDA     = 2, 6, 10, 14, 18, 22, 26)
> SCL (I2C sensors) . . . . =   15  (SCL     = 3, 7, 11, 15, 19, 23, 27)
> PWM Channels 1, 2, 3 ,4   =  255  255  255  255 (C1 / C16= 0, 1, 2, ..., 15)
> PWM Channels 5, 6, 7 ,8   =  255  255  255  255
> PWM Channels 9,10,11,12   =  255  255  255  255
> PWM Channels 13,14,15,16  =  255  255  255  255
> **Voltage 1, 2, 3, 4        =   26   27  255  255 (V1 / V4 = 26, 27, 28, 29**)
> 
> Protocol is Mpx
> CRSF baudrate = 420000
> Voltage parameters:
>     Scales : 1.000000 , 1.000000 , 1.000000 , 1.000000 
>     Offsets: 0.000000 , 0.000000 , 0.000000 , 0.000000 
>     No temperature sensors are connected on V3 and V4
> RPM multiplier = 1.000000
> Baro sensor is detected using MS5611
>     Sensitivity min = 100 (at 100)   , max = 300 (at 1000)
>     Hysteresis = 5 
> Acc/Gyro is not detected
> First analog to digital sensor is not detected
> Second analog to digital sensor is not detected
> **Foreseen GPS type is Ublox  :GPS is detected and has a fix**
> Failsafe type is HOLD
> 
> Config parameters are OK
> Press ? + Enter to get help about the commandsseen GPS type is Ublox  :GPS is detected and has a fix
> Failsafe type is HOLD
> 
> Config parameters are OK
> Press ? + Enter to get help about the commands

I don't see V1, V2 and GPS on the radio ;-(

mstrens, 2023-02-13:
And do you see Vspeed and Alt?

nostromo123, 2023-02-13:
Vsp and Alt shows up and, sorry for the confusion, also VFAS on V1 now is there*, V1 to Gnd give 0.0V on display, V1 to 1.5V shows 0.1V. I don't see GPS data (fix reported in terminal) and don't see V2

* I deleted all sensors and did several times a sensor search, but maybe the board needed a re powering for VFAS ..

mstrens, 2023-02-13:
in file mpx.cpp, there is currently
        case MVOLT:
            mpxValue= fields[fieldId].value / 100; // from mvolt to 0.1V
            break;

Perhaps you can try 
        case MVOLT:
            mpxValue= fields[fieldId].value / 10; // from 0.01V to 0.1V
            break;

I expect you should not see "V2" but "Current" and "capacity" (which are the fields calculated based on V2)

nostromo123, 2023-02-13:
Made this change in source

```
switch (fieldId) {
        case MVOLT:
          //  mpxValue= fields[fieldId].value / 100; // from mvolt to 0.1V
            mpxValue= fields[fieldId].value / 10; // from 0.01V to 0.1V
```

**GPS**, terminal shows (fast afer power ON) fix, but still not GPS values on transmitter. 

**Current**, still no sign on transmitter.

V1 - GND, VFAS 0.0V display

V1 - BAT (1.53V), VFAS 1.5V display

mstrens, 2023-02-13:
Please note that in version 1.0.6 I made a change in the way Volt1/4 are internally saved. As said in the doc, Volt1/4 are now really saved in mVolt (if scale =1). So for mpx, we still must have /100 because MPX expect a value in 0.1V (so 100 times smaller than mVolt)

mstrens, 2023-02-13:
Could it be that mpx protocol is now OK?
Is it some remaining issue for MPX protocol?

nostromo123, 2023-02-13:
So, I put 1.0.6 on the breadboard here

> Version = 1.0.6 
>     Function                Pin   Change entering XXX=yyy (yyy=255 to disable)
> Primary channels input    =  255  (PRI     = 5, 9, 21, 25)
> Secondary channels input  =  255  (SEC     = 1, 13, 17, 29)
> Telemetry . . . . . . . . =   10  (TLM     = 0, 1, 2, ..., 29)
> GPS Rx  . . . . . . . . . =   12  (GPS_RX  = 0, 1, 2, ..., 29)
> GPS Tx  . . . . . . . . . =   13  (GPS_TX  = 0, 1, 2, ..., 29)
> Sbus OUT  . . . . . . . . =  255  (SBUS_OUT= 0, 1, 2, ..., 29)
> RPM   . . . . . . . . . . =  255  (RPM     = 0, 1, 2, ..., 29)
> SDA (I2C sensors) . . . . =   14  (SDA     = 2, 6, 10, 14, 18, 22, 26)
> SCL (I2C sensors) . . . . =   15  (SCL     = 3, 7, 11, 15, 19, 23, 27)
> PWM Channels 1, 2, 3 ,4   =  255  255  255  255 (C1 / C16= 0, 1, 2, ..., 15)
> PWM Channels 5, 6, 7 ,8   =  255  255  255  255
> PWM Channels 9,10,11,12   =  255  255  255  255
> PWM Channels 13,14,15,16  =  255  255  255  255
> Voltage 1, 2, 3, 4        =   26   27  255  255 (V1 / V4 = 26, 27, 28, 29)
> 
> Protocol is Mpx
> CRSF baudrate = 420000
> Voltage parameters:
>     Scales : 1.000000 , 1.000000 , 1.000000 , 1.000000 
>     Offsets: 0.000000 , 0.000000 , 0.000000 , 0.000000 
>     No temperature sensors are connected on V3 and V4
> RPM multiplier = 1.000000
> Baro sensor is detected using MS5611
>     Sensitivity min = 100 (at 100)   , max = 300 (at 1000)
>     Hysteresis = 5 
> Acc/Gyro is not detected
> First analog to digital sensor is not detected
> Second analog to digital sensor is not detected
> **Foreseen GPS type is Ublox  :GPS is detected and has a fix**
> Failsafe type is HOLD
> 
> Config parameters are OK

IT DOES NOT LOOK GOOD! Sorry.

**No GPS, no current!**

I changed the setting for VFAS in the radio from 0.0 --> 0.00 and used a new AA battery

But have a look at that YT video and specially **watch ALT and VSpd and the dots in front of ALT and VSpd blinking** 

https://www.youtube.com/watch?v=lG68TlIz-QQ

After this test I changed back to 1.0.1.Now do the same, **watch ALT and VSpd.**

https://www.youtube.com/watch?v=2moF1cvGKiI

Is that normal???

mstrens, 2023-02-13:
it does not seems me normal.
The values should be refreshed more often.
Still I do not understand what happens. 
Do you have a digital scope like a saleae (or a clone)?
This would be useful to capture the signal on tlm pin and perhaps understand what happens.

nostromo123, 2023-02-13:
> it does not seems me normal. The values should be refreshed more often. Still I do not understand what happens. 

Therefore I recorded the videos

> Do you have a digital scope like a saleae (or a clone)? This would be useful to capture the signal on tlm pin and perhaps understand what happens.

Yes, I do. But I used it maybe 2 years ago last time and I am not expert on this, fore sure  ...

Some questions:

- Has the RX <> TX transmission be on or is it sufficient, when the breadboard RP2040 + sensorrs + RX are powered?

- Any proposals for those settings?
  
[image]

[image]

[image]

mstrens, 2023-02-13:
If you can, I suggest having the handset also connected to the RX for the test.
The Rx is supposed to send polling code to oXs every 6msec but I am not sure if this happens when Tx is not connected.
There is 99% chance it is not required but if you can it is better.

The settings for the capture and for the UART analyser seems me OK.
I have such an analyser. So if you make/save a capture and send me the file, I will probably be able to analysed it.
Please send me also the oXs config that you used for the test (so I can see if I find the expected sensor value in the capture).
My email is mstrens@yahoo.fr

nostromo123, 2023-02-13:
I connected for having a look, TX is OFF! But I get data on TLM.

[image]

I'll try to do tests with 1.0.1 and 1.0.6 with unchanged CONFIG as posted here.

Hope, I can help a bit .... But first I have to play a little with saleae to get used to it. ;-)

Good night - KH

mstrens, 2023-02-13:
Thanks.
If there is already a signal on TLM, it is probably not required having the handset connected.
Still it could perhaps help because it could say if some fields that are supposed to be transmitted are missing on the handset.
Having the handset connected or not will probably not help to understand why there is such a difference in the the refresh rates.

nostromo123, 2023-02-13:
[1 MHz, 30 M Samples [1].logicdata.zip](https://github.com/mstrens/oXs_on_RP2040/files/10726647/1.MHz.30.M.Samples.1.logicdata.zip)

**Only a short test**, I think, this is 1.0.1 and TX is OFF. You would prefer TX on and that should be no problem now. What in the saleae recording settings do I have to modify in your opinion, so you can analyze the data?

But now is bedtime!

mstrens, 2023-02-13:
I have probably found an explanation for the issue in 1.0.6.
I activated a debug option in order to simulate mpx Rx and I forgot to remove it in the 1.0.6 version.

In mpx.cpp you have a line
#define DEBUG_MPX_WITHOUT_RX

This line should be commented with // in front
//#define DEBUG_MPX_WITHOUT_RX

Sorry for the mistake.
Anyway I will also have at the capture you made.

mstrens, 2023-02-13:
Analyzing your capture already helped me to see a second issue.
So best is to wait for a new version to fix this before making a new capture (to avoid loosing you time)

nostromo123, 2023-02-14:
Michel, don't hesitate. I is quite easy to other captures here. I forgot, how easy it is. Just tell me, TX on, capture START, RP2040 on. Which firmware to use, any modification in source. I am ready to do the tests here. Just tell me, what you need ....

mstrens, 2023-02-14:
I just put 1.0.7 in test branch.
Thanks for making a capture with this version.
Can you make it using the faster sample rate that you logic analyser can support (to see if there are very short glitch in the signal).
Can you also confirm if the data are present on the handset display and if the refresh rate seems normal.

nostromo123, 2023-02-14:
OK, just saw it on github.

Sample rate, how fast?
[image]

Trigger?
[image]

30sec ok?

The board with gps, MS5611 ready to go!

[image]

nostromo123, 2023-02-14:
OK. But at the moment have to say it looks worse. NO GPS data!, And Alt, VSpd and VFAS now drop out temporary on the telemetry screen of the radio. If you want to see, I'll make a YT video.

Give me a minute and I'll send a capture ....

nostromo123, 2023-02-14:
Repowered the board and now Alt, VSpd and VFAS  show up stable .....

nostromo123, 2023-02-14:
Here is the capture 
[8 MHz, 160 M Samples [1].logicdata.zip](https://github.com/mstrens/oXs_on_RP2040/files/10730484/8.MHz.160.M.Samples.1.logicdata.zip)

mstrens, 2023-02-14:
I expect that(based on your config) you should also have current.
I think it is a bug (timing issue) in oXs.
I now put version 1.0.8.
Do you get current with this new version.

Was GPS activated in your config? I do not see GPS data in the capture.

nostromo123, 2023-02-14:
> Version = 1.0.8 
>     Function                Pin   Change entering XXX=yyy (yyy=255 to disable)
> Primary channels input    =  255  (PRI     = 5, 9, 21, 25)
> Secondary channels input  =  255  (SEC     = 1, 13, 17, 29)
> Telemetry . . . . . . . . =   10  (TLM     = 0, 1, 2, ..., 29)
> GPS Rx  . . . . . . . . . =   12  (GPS_RX  = 0, 1, 2, ..., 29)
> GPS Tx  . . . . . . . . . =   13  (GPS_TX  = 0, 1, 2, ..., 29)
> Sbus OUT  . . . . . . . . =  255  (SBUS_OUT= 0, 1, 2, ..., 29)
> RPM   . . . . . . . . . . =  255  (RPM     = 0, 1, 2, ..., 29)
> SDA (I2C sensors) . . . . =   14  (SDA     = 2, 6, 10, 14, 18, 22, 26)
> SCL (I2C sensors) . . . . =   15  (SCL     = 3, 7, 11, 15, 19, 23, 27)
> PWM Channels 1, 2, 3 ,4   =  255  255  255  255 (C1 / C16= 0, 1, 2, ..., 15)
> PWM Channels 5, 6, 7 ,8   =  255  255  255  255
> PWM Channels 9,10,11,12   =  255  255  255  255
> PWM Channels 13,14,15,16  =  255  255  255  255
> **Voltage 1, 2, 3, 4        =   26   27**  255  255 (V1 / V4 = 26, 27, 28, 29)
> 
> Protocol is Mpx
> CRSF baudrate = 420000
> Voltage parameters:
>     Scales : 1.000000 , 1.000000 , 1.000000 , 1.000000 
>     Offsets: 0.000000 , 0.000000 , 0.000000 , 0.000000 
>     No temperature sensors are connected on V3 and V4
> RPM multiplier = 1.000000
> Baro sensor is detected using MS5611
>     Sensitivity min = 100 (at 100)   , max = 300 (at 1000)
>     Hysteresis = 5 
> Acc/Gyro is not detected
> First analog to digital sensor is not detected
> Second analog to digital sensor is not detected
> Foreseen GPS type is Ublox  :**GPS is detected and has a fix**
> Failsafe type is HOLD

**No current, no GPS**

Capture? Nuke the RP2040?

mstrens, 2023-02-14:
Can you make the capture when you are sure having a fix (as said on the PC when printing the config).
Can you also join the result of a command FV just (taken before the capture).

mstrens, 2023-02-14:
Do you have a resistor (between 100 and 1000 ohm) between RX and TLM pin?
This would not help for the issue.
It is just for safety. The bus is driven by the Rx AND by oXs. If they are not well synchronized, there can be conflict. The resistor can help to avoid damaging one component. I am not sure it is absolutely required but it is safier.

nostromo123, 2023-02-14:
> 
> Version = 1.0.8 
>     Function                Pin   Change entering XXX=yyy (yyy=255 to disable)
> Primary channels input    =  255  (PRI     = 5, 9, 21, 25)
> Secondary channels input  =  255  (SEC     = 1, 13, 17, 29)
> Telemetry . . . . . . . . =   10  (TLM     = 0, 1, 2, ..., 29)
> GPS Rx  . . . . . . . . . =   12  (GPS_RX  = 0, 1, 2, ..., 29)
> GPS Tx  . . . . . . . . . =   13  (GPS_TX  = 0, 1, 2, ..., 29)
> Sbus OUT  . . . . . . . . =  255  (SBUS_OUT= 0, 1, 2, ..., 29)
> RPM   . . . . . . . . . . =  255  (RPM     = 0, 1, 2, ..., 29)
> SDA (I2C sensors) . . . . =   14  (SDA     = 2, 6, 10, 14, 18, 22, 26)
> SCL (I2C sensors) . . . . =   15  (SCL     = 3, 7, 11, 15, 19, 23, 27)
> PWM Channels 1, 2, 3 ,4   =  255  255  255  255 (C1 / C16= 0, 1, 2, ..., 15)
> PWM Channels 5, 6, 7 ,8   =  255  255  255  255
> PWM Channels 9,10,11,12   =  255  255  255  255
> PWM Channels 13,14,15,16  =  255  255  255  255
> Voltage 1, 2, 3, 4        =   26   27  255  255 (V1 / V4 = 26, 27, 28, 29)
> 
> Protocol is Mpx
> CRSF baudrate = 420000
> Voltage parameters:
>     Scales : 1.000000 , 1.000000 , 1.000000 , 1.000000 
>     Offsets: 0.000000 , 0.000000 , 0.000000 , 0.000000 
>     No temperature sensors are connected on V3 and V4
> RPM multiplier = 1.000000
> Baro sensor is detected using MS5611
>     Sensitivity min = 100 (at 100)   , max = 300 (at 1000)
>     Hysteresis = 5 
> Acc/Gyro is not detected
> First analog to digital sensor is not detected
> Second analog to digital sensor is not detected
> Foreseen GPS type is Ublox  :GPS is detected and has a fix
> Failsafe type is HOLD
> 
> Config parameters are OK
> Press ? + Enter to get help about the commands
> 
>  >> 
> processing cmd
> 
> Cmd to execute:   FV
> 
> GPS Latitude = 51.1881536 degree
> GPS Longitude = 6.4619788 degree
> GPS Groundspeed = 3 cm/s
> GPS Heading = 272.130000 degree
> GPS Altitude = 1913 cm
> GPS Num sat. = 108 cm
> GPS Date J M A = 14 2 23 
> GPS Time H M S = 9 57 55 
> GPS Pdop = 250 
> GPS Home bearing = 283 degree
> GPS Home distance = 2 m
> Volt 1 = 18 mVolt
> Current (Volt 2) = 19 mA
> Capacity (using current) = 0 mAh
> Vspeed = 0 cm/s
> Baro Rel altitude = 111 cm

No resistor!

I'll do a capture, give me some minutes.

Michel, I'll not join that troubleshooting done in the RCG thread by others, because:

This blows up the RCG thread immensely and for anyone interested in oXs-on_RP2040, this thread, as it is messed up at the moment, will be daunting to new users. An issue, if the issue has been found and solved, can be closed!

nostromo123, 2023-02-14:
Here is the capture
[8 MHz, 160 M Samples [1].logicdata.zip](https://github.com/mstrens/oXs_on_RP2040/files/10731243/8.MHz.160.M.Samples.1.logicdata.zip)

No GPS, no current on the radio!

Satcomix, 2023-02-14:
@nostromo123 
, think of the published GPS data, edit the post.

mstrens, 2023-02-14:
I have perhaps found why there was no GPS data. See version 1.0.9
For no current I still have to search.

nostromo123, 2023-02-14:
GPS is shown on the radio, still no current ....

mstrens, 2023-02-14:
New version 1.0.10 is uploaded
I hope I fixed the current and capacity issues

nostromo123, 2023-02-14:
About that resistor RX <> TLM.

This is the MPX RX with MPX Vario

[image]

I am not an electrician, but I don't think it is critical, we also are directly running attiny841 on MPX. And Reinhardt, who manages the project MPX is an expert ....

https://www.rc-network.de/threads/msb-s-port-telemetrie-konverter.11896883/

I'll load 1.0.10 and report ;-)

nostromo123, 2023-02-14:
Curr and Cap show up.

1.6V on V2 shows 1.6A!

That was a bit hard ;-)

Thank you, close this issue now ....

closed 2023-02-14 by nostromo123