mstrens/oXs_on_RP2040 #37: V1.0.1 MPX - GPS not detected

nostromo123, 2023-02-12:
Added GPS

> Version = 1.0.1 
>     Function                Pin   Change entering XXX=yyy (yyy=255 to disable)
> Primary channels input    =  255  (PRI     = 5, 9, 21, 25)
> Secondary channels input  =  255  (SEC     = 1, 13, 17, 29)
> Telemetry . . . . . . . . =   10  (TLM     = 0, 1, 2, ..., 29)
> **GPS Rx  . . . . . . . . . =   12**  (GPS_RX  = 0, 1, 2, ..., 29)
> **GPS Tx  . . . . . . . . . =   13**  (GPS_TX  = 0, 1, 2, ..., 29)
> Sbus OUT  . . . . . . . . =  255  (SBUS_OUT= 0, 1, 2, ..., 29)
> RPM   . . . . . . . . . . =  255  (RPM     = 0, 1, 2, ..., 29)
> SDA (I2C sensors) . . . . =   14  (SDA     = 2, 6, 10, 14, 18, 22, 26)
> SCL (I2C sensors) . . . . =   15  (SCL     = 3, 7, 11, 15, 19, 23, 27)
> PWM Channels 1, 2, 3 ,4   =  255  255  255  255 (C1 / C16= 0, 1, 2, ..., 15)
> PWM Channels 5, 6, 7 ,8   =  255  255  255  255
> PWM Channels 9,10,11,12   =  255  255  255  255
> PWM Channels 13,14,15,16  =  255  255  255  255
> Voltage 1, 2, 3, 4        =   26   27  255  255 (V1 / V4 = 26, 27, 28, 29)
> 
> Protocol is Mpx
> CRSF baudrate = 420000
> Voltage parameters:
>     Scales : 1.000000 , 1.000000 , 1.000000 , 1.000000 
>     Offsets: 0.000000 , 0.000000 , 0.000000 , 0.000000 
>     No temperature sensors are connected on V3 and V4
> RPM multiplier = 1.000000
> Baro sensor is detected using MS5611
>     Sensitivity min = 100 (at 100)   , max = 300 (at 1000)
>     Hysteresis = 5 
> Acc/Gyro is not detected
> First analog to digital sensor is not detected
> Second analog to digital sensor is not detected
> **Foreseen GPS type is Ublox  :GPS is detected**
> Failsafe type is HOLD
> 
> Config parameters are OK

GPS values not showing up on the radio ...

br - KH

pierrotm777, 2023-02-12:
I confirm, gps is not discovered but it was with older version .

mstrens, 2023-02-12:
Are your sure that you wait long enough in order to let gps get a 3D fix?
GPS is detected when some messages à received even if there is not yet a fix.
GPS data (except number of sat) are transmitted only when GPS has a 3D fix.
To clarify this, when printing on PC, I could add a message to say if there is already a fix or not.

pierrotm777, 2023-02-12:
I wait 1 hour without gps discovered.

nostromo123, 2023-02-12:
@pierrotm777 

Could you please leave that issue alone! Thank you.

pierrotm777, 2023-02-12:
@nostromo123
I use also Mpx.

mstrens, 2023-02-12:
If the GPS is inside the home, it could be that it never get the fix.
I put a version 1.0.2 on github in test branch.
In this version, when you print the config on pc, it should say if there is a fix or not when GPS is discovered.
This could help to see if there is really a bug. Looking just at the code, I do not see directly a bug.

nostromo123, 2023-02-12:
@mstrens

The board is an my window sill, blue LED slow blinking, after 5 sec fast blinking (code modification), after 30 sec red LED slow blinking. FIX.

Now after 3'  did I get GPS DIST, SPD and HDG. Can it be that fix conditions of GPS itself and on oXs are different?
BN-180 reports a fix with 3 SAT and oXs with SAT>3 ?

[BN-180 GPS Module Datasheet.pdf](https://github.com/mstrens/oXs_on_RP2040/files/10716844/BN-180.GPS.Module.Datasheet.pdf)

> @nostromo123 I use also Mpx.

@pierrotm777

.... then do your **own** tests and if you find something. open **your** own issues. **BUT DON'T MESS AROUND IN MINE!**

I'll close this issue now.

closed 2023-02-12 by nostromo123

mstrens, 2023-02-12:
There are several levels of FIX.
I think that Led blink with a lower level than the one used by oXs.
oXs wait for the highest level of fix because it expect to have a good Long and Lat in order to set a right "Home" position.