mstrens/oXs_on_RP2040 #38: MPUCAL value Reset to 0, after change Hardware

Satcomix, 2023-02-12:
Hello Mstrens,
I did some tests with the new MAIN Ver.1.0.1 with S.PORT and HOTT protocol. 
I noticed something that you could help me with.
Is there a way to reset the MPU calibration values, ​​stored in the board firmware to 0.
The GY521 boards with MP6050 are already calibrated at the factory. 
MPUCAL does not work on my GY521 boards. 
However, if I have connected and calibrated a GY86 in the meantime, these values ​​are available and are incorrectly displayed in the GY521  Terminal if I swapt the hardware.

Acc/Gyro is detected using MP6050 (GY521)
     Acceleration offsets X, Y, Z = 3274 , 328 , 1006 (Gy86)
     Gyro offsets         X, Y, Z = -2 , 41 , 49 (GY86)

br, 
Torsten

mstrens, 2023-02-12:
If you can compile by yourself, then there is an easy solution:

in file mpu.cpp you have in lines with 
mpu6050.setXAccelOffset(config.accOffsetX);
    mpu6050.setYAccelOffset(config.accOffsetY);
    mpu6050.setZAccelOffset(config.accOffsetZ);
    mpu6050.setXGyroOffset(config.gyroOffsetX);
    mpu6050.setYGyroOffset(config.gyroOffsetY);
    mpu6050.setZGyroOffset(config.gyroOffsetZ);
    

If you put those lines as comment (adding // in front), oXs will use the MP6050 with the factory offsets.

When I have time, I will look perhaps at another way of performing calibration. The curent code is found on internet and is quite complex to understand. So I can't change it directly in order to avoid/detect blocking.

Satcomix, 2023-02-12:
Hello Mstrens,
Thanks, i will comment those lines and make a build ( that i can put fast on the board, when changing Hardware)) with PlatformIO.
Its faster to make a new build and placed it on the board, as to change all GPIO Pins again in a new firmware.
Thanks,
br,
Torsten

closed 2023-02-12 by Satcomix

Satcomix, 2023-02-12:
Hello Mstrens,
I think that's a little tougher chunk.
First, as you suggested, I commented on the lines,
no visible result in the terminal!
After that I undid everything and set the numbers in these lines to 0/zero in the entire script, no visible result in the terminal!
I'll keep looking, it's not particularly important.
Greetings Torsten

mstrens, 2023-02-12:
With my proposal, oXs would not update the config and so, on PC you always get the lates values that where saved with MPUCAL.
Still with my proposal, the default values stored(and used)  in mp6050 are not overwritten by the values in config.
So normally you get the expected result even if the displayed values on the pc are wrong.

Satcomix, 2023-02-12:
Thank you for your quick response,
I also noticed that the saved values ​​in the board config are not overwritten as I was hoping.
So I'll take my time to look further tomorrow, because these values ​​irritate me a bit when I change hardware and I don't know exactly which hardware I connected to the many tests i do with S.PORT and HOTT. So many Features to proof, with so many different sensors to check all possibilities of an error.
Thanks again.
Greetings,
Torsten