mstrens/oXs_on_RP2040 #39: Graupner HOTT V3=T? V4=T1

Satcomix, 2023-02-13:
Hello Mstrens,
I found something in the HOTT protocol that doesn't match the "fields per protocol.txt".
V1 to V4 are activated.
1.63V are present at inputs V1 to V3, 
and 2.3V are present at V4
TEMP=2 was entered.
When I look at the handheld's display, I see it
ACC1
V1= 1.6V
T1= +23°C
AKK2
V2= 0.0V
T2= -20°C
if I change SCALE4 from 1 to 10, T1 changes to 230°C.
If I change SCALE3=10, nothing changes.
If I change Scale1 and Scale2, V1 changes to volts and V2 to amps.
The temperatur from T2 never changed.
Greeting,
Torsten

Satcomix, 2023-02-13:
Some more test with HOTT,
V1 26  V2 27  V3 28  V4 29
TEMP=1
Sensor1 Voltage1 16,5V =V1 Scale1=100
Sensor2 Power Con. 16A=V2 Scale2=100
Sensor3                                 Scale3=100 to 1000 Temperature1 varies from 120°C to 200°C
Sensor4 overall voltage 22,3V Scale4=1000
I cant setting the temperature, which i can see under Voltagesensor1.

Satcomix, 2023-02-13:
Next Test:
TEMP=0
Sensor1 Voltage1 =16,5V Scale 100
Sensor2 Ampere1 =16A   Scale 100
Sensor3 Voltage2 =16,4V Scale 100
Sensor4 OverallVolt=22,3V Scale=1000
Everything seems OK

mstrens, 2023-02-13:
I propose to analyse the issues in 2 steps.
First oXs makes different measurements and stores them in a table. This process is the same for all protocols.
In a second step, oXs read the data from this table and fill the frames requested by each protocol. During this second step, oXs skips some data not supported by the related protocol and also change the format (e.g. more or less decimal).

In order to easily analyse the content of the table with the measurement, I added a new command to be entered via the serial terminal on the PC. The command is FV (for Field Values). It print the content of the table for each field that has been calculated at least once. So for GPS, when a fix has not yet been received, some values will not appear in the list even if the GPS is detected.
I propose to first check that all measurements are correct in the table before checking if the value is correctly transmitted and display on the handset.
Note: I already saw a bug about Voltx. The values were stored in 0.01 V in the table while I had expected they were in 0.001V
This version is in 0.7.5 in test branch

Satcomix, 2023-02-13:
TEMP=0

GPS Latitude = 51.xxxxxxxx degree ok
GPS Longitude = 6.xxxxxxx degree ok
GPS Groundspeed = 5 cm/s ok
GPS Heading = 173.050000 degree ok
GPS Altitude = 3888 cm ok
GPS Num sat. = 111 cm ok
GPS Date J M A = 13 2 23 ok
GPS Time H M S = 17 26 23 ok
GPS Pdop = 275 ok
GPS Home bearing = 131 degree ok
GPS Home distance = 29 m ok
Volt 1 = 167449 mVolt       52,3V to 59,6V not ok
Current (Volt 2) = 164452 mA   164,4A ok
Volt 3 = 165056 mVolt ok
Volt 4 = 256586 mVolt ok
Capacity (using current) = 1943 mAh ok
Vspeed = -2 cm/s ok
Baro Rel altitude = -86 cm ok
Pitch = -5 degree ok
Roll = 0 degree ok

Sensor V1=67,4V
Sensor V2 =65,0V

Satcomix, 2023-02-13:
TEMP=1

cmd to execute:   FV

GPS Latitude = 51.xxxxxxx degree
GPS Longitude = 6.xxxxxxxx degree
GPS Groundspeed = 36 cm/s
GPS Heading = 151.920000 degree
GPS Altitude = 3045 cm
GPS Num sat. = 111 cm
GPS Date J M A = 13 2 23 
GPS Time H M S = 17 37 31 
GPS Pdop = 189 
GPS Home bearing = 341 degree
GPS Home distance = 36 m
Volt 1 = 167465 mVolt       67,4V not ok
Current (Volt 2) = 164452 mA  ok 164.4A
Volt 4 = 244820 mVolt  42-44.9V not ok
Capacity (using current) = 58 mAh
Temp 1 (Volt 3) = 165016 degree 176-200°C not ok
Vspeed = 0 cm/s ok
Baro Rel altitude = -1 cm ok 
Pitch = -5 degree ok 
Roll = 0 degree ok

Satcomix, 2023-02-13:
Temp=2

Cmd to execute:   FV

GPS Latitude = 51.xxxxxxxx degree
GPS Longitude = 6.xxxxxxxxx degree
GPS Groundspeed = 7 cm/s
GPS Heading = 160.920000 degree
GPS Altitude = 2374 cm
GPS Num sat. = 112 cm
GPS Date J M A = 13 2 23 
GPS Time H M S = 17 44 1 
GPS Pdop = 183 
GPS Home bearing = 167 degree
GPS Home distance = 29 m
Volt 1 = 167457 mVolt    67,5V not ok
Current (Volt 2) = 164476 mA  164.4A ok
Capacity (using current) = 114 mAh 
Temp 1 (Volt 3) = 165040 degree     -15 to 222°C not ok
Temp 2 (Volt 4) = 242564 degree    -20°C not ok
Vspeed = 2 cm/s
Baro Rel altitude = 1 cm
Pitch = -5 degree
Roll = 0 degree

Satcomix, 2023-02-13:
Its not a problem with GPS and Baro/Vario.
In Version 1.0.5  the values for Sensor1 Voltage1 and Temp1,Temp2 is very stange

mstrens, 2023-02-13:
For V1:
I presume that the real voltage on the pin was 1.65V
So you should get a value 1650 on PC.
Did you used a scale 1.0 or scale 100.0?
If you had scale = 100, then it is normal that the displayed value is about 165000 (= 165V)
This value 165 V is to big for the handset.

Satcomix, 2023-02-13:
Protocol is Hott
CRSF baudrate = 420000
Voltage parameters:
    Scales : 10.000000 , 10.000000 , 10.000000 , 10.000000 
    Offsets: 0.000000 , 0.000000 , 0.000000 , 0.000000 
    No temperature sensors are connected on V3 and V4
RPM multiplier = 1.000000
Baro sensor is detected using MS5611
    Sensitivity min = 100 (at 100)   , max = 300 (at 1000)
    Hysteresis = 5 
Acc/Gyro is detected using MP6050
     Acceleration offsets X, Y, Z = 3134 , 346 , 990
     Gyro offsets         X, Y, Z = -2 , 40 , 49
First analog to digital sensor is not detected
Second analog to digital sensor is not detected
Foreseen GPS type is Ublox  :GPS is detected and has a fix
Failsafe uses predefined values
     Chan 1...4  = 992 , 992 , 992 , 992
     Chan 5...8  = 992 , 992 , 992 , 992
     Chan 9...12 = 992 , 992 , 992 , 992
     Chan 13...16= 992 , 992 , 992 , 992

Config parameters are OK
Press ? + Enter to get help about the commands

 >> 
processing cmd

Cmd to execute:   FV

GPS Latitude = 51.xxxxxx degree
GPS Longitude = 6.xxxxxx degree
GPS Groundspeed = 6 cm/s
GPS Heading = 144.420000 degree
GPS Altitude = 4157 cm
GPS Num sat. = 110 cm
GPS Date J M A = 13 2 23 
GPS Time H M S = 18 6 31 
GPS Pdop = 188 
GPS Home bearing = 176 degree
GPS Home distance = 47 m
Volt 1 = 16748 mVolt
Current (Volt 2) = 16448 mA
Volt 3 = 16503 mVolt
Volt 4 = 2384 mVolt
Capacity (using current) = 3 mAh
Vspeed = 2 cm/s
Baro Rel altitude = -2 cm
Pitch = -5 degree
Roll = 0 degree

Satcomix, 2023-02-13:
Voltage parameters:
    Scales : 10.000000 , 10.000000 , 10.000000 , 10.000000 
    Offsets: 0.000000 , 0.000000 , 0.000000 , 0.000000 
    Temperature sensors are connected on V3 and V4
RPM multiplier = 1.000000
Baro sensor is detected using MS5611
    Sensitivity min = 100 (at 100)   , max = 300 (at 1000)
    Hysteresis = 5 
Acc/Gyro is detected using MP6050
     Acceleration offsets X, Y, Z = 3134 , 346 , 990
     Gyro offsets         X, Y, Z = -2 , 40 , 49
First analog to digital sensor is not detected
Second analog to digital sensor is not detected
Foreseen GPS type is Ublox  :GPS is detected and has a fix
Failsafe uses predefined values
     Chan 1...4  = 992 , 992 , 992 , 992
     Chan 5...8  = 992 , 992 , 992 , 992
     Chan 9...12 = 992 , 992 , 992 , 992
     Chan 13...16= 992 , 992 , 992 , 992

Config parameters are OK
Press ? + Enter to get help about the commands

 >> 
processing cmd

Cmd to execute:   FV

GPS Latitude = 51.xxxxxxxxxx degree
GPS Longitude = 6.xxxxxxxxxx degree
GPS Groundspeed = 8 cm/s
GPS Heading = 138.150000 degree
GPS Altitude = 4170 cm
GPS Num sat. = 113 cm
GPS Date J M A = 13 2 23 
GPS Time H M S = 18 9 28 
GPS Pdop = 156 
GPS Home bearing = 167 degree
GPS Home distance = 5 m
Volt 1 = 16746 mVolt
Current (Volt 2) = 16446 mA
Capacity (using current) = 14 mAh
Temp 1 (Volt 3) = 16504 degree
Temp 2 (Volt 4) = 2444 degree
Vspeed = 1 cm/s
Baro Rel altitude = -4 cm
Pitch = -5 degree
Roll = 0 degree

Temp1 turns from +69°C to +130°C
Temp2= -20°C

mstrens, 2023-02-13:
So with scale = 10, it is normal that a real voltage of 1.65V is displayed on the PC as 16504 because it is 10 X (= scale) 1650mV.

What was the real voltage applied on pin for V4. If the displayed value is 2444 (with a scale = 10) it look like the real voltage on pin V4 was 2444/10 mv = 0.244V. Is that true (or was the voltage 2.4V)

Satcomix, 2023-02-13:
Protocol is Hott
CRSF baudrate = 420000
Voltage parameters:
    Scales : 1.000000 , 1.000000 , 0.100000 , 0.100000 
    Offsets: 0.000000 , 0.000000 , 0.000000 , 0.000000 
    Temperature sensors are connected on V3 and V4
RPM multiplier = 1.000000
Baro sensor is detected using MS5611
    Sensitivity min = 100 (at 100)   , max = 300 (at 1000)
    Hysteresis = 5 
Acc/Gyro is detected using MP6050
     Acceleration offsets X, Y, Z = 3134 , 346 , 990
     Gyro offsets         X, Y, Z = -2 , 40 , 49
First analog to digital sensor is not detected
Second analog to digital sensor is not detected
Foreseen GPS type is Ublox  :GPS is detected and has a fix
Failsafe uses predefined values
     Chan 1...4  = 992 , 992 , 992 , 992
     Chan 5...8  = 992 , 992 , 992 , 992
     Chan 9...12 = 992 , 992 , 992 , 992
     Chan 13...16= 992 , 992 , 992 , 992

Config parameters are OK
Press ? + Enter to get help about the commands

 >> 
processing cmd

Cmd to execute:   FV

GPS Latitude = 51.xxxxxxxxx degree
GPS Longitude = 6.xxxxxxxxx degree
GPS Groundspeed = 27 cm/s
GPS Heading = 182.540000 degree
GPS Altitude = 2039 cm
GPS Num sat. = 112 cm
GPS Date J M A = 13 2 23 
GPS Time H M S = 18 17 6 
GPS Pdop = 162 
GPS Home bearing = 213 degree
GPS Home distance = 6 m
Volt 1 = 1674 mVolt
Current (Volt 2) = 1644 mA
Capacity (using current) = 0 mAh
Temp 1 (Volt 3) = 165 degree
Temp 2 (Volt 4) = 23 degree
Vspeed = 3 cm/s
Baro Rel altitude = 1 cm
Pitch = -5 degree
Roll = 0 degree

Satcomix, 2023-02-13:
The handheld can show 165°C in Temp1 but Temp2 shows everytime -20°C

Satcomix, 2023-02-13:
Now i see it, temp1 and temp2 are reversed

mstrens, 2023-02-13:
I just see a copy/paste error in hott.cpp
There is 
    if ( ( fields[TEMP2].available) && (fields[TEMP2].value > -20) ) {
        TxHottData.gamMsg.temperature1 = fields[TEMP2].value + 20 ; // Hott applies an offset of 20. A value of 20 = 0°C
    } else {
        TxHottData.gamMsg.temperature2 = 20 ; // Hott applies an offset of 20. A value of 20 = 0°C    

You should have (temperature2 instead of temperature1)
    if ( ( fields[TEMP2].available) && (fields[TEMP2].value > -20) ) {
        TxHottData.gamMsg.temperature2 = fields[TEMP2].value + 20 ; // Hott applies an offset of 20. A value of 20 = 0°C
    } else {
        TxHottData.gamMsg.temperature2 = 20 ; // Hott applies an offset of 20. A value of 20 = 0°C    

I will fix it in next version

Satcomix, 2023-02-13:
Protocol is Hott
CRSF baudrate = 420000
Voltage parameters:
    Scales : 1.000000 , 1.000000 , 0.010000 , 0.100000 
    Offsets: 0.000000 , 0.000000 , 0.000000 , 0.000000 
    Temperature sensors are connected on V3 and V4
RPM multiplier = 1.000000
Baro sensor is detected using MS5611
    Sensitivity min = 100 (at 100)   , max = 300 (at 1000)
    Hysteresis = 5 
Acc/Gyro is detected using MP6050
     Acceleration offsets X, Y, Z = 3134 , 346 , 990
     Gyro offsets         X, Y, Z = -2 , 40 , 49
First analog to digital sensor is not detected
Second analog to digital sensor is not detected
Foreseen GPS type is Ublox  :GPS is detected and has a fix
Failsafe uses predefined values
     Chan 1...4  = 992 , 992 , 992 , 992
     Chan 5...8  = 992 , 992 , 992 , 992
     Chan 9...12 = 992 , 992 , 992 , 992
     Chan 13...16= 992 , 992 , 992 , 992

Config parameters are OK
Press ? + Enter to get help about the commands

 >> 
processing cmd

Cmd to execute:   FV

GPS Latitude = 51.xxxxxxxx degree
GPS Longitude = 6.xxxxxxxxx degree
GPS Groundspeed = 3 cm/s
GPS Heading = 0.000000 degree
GPS Altitude = 4136 cm
GPS Num sat. = 112 cm
GPS Date J M A = 13 2 23 
GPS Time H M S = 18 35 0 
GPS Pdop = 198 
GPS Home bearing = 176 degree
GPS Home distance = 40 m
Volt 1 = 1674 mVolt
Current (Volt 2) = 1644 mA
Capacity (using current) = 2 mAh
Temp 1 (Volt 3) = 16 degree
Temp 2 (Volt 4) = 23 degree
Vspeed = 1 cm/s
Baro Rel altitude = -3 cm
Pitch = -5 degree
Roll = 0 degree

Now everything seems ok, thank you very much for the time and work.
I will make some more tests with HOTT.
br,
Torsten

mstrens, 2023-02-13:
> When you set TEMP=1 the settings were set to Temp2 V4 and not Temp1V3 The same is in TEMP=2 V3 is V4 and V4 is V3 but shows -20°C

I only saw one bug that I fixed in a version 1.0.6 that I just put on github.
Value -20 is "normal": when OXS fill the frame with 0, the handset consider it is -20 (Hott works with an offset of 20 degree in order to keep values as positive up to -20)

Satcomix, 2023-02-13:
The last post is the new version!!!

mstrens, 2023-02-13:
yes.

Satcomix, 2023-02-13:
Now,
TEMP=0
TEMP=1
TEMP=2
works without issues.
Thanks again

closed 2023-02-13 by Satcomix