mstrens/oXs_on_RP2040 #40: GPS not discovered with Mpx mode

pierrotm777, 2023-02-14:
I use the version 1.1.0 but my gps isn't discovered.
Version = 1.1.0 
    Function                Pin   Change entering XXX=yyy (yyy=255 to disable)
Primary channels input    =    5  (PRI     = 5, 9, 21, 25)
Secondary channels input  =   13  (SEC     = 1, 13, 17, 29)
Telemetry . . . . . . . . =    0  (TLM     = 0, 1, 2, ..., 29)
GPS Rx  . . . . . . . . . =    8  (GPS_RX  = 0, 1, 2, ..., 29)
GPS Tx  . . . . . . . . . =    7  (GPS_TX  = 0, 1, 2, ..., 29)
Sbus OUT  . . . . . . . . =  255  (SBUS_OUT= 0, 1, 2, ..., 29)
RPM   . . . . . . . . . . =   14  (RPM     = 0, 1, 2, ..., 29)
SDA (I2C sensors) . . . . =    2  (SDA     = 2, 6, 10, 14, 18, 22, 26)
SCL (I2C sensors) . . . . =    3  (SCL     = 3, 7, 11, 15, 19, 23, 27)
PWM Channels 1, 2, 3 ,4   =  255  255  255  255 (C1 / C16= 0, 1, 2, ..., 15)
PWM Channels 5, 6, 7 ,8   =  255  255  255  255
PWM Channels 9,10,11,12   =  255  255  255  255
PWM Channels 13,14,15,16  =  255  255  255  255
Voltage 1, 2, 3, 4        =  255  255  255  255 (V1 / V4 = 26, 27, 28, 29)

Protocol is Mpx
CRSF baudrate = 420000
Voltage parameters:
    Scales : 34.000000 , 1.000000 , 1.000000 , 1.000000 
    Offsets: 0.000000 , 0.000000 , 0.000000 , 0.000000 
    No temperature sensors are connected on V3 and V4
RPM multiplier = 1.000000
Baro sensor is detected using MS5611
    Sensitivity min = 100 (at 100)   , max = 300 (at 1000)
    Hysteresis = 5 
Acc/Gyro is detected using MP6050
     Acceleration offsets X, Y, Z = -6502 , 1062 , 2346
     Gyro offsets         X, Y, Z = 185 , -36 , -10
First analog to digital sensor is not detected
Second analog to digital sensor is not detected
Foreseen GPS type is Ublox  :GPS is detected and has a fix
Failsafe type is HOLD

Config parameters are OK
Press ? + Enter to get help about the commands

 >> 

Mu FV values:
Cmd to execute:   FV

GPS Latitude = 44.xxxxxxx degree
GPS Longitude = -0.xxxxxxx degree
GPS Groundspeed = 6 cm/s
GPS Heading = 189.450000 degree
GPS Altitude = 3788 cm
GPS Num sat. = 112 cm
GPS Date J M A = 14 2 23 
GPS Time H M S = 11 44 59 
GPS Pdop = 155 
GPS Home bearing = 248 degree
GPS Home distance = 32 m
Vspeed = 2 cm/s
Baro Rel altitude = 136 cm
Pitch = 0 degree
Roll = 4 degree

mstrens, 2023-02-14:
There is probably a misunderstanding.
In your screen shot of the handset, I see Dist, Spd, Hdg which are data from the GPS.
Please note that the Mpx protocol does not allow to get all data from the GPS (e.g. long and lat are not part of the protocol).
In doc folder, there is a file that says which field is supposed to be present in which protocol.

pierrotm777, 2023-02-14:
Ha, in fact, gps return well something.
So i close this issue ?

mstrens, 2023-02-14:
Yes, from my point of view  it is OK and issue can be closed.

closed 2023-02-14 by pierrotm777