mstrens/oXs_on_RP2040 #42: Further testing here, seeing a strange effect with GPS

nostromo123, 2023-02-15:
Was doing some tests about V1/V2 here on ELRS and MPX. For my records wanted to make screenshots from the radio telemetry screen and om MPX the add. GPS data was a bit annoying. So I disconnected the GPS power, the GPS data in display got static, but the dots still blinking,  then RX and TX, then RESET and then re powered the RP2040. But the static blinking GPS data remains. This sound a bit strange, so again a YT video ....

https://www.youtube.com/watch?v=7VDTj2SD7c8

Did the same test with the ELRS board, but there a simple RESET on RP2040 finished the spook.

> 
> Version = 1.1.0 
>     Function                Pin   Change entering XXX=yyy (yyy=255 to disable)
> Primary channels input    =  255  (PRI     = 5, 9, 21, 25)
> Secondary channels input  =  255  (SEC     = 1, 13, 17, 29)
> Telemetry . . . . . . . . =   10  (TLM     = 0, 1, 2, ..., 29)
> **GPS Rx  . . . . . . . . . =   12  (GPS_RX  = 0, 1, 2, ..., 29)
> GPS Tx  . . . . . . . . . =   13  (GPS_TX  = 0, 1, 2, ..., 29)**
> Sbus OUT  . . . . . . . . =  255  (SBUS_OUT= 0, 1, 2, ..., 29)
> RPM   . . . . . . . . . . =  255  (RPM     = 0, 1, 2, ..., 29)
> SDA (I2C sensors) . . . . =   14  (SDA     = 2, 6, 10, 14, 18, 22, 26)
> SCL (I2C sensors) . . . . =   15  (SCL     = 3, 7, 11, 15, 19, 23, 27)
> PWM Channels 1, 2, 3 ,4   =  255  255  255  255 (C1 / C16= 0, 1, 2, ..., 15)
> PWM Channels 5, 6, 7 ,8   =  255  255  255  255
> PWM Channels 9,10,11,12   =  255  255  255  255
> PWM Channels 13,14,15,16  =  255  255  255  255
> Voltage 1, 2, 3, 4        =   26   27  255  255 (V1 / V4 = 26, 27, 28, 29)
> 
> **Protocol is Mpx**
> CRSF baudrate = 420000
> Voltage parameters:
>     Scales : 1.000000 , 1.000000 , 1.000000 , 1.000000 
>     Offsets: 0.000000 , 0.000000 , 0.000000 , 0.000000 
>     No temperature sensors are connected on V3 and V4
> RPM multiplier = 1.000000
> Baro sensor is detected using MS5611
>     Sensitivity min = 100 (at 100)   , max = 300 (at 1000)
>     Hysteresis = 5 
> Acc/Gyro is not detected
> First analog to digital sensor is not detected
> Second analog to digital sensor is not detected
> **Foreseen GPS type is Ublox  :GPS is detected but has not (yet) a fix**
> Failsafe type is HOLD
> 
> Config parameters are OK
> 
> Version = 1.1.0 
>     Function                Pin   Change entering XXX=yyy (yyy=255 to disable)
> Primary channels input    =    9  (PRI     = 5, 9, 21, 25)
> Secondary channels input  =  255  (SEC     = 1, 13, 17, 29)
> Telemetry . . . . . . . . =   10  (TLM     = 0, 1, 2, ..., 29)
> **GPS Rx  . . . . . . . . . =   12  (GPS_RX  = 0, 1, 2, ..., 29)
> GPS Tx  . . . . . . . . . =   13  (GPS_TX  = 0, 1, 2, ..., 29)**
> Sbus OUT  . . . . . . . . =  255  (SBUS_OUT= 0, 1, 2, ..., 29)
> RPM   . . . . . . . . . . =  255  (RPM     = 0, 1, 2, ..., 29)
> SDA (I2C sensors) . . . . =   14  (SDA     = 2, 6, 10, 14, 18, 22, 26)
> SCL (I2C sensors) . . . . =   15  (SCL     = 3, 7, 11, 15, 19, 23, 27)
> PWM Channels 1, 2, 3 ,4   =    0    1    2    3 (C1 / C16= 0, 1, 2, ..., 15)
> PWM Channels 5, 6, 7 ,8   =  255  255  255  255
> PWM Channels 9,10,11,12   =  255  255  255  255
> PWM Channels 13,14,15,16  =  255  255  255  255
> Voltage 1, 2, 3, 4        =   26   27  255  255 (V1 / V4 = 26, 27, 28, 29)
> 
> **Protocol is CRSF (=ELRS)**
> CRSF baudrate = 420000
> Voltage parameters:
>     Scales : 1.000000 , 1.000000 , 1.000000 , 1.000000 
>     Offsets: 0.000000 , 0.000000 , 0.000000 , 0.000000 
>     No temperature sensors are connected on V3 and V4
> RPM multiplier = 1.000000
> Baro sensor is detected using BMP280
>     Sensitivity min = 100 (at 100)   , max = 300 (at 1000)
>     Hysteresis = 5 
> Acc/Gyro is not detected
> First analog to digital sensor is not detected
> Second analog to digital sensor is not detected
> **Foreseen GPS type is Ublox  :GPS is detected but has not (yet) a fix**
> Failsafe type is HOLD
> 
> Config parameters are OK

mstrens, 2023-02-15:
About MPX protocol:

As long as a data has been calculated at least once (since power on), oXs sent always the last known value.
This is because the Rx request each field about 16X6 = 96 msec and there are many data that are not updated within a short delay.
So when you disconnect GPS without powering RP2040 down, it is "normal" that oXs continues to send the last values.
Perhaps I could send a special value meaning NO_DATA_AVAILABLE when the value has not been updated in the meantime but I do not know how this will be reflected on the handset. 

About what happens after RP2040 power on:
A few days ago we had issues with Mpx protocol.
I suspected that GPS data could be missing because they were not send short after power on of the receiver and that therfore the receiver stops polling them.
Originally most GPS data were sent only when a fix exists what could take some delay.
So I implemented a logic where I send GPS data when a gps pin is defined (even if GPS does not yet provide data) . As long as the fix is not there, I sent a special value  foreseen by mpx saying "NO DATA AVAILABLE".
I expect that the handset detect this value and shows the GPS field on his display even if there is no real value provided by the GPS.
Now that mpx protocol seems to work, I expect that it is not necessary to sent NO_DATA_AVAILABLE as long as a real value as not been available. I could change the program.

nostromo123, 2023-02-15:
If this behavior is OK for you, it is also OK for me. whereupon in my eyes the MPX implementation of GPS is not very helpful in RC modelling.

Thank you again for having a look into it and giving your explanation.

I'll close this issue and will delete that YT video.

Regards - KH

nostromo123, 2023-02-16:
Please explain!?

mstrens, 2023-02-16:
in version 1.1.5 I made the changes explained here above.
I do not send a field as long it has not been "calculated" (e.g. I do not send the value NOT_AVAILABLE) when the GPS has no yet a fix.
When a field has been transmitted at least once (since power on), I continue to send it every 96ms but if the field has not been recalculated in the mean time I send a special value NOT_AVAILABLE instead of sending the previous value.

Perhaps you can test those changes and say it the result on the handset side is better (e.g. discovering of sensor lost).
I hope that sending NOT_AVAILABLE does not lead to data flickering on the handset when the sensor is not lost but just takes more than 96 msec before getting a new value.

nostromo123, 2023-02-16:
I'll check and report.

But I have to say, new versions come to fast ;-)

Regards - KH

nostromo123, 2023-02-16:
MPX 1.1.5

> Version = 1.1.5 
>     Function                Pin   Change entering XXX=yyy (yyy=255 to disable)
> Primary channels input    =  255  (PRI     = 5, 9, 21, 25)
> Secondary channels input  =  255  (SEC     = 1, 13, 17, 29)
> **Telemetry . . . . . . . . =   10  (TLM     = 0, 1, 2, ..., 29)**
> **GPS Rx  . . . . . . . . . =   12  (GPS_RX  = 0, 1, 2, ..., 29)**
> **GPS Tx  . . . . . . . . . =   13  (GPS_TX  = 0, 1, 2, ..., 29)**
> Sbus OUT  . . . . . . . . =  255  (SBUS_OUT= 0, 1, 2, ..., 29)
> RPM   . . . . . . . . . . =  255  (RPM     = 0, 1, 2, ..., 29)
> **SDA (I2C sensors) . . . . =   14  (SDA     = 2, 6, 10, 14, 18, 22, 26)**
> **SCL (I2C sensors) . . . . =   15  (SCL     = 3, 7, 11, 15, 19, 23, 27)**
> PWM Channels 1, 2, 3 ,4   =  255  255  255  255 (C1 / C16= 0, 1, 2, ..., 15)
> PWM Channels 5, 6, 7 ,8   =  255  255  255  255
> PWM Channels 9,10,11,12   =  255  255  255  255
> PWM Channels 13,14,15,16  =  255  255  255  255
> **Voltage 1, 2, 3, 4        =   26   27  255  255 (V1 / V4 = 26, 27, 28, 29)**
> 
> **Protocol is Mpx**
> CRSF baudrate = 420000
> Voltage parameters:
>     Scales : 1.000000 , 1.000000 , 1.000000 , 1.000000 
>     Offsets: 0.000000 , 0.000000 , 0.000000 , 0.000000 
>     No temperature sensors are connected on V3 and V4
> RPM multiplier = 1.000000
> **Baro sensor is detected using MS5611**
>     Sensitivity min = 100 (at 100)   , max = 300 (at 1000)
>     Hysteresis = 5 
> Acc/Gyro is not detected
> First analog to digital sensor is not detected
> Second analog to digital sensor is not detected
> **Foreseen GPS type is Ublox  :GPS is detected and has a fix**
> Failsafe type is HOLD
> 
> Config parameters are OK

This looks a mess here!

Telemetry screen

[image]

Hdg drops to -1368.4° momentary
Capa drops to -13684mAh momentary
RPM 163840rpm constant
Temp1 -1638.4°C constant
Spd -1638.4km/h momentary
Alt -1638.4m momentary
VFAS -1638.4V momentary
Curr -1638.4A momentary

Moreover several sensors drop OUT (red in display), but that goes very fast, for recording that, I'll need to make another YT video.

And YES, I deleted all sensors and did a new sensor search.

mstrens, 2023-02-16:
I expect that this is the result of sending a special value NOT AVAILABLE (0X8000) when a new measurement is not yet available.
So it is best to avoid this.
Then there remains 2 options:
- always send the last calculated value (if it has been calculated at least once)
- don't send a value (so avoid  sending NOT AVAILABLE) when a new measurement is not yet available.
Your could test this second option (not yet tested) and see if it is good.
 For this test you can just change one line in mpx.cpp
There is
if ( fields[fieldId].available == false) mpxValue = MB_NOVALUE; //overwrite with NO_VALUE as long this field has never been available (no reset in mpx protocol) 

It should be
if ( fields[fieldId].available == false) return false ;

nostromo123, 2023-02-16:
I'll test the second option, but with what version? I see, that other versions are out now .....

Is that also affecting ELRS somehow? At the moment I would like to concentrate on MPX here ...

nostromo123, 2023-02-16:
Made the mod with 1.1.5 and looks good now. Only a short check, have to walk our dog now and will back later .... ;-)

mstrens, 2023-02-16:
This does not change anything for ELRS

nostromo123, 2023-02-16:
From my side we can close this issue!

mstrens, 2023-02-16:
OK I made the change in next oXs version.

closed 2023-02-16 by nostromo123