mstrens/oXs_on_RP2040 #43: oXs Ver.1.1.5 ADV telemetry values,

Satcomix, 2023-02-16:
Good morning Mstrens,
One day no time and already 4 versions further.
I am currently testing Ver1.1.5 with FrSky X20S S.PORT.
With the "FV" command, I noticed that RPM is not displayed in the terminal. The display now shows 21 sensors from the oXs_RP2040 plus 8 sensors from the TD-R10 receiver.
The values ​​from the ADV1115 are displayed with FV, but not in the handheld display, not even with DIY sensors.
Greetings,
Torsten

processing cmd

Cmd to execute:   

Version = 1.1.5 
    Function                Pin   Change entering XXX=yyy (yyy=255 to disable)
Primary channels input    =    5  (PRI     = 5, 9, 21, 25)
Secondary channels input  =    1  (SEC     = 1, 13, 17, 29)
Telemetry . . . . . . . . =    0  (TLM     = 0, 1, 2, ..., 29)
GPS Rx  . . . . . . . . . =   12  (GPS_RX  = 0, 1, 2, ..., 29)
GPS Tx  . . . . . . . . . =   13  (GPS_TX  = 0, 1, 2, ..., 29)
Sbus OUT  . . . . . . . . =    2  (SBUS_OUT= 0, 1, 2, ..., 29)
RPM   . . . . . . . . . . =    4  (RPM     = 0, 1, 2, ..., 29)
SDA (I2C sensors) . . . . =   14  (SDA     = 2, 6, 10, 14, 18, 22, 26)
SCL (I2C sensors) . . . . =   15  (SCL     = 3, 7, 11, 15, 19, 23, 27)
PWM Channels 1, 2, 3 ,4   =  255  255    6  255 (C1 / C16= 0, 1, 2, ..., 15)
PWM Channels 5, 6, 7 ,8   =  255  255  255  255
PWM Channels 9,10,11,12   =  255  255  255  255
PWM Channels 13,14,15,16  =  255  255  255  255
Voltage 1, 2, 3, 4        =   26   27   28   29 (V1 / V4 = 26, 27, 28, 29)

Protocol is Sport (Frsky)
CRSF baudrate = 420000
Voltage parameters:
    Scales : 1.000000 , 1.000000 , 0.010000 , 0.100000 
    Offsets: 0.000000 , 0.000000 , 0.000000 , 0.000000 
    No temperature sensors are connected on V3 and V4
RPM multiplier = 1.000000
Baro sensor is detected using MS5611
    Sensitivity min = 100 (at 100)   , max = 300 (at 1000)
    Hysteresis = 5 
Acc/Gyro is detected using MP6050
     Acceleration offsets X, Y, Z = 0 , 0 , 0
     Gyro offsets         X, Y, Z = 0 , 0 , 0
First analog to digital sensor is detected using ads1115
    Measurement setup: 0 , 8 , 8 ,8
    Gains: 2 , 2 , 2 ,2
    Rates: 7 , 4 , 4 ,7
    Offsets: 0.000000 , 0.000000 , 0.000000 ,0.000000
    Scales: 1.000000 , 1.000000 , 1.000000 ,1.000000
    Averaged on: 10 , 20 , 30 ,50
Second analog to digital sensor is not detected
Foreseen GPS type is Ublox  :GPS is detected and has a fix
Failsafe uses predefined values
     Chan 1...4  = 976 , 992 , 992 , 992
     Chan 5...8  = 992 , 992 , 992 , 992
     Chan 9...12 = 992 , 992 , 992 , 992
     Chan 13...16= 992 , 992 , 992 , 992

Config parameters are OK
Press ? + Enter to get help about the commands

 >> 
processing cmd

Cmd to execute:   FV

GPS Latitude = xx.xxxxxxx degree
GPS Longitude = x.xxxxxxx degree
GPS Groundspeed = 27 cm/s
GPS Heading = 174.930000 degree
GPS Altitude = 5361 cm
GPS Num sat. = 108 cm
GPS Date J M A = 16 2 23 
GPS Time H M S = 8 40 7 
GPS Pdop = 262 
GPS Home bearing = 355 degree
GPS Home distance = 74 m
Volt 1 = 1675 mVolt
Current (Volt 2) = 1651 mA
Volt 3 = 16 mVolt
Volt 4 = 26 mVolt
Capacity (using current) = 6 mAh
Vspeed = 0 cm/s
Baro Rel altitude = 47 cm
Pitch = -9 degree
Roll = 0 degree
Ads 1 1 = -30 mVolt

Satcomix, 2023-02-16:
I make a  Ver.1.1.5 build with PlatformIO and now i can see DIY5132,5133,5134 but no DIY5131

mstrens, 2023-02-16:
strange that you do not get DIY5131.
Also strange that you had to compile it yourself.

Anyway I put version 1.1.6 on github.
I expect it will solve RPM.
Perhaps it helps for ADS.

mstrens, 2023-02-16:
I found the bug about ads 1 1
Version 1.1.7 on github

Satcomix, 2023-02-16:
Hello Mstrens,
RPM shows now in Terminal by FV command,
but i can not find DIY5131.
DIY5132, DIY5133, DIY5134, DIY5135, DIY5136, DIY5137, DIY5138 are shown in telemetry Display in Handheld.
There are no other Sensors with ID5131 in telemetry Sensors in Handheld.

processing cmd

Cmd to execute:   

Version = 1.1.6 
    Function                Pin   Change entering XXX=yyy (yyy=255 to disable)
Primary channels input    =    5  (PRI     = 5, 9, 21, 25)
Secondary channels input  =    1  (SEC     = 1, 13, 17, 29)
Telemetry . . . . . . . . =    0  (TLM     = 0, 1, 2, ..., 29)
GPS Rx  . . . . . . . . . =   12  (GPS_RX  = 0, 1, 2, ..., 29)
GPS Tx  . . . . . . . . . =   13  (GPS_TX  = 0, 1, 2, ..., 29)
Sbus OUT  . . . . . . . . =    2  (SBUS_OUT= 0, 1, 2, ..., 29)
RPM   . . . . . . . . . . =    4  (RPM     = 0, 1, 2, ..., 29)
SDA (I2C sensors) . . . . =   14  (SDA     = 2, 6, 10, 14, 18, 22, 26)
SCL (I2C sensors) . . . . =   15  (SCL     = 3, 7, 11, 15, 19, 23, 27)
PWM Channels 1, 2, 3 ,4   =  255  255    6  255 (C1 / C16= 0, 1, 2, ..., 15)
PWM Channels 5, 6, 7 ,8   =  255  255  255  255
PWM Channels 9,10,11,12   =  255  255  255  255
PWM Channels 13,14,15,16  =  255  255  255  255
Voltage 1, 2, 3, 4        =   26   27   28   29 (V1 / V4 = 26, 27, 28, 29)

Protocol is Sport (Frsky)
CRSF baudrate = 420000
Voltage parameters:
    Scales : 1.000000 , 1.000000 , 0.010000 , 0.100000 
    Offsets: 0.000000 , 0.000000 , 0.000000 , 0.000000 
    No temperature sensors are connected on V3 and V4
RPM multiplier = 1.000000
Baro sensor is detected using MS5611
    Sensitivity min = 100 (at 100)   , max = 300 (at 1000)
    Hysteresis = 5 
Acc/Gyro is detected using MP6050
     Acceleration offsets X, Y, Z = 0 , 0 , 0
     Gyro offsets         X, Y, Z = 0 , 0 , 0
First analog to digital sensor is detected using ads1115
    Measurement setup: 4 , 5 , 6 ,7
    Gains: 2 , 2 , 2 ,2
    Rates: 3 , 3 , 3 ,3
    Offsets: 0.000000 , 0.000000 , 0.000000 ,0.000000
    Scales: 1.000000 , 1.000000 , 1.000000 ,1.000000
    Averaged on: 10 , 10 , 10 ,10
Second analog to digital sensor is detected using ads1115
    Measurement setup: 4 , 5 , 6 ,7
    Gains: 2 , 2 , 2 ,2
    Rates: 3 , 3 , 3 ,3
    Offsets: 0.000000 , 0.000000 , 0.000000 ,0.000000
    Scales: 1.000000 , 1.000000 , 1.000000 ,1.000000
    Averaged on: 10 , 10 , 10 ,10
Foreseen GPS type is Ublox  :GPS is detected and has a fix
Failsafe uses predefined values
     Chan 1...4  = 976 , 992 , 992 , 992
     Chan 5...8  = 992 , 992 , 992 , 992
     Chan 9...12 = 992 , 992 , 992 , 992
     Chan 13...16= 992 , 992 , 992 , 992

Config parameters are OK
Press ? + Enter to get help about the commands

 >> 
processing cmd

Cmd to execute:   FV

GPS Latitude = xx.xxxxxxx degree
GPS Longitude = x.xxxxxxx degree
GPS Groundspeed = 5 cm/s
GPS Heading = 0.000000 degree
GPS Altitude = 4077 cm
GPS Num sat. = 110 cm
GPS Date J M A = 16 2 23 
GPS Time H M S = 9 47 1 
GPS Pdop = 229 
GPS Home bearing = 340 degree
GPS Home distance = 18 m
Volt 1 = 1676 mVolt
Current (Volt 2) = 1651 mA
Volt 3 = 10 mVolt
Volt 4 = 25 mVolt
Capacity (using current) = 0 mAh
Vspeed = 1 cm/s
Baro Rel altitude = 0 cm
Pitch = 72 degree
Roll = -47 degree
RPM = 996 Hertz
Ads 1 1 = 77 mVolt
Ads 1 2 = 77 mVolt
Ads 1 3 = 77 mVolt
Ads 1 4 = 77 mVolt
Ads 2 1 = 77 mVolt
Ads 2 2 = 77 mVolt
Ads 2 3 = 77 mVolt
Ads 2 4 = 77 mVolt

mstrens, 2023-02-16:
it is fixed in version 1.1.7 (ads 1 1 on handset)

Satcomix, 2023-02-16:
Hello Mstrens,
It works, now i see 8 ADS Sensors in my Handheld, that are 8+21+8=37 Telemetry values  (MONSTER-DISPLAY)
I must test the voltage, but i think i can close the issue.
Thank you very much.
Greetings,
Torsten

processing cmd

Cmd to execute:   

Version = 1.1.7 
    Function                Pin   Change entering XXX=yyy (yyy=255 to disable)
Primary channels input    =    5  (PRI     = 5, 9, 21, 25)
Secondary channels input  =    1  (SEC     = 1, 13, 17, 29)
Telemetry . . . . . . . . =    0  (TLM     = 0, 1, 2, ..., 29)
GPS Rx  . . . . . . . . . =   12  (GPS_RX  = 0, 1, 2, ..., 29)
GPS Tx  . . . . . . . . . =   13  (GPS_TX  = 0, 1, 2, ..., 29)
Sbus OUT  . . . . . . . . =    2  (SBUS_OUT= 0, 1, 2, ..., 29)
RPM   . . . . . . . . . . =    4  (RPM     = 0, 1, 2, ..., 29)
SDA (I2C sensors) . . . . =   14  (SDA     = 2, 6, 10, 14, 18, 22, 26)
SCL (I2C sensors) . . . . =   15  (SCL     = 3, 7, 11, 15, 19, 23, 27)
PWM Channels 1, 2, 3 ,4   =  255  255    6  255 (C1 / C16= 0, 1, 2, ..., 15)
PWM Channels 5, 6, 7 ,8   =  255  255  255  255
PWM Channels 9,10,11,12   =  255  255  255  255
PWM Channels 13,14,15,16  =  255  255  255  255
Voltage 1, 2, 3, 4        =   26   27   28   29 (V1 / V4 = 26, 27, 28, 29)

Protocol is Sport (Frsky)
CRSF baudrate = 420000
Voltage parameters:
    Scales : 1.000000 , 1.000000 , 0.010000 , 0.100000 
    Offsets: 0.000000 , 0.000000 , 0.000000 , 0.000000 
    No temperature sensors are connected on V3 and V4
RPM multiplier = 1.000000
Baro sensor is detected using MS5611
    Sensitivity min = 100 (at 100)   , max = 300 (at 1000)
    Hysteresis = 5 
Acc/Gyro is detected using MP6050
     Acceleration offsets X, Y, Z = 0 , 0 , 0
     Gyro offsets         X, Y, Z = 0 , 0 , 0
First analog to digital sensor is detected using ads1115
    Measurement setup: 4 , 5 , 6 ,7
    Gains: 1 , 1 , 1 ,1
    Rates: 3 , 3 , 3 ,3
    Offsets: 0.000000 , 0.000000 , 0.000000 ,0.000000
    Scales: 1.000000 , 1.000000 , 1.000000 ,1.000000
    Averaged on: 10 , 10 , 10 ,10
Second analog to digital sensor is detected using ads1115
    Measurement setup: 4 , 5 , 6 ,7
    Gains: 2 , 2 , 2 ,2
    Rates: 3 , 3 , 3 ,3
    Offsets: 0.000000 , 0.000000 , 0.000000 ,0.000000
    Scales: 1.000000 , 1.000000 , 1.000000 ,1.000000
    Averaged on: 10 , 10 , 10 ,10
Foreseen GPS type is Ublox  :GPS is detected and has a fix
Failsafe uses predefined values
     Chan 1...4  = 976 , 992 , 992 , 992
     Chan 5...8  = 992 , 992 , 992 , 992
     Chan 9...12 = 992 , 992 , 992 , 992
     Chan 13...16= 992 , 992 , 992 , 992

Config parameters are OK
Press ? + Enter to get help about the commands

 >> 
processing cmd

Cmd to execute:   FV

GPS Latitude = xx.xxxxxxx degree
GPS Longitude = x.xxxxxxx degree
GPS Groundspeed = 39 cm/s
GPS Heading = 181.840000 degree
GPS Altitude = 3987 cm
GPS Num sat. = 111 cm
GPS Date J M A = 16 2 23 
GPS Time H M S = 10 5 45 
GPS Pdop = 172 
GPS Home bearing = 325 degree
GPS Home distance = 54 m
Volt 1 = 1676 mVolt
Current (Volt 2) = 1651 mA
Volt 3 = 10 mVolt
Volt 4 = 25 mVolt
Capacity (using current) = 0 mAh
Vspeed = 28 cm/s
Baro Rel altitude = -4 cm
Pitch = 72 degree
Roll = -46 degree
RPM = 997 Hertz
Ads 1 1 = 120 mVolt
Ads 1 2 = 120 mVolt
Ads 1 3 = 120 mVolt
Ads 1 4 = 120 mVolt
Ads 2 1 = 77 mVolt
Ads 2 2 = 77 mVolt
Ads 2 3 = 77 mVolt
Ads 2 4 = 77 mVolt

mstrens, 2023-02-16:
Avoid having to many telemetry fields in practice. This reduces the refresh rate (not more that about 50 data/sec with frsky)

Satcomix, 2023-02-16:
Thanks for the Info, but i know about the problems with too much IDs, like Mike B. wrote in RCG and GitHub.
I hear it ,because the transmitter is lying next to me, while I'm doing a new build in PlatformIO and it keeps saying "sensor lost" from time to time.
Next time, i will make some test with the Kingst Logic Analyzer(200Mhz/16CH) to see more.
Thank you for your time and work.
greetings,
Torsten

Satcomix, 2023-02-16:
Hello Mstrens,
I have some problems with the ADS1115.
The telemetry Display at Handheld is ok with the 8 DIY Sensors, but "FV Terminal" show me peculiar values.
Normally you can use the 1115 for four single ended Ax_TO_GND or two differenzial A0_TO_A1 measurements.
When I only prog the first A0, i can measure at ADS1 and ADS2 with the correct values.
When i want the values from A0 and A1 or all from A0 to A3, i get wrong values, they are mixed, or doubled.
Greetings
Torsten

Satcomix, 2023-02-16:
some Tests:
MV4096 , Offset 0, Scale1, MS69, Average 1
Ads 1 1 = 3203 mVolt ---> A0 twice from A2 
Ads 1 3 = 1601 mVolt --->1605mV input
Ads 2 1 = 2044 mVolt ---> A0 twice from A2
Ads 2 3 = 1022 mVolt ---> 1024mV input

MV2048 , Offset 0, Scale1, MS69, Average 1
Ads 1 1 = 1022 mVolt ---->2048mV input
Ads 1 3 = 1022 mVolt ---->1022mV input
Ads 2 1 = 1601 mVolt ---->2048mV input
Ads 2 2 = 1601 mVolt ---->1601mV input

MV2048 , Offset 0, Scale1, MS69, Average 1
Ads 1 1 = 77 mVolt --->2048mV input
Ads 1 2 = 77 mVolt --->1605mV input
Ads 1 3 = 77 mVolt --->1024mV input
Ads 2 1 = 1601 mVolt-->2048mV input
Ads 2 2 = 1601 mVolt-->1605mV input
Ads 2 3 = 1601 mVolt-->1024mV input

mstrens, 2023-02-16:
Sorry but I do not understand.
Could you please provide just one test (with the number of canals you want from 1,2,3 or 4) for only one ADS that gives wrong values.
For this test, can you provide:
- the full setup used in the config.h files
- the real voltages applied on each ads1115 pin (A0,A1,A2,A3)
- the values provided by the FV command
- the values provided by the handset.
I can then better understand and analyse.

Satcomix, 2023-02-16:
Hello Mstrens,
I have no problems with the DIY Sensors in Handset.
When i use one channel on each ADS1115, for example only A0, or only A1,......on ADS1 and ADS2, the values in Terminal per FV match.
When i want to use two channels like A0 and A1on ADS1 and ADS2, you see the values in the post above.
Maybe i forget something to change in config.h, i dont know.

#define I2C_ADS_Add1 0x48 // I2C address of ads1115 when addr pin is connected to ground
#define I2C_ADS_Add2 0x49 // I2C address of ads1115 when addr pin is connected to vdd

#define ADS1_MEASURE ADS_OFF , A1_TO_GND , ADS_OFF , A3_TO_GND // select 4 values between A0_TO_A1, A0_TO_A3, A1_TO_A3, A2_TO_A3, A0_TO_GND, A1_TO_GND, A2_TO_GND, A3_TO_GND, ADS_OFF
#define ADS1_FULL_SCALE_VOLT  MV4096, MV4096, MV4096, MV4096 //  select between MV6144 MV4096 MV2048 MV1024 MV512 MV256
#define ADS1_OFFSET 0.0 , 0.0 , 0.0 , 0.0 // can be a float (positive or negative)
#define ADS1_SCALE 1.0, 1.0, 1.0, 1.0 // can be a float
#define ADS1_RATE  MS18 , MS18, MS18 , MS18 // select between MS137, MS69, MS35, MS18, MS9, MS5, MS3 , MS2
#define ADS1_AVERAGING_ON 10 , 10, 10, 10 // number of values used for averaging (must be between 1 and 254) 
//#define ADS_CURRENT_BASED_ON ADS_VOLT_1  // uncomment if current, and comsumption have to be calculated based on one of ADS voltage measurement; select then the voltage to be used between ADS_VOLT_1, ADS_VOLT_2, ADS_VOLT_3, ADS_VOLT_4
//#define ADS_AIRSPEED_BASED_ON ADS_VOLT_1  // uncomment if airspeed (and dte) have to be calculated based on one of ADS voltage measurement ; select then the voltage to be used between ADS_VOLT_1, ADS_VOLT_2, ADS_VOLT_3, ADS_VOLT_4

#define ADS2_MEASURE ADS_OFF ,  A1_TO_GND , ADS_OFF , A3_TO_GND // select 4 values between A0_TO_A1, A0_TO_A3, A1_TO_A3, A2_TO_A3, A0_TO_GND, A1_TO_GND, A2_TO_GND, A3_TO_GND, ADS_OFF
#define ADS2_FULL_SCALE_VOLT  MV4096, MV4096, MV4096, MV4096 //  select between MV6144 MV4096 MV2048 MV1024 MV512 MV256
#define ADS2_OFFSET 0.0 , 0.0 , 0.0 , 0.0 // can be a float (positive or negative)
#define ADS2_SCALE 1.0, 1.0, 1.0, 1.0 // can be a float
#define ADS2_RATE  MS18 , MS18, MS18 , MS18 // select between MS137, MS69, MS35, MS18, MS9, MS5, MS3 , MS2
#define ADS2_AVERAGING_ON 10 , 10, 10, 10 // number of values used for averaging (must be between 1 and 254) 
//#define ADS_CURRENT_BASED_ON ADS_VOLT_1  // uncomment if current, and comsumption have to be calculated based on one of ADS voltage measurement; select then the voltage to be used between ADS_VOLT_1, ADS_VOLT_2, ADS_VOLT_3, ADS_VOLT_4
//#define ADS_AIRSPEED_BASED_ON ADS_VOLT_1  // uncomment if airspeed (and dte) have to be calculated based on one of ADS voltage measurement ; select then the voltage to be used between ADS_VOLT_1, ADS_VOLT_2, ADS_VOLT_3, ADS_VOLT_4

Satcomix, 2023-02-16:
processing cmd

Cmd to execute:   

Version = 1.1.7 
    Function                Pin   Change entering XXX=yyy (yyy=255 to disable)
Primary channels input    =    5  (PRI     = 5, 9, 21, 25)
Secondary channels input  =    1  (SEC     = 1, 13, 17, 29)
Telemetry . . . . . . . . =    0  (TLM     = 0, 1, 2, ..., 29)
GPS Rx  . . . . . . . . . =   12  (GPS_RX  = 0, 1, 2, ..., 29)
GPS Tx  . . . . . . . . . =   13  (GPS_TX  = 0, 1, 2, ..., 29)
Sbus OUT  . . . . . . . . =    2  (SBUS_OUT= 0, 1, 2, ..., 29)
RPM   . . . . . . . . . . =    4  (RPM     = 0, 1, 2, ..., 29)
SDA (I2C sensors) . . . . =   14  (SDA     = 2, 6, 10, 14, 18, 22, 26)
SCL (I2C sensors) . . . . =   15  (SCL     = 3, 7, 11, 15, 19, 23, 27)
PWM Channels 1, 2, 3 ,4   =  255  255    6  255 (C1 / C16= 0, 1, 2, ..., 15)
PWM Channels 5, 6, 7 ,8   =  255  255  255  255
PWM Channels 9,10,11,12   =  255  255  255  255
PWM Channels 13,14,15,16  =  255  255  255  255
Voltage 1, 2, 3, 4        =   26   27   28   29 (V1 / V4 = 26, 27, 28, 29)

Protocol is Sport (Frsky)
CRSF baudrate = 420000
Voltage parameters:
    Scales : 1.000000 , 1.000000 , 0.010000 , 0.100000 
    Offsets: 0.000000 , 0.000000 , 0.000000 , 0.000000 
    No temperature sensors are connected on V3 and V4
RPM multiplier = 1.000000
Baro sensor is detected using MS5611
    Sensitivity min = 100 (at 100)   , max = 300 (at 1000)
    Hysteresis = 5 
Acc/Gyro is detected using MP6050
     Acceleration offsets X, Y, Z = 3068 , 350 , 1000
     Gyro offsets         X, Y, Z = -4 , 41 , 49
First analog to digital sensor is detected using ads1115
    Measurement setup: 8 , 5 , 8 ,7
    Gains: 1 , 1 , 1 ,1
    Rates: 3 , 3 , 3 ,3
    Offsets: 0.000000 , 0.000000 , 0.000000 ,0.000000
    Scales: 1.000000 , 1.000000 , 1.000000 ,1.000000
    Averaged on: 10 , 10 , 10 ,10
Second analog to digital sensor is detected using ads1115
    Measurement setup: 8 , 5 , 8 ,7
    Gains: 1 , 1 , 1 ,1
    Rates: 3 , 3 , 3 ,3
    Offsets: 0.000000 , 0.000000 , 0.000000 ,0.000000
    Scales: 1.000000 , 1.000000 , 1.000000 ,1.000000
    Averaged on: 10 , 10 , 10 ,10
Foreseen GPS type is Ublox  :GPS is detected and has a fix
Failsafe uses predefined values
     Chan 1...4  = 976 , 992 , 992 , 992
     Chan 5...8  = 992 , 992 , 992 , 992
     Chan 9...12 = 992 , 992 , 992 , 992
     Chan 13...16= 992 , 992 , 992 , 992

Config parameters are OK
Press ? + Enter to get help about the commands

 >> 
processing cmd

Cmd to execute:   FV

GPS Latitude = 51.1225184 degree
GPS Longitude = 6.7173224 degree
GPS Groundspeed = 10 cm/s
GPS Heading = 122.910000 degree
GPS Altitude = 3470 cm
GPS Num sat. = 111 cm
GPS Date J M A = 16 2 23 
GPS Time H M S = 19 12 24 
GPS Pdop = 198 
GPS Home bearing = 152 degree
GPS Home distance = 11 m
Volt 1 = 535 mVolt
Current (Volt 2) = 568 mA
Volt 3 = 6 mVolt
Volt 4 = 19 mVolt
Capacity (using current) = 0 mAh
Vspeed = 0 cm/s
Baro Rel altitude = -43 cm
Pitch = 9 degree
Roll = -7 degree
RPM = 1005 Hertz
Ads 1 2 = 1602 mVolt
Ads 1 4 = 1602 mVolt
Ads 2 2 = 2042 mVolt
Ads 2 4 = 2042 mVolt
processing cmd

Cmd to execute:   FV

GPS Latitude = xxxxxxxxxx degree
GPS Longitude = xxxxxxxxx degree
GPS Groundspeed = 12 cm/s
GPS Heading = 8.860000 degree
GPS Altitude = 4730 cm
GPS Num sat. = 112 cm
GPS Date J M A = 16 2 23 
GPS Time H M S = 19 22 25 
GPS Pdop = 167 
GPS Home bearing = 124 degree
GPS Home distance = 7 m
Volt 1 = 538 mVolt
Current (Volt 2) = 568 mA
Volt 3 = 6 mVolt
Volt 4 = 19 mVolt
Capacity (using current) = 9 mAh
Vspeed = 9 cm/s
Baro Rel altitude = -89 cm
Pitch = 10 degree
Roll = -7 degree
RPM = 997 Hertz
Ads 1 2 = 1602 mVolt
Ads 1 4 = 1602 mVolt
Ads 2 2 = 2042 mVolt
Ads 2 4 = 2042 mVolt

Satcomix, 2023-02-16:
complete config.h
i renamed it in .txt
[config.txt](https://github.com/mstrens/oXs_on_RP2040/files/10760394/config.txt)

mstrens, 2023-02-16:
OK. I see that ads 1 4 and ads 2 4 are not correct.
Could you make a small change to ads1115.cpp.
There is a line with
    dataToWrite[2] = ( ads_Rate[ads_idx][ads_CurrentIdx] << 5 | 0B11 );
 
Could you add one line just after.
   printf("config %x %x %x\n",ads_CurrentIdx , dataToWrite[1] , dataToWrite[2]);

Please run the test with only the first ads1115 (so disconnect the second one) and the same setup as before.
On the PC you should get a list of messages beginning with "config".
Make a copy/paste of about 20 lines (I expect it is always the same messages).
Thanks.

Satcomix, 2023-02-17:
Good morning Mstrens,
Here is the list as requested. The ports are queried in sequence, seems ok.
config 0 c3 63
config 1 d3 63
config 2 e3 63
config 3 f3 63
config 0 c3 63
config 1 d3 63
config 2 e3 63
config 3 f3 63
config 0 c3 63
config 1 d3 63
config 2 e3 63
config 3 f3 63
config 0 c3 63
config 1 d3 63
config 2 e3 63
config 3 f3 63
config 0 c3 63
config 1 d3 63
config 2 e3 63
config 3 f3 63
config 0 c3 63
config 1 d3 63
config 2 e3 63
config 3 f3 63
config 0 c3 63
config 1 d3 63
config 2 e3 63
config 3 f3 63
config 0 c3 63
config 1 d3 63
config 2 e3 63
config 3 f3 63
config 0 c3 63
config 1 d3 63
config 2 e3 63
config 3 f3 63
config 0 c3 63
config 1 d3 63
config 2 e3 63
config 3 f3 63
config 0 c3 63
config 1 d3 63
config 2 e3 63
config 3 f3 63

Satcomix, 2023-02-17:
**Only one ADS1115 and one GY86 at I2C1**

processing cmd

Cmd to execute:   

Version = 1.1.7 
    Function                Pin   Change entering XXX=yyy (yyy=255 to disable)
Primary channels input    =    5  (PRI     = 5, 9, 21, 25)
Secondary channels input  =    1  (SEC     = 1, 13, 17, 29)
Telemetry . . . . . . . . =    0  (TLM     = 0, 1, 2, ..., 29)
GPS Rx  . . . . . . . . . =   12  (GPS_RX  = 0, 1, 2, ..., 29)
GPS Tx  . . . . . . . . . =   13  (GPS_TX  = 0, 1, 2, ..., 29)
Sbus OUT  . . . . . . . . =    2  (SBUS_OUT= 0, 1, 2, ..., 29)
RPM   . . . . . . . . . . =    4  (RPM     = 0, 1, 2, ..., 29)
SDA (I2C sensors) . . . . =   14  (SDA     = 2, 6, 10, 14, 18, 22, 26)
SCL (I2C sensors) . . . . =   15  (SCL     = 3, 7, 11, 15, 19, 23, 27)
PWM Channels 1, 2, 3 ,4   =  255  255    6  255 (C1 / C16= 0, 1, 2, ..., 15)
PWM Channels 5, 6, 7 ,8   =  255  255  255  255
PWM Channels 9,10,11,12   =  255  255  255  255
PWM Channels 13,14,15,16  =  255  255  255  255
Voltage 1, 2, 3, 4        =   26   27   28   29 (V1 / V4 = 26, 27, 28, 29)

Protocol is Sport (Frsky)
CRSF baudrate = 420000
Voltage parameters:
    Scales : 1.000000 , 1.000000 , 0.010000 , 0.100000 
    Offsets: 0.000000 , 0.000000 , 0.000000 , 0.000000 
    No temperature sensors are connected on V3 and V4
RPM multiplier = 1.000000
Baro sensor is not detected
Acc/Gyro is detected using MP6050
     Acceleration offsets X, Y, Z = 3068 , 350 , 1000
     Gyro offsets         X, Y, Z = -4 , 41 , 49
First analog to digital sensor is not detected
Second analog to digital sensor is detected using ads1115
    Measurement setup: 4 , 5 , 6 ,7
    Gains: 1 , 1 , 1 ,1
    Rates: 3 , 3 , 3 ,3
    Offsets: 0.000000 , 0.000000 , 0.000000 ,0.000000
    Scales: 1.000000 , 1.000000 , 1.000000 ,1.000000
    Averaged on: 10 , 10 , 10 ,10
Foreseen GPS type is Ublox  :GPS is detected and has a fix
Failsafe uses predefined values
     Chan 1...4  = 976 , 992 , 992 , 992
     Chan 5...8  = 992 , 992 , 992 , 992
     Chan 9...12 = 992 , 992 , 992 , 992
     Chan 13...16= 992 , 992 , 992 , 992

Config parameters are OK
Press ? + Enter to get help about the commands

 >> 
processing cmd

Cmd to execute:   FV

GPS Latitude = xx.xxxxxxxx degree
GPS Longitude = x.xxxxxxxxx degree
GPS Groundspeed = 74 cm/s
GPS Heading = 273.570000 degree
GPS Altitude = 5193 cm
GPS Num sat. = 110 cm
GPS Date J M A = 17 2 23 
GPS Time H M S = 6 48 56 
GPS Pdop = 172 
GPS Home bearing = 333 degree
GPS Home distance = 3 m
Volt 1 = 549 mVolt
Current (Volt 2) = 582 mA
Volt 3 = 6 mVolt
Volt 4 = 17 mVolt
Capacity (using current) = 0 mAh
Pitch = 1 degree
Roll = -12 degree
RPM = 0 Hertz
**Ads 2 1 = 1595 mVolt ---->1021mV Input
Ads 2 2 = 1595 mVolt ---->1600mV Input
Ads 2 3 = 1595 mVolt ---->2041mV Input
Ads 2 4 = 1595 mVolt ---->1028mV Input**

**If i put only one Port to measurement, only A0 than i get the right result A0=1020mV**

Satcomix, 2023-02-17:
**Next Tests with one ADS1115:**

**1.Test: Only A0_TO_GND**
Roll = -13 degree
RPM = 0 Hertz
Ads 1 1 = **1018 mVolt**

**2.Test: A0 and A1**
Roll = -13 degree
RPM = 0 Hertz
Ads 1 1 = 1595 mVolt
Ads 1 2 = 1595 mVolt

**3.Test: Only A1_TO_GND**
Pitch = 1 degree
Roll = -13 degree
RPM = 0 Hertz
Ads 1 2 = **1595 mVolt**

**4.Test: Only A1 and A2**
Pitch = 1 degree
Roll = -13 degree
RPM = 0 Hertz
Ads 1 2 = 1595 mVolt
Ads 1 3 = 1595 mVolt

**5.Test: Only A2**
Pitch = 1 degree
Roll = -13 degree
RPM = 0 Hertz
Ads 1 3 = **2035 mVolt**

**6.Test: Only A2 and A3**
Pitch = 1 degree
Roll = -13 degree
RPM = 0 Hertz
Ads 1 3 = 2035 mVolt
Ads 1 4 = 2035 mVolt

**7.Test: Only A3**
Pitch = 1 degree
Roll = -13 degree
RPM = 0 Hertz
Ads 1 4 = **1024 mVolt**

mstrens, 2023-02-17:
I do not see the bug.
can you try adding one more line of code:
There is already (in ads1115.cpp)
ads_Value[ads_CurrentIdx] = round( ((float) ads_SumOfConv[ads_CurrentIdx] / (float) ads_MaxCount[ads_idx][ads_CurrentIdx] * adcToMvoltScaling ) * ads_Scale[ads_idx][ads_CurrentIdx] ) + ads_Offset[ads_idx][ads_CurrentIdx];
                       
Can you add just after this line (and test again using only one ADS)
printf("Adc=%d pin=%d : %d mVolt at%d\n", ads_idx , ads_CurrentIdx , ads_Value[ads_CurrentIdx], millis());

Satcomix, 2023-02-17:
**Next Test with A2_TO_A3
A2=2035mV Input
A3=1024mV Input**

**Test: Only A2**
Pitch = 1 degree
Roll = -13 degree
RPM = 0 Hertz
Ads 1 3 = **2035 mVolt**

**Test: Only A2 and A3**
Pitch = 1 degree
Roll = -13 degree
RPM = 0 Hertz
Ads 1 3 = 2035 mVolt
Ads 1 4 = 2035 mVolt

**Test: Only A3**
Pitch = 1 degree
Roll = -13 degree
RPM = 0 Hertz
Ads 1 4 = **1024 mVolt**

**Test: Only A0_TO_A1**
Pitch = 1 degree
Roll = -13 degree
RPM = 0 Hertz
Ads 1 1 = **-577 mVolt**

Test A0_TO_A1 and A2_TO_A3
Pitch = 1 degree
Roll = -13 degree
RPM = 0 Hertz
Ads 1 1 = 1012 mVolt
Ads 1 3 = 1012 mVolt

**Test: Only A2_TO_A3**
Pitch = 1 degree
Roll = -14 degree
RPM = 0 Hertz
Ads 1 3 = **1011 mVolt**

Adc=0 pin=2 : 1008 mVolt at2906
Adc=0 pin=2 : 1008 mVolt at3096
Adc=0 pin=2 : 1008 mVolt at3291
Adc=0 pin=2 : 1008 mVolt at3481
Adc=0 pin=2 : 1008 mVolt at3672
Adc=0 pin=2 : 1008 mVolt at3866
Adc=0 pin=2 : 1008 mVolt at4056
Adc=0 pin=2 : 1008 mVolt at4250
Adc=0 pin=2 : 1008 mVolt at4444
Adc=0 pin=2 : 1008 mVolt at4635
Adc=0 pin=2 : 1008 mVolt at4825
Adc=0 pin=2 : 1008 mVolt at5016
Adc=0 pin=2 : 1008 mVolt at5216
Adc=0 pin=2 : 1008 mVolt at5407
Adc=0 pin=2 : 1008 mVolt at5599
Adc=0 pin=2 : 1008 mVolt at5790
Adc=0 pin=2 : 1008 mVolt at5985
Adc=0 pin=2 : 1008 mVolt at6181
Adc=0 pin=2 : 1008 mVolt at6371
Adc=0 pin=2 : 1008 mVolt at6562
Adc=0 pin=2 : 1008 mVolt at6755
Adc=0 pin=2 : 1008 mVolt at6945
Adc=0 pin=2 : 1008 mVolt at7135
Adc=0 pin=2 : 1008 mVolt at7325
Adc=0 pin=2 : 1008 mVolt at7518
Adc=0 pin=2 : 1008 mVolt at7708
Adc=0 pin=2 : 1009 mVolt at7901
Adc=0 pin=2 : 1008 mVolt at8091
Adc=0 pin=2 : 1008 mVolt at8283
Adc=0 pin=2 : 1008 mVolt at8475

Satcomix, 2023-02-17:
When i make only one value measurement,
A0 only,
A1 only,
A2 only,
A3 only,
A0_TOA1,
A2_TO_A3,
everything is ok!!!!
But when i measure two Ports, than the second value get the first.

mstrens, 2023-02-17:
can you provide the trace with Adc= XXXXXXXXXXX when it runs with a set up that requires 4 conversions on only one ADS?

Satcomix, 2023-02-17:
If I understand you correctly, should I activate A0, A1, A2, A3 and show them via VF Terminal???

mstrens, 2023-02-17:
As an additional test, can you also put this line as comment
        ads_requestNextConv() ; // new request for conversion in case of error 
Take care that there are 2 lines with ads_requestNextConv(); The line to put as comment is the second one (having the comment "// new request for conversion in case of error")

Satcomix, 2023-02-17:
Next Test with A0_TO_GND  to A3_TO_GND
with Trace....
Attention: setup on core 1 did not ended within timeout
Attention: setup on core 1 did not ended within timeout
Attention: setup on core 1 did not ended within timeout
Attention: setup on core 1 did not ended within timeout
Attention: setup on core 1 did not ended within timeout
Attention: setup on core 1 did not ended within timeout
Attention: setup on core 1 did not ended within timeout
Attention: setup on core 1 did not ended within timeout
Attention: setup on core 1 did not ended within timeout
Attention: setup on core 1 did not ended within timeout
Attention: setup on core 1 did not ended within timeout
Attention: setup on core 1 did not ended within timeout
Attention: setup on core 1 did not ended within timeout
Attention: setup on core 1 did not ended within timeout
Attention: setup on core 1 did not ended within timeout
Attention: setup on core 1 did not ended within timeout
Attention: setup on core 1 did not ended within timeout
Attention: setup on core 1 did not ended within timeout
Attention: setup on core 1 did not ended within timeout
Attention: setup on core 1 did not ended within timeout
Attention: setup on core 1 did not ended within timeout

mstrens, 2023-02-17:
If I understand you correctly, should I activate A0, A1, A2, A3 and show them via VF Terminal???

Yes activate A0,, A1, A2, A3 + keep adding the line printf("Adc=%d pin=%d : %d mVolt at%d\n", ads_idx , ads_CurrentIdx , ads_Value[ads_CurrentIdx], millis());
And then no need to make FV but provide some lines of debugging like those:
Adc=0 pin=2 : 1008 mVolt at5216
Adc=0 pin=2 : 1008 mVolt at5407
Adc=0 pin=2 : 1008 mVolt at5599
Adc=0 pin=2 : 1008 mVolt at5790
Adc=0 pin=2 : 1008 mVolt at5985
Adc=0 pin=2 : 1008 mVolt at6181
Adc=0 pin=2 : 1008 mVolt at6371
Adc=0 pin=2 : 1008 mVolt at6562
Adc=0 pin=2 : 1008 mVolt at6755
Adc=0 pin=2 : 1008 mVolt at6945
Adc=0 pin=2 : 1008 mVolt at7135

mstrens, 2023-02-17:
To avoid all the "Atttention" messages, you can change one line in main.cpp
There is 
if ((micros() - setup1StartUs) > 2000000) {
            printf("Attention: setup on core 1 did not ended within timeout\n");
            continue;
        }

You can change "continue" by "break"

Satcomix, 2023-02-17:
As you see, i activate A0 to A3 with the your line in the script.

Im searching the right line for comment.
Line 107
or
Line 121
in ads1115.ccp

    #endif
                    }
                    ads_requestNextConv() ;
                    return true ;
                } else {
                    printf("Read error on ads1115\n");
                }
            }
        }    
//    resetI2c() ;
    ads_requestNextConv() ; // new request for conversion in case of error 
    }
    return false ;
} // end of readSensor

void ADS1115::ads_requestNextConv(void) {

mstrens, 2023-02-17:
The line to be put as comment is the one that follows //resetI2c();

Satcomix, 2023-02-17:
I command the Line, but....
Attention: setup on core 1 did not ended within timeout
Attention: setup on core 1 did not ended within timeout
Attention: setup on core 1 did not ended within timeout
Attention: setup on core 1 did not ended within timeout
Attention: setup on core 1 did not ended within timeout
Attention: setup on core 1 did not ended within timeout
Attention: setup on core 1 did not ended within timeout
Attention: setup on core 1 did not ended within timeout
Attention: setup on core 1 did not ended within timeout
Attention: setup on core 1 did not ended within timeout

**I think i put a fresh Ver.1.0.7 in Platformio Projects!**

mstrens, 2023-02-17:
I will put a new version on github if you want even if I did not yet find the bug.
So you have nothing to change in the code.
It will take 5 min

Satcomix, 2023-02-17:
I make a new build , but....
Attention: setup on core 1 did not ended within timeout
Attention: setup on core 1 did not ended within timeout
Attention: setup on core 1 did not ended within timeout
Attention: setup on core 1 did not ended within timeout
Attention: setup on core 1 did not ended within timeout
I will nuke the board and see....
Than i will put your new Version 1.1.8 on it.
Greetings 
Torsten

mstrens, 2023-02-17:
I put version 1.1.8.
Normally you can use it without changes (I added lines to get some debug messages).
I reorganized also the code for ads1115.
I activated already the request to make conversion for A0,A1,A2 and A4 to grnd.

Satcomix, 2023-02-17:
I nuke the board and put the new version on it.

request 1 d5 23 idx=1 at16025
request 2 e5 23 idx=2 at16095
Adc=0 pin=2 : -1 mVolt at16095
request 3 f5 23 idx=3 at16165
Adc=0 pin=3 : -1 mVolt at16165
request 0 85 23 idx=0 at16235
Adc=0 pin=0 : -1 mVolt at16235
request 1 d5 23 idx=1 at16305
Adc=0 pin=1 : -1 mVolt at16305
request 2 e5 23 idx=2 at16375
request 3 f5 23 idx=3 at16446
request 0 85 23 idx=0 at16516
request 1 d5 23 idx=1 at16586
request 2 e5 23 idx=2 at16656
request 3 f5 23 idx=3 at16726
request 0 85 23 idx=0 at16796
request 1 d5 23 idx=1 at16866
request 2 e5 23 idx=2 at16936
request 3 f5 23 idx=3 at17007
request 0 85 23 idx=0 at17077
request 1 d5 23 idx=1 at17147
request 2 e5 23 idx=2 at17217
request 3 f5 23 idx=3 at17287
request 0 85 23 idx=0 at17357
request 1 d5 23 idx=1 at17428
request 2 e5 23 idx=2 at17498
request 3 f5 23 idx=3 at17568
request 0 85 23 idx=0 at17638
request 1 d5 23 idx=1 at17708
request 2 e5 23 idx=2 at17778
request 3 f5 23 idx=3 at17848
request 0 85 23 idx=0 at17918
request 1 d5 23 idx=1 at17988

mstrens, 2023-02-17:
I think there was no need to change the config.h 
I already did it

Satcomix, 2023-02-17:
Adc=0 pin=2 : 1596 mVolt at32927
request 3 f5 23 idx=3 at32997
Adc=0 pin=3 : 1596 mVolt at32997
request 0 85 23 idx=0 at33067
Adc=0 pin=0 : 1596 mVolt at33067
request 1 d5 23 idx=1 at33137
Adc=0 pin=1 : 1596 mVolt at33137

**At:
Pin0 A0 1020mV input
Pin1 A1 1600mV input
Pin2 A2 2040mV input
Pin3 A3 1030mV input**

mstrens, 2023-02-17:
So, it requests a new conversion every 70 msec (that is OK with my config using MS69).
And still the values (3 on 4) are wrong.
I do not see why.
Anyway there is clearly an issue.
The good new is that I found back my ads1115 (not used in the last 8 years I think) and so I can now make some more tests on my own side.
I will come back when I found the solution.
Thanks a lot for all your time for those tests.

Satcomix, 2023-02-17:
**I have to thank you for all the work and effort that you put into this wonderful project for all of us.**
I like testing and learning.
Maybe my two ADS1115 have an error, who knows, I bought them in Germany from a well-known dealer, but who knows....
Greetings,
Torsten

mstrens, 2023-02-17:
I do not think that your ads1115 have an error.
I could reproduce the issue.
I made some changes and it seems ok on my side.
Can you try again (version 1.1.9 on github)

Satcomix, 2023-02-17:
Some more problems!
Sorry
processing cmd

Cmd to execute:   

Version = 1.1.9 
    Function                Pin   Change entering XXX=yyy (yyy=255 to disable)
Primary channels input    =    5  (PRI     = 5, 9, 21, 25)
Secondary channels input  =    1  (SEC     = 1, 13, 17, 29)
Telemetry . . . . . . . . =    0  (TLM     = 0, 1, 2, ..., 29)
GPS Rx  . . . . . . . . . =   12  (GPS_RX  = 0, 1, 2, ..., 29)
GPS Tx  . . . . . . . . . =   13  (GPS_TX  = 0, 1, 2, ..., 29)
Sbus OUT  . . . . . . . . =    2  (SBUS_OUT= 0, 1, 2, ..., 29)
RPM   . . . . . . . . . . =    4  (RPM     = 0, 1, 2, ..., 29)
SDA (I2C sensors) . . . . =   14  (SDA     = 2, 6, 10, 14, 18, 22, 26)
SCL (I2C sensors) . . . . =   15  (SCL     = 3, 7, 11, 15, 19, 23, 27)
PWM Channels 1, 2, 3 ,4   =  255  255    6  255 (C1 / C16= 0, 1, 2, ..., 15)
PWM Channels 5, 6, 7 ,8   =  255  255  255  255
PWM Channels 9,10,11,12   =  255  255  255  255
PWM Channels 13,14,15,16  =  255  255  255  255
Voltage 1, 2, 3, 4        =   26   27   28   29 (V1 / V4 = 26, 27, 28, 29)

Protocol is Sport (Frsky)
CRSF baudrate = 420000
Voltage parameters:
    Scales : 1.000000 , 1.000000 , 1.000000 , 1.000000 
    Offsets: 0.000000 , 0.000000 , 0.000000 , 0.000000 
    No temperature sensors are connected on V3 and V4
RPM multiplier = 1.000000
Baro sensor is detected using MS5611
    Sensitivity min = 100 (at 100)   , max = 300 (at 1000)
    Hysteresis = 5 
Acc/Gyro is detected using MP6050
     Acceleration offsets X, Y, Z = 3024 , 356 , 1006
     Gyro offsets         X, Y, Z = -3 , 41 , 48
First analog to digital sensor is detected using ads1115
    Measurement setup: 4 , 5 , 6 ,7
    Gains: 2 , 2 , 2 ,2
    Rates: 1 , 1 , 1 ,1
    Offsets: 0.000000 , 0.000000 , 0.000000 ,0.000000
    Scales: 1.000000 , 1.000000 , 1.000000 ,1.000000
    Averaged on: 10 , 10 , 10 ,10
Second analog to digital sensor is not detected
Foreseen GPS type is Ublox  :GPS is detected and has a fix
Failsafe uses predefined values
     Chan 1...4  = 992 , 992 , 992 , 992
     Chan 5...8  = 992 , 992 , 992 , 992
     Chan 9...12 = 992 , 992 , 992 , 992
     Chan 13...16= 992 , 992 , 992 , 992

Config parameters are OK
Press ? + Enter to get help about the commands

 >> 
processing cmd

Cmd to execute:   FV

GPS Latitude = xxxxxxxxxx degree
GPS Longitude = xxxxxxxxx degree
GPS Groundspeed = 9 cm/s
GPS Heading = 174.760000 degree
GPS Altitude = 3702 cm
GPS Num sat. = 111 cm
GPS Date J M A = 17 2 23 
GPS Time H M S = 11 18 56 
GPS Pdop = 177 
GPS Home bearing = 286 degree
GPS Home distance = 11 m
Volt 1 = 541 mVolt
Current (Volt 2) = 570 mA
Volt 3 = 619 mVolt
Volt 4 = 193 mVolt
Capacity (using current) = 1 mAh
Vspeed = 0 cm/s
Baro Rel altitude = -7 cm
Pitch = 4 degree
Roll = 1 degree
RPM = 996 Hertz
**Ads 1 1 = -1 mVolt ---->1021mV Input
Ads 1 2 = -1 mVolt ---->1600mV Input
Ads 1 3 = -1 mVolt ---->2040mV Input
Ads 1 4 = -1 mVolt ---->1038mV Input**

Satcomix, 2023-02-17:
I have build a new version with MV4096, after that the FV Terminal command respond nothing???

mstrens, 2023-02-17:
did you make a power down of all sensors (it happens that I2C is locked).
Here what I get:
It seems me OK (A0 connected to Vcc, A1 connected to grnd, the 2 others are floating.

Function                Pin   Change entering XXX=yyy (yyy=255 to disable)
Primary channels input    =  255  (PRI     = 5, 9, 21, 25)
Secondary channels input  =  255  (SEC     = 1, 13, 17, 29)
Telemetry . . . . . . . . =    0  (TLM     = 0, 1, 2, ..., 29)
GPS Rx  . . . . . . . . . =  255  (GPS_RX  = 0, 1, 2, ..., 29)
GPS Tx  . . . . . . . . . =  255  (GPS_TX  = 0, 1, 2, ..., 29)
Sbus OUT  . . . . . . . . =  255  (SBUS_OUT= 0, 1, 2, ..., 29)
RPM   . . . . . . . . . . =   10  (RPM     = 0, 1, 2, ..., 29)
SDA (I2C sensors) . . . . =    2  (SDA     = 2, 6, 10, 14, 18, 22, 26)
SCL (I2C sensors) . . . . =    3  (SCL     = 3, 7, 11, 15, 19, 23, 27)
PWM Channels 1, 2, 3 ,4   =  255  255  255  255 (C1 / C16= 0, 1, 2, ..., 15)
PWM Channels 5, 6, 7 ,8   =  255  255  255  255
PWM Channels 9,10,11,12   =  255  255  255  255
PWM Channels 13,14,15,16  =  255  255  255  255
Voltage 1, 2, 3, 4        =   26   27   28   29 (V1 / V4 = 26, 27, 28, 29)

Protocol is Sport (Frsky)
CRSF baudrate = 115200
Voltage parameters:
    Scales : 1.000000 , 1.000000 , 1.000000 , 1.000000
    Offsets: 0.000000 , 0.000000 , 0.000000 , 0.000000
    No temperature sensors are connected on V3 and V4
RPM multiplier = 1.000000
Baro sensor is detected using MS5611
    Sensitivity min = 100 (at 100)   , max = 300 (at 1000)
    Hysteresis = 5
Acc/Gyro is detected using MP6050
     Acceleration offsets X, Y, Z = 308 , 15 , 1170
     Gyro offsets         X, Y, Z = 69 , -15 , -11
First analog to digital sensor is detected using ads1115
    Measurement setup: 4 , 5 , 6 ,7
    Gains: 1 , 1 , 1 ,1
    Rates: 1 , 1 , 1 ,1
    Offsets: 0.000000 , 0.000000 , 0.000000 ,0.000000
    Scales: 1.000000 , 1.000000 , 1.000000 ,1.000000
    Averaged on: 10 , 10 , 10 ,10
Second analog to digital sensor is not detected
Foreseen GPS type is Ublox  :GPS is not (yet) detected
Failsafe type is HOLD

Config parameters are OK
Press ? + Enter to get help about the commands

 >>
processing cmd

Cmd to execute:   FV

Volt 1 = 676 mVolt
Current (Volt 2) = 656 mA
Volt 3 = 655 mVolt
Volt 4 = 609 mVolt
Capacity (using current) = 1 mAh
Vspeed = 0 cm/s
Baro Rel altitude = 30 cm
Pitch = 4 degree
Roll = -178 degree
RPM = 0 Hertz
Ads 1 1 = 3305 mVolt
Ads 1 2 = 0 mVolt
Ads 1 3 = 600 mVolt
Ads 1 4 = 601 mVolt

mstrens, 2023-02-17:
Just one comment: in your last test, it does not seems that you used MV4096 because gain was reported as 2 instead of 1.
Anyway this can't explain why you did not got a reply to FV command

Satcomix, 2023-02-17:
I take the original Ver.1.1.9 on Board.
3.3V to A0
GND to A1

Ads 1 1 = 0 mVolt
Ads 1 2 = 0 mVolt
Ads 1 3 = 0 mVolt
Ads 1 4 = 0 mVolt

Satcomix, 2023-02-17:
> Just one comment: in your last test, it does not seems that you used MV4096 because gain was reported as 2 instead of 1. Anyway this can't explain why you did not got a reply to FV command
**I use MV4096 for the Terminal output!**
#define MV6144 0 
#define MV4096 1
#define MV2048 2
#define MV1024 3
#define MV512 4
#define MV256 5

mstrens, 2023-02-17:
I do not understand what happens on your board.
With my board all seems OK.

When you change to MV4096 and ask for the current config (ENTER), do you get a line with 
Gains: 1 , 1 , 1 ,1

Satcomix, 2023-02-17:
Change to MV4096 A0_TO_GND, ADS_OFF,ADS_OFF, ADS_OFF
I want only A0, but nothing.
Also no output to TLM,
No Value at Handheld, after Power down several times.

processing cmd

Cmd to execute:   

Version = 1.1.9 
    Function                Pin   Change entering XXX=yyy (yyy=255 to disable)
Primary channels input    =    5  (PRI     = 5, 9, 21, 25)
Secondary channels input  =    1  (SEC     = 1, 13, 17, 29)
Telemetry . . . . . . . . =    0  (TLM     = 0, 1, 2, ..., 29)
GPS Rx  . . . . . . . . . =   12  (GPS_RX  = 0, 1, 2, ..., 29)
GPS Tx  . . . . . . . . . =   13  (GPS_TX  = 0, 1, 2, ..., 29)
Sbus OUT  . . . . . . . . =    2  (SBUS_OUT= 0, 1, 2, ..., 29)
RPM   . . . . . . . . . . =    4  (RPM     = 0, 1, 2, ..., 29)
SDA (I2C sensors) . . . . =   14  (SDA     = 2, 6, 10, 14, 18, 22, 26)
SCL (I2C sensors) . . . . =   15  (SCL     = 3, 7, 11, 15, 19, 23, 27)
PWM Channels 1, 2, 3 ,4   =  255  255    6  255 (C1 / C16= 0, 1, 2, ..., 15)
PWM Channels 5, 6, 7 ,8   =  255  255  255  255
PWM Channels 9,10,11,12   =  255  255  255  255
PWM Channels 13,14,15,16  =  255  255  255  255
Voltage 1, 2, 3, 4        =   26   27   28   29 (V1 / V4 = 26, 27, 28, 29)

Protocol is Sport (Frsky)
CRSF baudrate = 420000
Voltage parameters:
    Scales : 1.000000 , 1.000000 , 1.000000 , 1.000000 
    Offsets: 0.000000 , 0.000000 , 0.000000 , 0.000000 
    No temperature sensors are connected on V3 and V4
RPM multiplier = 1.000000
Baro sensor is detected using MS5611
    Sensitivity min = 100 (at 100)   , max = 300 (at 1000)
    Hysteresis = 5 
Acc/Gyro is not detected
First analog to digital sensor is detected using ads1115
    Measurement setup: 4 , 8 , 8 ,8
    Gains: 1 , 1 , 1 ,1
    Rates: 1 , 1 , 1 ,1
    Offsets: 0.000000 , 0.000000 , 0.000000 ,0.000000
    Scales: 1.000000 , 1.000000 , 1.000000 ,1.000000
    Averaged on: 10 , 10 , 10 ,10
Second analog to digital sensor is not detected
Foreseen GPS type is Ublox  :GPS is not (yet) detected
Failsafe uses predefined values
     Chan 1...4  = 992 , 992 , 992 , 992
     Chan 5...8  = 992 , 992 , 992 , 992
     Chan 9...12 = 992 , 992 , 992 , 992
     Chan 13...16= 992 , 992 , 992 , 992

Config parameters are OK
Press ? + Enter to get help about the commands

 >> 
processing cmd

Cmd to execute:   FV

mstrens, 2023-02-17:
Do you get an answer when you enter the command FV?
Your screen shot stops just after Cmd to execute: FV

Satcomix, 2023-02-17:
I change only A0 to MV4096,
and when i put FV to Terminal there is nothing to see.
And Telemetry sensors are lost on Handheld, but green LED is blinking (SBUS)
processing cmd

Cmd to execute:   

Version = 1.1.9 
    Function                Pin   Change entering XXX=yyy (yyy=255 to disable)
Primary channels input    =    5  (PRI     = 5, 9, 21, 25)
Secondary channels input  =    1  (SEC     = 1, 13, 17, 29)
Telemetry . . . . . . . . =    0  (TLM     = 0, 1, 2, ..., 29)
GPS Rx  . . . . . . . . . =   12  (GPS_RX  = 0, 1, 2, ..., 29)
GPS Tx  . . . . . . . . . =   13  (GPS_TX  = 0, 1, 2, ..., 29)
Sbus OUT  . . . . . . . . =    2  (SBUS_OUT= 0, 1, 2, ..., 29)
RPM   . . . . . . . . . . =    4  (RPM     = 0, 1, 2, ..., 29)
SDA (I2C sensors) . . . . =   14  (SDA     = 2, 6, 10, 14, 18, 22, 26)
SCL (I2C sensors) . . . . =   15  (SCL     = 3, 7, 11, 15, 19, 23, 27)
PWM Channels 1, 2, 3 ,4   =  255  255    6  255 (C1 / C16= 0, 1, 2, ..., 15)
PWM Channels 5, 6, 7 ,8   =  255  255  255  255
PWM Channels 9,10,11,12   =  255  255  255  255
PWM Channels 13,14,15,16  =  255  255  255  255
Voltage 1, 2, 3, 4        =   26   27   28   29 (V1 / V4 = 26, 27, 28, 29)

Protocol is Sport (Frsky)
CRSF baudrate = 420000
Voltage parameters:
    Scales : 1.000000 , 1.000000 , 1.000000 , 1.000000 
    Offsets: 0.000000 , 0.000000 , 0.000000 , 0.000000 
    No temperature sensors are connected on V3 and V4
RPM multiplier = 1.000000
Baro sensor is detected using MS5611
    Sensitivity min = 100 (at 100)   , max = 300 (at 1000)
    Hysteresis = 5 
Acc/Gyro is not detected
First analog to digital sensor is detected using ads1115
    Measurement setup: 4 , 8 , 8 ,8
    Gains: 1 , 2 , 2 ,2
    Rates: 1 , 1 , 1 ,1
    Offsets: 0.000000 , 0.000000 , 0.000000 ,0.000000
    Scales: 1.000000 , 1.000000 , 1.000000 ,1.000000
    Averaged on: 10 , 10 , 10 ,10
Second analog to digital sensor is not detected
Foreseen GPS type is Ublox  :GPS is not (yet) detected
Failsafe uses predefined values
     Chan 1...4  = 992 , 992 , 992 , 992
     Chan 5...8  = 992 , 992 , 992 , 992
     Chan 9...12 = 992 , 992 , 992 , 992
     Chan 13...16= 992 , 992 , 992 , 992

Config parameters are OK
Press ? + Enter to get help about the commands

 >> 
processing cmd

Cmd to execute:   FV

mstrens, 2023-02-17:
Did you already tried to upload again the version 1.1.9 from github without making any change.
Normally there is no need to erase every thing and so you do not have to reenter your parameters.
Perhaps you made typo fout somewhere without noticing it. That the reason why I propose to start from a new download from github.

Satcomix, 2023-02-17:
When i put the original 1.1.9 on the board , i dont see any voltage to A0-A3
and when i make the change to MV4096, i have no Telemetry but SBUS is blinking green.
PlatformIO Error????
By version 1.1.9 there are many warnings of EMF Button, but i can make a build and upload.

mstrens, 2023-02-17:
When you put the original 1.1.9 on the board, did you compiled it your self or did you just make a copy/paste of the version I put in uf2 folder.
If you did not do it, I suggest a test with the version I put in uf2 folder.

The warnings about EMF button should not be an issue. This part of the code is currently compiled but not activated.

Perhaps make also a test without connecting oXs to the SBUS (because it is the way I tested it on my side)

mstrens, 2023-02-17:
I really do not understand why your oXs seems replies to reply to ENTER command but not to FV command.
Does it reply to the "?" (help) command?

Satcomix, 2023-02-17:
I will explain the way i do it every time,when a new version is out in GitHub.
I go to CODE and make a .zip Download. Than i make a new folder with the Name of the version. I extract the new version in that folder and make a copy to the Projects Folder from PlatformIO and copy and replace the old -test folder.
I think thats the right way i do it.

mstrens, 2023-02-17:
The way you do it should be OK.
Still can you try (as test) to just do a drag and drop of the file uf2/APPLICATION.uf2 that is part of the zip.

Satcomix, 2023-02-17:
I do not understand, when i put the .uf2 from my fresh extracted folder to the board, i make a drag and drop, or what do you mean?

Satcomix, 2023-02-17:
I make a download from Git of the APPLICATION.uf2 and copy and paste it to the board.

processing cmd

Cmd to execute:   

Version = 1.1.9 
    Function                Pin   Change entering XXX=yyy (yyy=255 to disable)
Primary channels input    =    5  (PRI     = 5, 9, 21, 25)
Secondary channels input  =    1  (SEC     = 1, 13, 17, 29)
Telemetry . . . . . . . . =    0  (TLM     = 0, 1, 2, ..., 29)
GPS Rx  . . . . . . . . . =   12  (GPS_RX  = 0, 1, 2, ..., 29)
GPS Tx  . . . . . . . . . =   13  (GPS_TX  = 0, 1, 2, ..., 29)
Sbus OUT  . . . . . . . . =    2  (SBUS_OUT= 0, 1, 2, ..., 29)
RPM   . . . . . . . . . . =    4  (RPM     = 0, 1, 2, ..., 29)
SDA (I2C sensors) . . . . =   14  (SDA     = 2, 6, 10, 14, 18, 22, 26)
SCL (I2C sensors) . . . . =   15  (SCL     = 3, 7, 11, 15, 19, 23, 27)
PWM Channels 1, 2, 3 ,4   =  255  255    6  255 (C1 / C16= 0, 1, 2, ..., 15)
PWM Channels 5, 6, 7 ,8   =  255  255  255  255
PWM Channels 9,10,11,12   =  255  255  255  255
PWM Channels 13,14,15,16  =  255  255  255  255
Voltage 1, 2, 3, 4        =   26   27   28   29 (V1 / V4 = 26, 27, 28, 29)

Protocol is Sport (Frsky)
CRSF baudrate = 420000
Voltage parameters:
    Scales : 1.000000 , 1.000000 , 1.000000 , 1.000000 
    Offsets: 0.000000 , 0.000000 , 0.000000 , 0.000000 
    No temperature sensors are connected on V3 and V4
RPM multiplier = 1.000000
Baro sensor is detected using MS5611
    Sensitivity min = 100 (at 100)   , max = 300 (at 1000)
    Hysteresis = 5 
Acc/Gyro is detected using MP6050
     Acceleration offsets X, Y, Z = 3024 , 356 , 1006
     Gyro offsets         X, Y, Z = -3 , 41 , 48
First analog to digital sensor is detected using ads1115
    Measurement setup: 4 , 5 , 6 ,7
    Gains: 2 , 2 , 2 ,2
    Rates: 1 , 1 , 1 ,1
    Offsets: 0.000000 , 0.000000 , 0.000000 ,0.000000
    Scales: 1.000000 , 1.000000 , 1.000000 ,1.000000
    Averaged on: 10 , 10 , 10 ,10
Second analog to digital sensor is not detected
Foreseen GPS type is Ublox  :GPS is detected and has a fix
Failsafe uses predefined values
     Chan 1...4  = 992 , 992 , 992 , 992
     Chan 5...8  = 992 , 992 , 992 , 992
     Chan 9...12 = 992 , 992 , 992 , 992
     Chan 13...16= 992 , 992 , 992 , 992

Config parameters are OK
Press ? + Enter to get help about the commands

 >> 
processing cmd

Cmd to execute:   FV

GPS Latitude = xxxxxxxxx degree
GPS Longitude = xxxxxxxxx degree
GPS Groundspeed = 67 cm/s
GPS Heading = 87.640000 degree
GPS Altitude = 3926 cm
GPS Num sat. = 109 cm
GPS Date J M A = 17 2 23 
GPS Time H M S = 13 24 35 
GPS Pdop = 199 
GPS Home bearing = 20 degree
GPS Home distance = 9 m
Volt 1 = 540 mVolt
Current (Volt 2) = 574 mA
Volt 3 = 627 mVolt
Volt 4 = 218 mVolt
Capacity (using current) = 0 mAh
Vspeed = 5 cm/s
Baro Rel altitude = 32 cm
Pitch = 4 degree
Roll = 1 degree
RPM = 0 Hertz
Ads 1 1 = 0 mVolt
Ads 1 2 = 0 mVolt
Ads 1 3 = 0 mVolt
Ads 1 4 = 0 mVolt

**A0=3.3V Input
A1=GND**

mstrens, 2023-02-17:
You have to press the "boot" button of the RP2040 when you apply the power on.
So the RP2040 enter automatically in bootloader mode and  the PC see it as a USB drive.
You can then do a drag and drop (or a copy/paste) from the uf2/APPLICATION.uf2 file to this drive usb drive.

This automatically upload the file (flash) and the RP2040 restart automatically.

mstrens, 2023-02-17:
OK I was to late with my last reply.
So now, you get a reply to the FV command but they are all 0mv.

Satcomix, 2023-02-17:
Thats the normal way a make it 100-200......... times the last weeks.
or you press RESET and hold the BOOT Button to get the RPi Folder.

[17 more replies]

closed 2023-02-18 by Satcomix