mstrens/oXs_on_RP2040 #45: GPS Speed in knots for S.PORT possible?

ParkerEde, 2023-02-16:
currently GSpd is transmitted in "km/h". Is it possible to transmit the speed also in "knots"? 
OpenTx and EdgeTx always assume that the speed is delivered in knots. If I then set the sensor unit to km/h, OTX and ETX convert this again and the result is a speed that is effectively too high. If I set the sensor unit to knots, the value is correct, but the unit is wrong. Thus, at least for FrSky (SPORT), the speed should be transmitted in knots by default.

nostromo123, 2023-02-16:
> OpenTx and EdgeTx always assume that the speed is delivered in knots. 

Interesting, please give a link to this. Under Edge here I can choose the dimensions  m/s, km/h mph and knots

Regards - KH

ParkerEde, 2023-02-16:
@nostromo123 
I am aware that you can choose the units, as I have already written. The point is that the units must also fit the values. And they don't if the sensor already delivers the data in km/h and I set the unit to km/h. In plain language, this means that the delivered value (which is already km/h in this case) is calculated into km/h again (because OTX and ETX assume that knots are present at the raw input. Logically, this results in a value that is too high.

ParkerEde, 2023-02-16:
To explain how I found out:
I use the RP2040 with a BP220 GPS module and ETX on a Horus X10. 

1. GPS Fix is done. After searching for all sensors in ETX, I set the unit for GSpd to km/h. 
2. I get into the car with the whole test set-up and drive exactly 50km/h. However, GSpd in ETX shows 93km/h (which is of course too much).
3. now I set the sensor GSpd in ETX to the unit "knots" and again drive 50km/h with the car. Now ETX shows 50kts. The value 50 for GSpd is correct, but the unit is knots is not correct.

The conclusion is that if the GSpd sensor unit is set to knots in ETX, the value is passed through 1:1. If the sensor unit in ETX for GSpd is set to km/h, ETX automatically multiplies the incomming value by 1.852 to convert to km/h. However, if the value is already supplied in km/h, as in this project, the result is incorrect.

nostromo123, 2023-02-16:
And OTX?

Remember the old oXs, I always had the oXs setting to km/h and in

https://openrcforums.com/forum/viewtopic.php?f=86&t=7393#p110909

found this old note here

> I presume that It is what openTX expects **and that openTx converts it in the units you set up on the Tx.**
> So even if oXs allows to select between knots and km, I think it should not be used for Frsky protocols.

OTX and Edge do handle this different?

But mstrens will tell us .... ;-)

ParkerEde, 2023-02-16:
I had exactly the same experience with OTX and oXs on Arduino with M8N. I also set it to km/h and tested it. But it was exactly the same, which is why I had to set oXs to kts again.

mstrens, 2023-02-16:
I checked in openTx code and indeed, it seems they expect a value in 0.001 kts.
I will change this in next version for Sport protocol

nostromo123, 2023-02-17:
But does it make sense to fix it on knots? It is just a number which is transferred and on the receiver side you are free to set the dimension ....?

And when pointed to the line in source, where GSpd is generated, could be set to a dimension ...
Not a big thing to do.

mstrens, 2023-02-18:
I presume that Opentx (and similar) expect the value in kts and performes already a conversion into the chose units for the display (or the "play value" function). So best is to provide the handset with the units it expects.

nostromo123, 2023-02-18:
@mstrens 

Do I understand you in the right way? The radio will display GSpd calculated in the unit selected? 

I see here this choice under OTX for the user and the same for EgdeTX.

[image]

and by the way, was adding this line in 'old' oXs and offering km/h a fault

> //#define GPS_SPEED_IN_KMH  // uncomment this line if GPS speed has to be sent in km/h instead of knot/h (only for Frsky protocol)

because this is against the protocol?

Have to admit, I don't understand this issue??

Regards - KH

mstrens, 2023-02-18:
I am not sure.
Best would be to make a test.
If I remember well (but it is some years ago and I am not sure), it could be that openTX did not work in the same was as today and it was expecting km/h. It could be that the comment in oXs was an old comment that is now wrong.

I expect that when oXs transmit in kts and if you select km/h on handset, then you get the rigth value on the handset display.
But again, the only way to be sure is to make a test.

ParkerEde, 2023-02-18:
I have already made the test (see above). OTX and ETX expect the value in kts as base value. If you then select e.g. km/h in OTX / ETX, OTX / ETX converts the input value from kts to km/h by multiplying the input value kts with 1.852 = km/h.
The unit selection in OTX/ETX is definitely not only a unit selection, it also converts the input value starting from kts to the selected unit. I have written that now nevertheless often enough. Karl Heinz should now accept that he is wrong.

nostromo123, 2023-02-18:
Then only a test like 'ParkerEde' did would give an answer!? And what protocols are affected? Sport, CRSF/ELRS, Jeti, Hott, Mpx?

@ParkerEde 

> I have written that now nevertheless often enough. Karl Heinz should now accept that he is wrong.

Sorry if you feel affronted by my comments here, that was not my intend! I thought, we are here to discuss ...

ParkerEde, 2023-02-18:
maybe you can make the desired unit in oXs "kts or km/h" configurable. Then those who set it to km/h in oXs and also set it to km/h in OTX/ETX can be happy about a speed display that is too high by a factor of 1.852 ;-)

ParkerEde, 2023-02-18:
@nostromo123 Yes, I'm sorry, I didn't mean anything by that. This ticket is only related to S.Port. Therefore, I think, you can make it already firmly to it. The easiest way will be to make it configurable like in oXs for arduino

mstrens, 2023-02-18:
Sending unit in kts concerns only Sport.
Each protocol expect the value in a specific unit and oXs makes the conversion from the "internal" unit to the unit expected by the protocol.
So bug in oXs on RP2040 resulted from a copy/paste from oXs on pro mini.

nostromo123, 2023-02-18:
@mstrens 

So the issue reported here comes from a 'fault' in 'old' oXs!?

> #define PROTOCOL  FRSKY_SPORT   // select between FRSKY_SPORT , FRSKY_HUB , FRSKY_SPORT_HUB , MULTIPLEX , HOTT, JETI

> //#define GPS_SPEED_IN_KMH  // uncomment this line if GPS speed has to be sent in km/h instead of knot/h (only for Frsky protocol)

@ParkerEde 

Thank you, now this 'old guy' here understood your issue. But please be calm, we are not discussing things like 'climate change' or the 'war in Ukraine' here. ;-)

mstrens, 2023-02-28:
In fact I changed it but on the wrong line (Vspeed instead groundspeed).
I will fix this in next version.

ParkerEde, 2023-02-28:
isn't it line 409-412?:
    case GROUNDSPEED:  // to do : test for the right value
        //uintValue =  ( ((uint32_t) uintValue) * 36 )  ; // convert cm/s in 1/100 of km/h (factor = 3.6)
        uintValue =  ( ((uint32_t) uintValue) * 700 ) / 36 ; // convert cm/s in 1/1000 of knots (factor = 19.44)
        break;

mstrens, 2023-02-28:
I presume sport.cpp is OK
I will just change the doc 'fields per protocol.txt
It should be (I think)
Vspeed      cm/sec           a baro is installed (and SCL/SDA defined)    Vspeed (0X0110)            cm/s
                             if a MP6050 is installed, Vpeed use it too
GPS longitude                GPS is installed (and GPS_TX, GPS_RX defined) long (0x0800)          special format
GPS lat                      Same as above                                 lat (0x0800)           special format
GPS speed  cm/sec            Same as above                                 GPS ground speed(0X0830)  0.01 kts/h

ParkerEde, 2023-04-04:
great work, thank you @mstrens

closed 2023-04-04 by ParkerEde