mstrens/oXs_on_RP2040 #48: Jeti protocol - only one voltage possible (no current measurement)

onki69, 2023-02-19:
Hi,

is there a chance to get "V2" on the board used as current sensor and add a calculated capacity as well?
With the current test version all Jeit stuff seems to be working (tested with MS5611, voltage and ublox GPS), but just one voltage is supported at the moment.

Best regards
Onki

mstrens, 2023-02-19:
Yes, it is possible.
I can do it when ibus (flysky) will be implemented

mstrens, 2023-02-20:
I just added it in version 1.2.0. (in test branch)
Please test it and let me know if it is ok

Satcomix, 2023-02-20:
Hello Mstrens,
Is it a pure trial for Jeti?
aaw=0 acc=-69 az=16248
aaw=0 acc=-143 az=16170
aaw=0 acc=-115 az=16198
aaw=0 acc=-13 az=16306
aaw=0 acc=-48 az=16272
aaw=0 acc=-78 az=16244
aaw=0 acc=-46 az=16276
aaw=0 acc=-37 az=16280
aaw=0 acc=-55 az=16260
aaw=0 acc=15 az=16330
aaw=0 acc=-10 az=16308
aaw=0 acc=47 az=16360
aaw=0 acc=-54 az=16258
aaw=0 acc=-116 az=16198
aaw=0 acc=-51 az=16266
aaw=0 acc=-61 az=16248
aaw=0 acc=-46 az=16274
aaw=0 acc=-54 az=16264
aaw=0 acc=-95 az=16218
aaw=0 acc=-140 az=16168

**In the next test versions, could you please provide the values ​​for both ADS1115**
ADS_Full_Scale_Volt: MV4096
ADS_Rate: 5ms,
ADS_Averaging: 10
**That would make testing easier for me in the future.**
Thanks,
Torsten

onki69, 2023-02-20:
I did a very quick test.
CXurrent at V2 is working and I was able to adjust the scale to get correct current values.
But Consumption always stays at 0.00Ah despite of the current that is detected.
I will test this more detailes later today.

mstrens, 2023-02-20:
In order to check if the bug is in the calculation or in the transmission, you can enter the command "FV" in order to know the actual internal values (= calculation)

onki69, 2023-02-20:
Hi again,

I could test a bit more. The consumption is available and I can see values.
The problem is that the calculated value is way too low. This is why I thought is is not working.
I am currently running a test with 2A constant current for 10 mins.
The expected consumption should be about 0.33Ah but I just get 1/10 of it displayed (33mAh).
I think there is a little error in the calculation formula.

mstrens, 2023-02-20:
The error is perhaps in the conversion of the values when they are sent to the Jeti Rx.
You can use the FV command (on PC) to display the values that are calculated internally to oXs.
First you have to get on the PC (with this FV) a good value for current (in mA) and then you should get a good value for consumption (in mAh).

Then, when this is ok, you can look at the Jeti display. If some values are to low/big on the jeti display compared to those of the FV command, I can apply some multiplier/divider while filling the frame with the values transmitted to the TX.
I expect that it is at this level that a change has to be made.

onki69, 2023-02-20:
When I checked the values I have used the console commands.
So the calculated capacity shown in the console was already wrong. The Jeti dispay values are identical to the values shown on the console.

mstrens, 2023-02-20:
Thanks.
I think I found the bug (in the way consumption was calculated).
It should be fixed in next version (not yet loaded on github)

mstrens, 2023-02-21:
I think it is now OK in version 1.2.13 (loaded on github)

onki69, 2023-02-21:
Thanks for efforts.

Just one last question on how to apply scale and offset with the current.
I have set my V2 to Scale 1 and offset 0. With no current I do get 16mA current (using the FV command).
When 1A is applied I do get 92mA, with 2A it is 168mA. Having the 16mA in mind it would be 76mA @1A and 152@2A.
How do I apply the offset. It is based on scale 1 or will it be applied to the calculated value?
In my case the scale should be 13.15 if I understand correctly.

mstrens, 2023-02-21:
Currently, during calculation offset is applied after scaling. This is perhaps not the most useful for the user and I could perhaps change it.
Currently the logic would be
First set scale to 1 and offset = 0
Apply current in the range you expect (e.g. Max and MAX/2).
Imagine max = 2A = 2000mA gives 168
Imagine max/2=1A = 1000mA gives 92.
So scale should be (2000 - 1000) / (168-92) = 13.15
Offset should be (92*13.15) - 1000 = 210

onki69, 2023-02-21:
My sensor can handle 90A/30V (APM 2.5 Sensor).
So it is quite impossible to do a calibration at max and max/2.

Therefore the offset subtraction of the scaled values ist not that handy.
My personal preference is using the offset with scale 1 to get a real zero value and apply the scale to this corrected values.

But maybe others may think different.

closed 2023-02-22 by onki69