mstrens/oXs_on_RP2040 #49: Improvements TEST-Ver.1.2.1

Satcomix, 2023-02-20:
Hello Mstrens
It was **tested** with FrSky X20S Ethos 1.4.7 **S.PORT** and Graupner MC28 **HOTT**.
**S.PORT Protocol:**
All 29 values ​​in the telemetry display are displayed without any recognizable errors.
Suggested improvement for sample rates:
GPS LAT/LON set to 300ms or 500ms.
Set GPS Time to 1 second.
Set Vspeed Vario to 300ms or 500ms.
That three values ​​are too fast, i think the query dot almost stands still and doesn't disappear.
**HOTT Protocol:**
During the tests in the HOTT protocol, I did not notice any problems with the 15 values ​​displayed. Everything works satisfactorily.
If you could also enter the RPM value in rpm in "fields_per_protocol.txt" under HOTT protocol, the value will be shown correctly on the display.
br,
Torsten

processing cmd

Cmd to execute:   

Version = 1.2.1 
    Function                Pin   Change entering XXX=yyy (yyy=255 to disable)
Primary channels input    =    5  (PRI     = 5, 9, 21, 25)
Secondary channels input  =    1  (SEC     = 1, 13, 17, 29)
Telemetry . . . . . . . . =    0  (TLM     = 0, 1, 2, ..., 29)
GPS Rx  . . . . . . . . . =   12  (GPS_RX  = 0, 1, 2, ..., 29)
GPS Tx  . . . . . . . . . =   13  (GPS_TX  = 0, 1, 2, ..., 29)
Sbus OUT  . . . . . . . . =    2  (SBUS_OUT= 0, 1, 2, ..., 29)
RPM   . . . . . . . . . . =    4  (RPM     = 0, 1, 2, ..., 29)
SDA (I2C sensors) . . . . =   14  (SDA     = 2, 6, 10, 14, 18, 22, 26)
SCL (I2C sensors) . . . . =   15  (SCL     = 3, 7, 11, 15, 19, 23, 27)
PWM Channels 1, 2, 3 ,4   =  255  255    6  255 (C1 / C16= 0, 1, 2, ..., 15)
PWM Channels 5, 6, 7 ,8   =  255  255  255  255
PWM Channels 9,10,11,12   =  255  255  255  255
PWM Channels 13,14,15,16  =  255  255  255  255
Voltage 1, 2, 3, 4        =   26   27   28   29 (V1 / V4 = 26, 27, 28, 29)

Protocol is Sport (Frsky)
CRSF baudrate = 420000
Voltage parameters:
    Scales : 10.000000 , 10.000000 , 0.100000 , 0.100000 
    Offsets: 0.000000 , 0.000000 , 0.000000 , 0.000000 
    No temperature sensors are connected on V3 and V4
RPM multiplier = 1.000000
Baro sensor is detected using MS5611
    Sensitivity min = 100 (at 100)   , max = 300 (at 1000)
    Hysteresis = 5 
Acc/Gyro is detected using MP6050
     Acceleration offsets X, Y, Z = 3024 , 356 , 1006
     Gyro offsets         X, Y, Z = -3 , 41 , 48
First analog to digital sensor is detected using ads1115
    Measurement setup: 4 , 5 , 6 ,7
    Gains: 1 , 1 , 1 ,1
    Rates: 5 , 5 , 5 ,5
    Offsets: 0.000000 , 0.000000 , 0.000000 ,0.000000
    Scales: 1.000000 , 1.000000 , 1.000000 ,1.000000
    Averaged on: 10 , 10 , 10 ,10
Second analog to digital sensor is detected using ads1115
    Measurement setup: 4 , 5 , 6 ,7
    Gains: 1 , 1 , 1 ,1
    Rates: 5 , 5 , 5 ,5
    Offsets: 0.000000 , 0.000000 , 0.000000 ,0.000000
    Scales: 1.000000 , 1.000000 , 1.000000 ,1.000000
    Averaged on: 10 , 10 , 10 ,10
Foreseen GPS type is Ublox  :GPS is detected and has a fix
Failsafe uses predefined values
     Chan 1...4  = 992 , 992 , 992 , 992
     Chan 5...8  = 992 , 992 , 992 , 992
     Chan 9...12 = 992 , 992 , 992 , 992
     Chan 13...16= 992 , 992 , 992 , 992

Config parameters are OK
Press ? + Enter to get help about the commands

 >> 
processing cmd

Cmd to execute:   FV

GPS Latitude = xxxxxxxxxxxxx degree
GPS Longitude = xxxxxxxxxxx degree
GPS Groundspeed = 11 cm/s
GPS Heading = 0.000000 degree
GPS Altitude = 4249 cm
GPS Num sat. = 109 cm
GPS Date J M A = 20 2 23 
GPS Time H M S = 10 15 33 
GPS Pdop = 194 
GPS Home bearing = 72 degree
GPS Home distance = 12 m
Volt 1 = 5373 mVolt
Current (Volt 2) = 5720 mA
Volt 3 = 62 mVolt
Volt 4 = 22 mVolt
Capacity (using current) = 11 mAh
Vspeed = 0 cm/s
Baro Rel altitude = 31 cm
Pitch = 4 degree
Roll = -5 degree
RPM = 992 Hertz
Ads 1 1 = 1601 mVolt
Ads 1 2 = 1708 mVolt
Ads 1 3 = 1800 mVolt
Ads 1 4 = 1070 mVolt
Ads 2 1 = 1600 mVolt
Ads 2 2 = 1708 mVolt
Ads 2 3 = 1799 mVolt
Ads 2 4 = 1070 mVolt

Satcomix, 2023-02-20:
Something changed in S.PORT in Ver.1.2.4
VF Terminal display:
Pitch = 0 degree
Roll = -14 degree
FrSky X20S Ethos 1.4.7 displayed 
DIY 5120 Pitch 0°
DIY 5121 Roll 0°
I make MPUCAL but the display in VF Terminal dont change.
br,
Torsten

mstrens, 2023-02-20:
I will have to test myself first.
I made some changes about calibration because sometime previous version did not work.
I tried to implement another version but did not tested (I am also debugging ibus)

Satcomix, 2023-02-21:
Hello Mstrens,
The problem with the telemetry **sensor ID:00 600 FUEL** still exists in test **version 1.2.13.**
In previous/older versions the FUEL sensor was on V4/GPIO29.
This value did not count up from 0% to 100%.
With FrSky X20S Ethos 1.4.7 there is **no way to change the information, only %** is displayed as the unit.
If the value now, depending on V2, counts up and has reached 100%, **"Sensor Lost"** sounds. 
A DIY sensor ID for the capacity/consumption would make more sense.

br, Torsten

mstrens, 2023-02-21:
1) About Sport and fuel:
I expect that the telemetry lost is done on handset side because the value exceed 100.
I presume Sport expect a value in %. This would suppose that oXs knows the capacity of the battery.
This is not the case.
I see 2 solutions:
 - I transmit the consumption in a "DIY" field (this is done in next version)
 - I just ignore the field and let the user defines it in the handset.

2) About Hott and RPM: I updated the file fields per protocol.txt

3) About MPUCAL.
I tested the calibration on my board.
It seems that it performs a calibration, that the result is saved in config and reused after power on.
I noticed that some values being displayed were abnormal. I think it was only an issue in the way the values were displayed and I hope I fix it in a new version
I still notice an issue I can't explain currently:
After a power on, I can run the MPUCAL once and it seems ok.
If I then try it again, I get the values before calibration but not value after calibration and my usb connection is lost.
After pressing enter, the usb reconnect and I can do the MPUCAL again.
Take care that when you perform the MPUCAL command, the MP6050 must be perfectly horizontal and may not move.

Newe version is currently 1.3.0

Satcomix, 2023-02-21:
Hello Mstrens,
Thank you for the reply.
I will test the 1.3.0 and look for issues :-))
In the last post, I saw something odd about Nick and Roll in the FV Terminal.
Pitch = 0 degree
Roll = -8 degree
In display from handheld both were at 0°
I have same problems with PlatformIO and Windows10Pro, after ca. 10 test/builds there are strange failure in the new builds. 
In those build i made, you can`t press FV and Telemetry is lost, but SBUS is working. Strange thing.
I could also observe strange phenomena with the ADS1115 and a reset of the board. The voltages applied to the ADS1115 also supply the board with a reset. You should always disconnect these voltages before playing a new setting into the board.
Now i will test the new release.
br,
Torsten

Satcomix, 2023-02-21:
No MPUCAL !!! after Reset and Power down.
And PITCH and ROLL value in FV Terminal not the same in Handheld display.
Cmd to execute:   FV

GPS Latitude = xxxxxxxxxx degree
GPS Longitude = xxxxxxxxx degree
GPS Groundspeed = 17 cm/s
GPS Heading = 180.960000 degree
GPS Altitude = 3214 cm
GPS Num sat. = 111 cm
GPS Date J M A = 21 2 23 
GPS Time H M S = 13 33 25 
GPS Pdop = 153 
GPS Home bearing = 189 degree
GPS Home distance = 18 m
Volt 1 = 5346 mVolt
Current (Volt 2) = 5646 mA
Volt 3 = 62 mVolt
Volt 4 = 20 mVolt
Capacity (using current) = 16 mAh
Vspeed = 0 cm/s
Baro Rel altitude = 1 cm
Pitch = 0 degree
Roll = -8 degree
RPM = 997 Hertz
Ads 1 1 = -1 mVolt
Ads 1 2 = -1 mVolt
Ads 1 3 = -1 mVolt
Ads 1 4 = -1 mVolt
Ads 2 1 = -1 mVolt
Ads 2 2 = -1 mVolt
Ads 2 3 = -1 mVolt
Ads 2 4 = -1 mVolt

Cmd to execute:   MPUCAL
Before calibration:
Offset Values in config:
Acc. X = 550, Y = -377, Z = -2678
Gyro. X = 17043, Y = -165, Z = -197
Write error for MPU6050 calibration
error while calibrating mp6050

Board RESET and POWER DOWN!
New MPUCAL

Cmd to execute:   MPUCAL
Before calibration:
Offset Values in config:
Acc. X = 550, Y = -377, Z = -2678
Gyro. X = 17043, Y = -165, Z = -197
Write error for MPU6050 calibration
error while calibrating mp6050

mstrens, 2023-02-21:
Sorry.
I think it is now ok in 1.3.1 (on github)

Satcomix, 2023-02-21:
**Good job,thank you very much.**
I find the new DIY 5123 Sensor and shut down the old 00 0600.
**Everything works on FrSky X20S Ethos 1.4.7 S.PORT.**
Did you read my post about the timing of the three sensors, what do you think about?
The **GPS time** (****before every second**, now every three seconds**) would be extremely interesting for me, since the internal clock of the X20S handheld is very unreliable.
br,
Torsten
processing cmd
Cmd to execute:   

Version = 1.3.1 
    Function                Pin   Change entering XXX=yyy (yyy=255 to disable)
Primary channels input    =    5  (PRI     = 5, 9, 21, 25)
Secondary channels input  =    1  (SEC     = 1, 13, 17, 29)
Telemetry . . . . . . . . =    0  (TLM     = 0, 1, 2, ..., 29)
GPS Rx  . . . . . . . . . =   12  (GPS_RX  = 0, 1, 2, ..., 29)
GPS Tx  . . . . . . . . . =   13  (GPS_TX  = 0, 1, 2, ..., 29)
Sbus OUT  . . . . . . . . =    2  (SBUS_OUT= 0, 1, 2, ..., 29)
RPM   . . . . . . . . . . =    4  (RPM     = 0, 1, 2, ..., 29)
SDA (I2C sensors) . . . . =   14  (SDA     = 2, 6, 10, 14, 18, 22, 26)
SCL (I2C sensors) . . . . =   15  (SCL     = 3, 7, 11, 15, 19, 23, 27)
PWM Channels 1, 2, 3 ,4   =  255  255    6  255 (C1 / C16= 0, 1, 2, ..., 15)
PWM Channels 5, 6, 7 ,8   =  255  255  255  255
PWM Channels 9,10,11,12   =  255  255  255  255
PWM Channels 13,14,15,16  =  255  255  255  255
Voltage 1, 2, 3, 4        =   26   27   28   29 (V1 / V4 = 26, 27, 28, 29)

Protocol is Sport (Frsky)
CRSF baudrate = 420000
Voltage parameters:
    Scales : 10.000000 , 10.000000 , 0.100000 , 0.100000 
    Offsets: 0.000000 , 0.000000 , 0.000000 , 0.000000 
    No temperature sensors are connected on V3 and V4
RPM multiplier = 1.000000
Baro sensor is detected using MS5611
    Sensitivity min = 100 (at 100)   , max = 300 (at 1000)
    Hysteresis = 5 
Acc/Gyro is detected using MP6050
     Acceleration offsets X, Y, Z = 633 , -432 , -2676
     Gyro offsets         X, Y, Z = 2 , -163 , -195
First analog to digital sensor is detected using ads1115
    Measurement setup: 4 , 5 , 6 ,7
    Gains: 1 , 1 , 1 ,1
    Rates: 5 , 5 , 5 ,5
    Offsets: 0.000000 , 0.000000 , 0.000000 ,0.000000
    Scales: 1.000000 , 1.000000 , 1.000000 ,1.000000
    Averaged on: 10 , 10 , 10 ,10
Second analog to digital sensor is detected using ads1115
    Measurement setup: 4 , 5 , 6 ,7
    Gains: 1 , 1 , 1 ,1
    Rates: 5 , 5 , 5 ,5
    Offsets: 0.000000 , 0.000000 , 0.000000 ,0.000000
    Scales: 1.000000 , 1.000000 , 1.000000 ,1.000000
    Averaged on: 10 , 10 , 10 ,10
Foreseen GPS type is Ublox  :GPS is detected and has a fix
Failsafe uses predefined values
     Chan 1...4  = 992 , 992 , 992 , 992
     Chan 5...8  = 992 , 992 , 992 , 992
     Chan 9...12 = 992 , 992 , 992 , 992
     Chan 13...16= 992 , 992 , 992 , 992

Config parameters are OK
Press ? + Enter to get help about the commands

 >> 
processing cmd

Cmd to execute:   FV

GPS Latitude = xxxxxxxxxx degree
GPS Longitude = xxxxxxxxx degree
GPS Groundspeed = 37 cm/s
GPS Heading = 320.260000 degree
GPS Altitude = 3051 cm
GPS Num sat. = 109 cm
GPS Date J M A = 21 2 23 
GPS Time H M S = 15 33 13 
GPS Pdop = 231 
GPS Home bearing = 319 degree
GPS Home distance = 87 m
Volt 1 = 5408 mVolt
Current (Volt 2) = 5681 mA
Volt 3 = 61 mVolt
Volt 4 = 23 mVolt
Capacity (using current) = 43 mAh
Vspeed = 1 cm/s
Baro Rel altitude = 0 cm
Pitch = 0 degree
Roll = 0 degree
RPM = 997 Hertz
Ads 1 1 = 1499 mVolt
Ads 1 2 = 1600 mVolt
Ads 1 3 = 1705 mVolt
Ads 1 4 = 1057 mVolt
Ads 2 1 = 1499 mVolt
Ads 2 2 = 1599 mVolt
Ads 2 3 = 1704 mVolt
Ads 2 4 = 1056 mVolt

closed 2023-02-21 by Satcomix

Satcomix, 2023-02-21:
Hello Mstrens,
I didn't want the GPS time to be updated either, because of the load on the entire telemetry system, **but the time on the FrSky X20S is really so bad that after a while (1 hour) there is already a difference of several seconds.**
I had to set the internal clock to match the GPS, it works, but not really.
The other values ​​don't matter to me from the timing, they just seemed very fast to me.
You make a good job.
br,
Torsten