mstrens/oXs_on_RP2040 #50: Set GPS_TIME to 1 second

Satcomix, 2023-02-22:
Hello Mstrens,
Short question:
In sport.cpp I can set the min_pooling and max_pooling for the sport_frame to change the query priority.
I have set the GPS_TIME from max=200 to 50 and the min=100 to 30.
I would like a GPS TIME resolution of 1 second, how do I set it?, as the GPS time is now being queried and displayed a little too quickly.
br,
Torsten

mstrens, 2023-02-22:
You can't exactly set the refresh rate to 1 sec.
For sport, oXs sent the data based on a set of rules. It is sent:
- only if a value is available from the sensor ; for GPS this happens 10 X per sec
- depending on the number of polling requests oXs get and on other data to be sent. 
- the number of polling requests depends on the number of other devices that are connected on sport bus. If oXs is alone, then oXs get 1000/24 = about 50 X per sec. If there is a second device on the sport bus, it becomes 1000/36 = about 30 X per sec
-e.g. if you put min = 50 and there is no other device, you will not get the field more often than 1 X per sec. Still the same setup (min 50) with another device on the bus will lead to 1 X per 1.5 sec.
- the max value says that oXs has to wait max pooling request before sending the value. So if you want to get it once per sec and there is only one device, then you can try max =49 and min 50.
- oXs will not respect this exactly because there could be some other fields that have got a higher priority and will be sent first  but it is the best you can get actually.

The oXs code could be improved if oXs would send the time only when it changes. You could then assign a higher priority to this field and still send it only once per sec but this is not possible currently.

Satcomix, 2023-02-22:
Hello Mstrens,
Thanks for the detailed answer.
I will do my best :-))
br,
Torsten

closed 2023-02-22 by Satcomix

Satcomix, 2023-02-24:
Hello Mstrens,
Now that a new 1.3.4 main release has been released, I noticed a few things that could perhaps be changed.
The arrangement of SCALES and OFFSETS on the ADS1115 could be reversed as with V1 to V3. First SCALE then OFFSET (is something for the eye)
The "ads1115.h" and "config.h" script still contains the settings for the ADs1115 MV6144, which could confuse the user, since a maximum voltage of 3300mVolt (plus max. 30mVolt according to the datasheet via supply voltage) may be applied with MV4096, and not 6144mVolt.
In the "fields per protocol.txt" the two lines GAM temperature 1 not transmitted GAM temperature 2 not transmitted
deleted, since these values ​​are transmitted with a specific setting of V3+V4=Temp1+2.
Thank you for your perfect work.
Greeting,
Torsten