mstrens/oXs_on_RP2040 #51: Setup Hall Current Sensor

doloebig, 2023-02-22:
The setup of the hall current sensor needs an offset and a scaling. 
The value is calculated by //fields[cntInit + MVOLT].value = ( ((float) sumVoltage[cntInit]) / (( float) SUM_COUNT_MAX_VOLTAGE) * mVoltPerStep[cntInit]) - offset[cntInit];
for my understanding it needs to be:
 //fields[cntInit + MVOLT].value = ( ((float) sumVoltage[cntInit]) / (( float) SUM_COUNT_MAX_VOLTAGE) - offset[cntInit] * mVoltPerStep[cntInit]) ;
first to subtracted the "0 Current" value (mosty 2.5 or 0.5V)and then scale by mV/A
it will be very nice if V2 is used as Currentsensor put the values (0Current and mV/A) direct in the setup instead offset and scale

mstrens, 2023-02-22:
This would require a specific setup for v2 which is more complex to code.
I do not think that scale and offset is so complex with a min of math.
E.G. if I take a typical sensor for 100A with (in data sheet) 0.1V offset per VCC volt and 40mV/A with Vcc = 5V
This means that with a vcc = 3.3V
- at 0A, the voltage on rp2040 pin would be 0.33V and with a scale = 1 and an offset = 0, oXs would give a value 330 (mV)
- a 100A, the voltage would be (100*40/5*3.3)+ 0.33V = 2,970V and so oXs would give 2970 (mv) 
Scale to apply is 100/(2970-330)  = 0.037879
Offset to apply = - (330* 0.037879) = -12.5

doloebig, 2023-02-23:
Hallo @mstrens 
in may case the sensor need 5V
setting scale 1 offset 0
May "0A" is 520mv FV=520mV, FV at 4.5A 694mv diff 174mv~38,7mV/A

When is put you values in 0.037879 as Scale and -12.5 offset
i get 0A 
fv
processing cmd

Cmd to execute:   FV

Volt 1 = 632 mVolt
Current (Volt 2) = 32 mA
Capacity (using current) = 0 mAh

an on 4,5A
processing cmd

Cmd to execute:   FV

Volt 1 = 634 mVolt
Current (Volt 2) = 38 mA
Capacity (using current) = 0 mAh

on oTx 0A

mstrens, 2023-02-23:
I am not sure I understand.
I expect that with scale = 1 and offset =0, 
with 0A, you get a FV value = 520
with 4.5A you get a FV value = 694.
If it is so, then 
Scale must then be 4.5/(694 - 520)= 4.5/174 = 0.025862.
Offset must be 520 *  0.025862 = 13,4483

doloebig, 2023-02-23:
yes thas right 
cale = 1 and offset =0,
with 0A, you get a FV value = 520
with 4.5A you get a FV value = 694.

Set Scale2 and Offset to 0.025862 and -13.4483

reading at 0A
Version = 1.3.2 
    Function                Pin   Change entering XXX=yyy (yyy=255 to disable)
Primary channels input    =  255  (PRI     = 5, 9, 21, 25)
Secondary channels input  =  255  (SEC     = 1, 13, 17, 29)
Telemetry . . . . . . . . =    7  (TLM     = 0, 1, 2, ..., 29)
GPS Rx  . . . . . . . . . =    3  (GPS_RX  = 0, 1, 2, ..., 29)
GPS Tx  . . . . . . . . . =    2  (GPS_TX  = 0, 1, 2, ..., 29)
Sbus OUT  . . . . . . . . =  255  (SBUS_OUT= 0, 1, 2, ..., 29)
RPM   . . . . . . . . . . =  255  (RPM     = 0, 1, 2, ..., 29)
SDA (I2C sensors) . . . . =  255  (SDA     = 2, 6, 10, 14, 18, 22, 26)
SCL (I2C sensors) . . . . =  255  (SCL     = 3, 7, 11, 15, 19, 23, 27)
PWM Channels 1, 2, 3 ,4   =  255  255  255  255 (C1 / C16= 0, 1, 2, ..., 15)
PWM Channels 5, 6, 7 ,8   =  255  255  255  255
PWM Channels 9,10,11,12   =  255  255  255  255
PWM Channels 13,14,15,16  =  255  255  255  255
Voltage 1, 2, 3, 4        =   26   27  255  255 (V1 / V4 = 26, 27, 28, 29)

Protocol is Sport (Frsky)
CRSF baudrate = 420000
Voltage parameters:
    Scales : 1.000000 , 0.025862 , 1.000000 , 1.000000 
    Offsets: 0.000000 , 13.448300 , 0.000000 , 0.000000 
    No temperature sensors are connected on V3 and V4
RPM multiplier = 1.000000
Baro sensor is not detected
Acc/Gyro is not detected
First analog to digital sensor is not detected
Second analog to digital sensor is not detected
Foreseen GPS type is Ublox  :GPS is not (yet) detected
Failsafe type is HOLD

Config parameters are OK
Press ? + Enter to get help about the commands

 >> 
FV
processing cmd

Cmd to execute:   FV

Volt 1 = 624 mVolt
Current (Volt 2) = 0 mA
Capacity (using current) = 0 mAh

At 4,3A
FV
processing cmd

Cmd to execute:   FV

Volt 1 = 634 mVolt
Current (Volt 2) = 4 mA
Capacity (using current) = 0 mAh

Handset show always 0.00A

doloebig, 2023-02-23:
in both currents, 0A and 4A, the Handset shows 0.00A

Doro

mstrens, 2023-02-23:
Sorry, I mixed A and mA.
The values should be

Scale must then be 4500/(694 - 520)= 4500/174 = 25.862.
Offset must be - 520 * 25.862 = -13448.3 (do not forget the sign - )

mstrens, 2023-02-23:
Just one question. Yo say that you use a 5V current sensor.
Do not forget to use a voltage divider if your sensor can deliver a voltage that exceed 3.3V (max voltage accepted by RP2040 on the ADC pins)

doloebig, 2023-02-23:
set values to:
Protocol is Sport (Frsky)
CRSF baudrate = 420000
Voltage parameters:
    Scales : 1.000000 , 25.862000 , 1.000000 , 1.000000 
    Offsets: 0.000000 , -13.448300 , 0.000000 , 0.000000 

Reading at **0A**
processing cmd

Cmd to execute:   FV

Volt 1 = 639 mVolt
Current (Volt 2) = 11488 mA
Capacity (using current) = 28 mAh
Handset show 13.4**A**

Reading at **4.3A**

Cmd to execute:   FV

Volt 1 = 641 mVolt
Current (Volt 2) = 17703 mA
Capacity (using current) = 559 mAh

Handset shows 17,6**A**

and yes i know the 3.3V limt. I´m not have more the 45A on this plane. 

THX for tireless helping me

Doro

mstrens, 2023-02-23:
Can you try with:
scale = 25.862 (25 and some decimals)
offset = 13448 (so positive and no comma/dot - so more than 13 thousand)

mstrens, 2023-02-23:
The formula I used remains the same (but oXs already substract the offset so the value to fill in must be positive)
Scale must then be 4500/(694 - 520)= 4500/174 = 25.862 
Offset must be 520 * 25.862 = 13448.3 
Dot is the decimal point.

doloebig, 2023-02-24:
Thas it, perfekt @mstrens 
FV
processing cmd

Cmd to execute:   FV
Reading Handset 0A
Volt 1 = 649 mVolt
Current (Volt 2) = -9 mA
Capacity (using current) = -3 mAh
FV
processing cmd

Reading Handset 4.1A
Cmd to execute:   FV

Volt 1 = 648 mVolt
Current (Volt 2) = 4214 mA
Capacity (using current) = 11 mAh

Voltage parameters:
    Scales : 1.000000 , 25.862000 , 1.000000 , 1.000000 
    Offsets: 0.000000 , 13448.299805 , 0.000000 , 0.000000 

THX a million
Doro

closed 2023-02-24 by doloebig