mstrens/oXs_on_RP2040 #53: Setting up build environment in Windows

andypnz, 2023-02-23:
Hi. I have Anaconda installed. Within Anaconda I created an environment for Platformio:
1. conda create --name platformio
2. conda activate platformio
3. python.exe get-platformio.py
4. made a directory for this project
5. initialised the project pio project init --board pico
6. copied the contents of the src + lib directories from github to the new directories
7. pio run gives these errors:

(platformio) PS C:\Users\Andrew\arp_osx> pio run
Processing pico (platform: raspberrypi; board: pico; framework: arduino)
------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/raspberrypi/pico.html
PLATFORM: Raspberry Pi RP2040 (1.7.0) > Raspberry Pi Pico
HARDWARE: RP2040 133MHz, 264KB RAM, 2MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, jlink, raspberrypi-swd)
PACKAGES:
 - framework-arduino-mbed @ 3.1.1
 - tool-rp2040tools @ 1.0.2
 - toolchain-gccarmnoneeabi @ 1.90201.191206 (9.2.1)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 37 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Compiling .pio\build\pico\src\BMP280.cpp.o
Compiling .pio\build\pico\src\EMFButton.cpp.o
Compiling .pio\build\pico\src\I2Cdev.cpp.o
Compiling .pio\build\pico\src\KalmanFilter.cpp.o
Compiling .pio\build\pico\src\KalmanFilter4D.cpp.o
Compiling .pio\build\pico\src\MPU6050.cpp.o
Compiling .pio\build\pico\src\MS5611.cpp.o
Compiling .pio\build\pico\src\SPL06.cpp.o
In file included from src\I2Cdev.h:37,
                 from src\MPU6050.h:39,
                 from src\MPU6050.cpp:37:
C:\Users\Andrew\.platformio\packages\framework-arduino-mbed\cores\arduino/mbed/targets/TARGET_RASPBERRYPI/TARGET_RP2040/pico-sdk/common/pico_stdlib/include/pico/stdlib.h:11:10: fatal error: pico/stdio.h: No such file or directory
   11 | #include "pico/stdio.h"
      |          ^~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\pico\src\MPU6050.cpp.o] Error 1
In file included from src\MS5611.cpp:2:
C:\Users\Andrew\.platformio\packages\framework-arduino-mbed\cores\arduino/mbed/targets/TARGET_RASPBERRYPI/TARGET_RP2040/pico-sdk/common/pico_stdlib/include/pico/stdlib.h:11:10: fatal error: pico/stdio.h: No such file or directory
   11 | #include "pico/stdio.h"
      |          ^~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\pico\src\MS5611.cpp.o] Error 1
In file included from src\I2Cdev.h:37,
                 from src\I2Cdev.cpp:32:
C:\Users\Andrew\.platformio\packages\framework-arduino-mbed\cores\arduino/mbed/targets/TARGET_RASPBERRYPI/TARGET_RP2040/pico-sdk/common/pico_stdlib/include/pico/stdlib.h:11:10: fatal error: pico/stdio.h: No such file or directory
   11 | #include "pico/stdio.h"
      |          ^~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\pico\src\I2Cdev.cpp.o] Error 1
In file included from src\SPL06.cpp:4:
C:\Users\Andrew\.platformio\packages\framework-arduino-mbed\cores\arduino/mbed/targets/TARGET_RASPBERRYPI/TARGET_RP2040/pico-sdk/common/pico_stdlib/include/pico/stdlib.h:11:10: fatal error: pico/stdio.h: No such file or directory
   11 | #include "pico/stdio.h"
      |          ^~~~~~~~~~~~~~
compilation terminated.
In file included from src\BMP280.cpp:2:
C:\Users\Andrew\.platformio\packages\framework-arduino-mbed\cores\arduino/mbed/targets/TARGET_RASPBERRYPI/TARGET_RP2040/pico-sdk/common/pico_stdlib/include/pico/stdlib.h:11:10: fatal error: pico/stdio.h: No such file or directory
   11 | #include "pico/stdio.h"
      |          ^~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\pico\src\SPL06.cpp.o] Error 1
*** [.pio\build\pico\src\BMP280.cpp.o] Error 1
In file included from src\tools.h:3,
                 from src\EMFButton.cpp:2:
C:\Users\Andrew\.platformio\packages\framework-arduino-mbed\cores\arduino/mbed/targets/TARGET_RASPBERRYPI/TARGET_RP2040/pico-sdk/common/pico_stdlib/include/pico/stdlib.h:11:10: fatal error: pico/stdio.h: No such file or directory
   11 | #include "pico/stdio.h"
      |          ^~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\pico\src\EMFButton.cpp.o] Error 1
============================================== [FAILED] Took 9.43 seconds ==============================================

andypnz, 2023-02-23:
Adding this to the build flag solved the above:

`build_flags = -D PICO_USB_DEVICE / PICO_USB_HOST`

But now this error at the end of the build:

`(platformio) PS C:\Users\Andrew\arp_osx> pio run
Processing raspberry-pi-pico (platform: wizio-pico; board: raspberry-pi-pico; framework: baremetal)
------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
<<<<<<<<<<<< WIZIO - RASPBERRY PI PICO ( PICOPROBE ) 2021 Georgi Angelov >>>>>>>>>>>>
CONFIGURATION: https://docs.platformio.org/page/boards/wizio-pico/raspberry-pi-pico.html
PLATFORM: WizIO - Raspberry Pi Pico (1.1.1+sha.35ad0e1) > WizIO - Raspberry Pi Pico ( PICOPROBE )
HARDWARE: RP2040 125MHz, 256KB RAM, 2MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, picoprobe)
PACKAGES:
 - framework-wizio-pico @ 1.1.1+sha.60881de
 - tool-wizio-pico @ 1.0.0+sha.3a504e1
 - toolchain-gccarmnoneeabi @ 1.70201.0 (7.2.1)

SDK RASPBERRYPI PI PICO RP2040 ( PICO - SDK )
  * OPTIMIZATION : -Os
  * STACK        : 2048
  * HEAP         : 2048
  * SPECS        : nano.specs
  * WIFI         : NO
  * BINARY TYPE  : default ['w25q080', 'memmap_default.ld', '0x10000000']
  * TINYUSB      : HOST

LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 8 compatible libraries
Scanning dependencies...
Dependency Graph
|-- wiring
Building in release mode
Linking .pio\build\raspberry-pi-pico\APPLICATION.elf
.pio\build\raspberry-pi-pico\src\main.o: In function `loop':
main.cpp:(.text.loop+0x3a): undefined reference to `tud_task'
.pio\build\raspberry-pi-pico\src\param.o: In function `handleUSBCmd()':
param.cpp:(.text._Z12handleUSBCmdv+0x4): undefined reference to `getchar_timeout_us'
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\raspberry-pi-pico\APPLICATION.elf] Error 1`

mstrens, 2023-02-23:
The code from Jeti reuse the code from oXs on pro mini which was based on a code found on internet.

I checked a little the code for Jeti GPS and in jeti.cpp I see a line of code with
longLatE7P = (longLat >= 0 ? longLat : longLat ) ;
The test has no effect and is probably wrong.
So I expect the line should be modified in
longLatE7P = (longLat >= 0 ? longLat : -longLat ) ;
I put on github a version 1.3.3 with this correction.
Let me know if this solved the issue.

mstrens, 2023-02-23:
Note about the compilation errors:
I am on windows and can't help but the missing files are normally part of the R2040 sdk.
I presume part of it missing or some path are wrong.

andypnz, 2023-02-23:
Thanks... I set up the VS code build environment, and WIZIO

It builds without errors now, but only a .elf file (not uf2).

Any ideas on creating a uf2?

I will test your modified jeti file shortly :-)

andypnz, 2023-02-23:
Your new uf2 file gives correct Lat and Lon for me - thank you....

Any ideas on creating a uf2 (rather than elf) file building on VS Code PlatformIO?

mstrens, 2023-02-23:
In vscode with platformio, I have an option build. THis does not create a uf2 but I also have an option upload and this creates always a uf2 even if the RP2040 is not connected

Satcomix, 2023-02-23:
Have you disconnected the Board incl.USB?
Normaly than it build in .uf2 and not in .elf
That happened to me, when i make my first build in Platformio on Windows 10 Pro.
Greetings, Torsten

andypnz, 2023-02-23:
> In vscode with platformio, I have an option build. THis does not create a uf2 but I also have an option upload and this creates always a uf2 even if the RP2040 is not connected

Thanks! Yep, found the upload option so all working - thanks again!

andypnz, 2023-02-23:
> Have you disconnected the Board incl.USB? Normaly than it build in .uf2 and not in .elf That happened to me, when i make my first build in Platformio on Windows 10 Pro. Greetings, Torsten

Humm... could not seem to generate an uf2, but the upload option worked :-)

closed 2023-02-23 by andypnz

mstrens, 2023-02-23:
In my case, having the board connected via usb or not does not change. Windows always creates the uf2 file when I select the upload option. For sure the upload it self fails when the board is not connected.
Note: the file is not created in uf2 folder but under a .pio folder.
I moved the file manually to a uf2 folder in order to avoid loading on github all compilation logs.