mstrens/oXs_on_RP2040 #54: Baro-data not transmitted by iBus protocol

Steini63, 2023-02-23:
Hi,

I tested iBus-protocol successful with V1, V2 and GPS (uBlox and CASIC), but can't get BMS280 to be recognized by EdgeTX

- Baro-data is not recognized after discovering sensors.
- V1-data is ok and displayed in Tx.
- Rx-telemetry is ok and displayed in Tx.
- Serial config data say: "Baro sensor is detected using BMP280 ..."
- I checked data-transfer via iBus: No Baro data transmitted.
- If I connect the same arrangement via SPort-protocol to X8R, then the baro-data is displayed in Tx.

Thank you for your efforts.
Frank

Full config:
```
Version = 1.3.3 
    Function                Pin   Change entering XXX=yyy (yyy=255 to disable)
Primary channels input    =  255  (PRI     = 5, 9, 21, 25)
Secondary channels input  =  255  (SEC     = 1, 13, 17, 29)
Telemetry . . . . . . . . =   21  (TLM     = 0, 1, 2, ..., 29)
GPS Rx  . . . . . . . . . =  255  (GPS_RX  = 0, 1, 2, ..., 29)
GPS Tx  . . . . . . . . . =  255  (GPS_TX  = 0, 1, 2, ..., 29)
Sbus OUT  . . . . . . . . =  255  (SBUS_OUT= 0, 1, 2, ..., 29)
RPM   . . . . . . . . . . =  255  (RPM     = 0, 1, 2, ..., 29)
SDA (I2C sensors) . . . . =   14  (SDA     = 2, 6, 10, 14, 18, 22, 26)
SCL (I2C sensors) . . . . =   15  (SCL     = 3, 7, 11, 15, 19, 23, 27)
PWM Channels 1, 2, 3 ,4   =  255  255  255  255 (C1 / C16= 0, 1, 2, ..., 15)
PWM Channels 5, 6, 7 ,8   =  255  255  255  255
PWM Channels 9,10,11,12   =  255  255  255  255
PWM Channels 13,14,15,16  =  255  255  255  255
Voltage 1, 2, 3, 4        =   26  255  255  255 (V1 / V4 = 26, 27, 28, 29)

Protocol is ibus(Flysky)
CRSF baudrate = 420000
Voltage parameters:
    Scales : 1.000000 , 1.000000 , 1.000000 , 1.000000 
    Offsets: 0.000000 , 0.000000 , 0.000000 , 0.000000 
    No temperature sensors are connected on V3 and V4
RPM multiplier = 1.000000
Baro sensor is detected using BMP280
    Sensitivity min = 100 (at 100)   , max = 300 (at 1000)
    Hysteresis = 5 
Acc/Gyro is not detected
First analog to digital sensor is not detected
Second analog to digital sensor is not detected
Foreseen GPS type is CADIS  :GPS is not (yet) detected
Failsafe type is HOLD

Config parameters are OK
Press ? + Enter to get help about the commands
```

mstrens, 2023-02-23:
I hope I find the bug.
I put a new version 1.3.4 on github in main branch.

Steini63, 2023-02-23:
Wow, THAT was fast!

Seems to be fixed, I get data now. 

Many Thanks
Frank

mstrens, 2023-02-23:
It was just a typo fout during a cut and paste

Steini63, 2023-02-24:
Fixed !

closed 2023-02-24 by Steini63