mstrens/oXs_on_RP2040 #55: Some questions about volatage and current under ELRS.

nostromo123, 2023-02-24:
We do have a discussion started by Doro here in a German forum about voltage and current under SPORT.

So I had the idea to do some tests here under ELRS.

**My 1st notice, both voltage and current are displayed in 1/10 V**. Here FV with both values

> Version = 1.3.3 
>     Function                Pin   Change entering XXX=yyy (yyy=255 to disable)
> Primary channels input    =    9  (PRI     = 5, 9, 21, 25)
> Secondary channels input  =   13  (SEC     = 1, 13, 17, 29)
> Telemetry . . . . . . . . =   10  (TLM     = 0, 1, 2, ..., 29)
> GPS Rx  . . . . . . . . . =   11  (GPS_RX  = 0, 1, 2, ..., 29)
> GPS Tx  . . . . . . . . . =   12  (GPS_TX  = 0, 1, 2, ..., 29)
> Sbus OUT  . . . . . . . . =  255  (SBUS_OUT= 0, 1, 2, ..., 29)
> RPM   . . . . . . . . . . =  255  (RPM     = 0, 1, 2, ..., 29)
> SDA (I2C sensors) . . . . =   14  (SDA     = 2, 6, 10, 14, 18, 22, 26)
> SCL (I2C sensors) . . . . =   15  (SCL     = 3, 7, 11, 15, 19, 23, 27)
> PWM Channels 1, 2, 3 ,4   =    0    1    2    3 (C1 / C16= 0, 1, 2, ..., 15)
> PWM Channels 5, 6, 7 ,8   =  255  255  255  255
> PWM Channels 9,10,11,12   =  255  255  255  255
> PWM Channels 13,14,15,16  =  255  255  255  255
> Voltage 1, 2, 3, 4        =   26   27  255  255 (V1 / V4 = 26, 27, 28, 29)
> 
> Protocol is CRSF (=ELRS)
> CRSF baudrate = 420000
> Voltage parameters:
>     Scales : 1.000000 , 1.000000 , 1.000000 , 1.000000 
>     Offsets: 0.000000 , 0.000000 , 0.000000 , 0.000000 
>     No temperature sensors are connected on V3 and V4
> RPM multiplier = 1.000000
> Baro sensor is detected using BMP280
>     Sensitivity min = 100 (at 100)   , max = 300 (at 1000)
>     Hysteresis = 5 
> Acc/Gyro is not detected
> First analog to digital sensor is not detected
> Second analog to digital sensor is not detected
> Foreseen GPS type is Ublox  :GPS is detected and has a fix
> Failsafe type is HOLD
> 
> Config parameters are OK
> Press ? + Enter to get help about the commands
> 
>  >> 
> processing cmd
> 
> Cmd to execute:   FV
> 
> GPS Latitude = 51.1881536 degree
> GPS Longitude = 6.4616280 degree
> GPS Groundspeed = 8 cm/s
> GPS Heading = 57.360000 degree
> GPS Altitude = 3980 cm
> GPS Num sat. = 108 cm
> GPS Date J M A = 24 2 23 
> GPS Time H M S = 12 55 18 
> GPS Pdop = 206 
> GPS Home bearing = 33 degree
> GPS Home distance = 56 m
> **Volt 1 = 2576 mVolt
> Current (Volt 2) = 2931 mA**
> Capacity (using current) = 232 mAh
> Vspeed = -2 cm/s
> Baro Rel altitude = 187 cm
> 

Here the voltage supplied and  the current

[image]
[image]

and the display on the radio.

[image]

I guess, the slight deviation between FV values and DVM values could be trimmed a bit with offset and scale in CONFIG?

**2nd. My aim here is to simulate a Lipo/current sensor situation on a RC model with those two voltage sources**
 
V1 a 3S Lipo in range 2.5V ≙ 10.5V to 3.0V ≙ 12.6V
and 
V2 an ACS768 50A Uni with 330mV ≙ 0A (Vcc 3.3V and at 0A Vcc *0.1) and with 3.0V ≙ 44.5A (60 mV/A)

and there are a lot of possibilities to make adjustments.

In oXs_on_RP2040 those two

Voltage parameters:
**Scales : 1.000000 , 1.000000** , 1.000000 , 1.000000
**Offsets: 0.000000 , 0.000000** , 0.000000 , 0.000000 

and on radio side

[image]

**precision, ratio and offset.**

Could you please explain, what your ideas for setting up such a simulation are. Only on oXs_on_RP2040 side or in a combination with the radio settings?

On oXs I did voltage and current measurement adjustment in iterations and had very good results. sometimes 3 or 4 loops were enough, sometimes up to 12 loops, but the results were worth it! 

br - KH

mstrens, 2023-02-24:
The logic to calculate Volt and Current scale and offset are similar if you have both a Voltmeter and an Ampmeter.
If you do not have an Ampmeter, you have to use the specifications found in the datasheet of the sensor (and so it can be less accurate).

Take the example of the the voltage first.
For sure, you have to use voltage divider to reduce the voltage on the pin to max about 3V.
Then best is to make 2 measurements: one with a voltage nearly equal to the min of the expected range (e.g. about 9 or 10V in your case) and one with the max expected voltage (e.g. 13V).
So you first set scale = 1 and offset = 0 and 
- you first apply low voltage. Imagine the voltmeter gives you 9100mV and the FV function gives you 2120
- you apply high voltage. Imagine the voltmeter five you 12500mV and the FV function gives you 2900
Then you can calculate 
- scale : it is (12500-9100)/(2900-2120) = 4,3589
- offset = (2120*scale)- 91200 = 141.

For current it is exactly the same if you can measure the current.
If not, you have first to calculate the voltage that the current sensor would deliver.
For a V2 an ACS768 50A Uni with 330mV ≙ 0A (Vcc 3.3V and at 0A Vcc *0.1) and with 3.0V ≙ 44.5A (60 mV/A), I expect it should be
- 330 mV (for 0 A)
- 330mV + 60mV/A /5V * 3.3 V * 50A (for 50A) = 2310mV for 50.
Note: I think the sensitivity depends on Vcc too so it is less at 3.3V than at 5V.
So it means that we expect that FV will be 330 for 0A and 2310 for 50A.
Using the same formules as before you have
- scale : (5000-0)/(2310-330) = 2,52
- offset = (330*scale)- 0 = 131.
In this case, we apply a theoretical calculation assuming that both ACS and RP2040 are just perfect.
The only way to take care of deviations is to be able to measure the current with an ampmeter and to proceed just like for voltage.

mstrens, 2023-02-24:
If you want, you can write a text to explain with fotos, ... and I could add it to the doc folder.
This could perhaps help other people too

nostromo123, 2023-02-24:
Sorry for the confusion.

My 1st notice was about the voltage display on the radio. I see 1/10 volt. I don't see 2.53 V, I only see 2.50V 2.60V 2.70V and not 1/100.

Here is an old sheet I used with the old oXs for adjusting voltage measurement.

[image]

2nd black arrow from above. I started with offset 0 and scale 1. I supply with 12.60V (blue) and see on the radio 12.52V (red), then 10.50V and see 10.43V. Then I adjust offset to 36 and scale 0.999 and test next. The RP2040 has under ELRS 2 high resolution voltage inputs and I only see  2.5, 2.6 and 2.7V on the radio ????

You stated this in the documentation and I don't see neither 0.010V nor 0.001A

VOLT1       0.001V           V1 pin is defined                            Volt                       0.010V
VOLT2       0.001V=0.001A    V2 pin is defined                            Current                    0.001A

br KH

mstrens, 2023-02-24:
I am afraid I made some error or in the code or in the doc.
I see in the code that I divide the value (measured in mv ) by 100 when I send it to ELRS (so in 0.1V)
In the doc I say it is sent in 0.01V.
This is inconsistent.
Before we continue to discuss this, could you make a simple test.
Set offset =0 and scale = 1 and apply a voltage and measure the voltage on the RP2040 pin. Ideally chose the voltage in order to get between 1 and 3V on the rp2040 pin.
Say me what is the value given by FV command and what is the value on the display of the transmitter.
So I will know what is wrong (the code or the doc).

nostromo123, 2023-02-24:
The values in my first post here, both FV values, the displays on the DVMs and the display on the radio are those values requested. The DVMs are directly on V1 and V2 input. So you are transmitting in 0.1V. 

I use 2 voltage sources 0 .. 30V adjustable each with 10 turn potentiometers for fine adjustment and 1:10 voltage dividers .....

mstrens, 2023-02-24:
So, with offset = 0 and scale = 1, the value on the display matches the real voltage.
Dividing the FV value by 100 while sending is correct and there is a bug in the doc (values are in 0.1V (and 0.1A).
So there is no solution for having always 0 on the second decimals.
One improvement I could make is to add a rounding (so 2.590 V would be sent as 2.6 instead of 2.5

mstrens, 2023-02-24:
About scale and offset:
if 12.6V (real) gives 12,52 (on display) and 10.5 (real) gives 10.43.
Then I calculate (with the formula I provide above)
Scale = 1.0048
Offset = -20.1.
Did you try those values?

nostromo123, 2023-02-24:
> About scale and offset: if 12.6V (real) gives 12,52 (on display) and 10.5 (real) gives 10.43. Then I calculate (with the formula I provide above) Scale = 1.0048 Offset = -20.1. Did you try those values?

About your question. I joint oXs in 2015 and in the beginning used your formula, but found it easier for me to do it by iteration. I even did target comparison as graph on a piece of paper, as both lines were close to each other.
I'll have a second look with your formula (same as in oXs) here, but a rounded voltage in radio display and then also in radio telemetry is a bit disappointing! Say, as it is, voltage in 0.1V. And what about  '0.001A'??

mstrens, 2023-02-24:
Amp is internally measured in mAmp in oXs but is converted to 0.1A when transmitted to ELRS. It seems that this is the format required by ELRS/CRSF

nostromo123, 2023-02-26:
Could just make a test with simulated current and voltage.

> V1 voltage and V2 current under ELRS.
> 3S Lipo in range 10.5 - 12.6V and one ACS758 50A/U.
> 
> ACS with 5V Vcc, so 50mV ≜ 0A and 3500mV ≜ 50A
> 
> V1 input 10.50V --> TX display 10.3V
>      input 12.60V --> TX display 12.4V
> 
> V2 input 500mV --> TX display 0.2A
>      input 3500mV --> TX display 50.3A

But Michel let me make some remarks:

With these lines in oXS 

> 
> #define RESISTOR_TO_GROUND
> #define RESISTOR_TO_VOLTAGE
> #define OFFSET_VOLTAGE
> #define SCALE_VOLTAGE
> 
> 
> #define MVOLT_AT_ZERO_AMP              
> #define MVOLT_PER_AMP 
> #define RESISTOR_TO_GROUND_FOR_CURRENT
> #define RESISTOR_TO_CURRENT_SENSOR

things were much easy to set up, as the voltage divider was directly processed in the code from start. When the voltage dividers was calculated good, you even could skip further adjustment, which I assume, most users did. And you could read your data in the TX, not in mV/mA on a terminal, were you had to enter 'FV' to see, what is going on ....

Regards - KH

nostromo123, 2023-02-26:
About voltages, it is a bit disappointing, that under the  CRSF protocol only 0.1V resolution is on the TX side. Can on your side 'RP2040' a modification be implemented for rounding the send voltages a bit? You mentioned something like that.

br - KH

mstrens, 2023-02-27:
I added rounding in version 1.4.1 for all fields/all protocols where it makes sense

closed 2023-03-04 by nostromo123