mstrens/oXs_on_RP2040 #56: Ver.1.4.1_MAIN-HOTT Protocol

Satcomix, 2023-02-28:
Hello Mstrens,
There is something going wrong with Ver.1.4.1 Main in HOTT Protocol.
br,
Torsten

processing cmd

Cmd to execute:   

Version = 1.4.1 
    Function                Pin   Change entering XXX=yyy (yyy=255 to disable)
Primary channels input    =  255  (PRI     = 5, 9, 21, 25)
Secondary channels input  =  255  (SEC     = 1, 13, 17, 29)
Telemetry . . . . . . . . =    0  (TLM     = 0, 1, 2, ..., 29)
GPS Rx  . . . . . . . . . =   12  (GPS_RX  = 0, 1, 2, ..., 29)
GPS Tx  . . . . . . . . . =   13  (GPS_TX  = 0, 1, 2, ..., 29)
Sbus OUT  . . . . . . . . =  255  (SBUS_OUT= 0, 1, 2, ..., 29)
RPM   . . . . . . . . . . =    1  (RPM     = 0, 1, 2, ..., 29)
SDA (I2C sensors) . . . . =   14  (SDA     = 2, 6, 10, 14, 18, 22, 26)
SCL (I2C sensors) . . . . =   15  (SCL     = 3, 7, 11, 15, 19, 23, 27)
PWM Channels 1, 2, 3 ,4   =  255  255  255  255 (C1 / C16= 0, 1, 2, ..., 15)
PWM Channels 5, 6, 7 ,8   =  255  255  255  255
PWM Channels 9,10,11,12   =  255  255  255  255
PWM Channels 13,14,15,16  =  255  255  255  255
Voltage 1, 2, 3, 4        =   26   27   28   29 (V1 / V4 = 26, 27, 28, 29)

Protocol is Hott
CRSF baudrate = 420000
Voltage parameters:
    Scales : 1.000000 , 1.000000 , 1.000000 , 1.000000 
    Offsets: 0.000000 , 0.000000 , 0.000000 , 0.000000 
    No temperature sensors are connected on V3 and V4
RPM multiplier = 1.000000
Baro sensor is detected using MS5611
    Sensitivity min = 100 (at 100)   , max = 300 (at 1000)
    Hysteresis = 5 
Acc/Gyro is not detected
First analog to digital sensor is not detected
Second analog to digital sensor is not detected
Foreseen GPS type is Ublox  :GPS is detected and has a fix
Failsafe type is HOLD

Config parameters are OK
Press ? + Enter to get help about the commands

 >> 
processing cmd

Cmd to execute:   FV

GPS Latitude = xx.xxxxxxx degree
GPS Longitude = x.xxxxxxx degree
GPS Groundspeed = 7 cm/s
GPS Heading = 193.550000 degree
GPS Altitude = 4602 cm
GPS Num sat. = 110 cm
GPS Date J M A = 28 2 23 
GPS Time H M S = 6 52 15 
GPS Pdop = 157 
GPS Home bearing = 323 degree
GPS Home distance = 12 m
**Volt 1 = 607 mVolt -----------------> 57.0 Volt Sensor1
Current (Volt 2) = 632 mA ----------> 457,0 Ampere
Volt 3 = 669 mVolt -----------------> 57,0 Volt
Volt 4 = 219 mVolt -----------------> 56,6 Volt
Capacity (using current) = 8 mAh ---> 24320 mAh** 
Vspeed = -2 cm/s
Baro Rel altitude = -15 cm
**RPM = 999 Hertz --------------------> 525280 U/min**

mstrens, 2023-02-28:
It is perhaps a bug in the way I made the rounding.
I made a change. It is in 1.4.2 on github main.
I did not tested it.

Satcomix, 2023-02-28:
Hello Mstrens,
i installed Ver.1.4.2, but nothing changed, all values are very strange.
**Volt 1 = 607 mVolt -----------------> 57.0 Volt Sensor1
Current (Volt 2) = 632 mA ----------> 457,0 Ampere
Volt 3 = 669 mVolt -----------------> 57,0 Volt Sensor2
Volt 4 = 219 mVolt -----------------> 56,6 Volt Main Voltage
Capacity (using current) = 8 mAh ---> 24320 mAh
Vspeed = -2 cm/s
Baro Rel altitude = -15 cm -------------> goes to -9999 Meter after 0 Meter to +9999 Meter
RPM = 999 Hertz --------------------> 525280 U/min**

Satcomix, 2023-02-28:
**To check the Graupner MC28 HOTT, I installed version 1.3.4 MAIN again and everything is displayed correctly.**
Volt 1 = 607 mVolt -----------------> 0.6 Volt Sensor1
Current (Volt 2) = 632 mA ----------> 0.6 Ampere
Volt 3 = 669 mVolt -----------------> 0.6 Volt Sensor2
Volt 4 = 219 mVolt -----------------> 0.2 Volt Main Voltage
Capacity (using current) = 10 mAh --->  10mAh
Vspeed = -0 cm/s
Baro Rel altitude = -0 cm -------------> 0
RPM = 999 Hertz --------------------> 990-1000 U/min

mstrens, 2023-02-28:
I think I found the bug (in rounding)
See new version 1.4.3 in main

Satcomix, 2023-02-28:
Hello Mstrens,
Now **everything is displayed correctly** again in **Version 1.4.3_MAIN** with **HOTT** protocol, and I can also see your rounding change.
From 0.05 it is rounded up.
Example 0.672V is displayed in the FV at V2/Ampere, in the handheld 0.7 Ampere appears.
In a test with FrSky X20S
I could also not find any abnormalities in the **S.PORT** Protocol.
Thank you for the work.
Greetings,
Torsten

closed 2023-02-28 by Satcomix