mstrens/oXs_on_RP2040 #58: F.PORT2 Ver.1.4.x_TEST

Satcomix, 2023-03-02:
Hello Mstrens,
The first test with Ver.1.4.5 does not bring any connection to 16CH PWM(SBUS), and no connection to telemetry (V1-V4).
Two error messages appear:
When TLM=0 and PRI=255
Error in parameters: at least one PWM pin is defined but no pin for Primary nor Secondary channels input (PRI or SEC)
WhenTLM=0 and PRI=5
Error in parameters: For Fport2, primary channel input (PRI) pin may not be defined (but TLM must be defined)
br, 
Torsten

processing cmd

Cmd to execute:   

Version = 1.4.5 
    Function                Pin   Change entering XXX=yyy (yyy=255 to disable)
Primary channels input    =  255  (PRI     = 5, 9, 21, 25)
Secondary channels input  =  255  (SEC     = 1, 13, 17, 29)
Telemetry . . . . . . . . =    0  (TLM     = 0, 1, 2, ..., 29)
GPS Rx  . . . . . . . . . =  255  (GPS_RX  = 0, 1, 2, ..., 29)
GPS Tx  . . . . . . . . . =  255  (GPS_TX  = 0, 1, 2, ..., 29)
Sbus OUT  . . . . . . . . =  255  (SBUS_OUT= 0, 1, 2, ..., 29)
RPM   . . . . . . . . . . =  255  (RPM     = 0, 1, 2, ..., 29)
SDA (I2C sensors) . . . . =  255  (SDA     = 2, 6, 10, 14, 18, 22, 26)
SCL (I2C sensors) . . . . =  255  (SCL     = 3, 7, 11, 15, 19, 23, 27)
PWM Channels 1, 2, 3 ,4   =    1    2    3    4 (C1 / C16= 0, 1, 2, ..., 15)
PWM Channels 5, 6, 7 ,8   =  255    6    7    8
PWM Channels 9,10,11,12   =    9   10   11   12
PWM Channels 13,14,15,16  =   13   14   15  255
Voltage 1, 2, 3, 4        =   26   27   28   29 (V1 / V4 = 26, 27, 28, 29)

Protocol is Fport2(Frsky)
CRSF baudrate = 420000
Voltage parameters:
    Scales : 1.000000 , 1.000000 , 1.000000 , 1.000000 
    Offsets: 0.000000 , 0.000000 , 0.000000 , 0.000000 
    No temperature sensors are connected on V3 and V4
RPM multiplier = 1.000000
Baro sensor is not detected
Acc/Gyro is not detected
First analog to digital sensor is not detected
Second analog to digital sensor is not detected
Foreseen GPS type is Ublox  :GPS is not (yet) detected
Failsafe type is HOLD
Error in parameters: at least one PWM pin is defined but no pin for Primary nor Secondary channels input (PRI or SEC)

Attention: error in config parameters
Press ? + Enter to get help about the commands

 >>

mstrens, 2023-03-02:
I put a new version of github (test).
For fport2, I changed the logic:
- TLM must be = 255
- PRI must be defined.
There were less checks to change when I do it in this way.
So now, I can at least get a validation of the config.
I still do not know it the rest is OK

Satcomix, 2023-03-02:
OK, so i put FBUS to PRI=5 with 1K resistor.

Satcomix, 2023-03-02:
Next Test Ver.1.4.6
No results with SBUS and Telemetry

Cmd to execute:   

Version = 1.4.6 
    Function                Pin   Change entering XXX=yyy (yyy=255 to disable)
Primary channels input    =    5  (PRI     = 5, 9, 21, 25)
Secondary channels input  =  255  (SEC     = 1, 13, 17, 29)
Telemetry . . . . . . . . =  255  (TLM     = 0, 1, 2, ..., 29)
GPS Rx  . . . . . . . . . =  255  (GPS_RX  = 0, 1, 2, ..., 29)
GPS Tx  . . . . . . . . . =  255  (GPS_TX  = 0, 1, 2, ..., 29)
Sbus OUT  . . . . . . . . =  255  (SBUS_OUT= 0, 1, 2, ..., 29)
RPM   . . . . . . . . . . =  255  (RPM     = 0, 1, 2, ..., 29)
SDA (I2C sensors) . . . . =  255  (SDA     = 2, 6, 10, 14, 18, 22, 26)
SCL (I2C sensors) . . . . =  255  (SCL     = 3, 7, 11, 15, 19, 23, 27)
PWM Channels 1, 2, 3 ,4   =    1    2    3    4 (C1 / C16= 0, 1, 2, ..., 15)
PWM Channels 5, 6, 7 ,8   =  255    6    7    8
PWM Channels 9,10,11,12   =    9   10   11   12
PWM Channels 13,14,15,16  =   13   14   15  255
Voltage 1, 2, 3, 4        =   26   27   28   29 (V1 / V4 = 26, 27, 28, 29)

Protocol is Fport2(Frsky)
CRSF baudrate = 420000
Voltage parameters:
    Scales : 1.000000 , 1.000000 , 1.000000 , 1.000000 
    Offsets: 0.000000 , 0.000000 , 0.000000 , 0.000000 
    No temperature sensors are connected on V3 and V4
RPM multiplier = 1.000000
Baro sensor is not detected
Acc/Gyro is not detected
First analog to digital sensor is not detected
Second analog to digital sensor is not detected
Foreseen GPS type is Ublox  :GPS is not (yet) detected
Failsafe type is HOLD

Config parameters are OK
Press ? + Enter to get help about the commands

 >>

mstrens, 2023-03-02:
In fport2.cpp you have now
while (! queue_is_empty(&fportRxQueue)) {
            // we get the value in the queue
            queue_try_remove (&fportRxQueue,&data);
....

Can you add one line to get
while (! queue_is_empty(&fportRxQueue)) {
            // we get the value in the queue
            queue_try_remove (&fportRxQueue,&data);
            printf("%x\n", (uint8_t) data);
         
So, if oXs can detect that it receives some char from the RX, it will print them.
This is the first point to debug.

            print("%x\n", (uint8_t) data);

Satcomix, 2023-03-02:
d3
ff
d9
ff
d9
fe
d9
fe
ff
d3
ff
fe
d9
ff
d3
fe
d3
fe
ff
d9
fe
d9
ff
d3
ff
d3
fe
d9
ff
fe
d3
fe
d9
ff
d3
fe
d9
ff
d9
ff
d9
fe
d3
ff
d9
fe
d3
fe
d9
ff
d3
fe
d3
ff
d3
ff
d9
fe
d9
fe
ff
d9
ff
d9
fe
d9
ff
d3
fe
d9
fe
d3
ff
d3
fe
ff
d3
ff
d3
fe
d9
ff
d9
fe
d3
fe
d9
ff
d9
fe
d9
ff
d3
ff
d3
fe
d3
ff
fe
fe
d9
ff
d9
fe
d9
ff
d3
ff
d9
fe
d9
fe
d3
ff
d3
ff
d3
fe
d3
ff
fe
d3
fe
d3
ff
d9
fe
d9
ff
d9
ff
d9
fe
d9
ff
d3
fe
d9
fe
d9
ff
d9
fe
d3
ff
d9
ff
d3
fe
d9
ff
d3
fe
d3
fe
ff
d9
fe
d9
ff
d9
ff
d9
fe
d9
fe
d9
ff
d9
ff
fe
d9
ff
d9
fe
d3
fe
d9
ff
d3
fe
d3
ff
ff
d3
fe
d3
ff
d9
fe
d3
fe
d9
ff

mstrens, 2023-03-02:
It is strange that it is always the same char.
Could you provide also a capture?

When the data are capture, does the receiver have a connection with the handset.
I think it would be better in order to see also the rc channels frames.

Satcomix, 2023-03-02:
Hello Mstrens,
Some questions:
the settings for the port,
115200 or 460800?
Inverted (i think no?)
8 bit transfer,
1 stop bit,
No parity,
Least significant bit send first (default)
Special mode: none

mstrens, 2023-03-02:
fport2 is 115200 (in oXs).
460800 is for Fbus.

As far I know, it is Inverted uart and 8N1

mstrens, 2023-03-02:
if level = LOW when it is iddle, then it is inverted.

Satcomix, 2023-03-02:
[2023-03-02_12-41-46.txt](https://github.com/mstrens/oXs_on_RP2040/files/10870889/2023-03-02_12-41-46.txt)
 settings: 115200 8N1 Inverted, 
When i set to autobaud, there is something with 460800
I think, i must look for informations about X20S and TD-R10, because i can only set Rceiver to SBUS, S.PORT, FBUS and not F.PORT2

mstrens, 2023-03-02:
When oXs is not connected, I expect (based on what I read) that, when connected to the handset, the receiver should generate a long frame of 24 bytes (rc channels) immediately followed by a shorter frame of 8 bytes. So a total of 32 bytes and then it should wait for a reply.

I expect that if the RX has not connection with handset then only the short frame (8 bytes) should exist (still not sure this is so)

mstrens, 2023-03-02:
The problem is indeed the baudrate.
I do not know how to get Fport2.
Mike on rcgroups must know for sure.

Satcomix, 2023-03-02:
[2023-03-02_12-51-26.txt](https://github.com/mstrens/oXs_on_RP2040/files/10870971/2023-03-02_12-51-26.txt)
Result, when oXs is not connected, only data( 115200) from TD-R10 Receiver FBUS

Satcomix, 2023-03-02:
[2023-03-02_12-54-14.txt](https://github.com/mstrens/oXs_on_RP2040/files/10870996/2023-03-02_12-54-14.txt)
The receiver talks with 460800 not inverted 8N1

mstrens, 2023-03-02:
At 460800 baud I see the expected frames.
RC channel begins with 0X18 + 0XFF; len =26 (0X18 +2)
Pooling begins with 0X08; len = 10 (8+2)

So the question is : how to activate FPORT2 instead of Fbus?
Could it be that Fport2 is not supported anymore by frsky?

Satcomix, 2023-03-02:
Hello Mstrens,
I think Mike mentioned something that FPORT2 is now calling FBUS, but with some changes.
I can set my receiver to SBUS with 16CH or 24CH,
and to FBUS, not FPORt2.
Sorry....
Let's hear what Mike thinks about it.
br,
Torsten

mstrens, 2023-03-02:
I do not understand.
In ardupilot, I find code for fport2. 
The code says that baudrate = 115200.
As far I understood Mike post Fport2 is 115200 and Fbus similar but with a higher speed.
Perhaps can you put a question on a frsky forum with a mention about the type of Tx and Rx you have.

Satcomix, 2023-03-02:
I find some Informations for FPORT2 and FBUS on RCG:
https://www.rcgroups.com/forums/showthread.php?3395177-Official-OpenTX-version-2-3-Discussion-Thread/page519
br,
Torsten

mstrens, 2023-03-02:
here what I read on rcgroups (from Mike)
My understanding is we have:
SPort - 57600 baud.
FPort1 - 115200 baud single connection to a flight controller.
FPort2 - 115200 baud supports multiple sensors and servos.
FBUS - 460800 baud supports multiple sensors and servos (not seen this yet)

Originally firmware for the Archer receivers did FPort2, but it seems recent firmware changes this to FBUS.
I have an Archer R8 PRO with 2.1.8 firmware on it. On ETHOS I have the options of SPort, FPort, and FBUS.
SPort gives SPort.
FPort gives FPort1.
FBUS gives FPort2 (115200).

I feel ETHOS should look at the firmware on the Rx and show FPort2 or FBUS, depending on which is available.

I don't (currently) have any ADV sensors, so I cannot test if they work with FPort2 (115200 baud), as well as SPort and FBUS (460800 baud).
I would be interested if anyone can test this.

I have an openXsensor configured to work on FPort2 now.

mstrens, 2023-03-02:
I would have to check if I can handle a baudrate of 460800 in a RP2040 pio (which is not a hardware uart)

mstrens, 2023-03-02:
in fport2.cpp there are 2 lines with 115200.
You can try to change 115200 with 460800.

Note: I have to leave for a few hours

Satcomix, 2023-03-02:
Hello Mstrens,
I also have no ADV sensors, only S.PORT Sensors like SBEC, FLVSS, MLVSS......
Ok i change it and make a build.
See you in a few hours.
br,
Torsten

mstrens, 2023-03-02:
When you use 460800 bauds, do you still get some some messages on the serial terminal?
If yes, can you make a copy/paste of part of it.

When you make a test, can you use the option "capture" and not the option "export" of the logic analyser.
Then I can also view the trace in a graphical way.

mstrens, 2023-03-02:
I put on github in test version 1.4.7
I changed the baudrate (to 460800) and I added some debug messages (to be sent to PC via usb).
Can you test it and provide:
- a logic analyser caapture
- extract of the messages on the serial terminal

Satcomix, 2023-03-02:
I can only save data as .txt, .csv, .bin and .kvdat
.kvdat was the last .zip file
here is the data .bin file as .zip
[2023-03-02_19-01-09.zip](https://github.com/mstrens/oXs_on_RP2040/files/10874334/2023-03-02_19-01-09.zip)

Satcomix, 2023-03-02:
Here is the link for Kingst VIS software, normally than you can open and show the .kvdat files
http://www.qdkingst.com/en/download
[2023-03-02_19-13-58.zip](https://github.com/mstrens/oXs_on_RP2040/files/10874468/2023-03-02_19-13-58.zip)

mstrens, 2023-03-02:
OK I am currently installing the soft on my pc.
Looking at the timestamp in the txt files, I already saw that 
- the 2 frames (rc channels + pooling request) are just sent in one set
-  there is 8 ms interval between the first bytes of 2 successive frames.
At this stage this is already good to know.

On the PC, with the latest version 1.4.7)do you get some debug messages or just nothing?

Satcomix, 2023-03-02:
[Ver.1.4.7_TERMINAL.txt](https://github.com/mstrens/oXs_on_RP2040/files/10874789/Ver.1.4.7_TERMINAL.txt)
Sorry i forget the .txt file

mstrens, 2023-03-02:
OK, this helped.
I found at least one bug.
New version is 1.4.8

mstrens, 2023-03-02:
OK we make progress.
Rc channels frames seem to be read.
I fixed now another bug for the pooling request frame.
Version is 1.4.9

Satcomix, 2023-03-02:
I make some movements with CH3 and CH4
[2023-03-02_20-17-07.zip](https://github.com/mstrens/oXs_on_RP2040/files/10875100/2023-03-02_20-17-07.zip)

mstrens, 2023-03-02:
I put version 1.4.10 where I 
- removed most debug message
- added one that should show the telemetry frame sent by oXs (if any).
To test it, you should at least activate a pin that should generates telemetry (e.g. V1 ).
You can activate several sensors if you want. 
Just say me which one are supposed to be sent.
The file TERMINAL.txt you should make should contains at least 3 or 5 sec of messages (I expect it should not be too long because I removed the lines that contains only one char)

Satcomix, 2023-03-02:
I can`t get into the Terminal to change values, but i know in the first version i activate V1,V2,V3,V4 at GPio26-29
At handheld display is nothing shown in telemetry data, except the values from receiver.

[2023-03-02_20-44-46.zip](https://github.com/mstrens/oXs_on_RP2040/files/10875277/2023-03-02_20-44-46.zip)
[Ver.1.4.10_TERMINAL.txt](https://github.com/mstrens/oXs_on_RP2040/files/10875279/Ver.1.4.10_TERMINAL.txt)

mstrens, 2023-03-02:
Sorry, I probably make a mistake with previous version.
Normally all lines with just 1 char should not be there.
For safety, I just recompiled as 1.4.11
Please try again

Satcomix, 2023-03-02:
What i can see in VIS is CH1 is FBUS, CH2 is PWM CH1 1049us, Ch3 is PWM CH2 with 968us, and the rest of PWM Channels have also 968us instead of 1500us for CH1-7

mstrens, 2023-03-02:
FYI: in my case (using vscode monitor) I can type commands even if the display is scrolling very fast.
In fact with vscode monitor, I click in the output part of the screen and then I can enter command. The main drawback is that I never see what I type.

Satcomix, 2023-03-02:
No entry to terminal values
[Ver.1.4.11_TERMINAL.txt](https://github.com/mstrens/oXs_on_RP2040/files/10875385/Ver.1.4.11_TERMINAL.txt)
[2023-03-02_21-05-35.zip](https://github.com/mstrens/oXs_on_RP2040/files/10875387/2023-03-02_21-05-35.zip)

Satcomix, 2023-03-02:
> FYI: in my case (using vscode monitor) I can type commands even if the display is scrolling very fast. In fact with vscode monitor, I click in the output part of the screen and then I can enter command. The main drawback is that I never see what I type.

I`m not a Softworker, i work with Satellte Communication Systems, and i must see wht i type, that makes it a "little" bit slower.
I am amazed by your speed, your skills and what you have already created with openXsensor for all the model builders around the world over the years.

mstrens, 2023-03-02:
I am currently looking why you do not get telemetry data on the handset.

I see on the capture (logic analyser) that oXs generates frames for telemetry.
The frame seems to follow the specifications given by Mike.
So I do not know (at this stage) why the value does not appears on the handset.

Satcomix, 2023-03-02:
Normally i must see:
VFAS= V1
FASS= V2
DiY5123 Fuel/Capacity
DIY5113 V3
DIY5114 V4

mstrens, 2023-03-02:
The txt file shows that oXs created at least one frame for DIY5114.
The file is to short to see if there are other.
If you want, I could remove about 7 debug messages on 8.

Satcomix, 2023-03-02:
Its a little bit longer, i hope it will help.

[Ver.1.4.11_TERMINAL_Long.txt](https://github.com/mstrens/oXs_on_RP2040/files/10875595/Ver.1.4.11_TERMINAL_Long.txt)

mstrens, 2023-03-02:
I can confirm that oXs generates frames for 
VFAS (0210)
Curr(0200)
DIY 5113
DIY 5114
DIY 5123
Those codes exist in the txt file.

At this stage, best would be to find someone that has
- a receiver that support FBUS
- a sensor working with FBUS
- a logic analyser 
So it would be possible to be sure on the structure of the frame to be sent and on their timing

Satcomix, 2023-03-02:
OK,it was along day.
Maybe there will be someone who can fulfill all these qualities, but thank you very much for all the work and time.
Thank you very much.
Greetings,
Torsten

Satcomix, 2023-03-03:
Hello Mstrens, 
This morning i make a order about two ADV FBUS Sensors, one for Current, the other for LIPO.
I hope, that could help a little bit for the new FBUS protocol.
Greetings, 
Torsten

mstrens, 2023-03-03:
Thanks.
It will probably help.
I still do not understand why you do not get the telemetry data on the handset.
I noticed in the txt files that the frsky Rx recognize at least some of the reply from oXs because when the rx detect that a device replies, it increases the frequency of pooling of this device and this is the case.
So perhaps, that the Rx was able to decode some replies from oXs but not all of them.

This could be because the signal on the FBUS pin of the receiver is not "square" enough and is corrupted for the RX uart
The FBUS use a high baudrate and this is perhaps not compatible with the presence of a 1K resistor.

Perhaps you can try to reduce the value of the resistor to e.g. 100 ohm.

mstrens, 2023-03-03:
OK
Will perhaps work with just the 180 ohm. Let see!

Satcomix, 2023-03-04:
Hello Mstrens,
The two FBUS sensors have just been delivered.
It is FrSky FAS 40A ADV and FLVS ADV Lipo Sensor without display.
See Appendix.
Greeting,
Torsten
[2023-03-04_12-12-48.zip](https://github.com/mstrens/oXs_on_RP2040/files/10888129/2023-03-04_12-12-48.zip)
[Ver.1.5.0.txt](https://github.com/mstrens/oXs_on_RP2040/files/10888131/Ver.1.5.0.txt)

Satcomix, 2023-03-04:
As I have already written, I cannot get into the terminal menu with the oXs_RP2040 connected to the receiver.
Im Anhang befindet sich der Terminal Ausdruck von Ver.1.5.0 ohne angeschlossenem Receiver
[Ver.1.5.0_TERMINAL.txt](https://github.com/mstrens/oXs_on_RP2040/files/10888198/Ver.1.5.0_TERMINAL.txt)

mstrens, 2023-03-04:
I put a version 1.5.1 where I remove most debug msg for Fport2.
I also reduce a little the delay between the last char of a polling request and the first byte of the oXs answer (based on what I saw in your capture.
Still I am not convince this will help.
Did you already try to remove the 1K resistor or look at the quality of the signal with a scope?

Satcomix, 2023-03-04:
Yes in the last Tests the 1K resistor allready remove, only 180 Ohm at GPIO, and the signal edge is very good, with or without 1K.

mstrens, 2023-03-04:
And I presume that
- the original frsky sensors are discovered
- oXs is not discovered
- If so the bug is clearly in oXs

Satcomix, 2023-03-04:
The timing has gotten better, but still needs some improvement.
My measurement with FAS sensor:
7.17508ms
302.98us
6.67182ms
329.13us
6.67ms
7.17ms
303us
6.66ms
328us
6.66ms
7.17ms
[2023-03-04_13-38-02.zip](https://github.com/mstrens/oXs_on_RP2040/files/10888336/2023-03-04_13-38-02.zip)

processing cmd

Cmd to execute:   

Version = 1.5.1 
    Function                Pin   Change entering XXX=yyy (yyy=255 to disable)
Primary channels input    =    5  (PRI     = 5, 9, 21, 25)
Secondary channels input  =  255  (SEC     = 1, 13, 17, 29)
Telemetry . . . . . . . . =  255  (TLM     = 0, 1, 2, ..., 29)
GPS Rx  . . . . . . . . . =  255  (GPS_RX  = 0, 1, 2, ..., 29)
GPS Tx  . . . . . . . . . =  255  (GPS_TX  = 0, 1, 2, ..., 29)
Sbus OUT  . . . . . . . . =  255  (SBUS_OUT= 0, 1, 2, ..., 29)
RPM   . . . . . . . . . . =  255  (RPM     = 0, 1, 2, ..., 29)
SDA (I2C sensors) . . . . =  255  (SDA     = 2, 6, 10, 14, 18, 22, 26)
SCL (I2C sensors) . . . . =  255  (SCL     = 3, 7, 11, 15, 19, 23, 27)
PWM Channels 1, 2, 3 ,4   =    1    2    3    4 (C1 / C16= 0, 1, 2, ..., 15)
PWM Channels 5, 6, 7 ,8   =  255    6    7    8
PWM Channels 9,10,11,12   =    9   10   11   12
PWM Channels 13,14,15,16  =   13   14   15  255
Voltage 1, 2, 3, 4        =   26   27   28   29 (V1 / V4 = 26, 27, 28, 29)

Protocol is Fport2(Frsky)
CRSF baudrate = 420000
Voltage parameters:
    Scales : 1.000000 , 1.000000 , 1.000000 , 1.000000 
    Offsets: 0.000000 , 0.000000 , 0.000000 , 0.000000 
    No temperature sensors are connected on V3 and V4
RPM multiplier = 1.000000
Baro sensor is not detected
Acc/Gyro is not detected
First analog to digital sensor is not detected
Second analog to digital sensor is not detected
Foreseen GPS type is Ublox  :GPS is not (yet) detected
Failsafe type is HOLD

Config parameters are OK
Press ? + Enter to get help about the commands

 >> 
processing cmd

Cmd to execute:   FV

Volt 1 = 648 mVolt
Current (Volt 2) = 556 mA
Volt 3 = 679 mVolt
Volt 4 = 252 mVolt
Capacity (using current) = 75 mAh

Satcomix, 2023-03-04:
I make a new Test only with original FrSky FAS 40A ADV.
At the Handheld Display there are only 02 0210 VFAS with 8,31V and 02 0200 FASS with 0,0A shown.
[2023-03-04_13-54-11.zip](https://github.com/mstrens/oXs_on_RP2040/files/10888369/2023-03-04_13-54-11.zip)

Satcomix, 2023-03-04:
The next test with oXs 1.5.1 and only V1 and V2 activatet, as VFAS and FASS.
[2023-03-04_13-59-19.zip](https://github.com/mstrens/oXs_on_RP2040/files/10888378/2023-03-04_13-59-19.zip)

mstrens, 2023-03-04:
I made a new version 1.5.2 where I fill most of the frames with fixed data 0X0210= VFAS(copied from one of your capture).
I also change the device id in order to use the same as an original (FAS = 0X22)
So do not connect oXs and an original simultaneously. 
Note; in this version, oXs reply to all pooling request with not NULL frame and so it could be that it saturates the RF link.
I presume that the only consequence is that some frames would be lost what is not an issue for this test

Satcomix, 2023-03-04:
Nothing seen on handheld.

[2023-03-04_14-27-56.zip](https://github.com/mstrens/oXs_on_RP2040/files/10888446/2023-03-04_14-27-56.zip)
processing cmd

Cmd to execute:   

Version = 1.5.2 
    Function                Pin   Change entering XXX=yyy (yyy=255 to disable)
Primary channels input    =    5  (PRI     = 5, 9, 21, 25)
Secondary channels input  =  255  (SEC     = 1, 13, 17, 29)
Telemetry . . . . . . . . =  255  (TLM     = 0, 1, 2, ..., 29)
GPS Rx  . . . . . . . . . =  255  (GPS_RX  = 0, 1, 2, ..., 29)
GPS Tx  . . . . . . . . . =  255  (GPS_TX  = 0, 1, 2, ..., 29)
Sbus OUT  . . . . . . . . =  255  (SBUS_OUT= 0, 1, 2, ..., 29)
RPM   . . . . . . . . . . =  255  (RPM     = 0, 1, 2, ..., 29)
SDA (I2C sensors) . . . . =  255  (SDA     = 2, 6, 10, 14, 18, 22, 26)
SCL (I2C sensors) . . . . =  255  (SCL     = 3, 7, 11, 15, 19, 23, 27)
PWM Channels 1, 2, 3 ,4   =    1    2    3    4 (C1 / C16= 0, 1, 2, ..., 15)
PWM Channels 5, 6, 7 ,8   =  255    6    7    8
PWM Channels 9,10,11,12   =  255  255  255  255
PWM Channels 13,14,15,16  =  255  255  255  255
Voltage 1, 2, 3, 4        =   26   27  255  255 (V1 / V4 = 26, 27, 28, 29)

Protocol is Fport2(Frsky)
CRSF baudrate = 420000
Voltage parameters:
    Scales : 1.000000 , 1.000000 , 1.000000 , 1.000000 
    Offsets: 0.000000 , 0.000000 , 0.000000 , 0.000000 
    No temperature sensors are connected on V3 and V4
RPM multiplier = 1.000000
Baro sensor is not detected
Acc/Gyro is not detected
First analog to digital sensor is not detected
Second analog to digital sensor is not detected
Foreseen GPS type is Ublox  :GPS is not (yet) detected
Failsafe type is HOLD

Config parameters are OK
Press ? + Enter to get help about the commands

 >> 
fport2: first pos not a valid length frame
fport2: first pos not a valid length frame
fport2: first pos not a valid length frame
fport2: first pos not a valid length frame
fport2: first pos not a valid length frame
fport2: first pos not a valid length frame
fport2: first pos not a valid length frame
fport2: first pos not a valid length frame
fport2: first pos not a valid length frame
fport2: first pos not a valid length frame
fport2: first pos not a valid length frame
fport2: first pos not a valid length frame
fport2: first pos not a valid length frame
fport2: first pos not a valid length frame
fport2: first pos not a valid length frame
fport2: first pos not a valid length frame
fport2: first pos not a valid length frame
fport2: first pos not a valid length frame
fport2: first pos not a valid length frame

mstrens, 2023-03-04:
I think I found the bug looking at the capture.
I forgot to send the last byte of each frame (crc).
New version is 1.5.3. 

Still your last post with fport2: first pos not a valid length frame is strange.
I will look at this capture

mstrens, 2023-03-04:
For info, 
fport2: first pos not a valid length frame
is a normal message if it happens only at startup. 
oXs tries to recognize the first char of a frame.
As long it is not synchro, oXs discards the char.
Those messages will not always exist (do not exist when oXs starts listening when Rx is not sending a frame)

Satcomix, 2023-03-04:
Yeahhhhh.....you got it!!!!!!! I have VFAS and FASS in my Handheld display with version 1.5.3
[2023-03-04_14-35-23.zip](https://github.com/mstrens/oXs_on_RP2040/files/10888466/2023-03-04_14-35-23.zip)

[19 more replies]

closed 2023-03-05 by Satcomix