mstrens/oXs_on_RP2040 #59: GPS PDOP FBUS/S.PORT Ver.1.5.7

Satcomix, 2023-03-06:
Hello Mstrens,
I did some tests with different GPS systems today to check the information provided by the CN manufacturer.
The results were sobering.
The price makes the CN quality or clones.
For this I ran my FrSky X20S FBUS protocol with oXs 1.5.7 parallel to U-Center and was therefore always able to compare two GPS systems with each other.
I noticed that the GPS PDOP value for oXs S.PORT/FBUS is too high, it would have to be divided by 100. 
(Display:146, U-Center 1.4/1.5)
Would it be possible to change this?
Why am I doing all these tests? Quite simply: I am writing a manual for oXs Beginners in German as a PDF file, which will also be translated into English.( for the DOC folder?)
Thank you very much for your effort.
Greeting,
Torsten

mstrens, 2023-03-06:
Just like in other protocols, Sport and Fbus frames contains always integers (so never numbers with decimals).
It is the handset that put decimals (and some time make also some unit conversions).
The handset can do it when the field has a specific ID (e.g. altitude) but not when it is a "DIY" field.
For PDOP I created a DIY field that contains the value as provided by the GPS.
So, if you want decimals, you have to define them in the handset (customizing the sensor)  

FYI, I am currently adding Airspeed (based on a MS4525D0) and a compensated Vspeed (Vspeed based on baro alt+ airspeed).
Therefore there will be some change in existing alt and vspeed (and so a new risk for bug).
The code is not yet ready.

Satcomix, 2023-03-06:
Hello Mstrens,
Thank you for your quick response.
I can't set it on my FrSky X20S.
VFAS/V1, FASS/V2 are displayed correctly. At V3 and V4 or T1 and T2
I get the measured 1600mV/1600° displayed, but I can't change the decimals, only mV to V, but then I get 1600V.
With PDOP I can't change anything in such a way that the value is correct.
OK, then there's probably nothing you can do.
Thanks anyway and good luck with Airspeed Sensor MS4525D0
Greeting,
Torsten

closed 2023-03-06 by Satcomix

Satcomix, 2023-03-07:
Hello Mstrens,
Since this problem with the transmission of the values ​​as integers and not as decimals will also affect others, I have described a suggestion in the FrSky Ethos GitHub.
I know that I could also change these values ​​in the oXs SCALE, but that's not the solution, since this function is used to calculate the voltage values. Two separate things.
https://github.com/FrSkyRC/ETHOS-Feedback-Community/issues/2480
Greetings,
Torsten