mstrens/oXs_on_RP2040 #60: Telemetry FBUS Ver.1.6.x

Satcomix, 2023-03-07:
Hello Mstrens,
I don't get any telemetry values ​​on the handheld X20S FBUS with Ver.1.6.0
In FV Terminal everything seems ok.
Greeting,
Torsten

mstrens, 2023-03-07:
Sorry, I forgot to mention a big change in 1.6.0
Now Fbus requires to use "protocol=F" (f from fbus)
And Sbus2 requires protocol=2

So there is an inversion in the 2 codes

Satcomix, 2023-03-07:
Hello Mstrens,
Also with PROTOCOL=F (FBUS) , i don`t get anything on my handheld.

Clean boot
ADS1115 sensor I2C Addr=48
Set up Ads1115 done and OK.
ADS1115 sensor I2C Addr=49
Set up Ads1115 done and OK.
gps baudrate not detected; will be set on 9600
error read MS4525 (airspeed sensor) at startup 
Start detecting baudrate
gps baudrate detected = 9600
Attention: setup on core 1 did not ended within timeout
Setup1 takes 5000177 usec

Version = 1.6.0 
    Function                Pin   Change entering XXX=yyy (yyy=255 to disable)
Primary channels input    =    5  (PRI     = 5, 9, 21, 25)
Secondary channels input  =  255  (SEC     = 1, 13, 17, 29)
Telemetry . . . . . . . . =  255  (TLM     = 0, 1, 2, ..., 29)
GPS Rx  . . . . . . . . . =   12  (GPS_RX  = 0, 1, 2, ..., 29)
GPS Tx  . . . . . . . . . =   13  (GPS_TX  = 0, 1, 2, ..., 29)
Sbus OUT  . . . . . . . . =    2  (SBUS_OUT= 0, 1, 2, ..., 29)
RPM   . . . . . . . . . . =    4  (RPM     = 0, 1, 2, ..., 29)
SDA (I2C sensors) . . . . =   14  (SDA     = 2, 6, 10, 14, 18, 22, 26)
SCL (I2C sensors) . . . . =   15  (SCL     = 3, 7, 11, 15, 19, 23, 27)
PWM Channels 1, 2, 3 ,4   =  255  255    6  255 (C1 / C16= 0, 1, 2, ..., 15)
PWM Channels 5, 6, 7 ,8   =  255  255  255  255
PWM Channels 9,10,11,12   =  255  255  255  255
PWM Channels 13,14,15,16  =  255  255  255  255
Voltage 1, 2, 3, 4        =   26   27   28   29 (V1 / V4 = 26, 27, 28, 29)

Protocol is Fbus(Frsky)
CRSF baudrate = 420000
Voltage parameters:
    Scales : 1.000000 , 1.000000 , 1.000000 , 1.000000 
    Offsets: 0.000000 , 0.000000 , 0.000000 , 0.000000 
    No temperature sensors are connected on V3 and V4
RPM multiplier = 1.000000
Baro sensor is detected using MS5611
    Sensitivity min = 100 (at 100)   , max = 300 (at 1000)
    Hysteresis = 5 
Acc/Gyro is detected using MP6050
     Acceleration offsets X, Y, Z = 703 , 247 , -2586
     Gyro offsets         X, Y, Z = 17 , -161 , -193
First analog to digital sensor is detected using ads1115
    Measurement setup: 4 , 5 , 6 ,7
    Gains: 1 , 1 , 1 ,1
    Rates: 5 , 5 , 5 ,5
    Offsets: 0.000000 , 0.000000 , 0.000000 ,0.000000
    Scales: 1.000000 , 1.000000 , 1.000000 ,1.000000
    Averaged on: 10 , 10 , 10 ,10
Second analog to digital sensor is detected using ads1115
    Measurement setup: 4 , 5 , 6 ,7
    Gains: 1 , 1 , 1 ,1
    Rates: 5 , 5 , 5 ,5
    Offsets: 0.000000 , 0.000000 , 0.000000 ,0.000000
    Scales: 1.000000 , 1.000000 , 1.000000 ,1.000000
    Averaged on: 10 , 10 , 10 ,10
Foreseen GPS type is Ublox  :GPS is detected and has a fix
Failsafe uses predefined values
     Chan 1...4  = 992 , 992 , 992 , 992
     Chan 5...8  = 992 , 992 , 992 , 992
     Chan 9...12 = 992 , 992 , 992 , 992
     Chan 13...16= 992 , 992 , 992 , 992

Config parameters are OK
Press ? + Enter to get help about the commands
processing cmd

mstrens, 2023-03-07:
I put a new version 1.6.1

Satcomix, 2023-03-07:
No improvement with 1.6.1
SBUS works, FBUS doesn't, or do I have to switch to TLM=pin?

processing cmd

Cmd to execute:   

Version = 1.6.1 
    Function                Pin   Change entering XXX=yyy (yyy=255 to disable)
Primary channels input    =    5  (PRI     = 5, 9, 21, 25)
Secondary channels input  =  255  (SEC     = 1, 13, 17, 29)
Telemetry . . . . . . . . =  255  (TLM     = 0, 1, 2, ..., 29)
GPS Rx  . . . . . . . . . =   12  (GPS_RX  = 0, 1, 2, ..., 29)
GPS Tx  . . . . . . . . . =   13  (GPS_TX  = 0, 1, 2, ..., 29)
Sbus OUT  . . . . . . . . =    2  (SBUS_OUT= 0, 1, 2, ..., 29)
RPM   . . . . . . . . . . =    4  (RPM     = 0, 1, 2, ..., 29)
SDA (I2C sensors) . . . . =   14  (SDA     = 2, 6, 10, 14, 18, 22, 26)
SCL (I2C sensors) . . . . =   15  (SCL     = 3, 7, 11, 15, 19, 23, 27)
PWM Channels 1, 2, 3 ,4   =  255  255    6  255 (C1 / C16= 0, 1, 2, ..., 15)
PWM Channels 5, 6, 7 ,8   =  255  255  255  255
PWM Channels 9,10,11,12   =  255  255  255  255
PWM Channels 13,14,15,16  =  255  255  255  255
Voltage 1, 2, 3, 4        =   26   27   28   29 (V1 / V4 = 26, 27, 28, 29)

Protocol is Fbus(Frsky)
CRSF baudrate = 420000
Voltage parameters:
    Scales : 1.000000 , 1.000000 , 1.000000 , 1.000000 
    Offsets: 0.000000 , 0.000000 , 0.000000 , 0.000000 
    No temperature sensors are connected on V3 and V4
RPM multiplier = 1.000000
Baro sensor is detected using MS5611
    Sensitivity min = 100 (at 100)   , max = 300 (at 1000)
    Hysteresis = 5 
Acc/Gyro is detected using MP6050
     Acceleration offsets X, Y, Z = 703 , 247 , -2586
     Gyro offsets         X, Y, Z = 17 , -161 , -193
First analog to digital sensor is detected using ads1115
    Measurement setup: 4 , 5 , 6 ,7
    Gains: 1 , 1 , 1 ,1
    Rates: 5 , 5 , 5 ,5
    Offsets: 0.000000 , 0.000000 , 0.000000 ,0.000000
    Scales: 1.000000 , 1.000000 , 1.000000 ,1.000000
    Averaged on: 10 , 10 , 10 ,10
Second analog to digital sensor is detected using ads1115
    Measurement setup: 4 , 5 , 6 ,7
    Gains: 1 , 1 , 1 ,1
    Rates: 5 , 5 , 5 ,5
    Offsets: 0.000000 , 0.000000 , 0.000000 ,0.000000
    Scales: 1.000000 , 1.000000 , 1.000000 ,1.000000
    Averaged on: 10 , 10 , 10 ,10
Foreseen GPS type is Ublox  :GPS is detected and has a fix
Failsafe uses predefined values
     Chan 1...4  = 992 , 992 , 992 , 992
     Chan 5...8  = 992 , 992 , 992 , 992
     Chan 9...12 = 992 , 992 , 992 , 992
     Chan 13...16= 992 , 992 , 992 , 992

Config parameters are OK
Press ? + Enter to get help about the commands

 >> 
processing cmd

Cmd to execute:   FV

GPS Latitude = xx.xxxxxxxxxxx degree
GPS Longitude = x.xxxxxxxxxx degree
GPS Groundspeed = 8 cm/s
GPS Heading = 173.220000 degree
GPS Altitude = 2125 cm
GPS Num sat. = 109 cm
GPS Date J M A = 7 3 23 
GPS Time H M S = 15 22 6 
GPS Pdop = 261 
GPS Home bearing = 161 degree
GPS Home distance = 14 m
Volt 1 = 1557 mVolt
Current (Volt 2) = 1652 mA
Volt 3 = 1601 mVolt
Volt 4 = 274 mVolt
Capacity (using current) = 58 mAh
Vspeed = -1 cm/s
Baro Rel altitude = 4 cm
Pitch = 0 degree
Roll = 0 degree
RPM = 999 Hertz
Ads 1 1 = 499 mVolt
Ads 1 2 = 1003 mVolt
Ads 1 3 = 1502 mVolt
Ads 1 4 = 2001 mVolt
Ads 2 1 = 120 mVolt
Ads 2 2 = 121 mVolt
Ads 2 3 = 121 mVolt
Ads 2 4 = 121 mVolt

mstrens, 2023-03-07:
it is supposed to work on PRI pin (and not TLM).

Does Sport still works?

mstrens, 2023-03-07:
Ok I continue to look for changes on Fbus.
Perhaps I will have to make some debug versions.

Satcomix, 2023-03-07:
OK,
You make the debug-builds and i test it :-))

mstrens, 2023-03-07:
debug version 1.6.2 is on github.
When Fbus is activated and oXs is connected to the receiver, oXs is supposed to 
- print on the PC (via usb) one line for each frame received from the receiver (with the content of the frame)
- print  one line for each frame sent to the receiver in reply from the request.

Probably a capture with logic analyser can also help.

mstrens, 2023-03-07:
Thanks,
I think I found the bug.
New version is on github.
It it is OK , then I will remove the debug messages

Satcomix, 2023-03-07:
Ver.1.6.3 
FBUS works, all telemetry is shown in display and SBUS 16CH works too.
debug messages works too :-))
Greeting,
Torsten

mstrens, 2023-03-07:
OK
I will remove debug messages in next version.

In your test, did you had a GPS connected?
I am not sure because I think I saw a message says that the baudrate was not automatically detected.
GPS is another part that I modified significantly (to implement non blocking autodetect baudrate)

Satcomix, 2023-03-07:
GPS was connected and has a 3D-FIX (10 SATS)

mstrens, 2023-03-07:
I put a version without the sbus debug message.
If you try it with a gps being connected, do you get a message saying that baudrate is autodetected or just the opposite?

Satcomix, 2023-03-07:
I make some more tests, the GPS is everytime online and has a FIX after shutdown and startup the board.

Satcomix, 2023-03-07:
OK,
GPS is erverytime detected after bootup (2sec).
The boottime is faster, from 5000000us to 13504 usec

Satcomix, 2023-03-08:
Hello Mstrens,
After you were busy again, I tested the new version 1.6.6 this morning. Once with FrSky x20S with FBUS protocol and Graupner MC28 with HOTT protocol.
The SBUS values ​​in the FBUS protocol for the 16 PWM channels fit.
The telemetry values ​​are all displayed in both systems.
With Altitude and Vspeed, I could see a constant increase in flight altitude on both systems, although the GY86 incl. MS5611 and MPU6050(FBUS) and GY63 MS5611(HOTT) lay motionless on the table and there was no wind everywhere.
After about 5 minutes it was 2-3 meters, Vspeed oscillated between +/- 0.1-0.4m/s.
The BeiTian GPS BN-220 is apparently not correctly recognized on both systems during the boot process, although it already has a 3D FIX with 9 sats during initialization (see terminal).

[Ver1.6.6_TERMINAL.txt](https://github.com/mstrens/oXs_on_RP2040/files/10918184/Ver1.6.6_TERMINAL.txt)

Greetings,
Torsten

closed 2023-03-08 by Satcomix