mstrens/oXs_on_RP2040 #61: There are no values displayed from the MS5611.

holtho70, 2023-03-08:
I have worked with version 1. 3. 1 (Hott-version) in the past. Everything worked very well.

The values of the MS5611 were reasonably displayed in the telemetry.

I tested the current version (main-version) yesterday. I noticed that the values of the MS5611 are no longer displayed in the telemetry.
I use the board of ddano007 (Post #19, RCgroups. com)

In addition, an ELRS receiver works together with my transmitter Jumper T18 pro.
I hope I've made myself clear enough. If not, please give me a brief response.

Greetings

Th. Hollatz

Satcomix, 2023-03-08:
Hello Thomas,

Today I tested version 1.6.6 with FrSky X20S with FBUS, S.PORT with SBUS and Graupner MC28 HOTT.
All 29 possible telemetry values ​​were displayed for FBUS and S.PORT, the SBUS_OUT also delivered all 16 PWM RC channels.
With the HOTT protocol, all currently possible telemetry values ​​were also displayed to me.
For FrSky I use a GY86 sensor with MS5611 and MP6050 onboard, for HOTT a Gy63 with MS5611
Greeting,
 Torsten

[Ver.1.6.6_Terminal_S.PORT.txt](https://github.com/mstrens/oXs_on_RP2040/files/10918962/Ver.1.6.6_Terminal_S.PORT.txt)

mstrens, 2023-03-08:
When version in test is stable, I will copy it to main branch.
Currently I still have quite many changes to do for airspeed sensor.

holtho70, 2023-03-08:
Just tested with version 1. 6. 8 TEST. With this version it works. I can see ALT and V-Speed in the telemetry-screen again.

closed 2023-03-09 by holtho70