mstrens/oXs_on_RP2040 #62: Telemetry Values Ver.1.6.x

Satcomix, 2023-03-09:
Hello Mstrens,
I did some tests again with the new version 1.6.8.
I noticed with HOTT that the altitude display shows -9999m all the time, Vspeed oscillates around 0.0m/s +/-0.2m/s.
A BMP280 or an MS5611 is used.
If I move the sensor and create a negative pressure, the baro also shows the correct values, but then immediately jumps back to -9999m.
With the FrSky X20S with FBUS or S.PORT with SBUS, all telemetry values ​​and PWM channels are displayed correctly in Version 1.6.8.

[Ver.1.6.8_Terminal.txt](https://github.com/mstrens/oXs_on_RP2040/files/10928882/Ver.1.6.8_Terminal.txt)

Greetings,
Torsten

mstrens, 2023-03-09:
It should be fixed now in version 1.6.9.
It was an issue in rounding negative value (bug was present since several days)

Satcomix, 2023-03-09:
I know that the error has been there for a few days, I already wrote about it in version Ver1.4.1HOTT Protocol :-))

closed 2023-03-09 by Satcomix