mstrens/oXs_on_RP2040 #66: Fligt Test

Klippeneck, 2023-03-19:
Hello mstrens,
over the weekend I was able to flight test the OXS Rp2040 sensor with MS5611 and GPS BN220. Function so far i. O., but the telemetry values of altitude, vario and GPS altitude are not correct. 
with another sensor. The values of the comparison sensor I have compared with a Jeti Vario 2, agree very well. The OXS baro height is about a factor of 1.25 too low.
At -10m altitude both sensors are the same and diverge again below -10m.(See log).
The GPS altitude oscillates between 81 and 98m. 
The climb values of the vario also differ strongly. Probably there is a bug in the altitude calculation.
Greetings Jürgen 

[image]

mstrens, 2023-03-19:
For GPS, I saw the bug. The values from oXs where 10 X to small (I divided by 1000 instead of by 100 when I filled the data to be sent).
This is fixed in version 1.8.7.

I checked the formula to calculate baro altitude and I do not see immediately the bug.
Could you make a capture of the 3 altitudes with this new version.
With the right GPS altitude, it will be possible to have a better comparison. 
Could you also provide a graph of the 2 Vspeed (the one from oXs and the one from jeti vario) in the same flight.

Klippeneck, 2023-03-19:
I can test GPS tomorrow with SW 1.87.
Vspeed from Jeti Vario and OXS I have attached.
[image]

mstrens, 2023-03-19:
Based on the graph, Vspeed of oXs and Jeti seems me quite well correlated.
It seems that oXS provides higher value but this is not abnormal because I also note that it reacts faster.
So there is less damping.
It a choice I made.
It is possible to change the sensitivity of oXs (in config file) if you want.

Klippeneck, 2023-03-20:
Hello mstrens,
attached are two logs, the first shows the takeoff, the second the whole flight.
The GPS altitude is shown as altitude above sea level, so I have adjusted the scaling in the diagram accordingly.
accordingly. In the altitude there are again differences between OXS and Jeti Vario. It is strange that at an altitude of -10m no deviations are to be seen.  

[image]
[image]

Klippeneck, 2023-03-20:
And now with GPS altitude, OXS altitude and Jeti Vario altitude
[image]

Klippeneck, 2023-03-22:
Hello,
I already had the assumption, especially since the sensors are from China.
I will change the MS5611 and report again.
Thanks a lot .

mstrens, 2023-11-21:
I presume this can be closed

closed 2023-11-21 by mstrens