mstrens/oXs_on_RP2040 #67: Baro/Vario on SBUS2 protocol

fourfingers4, 2023-03-25:
Both using FVP command to feed fixed positive values, and using real sensors, display "0" on handset for altitude and vario. 
Tried using both  F1672 and F1712 available on my transmitter.
The rest of sensors seem to work as expected.
Last try on 1.8.7
Hardware: Tx T32MZ, Rx R3006SB on T-FHSS protocol.

Here is a dump of FVP command and settings:
[sbus2 vario.txt](https://github.com/mstrens/oXs_on_RP2040/files/11069813/sbus2.vario.txt)

mstrens, 2023-03-25:
Can you try the latest version.
I made some changes today for vario on SBUS2

fourfingers4, 2023-03-25:
Just tried 1.9.0, and using F1672 live vario displays a fixed value of  163.84m/s  (0.02-0.04m/s when sensors are moved) and altitude displays 0m( +1,+2m when sensors are moved)

Using FVP command 
20:22:09.670 -> Vspeed = 258 cm/s                     >>>>>>>>>>>>>>164.10m/s
20:22:09.670 -> Baro Rel altitude = 246821 cm   >>>>>>>>>>>>>>2468m

mstrens, 2023-03-25:
Can you now try version 1.93 I just uploaded based on your last test

fourfingers4, 2023-03-25:
Eureka!
That was fast. Congratulations.
1.9.3
So, live now displays 0.00m/s and positive and negative values corresponding to sensor movement, and so does altitude.

FVP values correspond
20:45:06.412 -> Vspeed = 258 cm/s >>>>>>>>>>>>>>>0.26m/s
20:45:06.412 -> Baro Rel altitude = 246821 cm>>>>>>>>>2468m.

I think I currently have only one more sensor to check >IAS. I don´t have a diff pressure setup right now, but if you want me to try with FVP, please let me know what slot number to assign and I will report.

mstrens, 2023-03-25:
So, there was still a bug in the conversion of Vspeed.
I fixed it in the version 1.9.4

Aispeed uses slot 26 (you find them in config.h)

fourfingers4, 2023-03-25:
Ah yes, now I see, a whole decimal place.
1.9.4 Now it is reporting as it should, 2.58m/s when sending FVP.
No luck with Airspeed, showing 0km/h on slot 26.

mstrens, 2023-03-25:
Version 1.9.5 will solve the issue about Airspeed( I hope).
There is perhaps still an error in the conversion to get the expected result on the display.
Let me know.

fourfingers4, 2023-03-25:
No luck. still showing 0km/h. 
I have tried both FPV AND FPN, in case the handset was re-zeroing for the first value recieved.
I can only assign one type of futaba sensor for airspeed, no other options available in this handset.

mstrens, 2023-03-25:
Perhaps now with version 1.9.6

fourfingers4, 2023-03-25:
1.9.6
22:10:01.387 -> Airspeed = 15151 cm/s  showing 15151km/h so just need to scale it.

On a side note, playing with FVN just noticed both temp sensors will remain on the last positive value, instead of displaying the negative one that has been sent. (Don´t know if futaba protocol is able to display - values though)

mstrens, 2023-03-25:
1.9.7 : I added conversion to km/h.
About FVN, I do not understand. For Temp1 and Temp2, I use a sensor type SBS/01T.
For this sensor, I apply an offset of 100. So -20° bomes 80° in the message and the handset should remove the offset and display -20°. I do not know why it is not the case.
For positive value, does the value on handset matches the FVP value?

fourfingers4, 2023-03-25:
Yes, positive values sent with FVP read as they should on SBS/01T , 139 and 148 if my memory doesn´t fail.
May I suggest to change line 40 of confi.h ? just to make it clear what sensor to use, as in my previous tests sensor Temp125-F1713 has never worked (there is no sensor named F125 that I am aware of, proper name is Temp125-F1713 ).

40 // V3 and V4 can be get using F125 or F1713   
  to
40 // V3 and V4 can be get using SBS/01T

I will test 1.9.7

fourfingers4, 2023-03-25:
19.7
Spot on!
545km/h showing on the handset,
Thanks for your work Mstrens.

mstrens, 2023-03-28:
I presume this issue can be closed?

closed 2023-03-29 by fourfingers4