mstrens/oXs_on_RP2040 #68: S.Port GPS-Time

onki69, 2023-03-26:
Some transmitters set the internal time when a GPS-Time-protocol is received from a sensor.
Is this something that can be added to the protocol?

Best regards
Onki

mstrens, 2023-03-26:
When a GPS is connected, oXs transmits the date and time in Sport protocol.
It is then up to the handset firmware to handle it or not to update his internal clock.

onki69, 2023-03-27:
If I understand correctly the time is transfered but not available as telemetry parameter?
It does not seem to work correctly.
When I use a SM GPS-Logger 3 the time is a separate telemetry value and the transmitter clock is set automatically.
I have tested this with the 1.8.1 version and a valid 3D fix but my transmitter clock is still 1 hour behind.
Will double check this with the GPS-Logger 3

onki69, 2023-04-13:
Time zone is set correctly. I will try it with another receiver (Archer M+).
With a Radiomaster R16 it does not seem to work.
Why there is no telemetry value for the date/time such with the GPS-Logger3?
Is the this only mused internally?

Satcomix, 2023-04-25:
Hello Mstrens,
Since the latest version from yesterday, my X20S handset system clock no longer updates itself automatically from the GPS Date/Time Sensor 0x850.
Usually has nothing to do with the oXs system clock of 133MHz, right?
br,
Torsten

mstrens, 2023-04-25:
there is no relation between GPS time and oXs clock.
Do you get the other GPS data?

Satcomix, 2023-04-25:
Hello Mstrens,
I get all 34 values ​​displayed properly.
Only the time from the handset (very, very imprecise) no longer wants to update, although I have the button in the time menu set to update automatically.
I'll search the FrSky GitHub for possible reasons, there are enough unsolved problems at Ethos.
Thanks for your answer,
Greeting,
Torsten

onki69, 2023-05-08:
I did some testing about a week ago.
Still no success with my x20(S) and Ethos 1.5 Nightlies.
To check if this is Ethos related I tried it with a GPSLogger3 (SM-Modellbau.de).
As soon as I got a fix the time on the transmitter was corrected (with the GPSLogger3).
The only issue was a 2h shift but this was related to the settings in the GPSLogger3 as the time zone can be adjusted there.
The GPS module I use is a uBlox 6M. It was reset to factory settings (9k6). Time is shown in Ublox Software.

mstrens, 2023-05-08:
With oXs, when you ask for discovering the sensors, do you get the GPS data and time in the list of sensors?
Are the oXs values displayed in the telemetry list correct and in the same format as with GPSLogger3?

Satcomix, 2023-05-08:
I'm starting to suspect that the date/time data packet from an M6 GPS module is structured a little differently and oXs or Ethos can't understand it.

mstrens, 2023-05-08:
It is oXs that decodes the date and time values received from GPS in some frame types.
I know that Ublox added/deleted frame types but I do not expect that they changed the frame format when moving from M6 to M10.

Satcomix, 2023-05-08:
I'm currently searching the net for M6N Time problems, and it looks like it's not the frame format, it's a specific setting in the M6. Unfortunately I don't have one here, only M8 and M10

Satcomix, 2023-05-08:
I make a test with Ver. 2.3.0_Test with FBUS Protocoll
and a FrSky X20S with TD-R10 Receiver and a Beitian BE-250 M10
The GPS time is UTC and ok and the Handheldtime is set from GPS with UTC+2h
Cmd to execute:   FV

GPS Latitude = XX.1225248 degree
GPS Longitude = X.7172880 degree
GPS Groundspeed = 4 cm/s
GPS Heading = 305.120000 degree
GPS Altitude = 3953 cm
GPS Num sat. = 122
GPS Date J M A = 8 5 23 
GPS Time H M S = 12 43 46 
GPS Pdop = 128 
GPS Home bearing = 164 degree
GPS Home distance = 9 m
Volt 1 = 1839 mVolt
Current (Volt 2) = 1656 mA
Volt 3 = 1601 mVolt
Volt 4 = 213 mVolt
Capacity (using current) = 13367 mAh
Vspeed = 1 cm/s
Baro Rel altitude = 122 cm
Pitch = -4 degree
Roll = -1 degree
RPM = 999 Hertz
Ads 1 1 = 119 mVolt
Ads 1 2 = 119 mVolt
Ads 1 3 = 119 mVolt
Ads 1 4 = 119 mVolt
Ads 2 1 = 119 mVolt
Ads 2 2 = 119 mVolt
Ads 2 3 = 119 mVolt
Ads 2 4 = 119 mVolt
Airspeed = 30 cm/s
Compensated Vspeed = -16 cm/s
Gps cumulative distance = 981
Vspeed compensation = 1.15

Satcomix, 2023-05-11:
Hello Mstrens,
Hello Onki,
FYI:
An issue with GPS Time was opened in the FrSky GitHub. The user still has Ethos version 1.4.7 installed.
I've done my testing on Ethos version 1.4.8 and Nightly and haven't noticed anything unusual. Very strange, since the user also receives untraceable date and time values ​​on the handset from an original FrSky ADV GPS.
https://github.com/FrSkyRC/ETHOS-Feedback-Community/issues/2632
Greeting,
Torsten

bsongis-frsky, 2023-05-11:
For dev purposes I can provide a special Ethos firmware, which will write a sport.log file with the raw S.Port stream. It helps to track such issues and know if Ethos or the custom sensor are involved

Satcomix, 2023-05-11:
> For dev purposes I can provide a special Ethos firmware, which will write a sport.log file with the raw S.Port stream. It helps to track such issues and know if Ethos or the custom sensor are involved

Hello Bertrand,
Thank you for your quick reply, here on oXs.
I think such an Ethos variant to read would be a very good thing.
But what I don't understand are the problems with the GPS date/time data,
I own an X20S with a TD-R10 and from June an X20PRO :-))
I had no other problems with original sensors or the oXs_RP2040, also not with the date/time, except for the telemetry problem in the Nightly (which was fixed directly), while others have these problems also on original FrSky sensors, very strange this behavior .
Greetings,
Torsten

jasc76, 2023-05-29:
I have these issues too...

Satcomix, 2023-05-31:
> I have these issues too... x20s ethos 1.4.8

Hello Jasc76,
Hall everyone,
I know about the GPS date/time problems that some users of original FrSky GPS-ADV or oXs_RP2040 GPS have.
Bertrand and his FrSky Ethos team are also aware of these issues.
I just can't explain it to myself because I don't have these problems, for whatever reason?
I tried to reproduce the error by only defining an oXs_RP2040 as GPS Phys.ID03 0x83, and no other telemetry data is transmitted in the frame.
I always get an exact time regardless of whether I switch on the transmitter first and then the GPS telemetry on the oXs or vice versa, switch on the oXs with GPS first and then the transmitter. I can also disconnect and reconnect the GPS, the time does not exist for the time of the disconnect, and reappears and syncs again after a successful connection.
Bertrand had spoken of an analysis software that he wanted to make available to those involved, but he did not comment on the GPS topic.
I found this software on FrSky GitHub at 08/05/2022 from Wismy Yao. I just can't promise success by evaluating the data as it will clearly be down to the ethos and not the different GPS systems.
https://github.com/FrSkyRC/ETHOS-Feedback-Community/issues/1827#issuecomment-1205998699
Best regards,
Torsten

jasc76, 2023-06-12:
[FUNJET-2023-05-20-19-07-06.csv](https://github.com/mstrens/oXs_on_RP2040/files/11724049/FUNJET-2023-05-20-19-07-06.csv)

Here is a log. At first the GPS Time is 2023 all good, but after a period with no reception it's suddenly 2015

Satcomix, 2023-06-12:
> [FUNJET-2023-05-20-19-07-06.csv](https://github.com/mstrens/oXs_on_RP2040/files/11724049/FUNJET-2023-05-20-19-07-06.csv)
> 
> Here is a log. At first the GPS Time is 2023 all good, but after a period with no reception it's suddenly 2015

Hello jasc76,
I could never find these problems with me, that the time/date shifts by a few years.
Is it maybe related to the ROLLOVER timestamp complement?
Which GPS board and which receiver do you use with your X20S.
I use the X20S with TD-R10 and FBUS protocol with Beitian M10 chip (BE-220 to 450)
br,
Torsten

jasc76, 2023-07-11:
this might help in future
https://github.com/FrSkyRC/ETHOS-Feedback-Community/issues/2513

romoloman, 2023-07-23:
> in this plane I use: I think its a BN220 REceiver S6R S.Port

I think that the main problem is the gps module you are using, It uses the M8 engine so no PVT messages out, 
To get the time in openXsensor I configure the GPS to transmit TIMEUTC on uart1 and then I parse the related packet.
oXs_on_RP2040  instead get time and date from PVT packet that doesn't exist in M6-M8 protocol.

romoloman, 2023-07-27:
In some receivers (fake/clone) the following test 
if (( _buffer.pvt.valid & 0x03) == 0X03 ) is true even with wrong date / time (that's why when it start receiving you may get the original date / time of the GPS receiver)

Instead getting the time from timeutc record using the following test always return a proper date/time
 if (_buffer.timeutc.flag & 0b111) {

That's what I'm implemented in the arduino version and also in my fork.