mstrens/oXs_on_RP2040 #73: Spektrum debug

pierrotm777, 2023-04-11:
All telemetry seems well received except GSpd that don't return 0 when the module don't move.
- RXA_ are well removed
- Alt and VSpd are no correct (not inverted).

mstrens, 2023-04-11:
After a FVP, what do you get for:
- GSpd
- Alt
- Vspeed
It is important to say that you are currently testing with EdgeTx and not with a real Spektrum device.
So it could be that the bug (if any) is in EdgeTx and not in oXs.
As far I remember, you tested Alt and Vspeed with a Spektrum TX and the value where OK. If so, the bug in at EdgeTx side.
For GSpd, it was not possible to check with Spektrum.
Am I right?

pierrotm777, 2023-04-11:
GSpd 133kmh
Alt 2468.2m
Vspd 2.6m/s

"As far I remember, you tested Alt and Vspd with a Spektrum TX and the value where OK."
If i remember, yes.
But i receive also a value named ASpd that return well 545kmh as with the Dx18.

"For GSpd, it was not possible to check with Spektrum."
Yes, the dx18 version not include the gps values.

pierrotm777, 2023-04-11:
For me VSpd and Alt are correctly received on my Tx16S.
This is just the GSpd who is curious.

mstrens, 2023-04-11:
I do not understand why there should be an issue with 
Alt 2468.2m
Vspd 2.6m/s

Those are the values foreseen with FVP.

Note: Aspd is the airspeed, so it is the value that should be provided by a pitot tube and a MS4525 (or SDP3x).
It is normal to get a airspeed when command FVP is used (even if there is not such a sensor connected). That is just the advantage of FVP: you can check the value on TX even if there is no sensor (except for Jeti protocol where there is some special cases)

mstrens, 2023-04-11:
Why did you say in your first post:
Alt and VSpd are no correct (not inverted).

pierrotm777, 2023-04-11:
Sorry i wanted say NOW and not NO.
Alt and VSpd are now correct (not inverted).

mstrens, 2023-04-11:
OK. 
"Now" I understand.

About GSpd, I propose not to change this currently.
I suspect that the bug is at the edgeTx side. I presume EdgeTx expect the value in the frame in another unit than km/h while spektrum expect it in km/h. So if I change in oXs, it could be OK for EdgeTx but not for spektrum.
I propose to wait that someone with a spektrum Tx using a recent firmware could test the GPS values.

pierrotm777, 2023-04-11:
I am ok :-)
I have just tested pwm command witch run very well over the SRXL2.

pierrotm777, 2023-04-22:
Hello mstrens,

I try the new 2.8.3 firmware.
ID 0000 to 000D now don't  exits.
But i discover now some ID from 2600 to 260C .
I don't see GPS values :-( !

mstrens, 2023-11-21:
I expect this can be closed and that Gps values are received on the handset

closed 2023-11-21 by mstrens

pierrotm777, 2023-11-21:
Hi mstrens,
If i test the Frsky protocol , i have all my sensors.