mstrens/oXs_on_RP2040 #74: MPX Protokoll GPS -  Alt, Heading, total Distance

Astor87, 2023-04-18:
Hi mstrens,
I just started my oXs_on_RP2040 this weekend following the German RC-Network Thread. 
I do only use MS5611 for Vario and an BN-180 for GPS data. 
Everything works absolutely perfect on an FrSky protocol with RM R168 via FrSky S.Bus with actual SW 2.1.4
However I do use MPX receivers for 99% of my models. Switching the protocol is really simple. Unfortunately there are currently only 3 GPS parameter transmitted, GPS Groundspeed, GPS Home bearing, GPS Home distance if I am right.  And this is totally clear for me consulting your file: "fields per protocol.txt" 
Reading a previous issue #40 here from Pierre and all 69 pages on RCG I understand that Position, Date and Time or Sats cannot be transmitted. On Arduino oXs there are 5 parameters. The 3 transmitted plus  GPS Altitude and GPS Heading. 

Would it be possible to add these two parameters in your MPX protocol transmission list ? 
And finally would it be possible to calculate total "flown" distance as a summary function of speed over time (maybe in m or km) ? I would assume that could be interesting for all protocols.

Sorry this shall not appear as a complain or issue at all!   I really do highly appreciate your great work and effort on this great project!!  
BR
Wolfram

mstrens, 2023-04-18:
I added GPS alt and GPS bearing to Mpx protocol in version 2.2.0.

For cumulative distance, it requires more work.

mstrens, 2023-04-18:
I also added now in 2.2.1 cumulative distance to sport and Mpx protocols.
The code is not tested. I hope it is OK.

Astor87, 2023-04-18:
Many Thanks !
I just loaded 2.2.1 (test) on the RP2040 Zero. With FV I can see the new Gps cumulative distance added. 
I could not find the GPS Alt, Heading, Cum. Distance on Sensor Search but I used my RM TX12 with a IRangeX Multimodule and MPX Protocol. I will check tomorrow with with my RM TX16S and Original MPX HF Module and further with my Zorro 4in1 . Will also test with Frsky Sport. Will keep you posted - I am sure it will work ! 
But anyway you are doing a miracle job !!! Really great !

Astor87, 2023-04-19:
Will do, but I am currently underway and will be back home this evening. Keep you posted!!

Astor87, 2023-04-19:
Hi mstrens,
just being back home I loaded  2.2.4
RM Zorro with Frsky X Protocol and RM 168 S.Bus all data available incl. Cumulative Distance! For me all ok !

on M-Link Protocol : today RM Zorro and yesterdays RM TX12 with IrangeX 4in1 (which failed)  now both transmitter read and show the 3 additional Data:  GPS Heading, GPS Altitude and GPS Cumulative Distance.  Flight Test for cumulative distance will be done after this heavy rain here :-) 
Superb Job and for me great ! 

**Just one more comment if I may:**
In M-Link Protocol:
GPS Distance from Home i.e.  Index = 13 is transmitted in "km"   
Wouldn't it be more logic to transmit GPS Distance from Home in "m"  ??? A Model should be always in sight which is 90% less than 1km  - at least for me. Just a thought.   

GPS Cumulative Distance i.e.  Index = 12  is transmitted in "km"    for me ok

Again many thanks for your great Job !!!
BR
Wolfram

Astor87, 2023-04-19:
Sorry it shall read :   RM Zorro with Frsky X Protocol and RM 168 **S.Port**     !!!!!!!!!

Astor87, 2023-04-19:
Again thanks for your great effort
BR
Wolfram

closed 2023-04-19 by Astor87